zigpy/zha-device-handlers #477: [Device Support Request] Hue Tap Switch Model 9290011152X by Philips

Hedda, 2020-09-07:
**Is your feature request related to a problem? Please describe.**

Has anyone here tested and can verify if the Hue Tap Switch Model 9290011152X by Philips works with ZHA in Home Assistant or if require a quirk device handler or since it is a ZGP (Zigbee Green Power) device if it perhaps even require modifications to zigpy and its radio library?

https://zigbee.blakadder.com/Philips_9290011152X.html

Disclaimer: I do not own one of these myself but created the issue on behalf of @dieneuser who posted in the ZGP (Zigbee Green Power) request for zigpy in https://github.com/zigpy/zigpy/issues/341

@dieneuser can hopefully fill in more information here, such as device signature in ZHA, plus which exact Zigbee hardware adapter and firmware he is using with ZHA in Home Assistant?

_Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line._

Anyway, he said he is currently using it deCONZ gateway software (Phoscon OS) which does support it but with these limitations:

_"Note that ZGP devices (like the Hue tap and FoH switches that harvest kinetic energy) currently don't work over the mesh network; they need to be in direct range of the coordinator. Also note that ZGP devices aren't shown in the deCONZ GUI."_

_"These switches use custom values for state.buttonevent please refer to the related GitHub issue pages for more information."_

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices

**Describe the solution you'd like**

Support for ZGP (Zigbee Green Power) devices like  the Hue Tap Switch Model 9290011152X by Philips.

See https://github.com/Koenkk/zigbee-herdsman/pull/92 for proof-of-concept used by zigbee-herdsman and Zigbee2mqtt project:

https://github.com/Koenkk/zigbee-herdsman/pull/92

**Additional context**

ZGP (Zigbee Green Power), a.k.a. "Zigbee GreenPower", is a subset feature of Zigbee that allows for energy-harvesting technology to be used directly with the Zigbee stack. Using Zigbee Green Power, certain home automation devices (end devices) may be powered by energy harvesting solutions, which when enabled introduces the potential of battery-less devices (end devices) operating in a Zigbee network.

https://zigbeealliance.org/solution/green-power/

Hedda, 2020-09-07:
Please also see the related device request https://github.com/zigpy/zha-device-handlers/issues/478 about EnOcean PTM 216Z ZGP Generic Switch + Wall Switches EWSxZ and EWSxZG

dieneuser, 2020-09-07:
Hey Hedda, 

thank you for the handling.

> @dieneuser can hopefully fill in more information here, such as device signature in ZHA, plus which exact Zigbee hardware adapter and firmware he is using with ZHA in Home Assistant?

I use a Silicon Labs EmberZNet based radio using the EZSP protocol in ZHA (Bitron Video/Smabit BV AV2010/10 USB-Stick with Silicon Labs Ember 3587 and firmware version 0x12217572).

> _Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line._

A pairing of the GreenPowerSwitch is not possible with ZHA. There is nothing happens except these lines in the logging window:

```
[0x6b76:1:0x0b04]: async_update
[0x605c:1:0x0b04]: async_update
[0x6b76:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=90 command_id=Command.Read_Attributes_rsp>
[0x605c:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=92 command_id=Command.Read_Attributes_rsp>
[0x04c1:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=129 command_id=Command.Report_Attributes>
[0x04c1:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=49251>>]]
[0x04c1:1:0x0400] Attribute report received: measured_value=49251
[0x9311:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=4 command_id=Command.Report_Attributes>
[0x9311:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=31090>>]]
[0x9311:1:0x0400] Attribute report received: measured_value=31090
[0x6b76:1:0x0b04]: async_update
[0x605c:1:0x0b04]: async_update
[0x605c:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=96 command_id=Command.Read_Attributes_rsp>
[0x6b76:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=94 command_id=Command.Read_Attributes_rsp>
[0x0e79:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=190 command_id=Command.Report_Attributes>
[0x0e79:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=6619>>]]
[0x0e79:1:0x0405] Attribute report received: measured_value=6619
[0x04c1:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=160 command_id=Command.Report_Attributes>
[0x04c1:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=49411>>]]
[0x04c1:1:0x0400] Attribute report received: measured_value=49411
[0xf1be:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=62 command_id=Command.Report_Attributes>
[0xf1be:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.false>>]]
[0xf1be:1:0x0006] Attribute report received: on_off=0
[0x9311:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=65 command_id=Command.Report_Attributes>
[0x9311:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=30973>>]]
[0x9311:1:0x0400] Attribute report received: measured_value=30973
[0x6b76:1:0x0b04]: async_update
[0x605c:1:0x0b04]: async_update
[0x6b76:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=99 command_id=Command.Read_Attributes_rsp>
[0x605c:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=101 command_id=Command.Read_Attributes_rsp>
[0x605c:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=101 command_id=Command.Read_Attributes_rsp>
[0x605c:1:0x0b04] ZCL request 0x0001: [[<ReadAttributeRecord attrid=1291 status=Status.SUCCESS value=0>]]
[0x04c1:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=191 command_id=Command.Report_Attributes>
[... cut ...]
[0x04c1:1:0x0400] Attribute report received: measured_value=41699
[0x9311:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=187 command_id=Command.Report_Attributes>
[0x9311:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=28489>>]]
[0x9311:1:0x0400] Attribute report received: measured_value=28489
```
I think there is a lack of support for this device category in ZHA/zigpy.

Thanks

Adminiuga, 2020-09-07:
I can't tell the emberznet version. Once ha beta is cut, new bellows should log version in the friendly format.

Zgp is on to-do list. Won't work on husbzb-1 sticks, not without a firmware update

Hedda, 2020-09-07:
@dieneuser Can you please try to find out the version of Silicon Labs EmberZNet NCP application? 

Silicon Labs EmberZNet NCP application should be 6.x.x.x something from the EmberZnet SDK with that same 6.x.x.x version.

The NCP runs the EmberZNet Zigbee stack and it is the version needed to tell if your current firmware even supports ZGP.

If it is an older version on your specific adapter then ZGP will not be supported by the firmware you are using now.

Similar to husbzb-1 stick it would probably have shipped with an older version that does not support ZGP.

Hedda, 2020-11-19:
FYI bellows and zigpy doesn't yet support ZGP (Zigbee Green Power) but @zoic21 started working on it now, see https://github.com/zigpy/zigpy/pull/519

closed 2021-08-13 by github-actions[bot]

Hedda, 2023-07-07:
> FYI, based on new pull requests for the zigpy project it looks like @nworbneb has now started working on ZGP support for ZHA:
> 
> [zigpy/zigpy#1213](https://github.com/zigpy/zigpy/pull/1213)
> 
> #2414

@nworbneb should this request maybe be reopened now that your initial code for ZGP has been merged?

klodeckl, 2024-06-23:
Any news here? It would be great to use the hue tap with ZHA.

Hedda, 2024-06-23:
> Any news here?

Sounds as if development stalled a bit -> https://github.com/zigpy/zigpy/pull/1282