zigpy/zha-device-handlers #478: [Device Support Request] EnOcean PTM 216Z ZGP Generic Switch Pushbutton Transmitter Module

Hedda, 2020-09-07:
**Is your feature request related to a problem? Please describe.**

Has anyone here tested and can verify if the EnOcean PTM 216Z Zigbee Green Power Generic Switch Pushbutton Transmitter Module works with ZHA in Home Assistant or if require a quirk device handler or since it is a ZGP (Zigbee Green Power) device if it perhaps even require modifications to zigpy and its radio library?

https://www.enocean.com/en/products/enocean_modules_24ghz/ptm-216z/

I believe that this are also the same module used in EnOcean Easyfit Batteryless Single / Double Rocker Wall Switch Model EWSxZG manufactured by EnOcean

Zigbee ID: "GreenPower_7"

https://zigbee.blakadder.com/EnOcean_EWSxZG.html

blakadder Zigbee Device Compatibility Repository mentions this in regards to Zigbee Green Power devices:

"_Green Power is a feature of Zigbee that allows for energy-harvesting technology to be used directly with the Zigbee stack. Green Power devices do not send standard Zigbee messages. To use a Green Power device in your Zigbee network you need a "translator". Currently only a Philips Hue device can serve as a translator. Your Zigbee Green Power device has to always be in range of a Philips Hue device to function._"

Disclaimer: I do not own one of these myself but created the issue on behalf of @dieneuser who posted in the ZGP (Zigbee Green Power) request for zigpy in https://github.com/zigpy/zigpy/issues/341

@dieneuser can hopefully fill in more information here, such as device signature in ZHA, plus which exact Zigbee hardware adapter and firmware he is using with ZHA in Home Assistant?

_Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line._

Anyway, he said he is currently using it deCONZ gateway software (Phoscon OS) which does support it but with these limitations:

_"Note that ZGP devices (like the Hue tap and FoH switches that harvest kinetic energy) currently don't work over the mesh network; they need to be in direct range of the coordinator. Also note that ZGP devices aren't shown in the deCONZ GUI."_

_"These switches use custom values for state.buttonevent please refer to the related GitHub issue pages for more information."_

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices

**Describe the solution you'd like**

Support for ZGP (Zigbee Green Power) devices like  the Hue Tap Switch Model 9290011152X by Philips.

See https://github.com/Koenkk/zigbee-herdsman/pull/92 for proof-of-concept used by zigbee-herdsman and Zigbee2mqtt project:

https://github.com/Koenkk/zigbee-herdsman/pull/92

**Additional context**

ZGP (Zigbee Green Power), a.k.a. "Zigbee GreenPower", is a subset feature of Zigbee that allows for energy-harvesting technology to be used directly with the Zigbee stack. Using Zigbee Green Power, certain home automation devices (end devices) may be powered by energy harvesting solutions, which when enabled introduces the potential of battery-less devices (end devices) operating in a Zigbee network.

https://zigbeealliance.org/solution/green-power/

Hedda, 2020-09-07:
Please also see the related device request https://github.com/zigpy/zha-device-handlers/issues/477 about Hue Tap Switch Model 9290011152X by Philip

Hedda, 2020-09-07:
@dieneuser Can you please try to find out the version of Silicon Labs EmberZNet NCP application? 

Silicon Labs EmberZNet NCP application should be 6.x.x.x something from the EmberZnet SDK with that same 6.x.x.x version.

The NCP runs the EmberZNet Zigbee stack and it is the version needed to tell if your current firmware even supports ZGP.

If it is an older version on your specific adapter then ZGP will not be supported by the firmware you are using now.

Similar to husbzb-1 stick it would probably have shipped with an older version that does not support ZGP.

sildehoop, 2020-10-16:
maybe this is helpful to you https://github.com/Koenkk/zigbee-herdsman/issues/22

sildehoop, 2020-10-16:
I really like to want this switch to be integrated, I have it myself and that is the only device that I cannot transfer to zha from deconz

Hedda, 2020-11-03:
@SildeHoop zigpy doesn't yet support ZGP (Zigbee Green Power) but someone started working on it recently, see https://github.com/zigpy/zigpy/pull/519

closed 2021-08-13 by github-actions[bot]

Hedda, 2023-06-01:
FYI, based on new pull requests for the zigpy project it looks like @nworbneb has now started working on ZGP support for ZHA:

https://github.com/zigpy/zigpy/pull/1213

https://github.com/zigpy/zha-device-handlers/pull/2414

Hedda, 2023-07-07:
> FYI, based on new pull requests for the zigpy project it looks like @nworbneb has now started working on ZGP support for ZHA:
> 
> [zigpy/zigpy#1213](https://github.com/zigpy/zigpy/pull/1213)
> 
> #2414

@nworbneb should this request maybe be reopened now that your initial code for ZGP has been merged?