zigpy/zha-device-handlers #479: [Device Support Request] Hue Bulb E14 Color (929001301301) power last state

vampywiz17, 2020-09-08:
Hello there,

it is possible ti add this type of hue bulb last power state value?  (0x4003 i think so)

```
{
  "node_descriptor": "<Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4107 maximum_buffer_size=71 maximum_incoming_transfer_size=45 server_mask=0 maximum_outgoing_transfer_size=45 descriptor_capability_field=0>",
  "endpoints": {
    "11": {
      "profile_id": 49246,
      "device_type": "0x0210",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "LCT012",
  "class": "zigpy.device.Device"
}
```

TheJulianJES, 2020-09-23:
This should already be supported:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/philips/zllextendedcolorlight.py#L40

Example of a LCT012 model:
[image]
(Clicked "Get Attribute")

When you want to set it to "Restore Last State", enter 0xFF:
[image]

0x00 is always OFF
0x01 is always ON (warm-white)

Adminiuga, 2020-09-25:
this one has a slightly different signature, it is missing the `64513` cluster, other than that looks exactly the same

Adminiuga, 2020-09-25:
the only difference https://github.com/zigpy/zha-device-handlers/blob/1f0bd91c746afe3d1576954ceefd74ee2e37fc63/zhaquirks/philips/zllextendedcolorlight.py#L65

TheJulianJES, 2020-09-26:
Oh, I didn't even look at it, because the model was the same.
I think something like ZLLExtendedColorLight2 (like ZHAExtendedColorLight2) could be done.
But since the model is the same - I'm just wondering, what firmware version are you running? @vampywiz17 
"1.50.2_r30933" seems like it's the latest firmware for the ZLL models.

vampywiz17, 2020-09-26:
@TheJulianJES 

softwareBuildID:  1.50.2_r30933

seems like latest.

PR #546 (2020-10-29): PhilipsOnOffCluster support: LWB014 and LCT012 without cluster 64513
Add ZLLExtendedColorLight2 for LCT012 without 64513 cluster (fixes #479)
Refactored Philips LWB010 quirk into ZLLDimmableLight and added LWB014 support (fixes #544)
commit 8792f183 (2020-10-29): PhilipsOnOffCluster support: LWB014 and LCT012 without cluster 64513 (#546)

* Add ZLLExtendedColorLight2 for LCT012 without 64513 cluster (fixes #479)

* Refactored Philips LWB010 quirk into ZLLDimmableLight and added LWB014 support (fixes #544)
closed 2020-10-29 by Adminiuga