zigpy/zha-device-handlers #480: [BUG] Aqara Battery not reporting for some devices

NobleWolf, 2020-09-11:
**Describe the bug**
Of the 7 Aqara devices I have only 2 are reporting battery level. One that is reporting is a contact sensor, but the other 2 are not reporting. The other that is reporting is a water sensor. They are each connected to a different zigbee repeater. Other non-reporting devices are connected to the same repeaters (Centralite plug) and directly to the coordinator (Conbee 2).

**To Reproduce**
Steps to reproduce the behavior:
1. Connect devices via ZHA
2. Allow devices to complete setup.
3. View Device listings

**Expected behavior**
Battery levels are expected to report.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
I have attempted to reinstall ZHA. After deleting, but before reinstalling I deleted the entities and devices from the core.entity_registry and core.device_registry files. I have restarted multiple times. I have waited 12+hours to allow time for something to happen, but no change.

Adminiuga, 2020-09-11:
do you get sensor entities for battery created?

[image]
[image]

BTW, I'm not sure Centralite plugs are good routers for xiaomi. Didn't have luck with mine, as it was always falling off those. 
Do you get device readings when you trigger the devices?

Adminiuga, 2020-09-11:
Confirm the devices are online and are sending the updates. They should report battery about every hour, but if they fall off the network, then you won't get the battery.
Confirm you get the sensor entities representing the battery

Adminiuga, 2020-09-11:
@dmulcahey should we add all entities associated with the device in the signature?

NobleWolf, 2020-09-11:
They are sensor battery types.
[image]

A couple more are reporting now. Maybe it is just because I have restarted several times?
[image]

I'm just now learning that Aqara devices don't play well with all repeaters. Hopefully they don't fall off. I've read that sometimes you can put the Aqara devices in pair mode and they'll reconnect without other actions.
I'll replace the Centralite plugs with Tradfri when they get back in stock. I found them on this list: https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623/5

Adminiuga, 2020-09-11:
after a restart it may take till next battery report (about an hour) for it to update and pop up in the device list.

dmulcahey, 2020-09-11:
> @dmulcahey should we add all entities associated with the device in the signature?

We could... up to you. Doesn’t matter to me

closed 2021-08-13 by github-actions[bot]