zigpy/zha-device-handlers #482: Device show only one entity

Georgebonas, 2020-09-12:
Hi, I use sonoff bridge tazmotized zha, I have pair it with many devices with no problem. I try to pair with alarm siren but show only 1 binary_sensor entity. Is there any chance to add the rest of it?

Device info
unk_model
by unk_manufacturer
Zigbee Coordinator
Zigbee info
IEEE: 08:6b:d7:ff:fe:fb:5c:03
Nwk: 0x6e35
Device Type: Router
LQI: 255
RSSI: -33
Last Seen: 2020-09-12T11:26:25
Power Source: Mains

Zigbee device signature

{
  "node_descriptor": "<Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4619 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0403",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0502",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0004",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "unk_manufacturer",
  "model": "unk_model",
  "class": "zigpy.device.Device"
}

dmulcahey, 2020-09-12:
There are 2 device actions and 2 services to interact with sirens. You won’t have other entities based on that signature. What do you think is missing?

Georgebonas, 2020-09-13:
Sorry but I new at zigbee. I'm afraid  that something obvious might be missing.

Georgebonas, 2020-09-15:
> There are 2 device actions and 2 services to interact with sirens. You won’t have other entities based on that signature. What do you think is missing?

Sorry but I new at zigbee. I'm afraid that something obvious might be missing.

dmulcahey, 2020-09-27:
There are no entities for the siren. Just services and device actions in HA to interact with them. You can research both items in the HA documentation.

closed 2020-09-27 by dmulcahey

Georgebonas, 2020-10-20:
The solution is that you call a service. Just replace ieee

home_alarm_on:
  sequence:
    service: zha.warning_device_warn
    data:
      duration: 65535
      duty_cycle: 2
      ieee: '08:6b:d7:ff:fe:fb:5c:03'
      intensity: 2
      level: 3
      mode: 1
      strobe: 3