zigpy/zha-device-handlers #489: [Device Support Request]  Zigbee 3 Gang Remote TS0043

stast1, 2020-09-15:
**Is your feature request related to a problem? Please describe.**
No zha_event events on button click

**Describe the solution you'd like**
Add support for this device

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0x000a"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_bi6lpsew",
  "model": "TS0043",
  "class": "zigpy.device.Device"
}
```
**Additional context**
```
2020-09-15 17:58:20 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Broadcast request seq 15
2020-09-15 17:58:20 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:00:00:00:00:00:ff:fc, <BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>, 0, 0, 54, 0, 0, 0, b'\x0f<\x00')
2020-09-15 17:58:20 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11/\x00\x00\x00\x00\x00\x00\xff\xfc\xff\xfc\x00\x00\x006\x00\x00\x00\x00\x0f<\x00'
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b/\xff\xfe\x00\x00\x00'
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0xfffe: TXStatus.SUCCESS after 0 tries. Discovery Status: DiscoveryStatus.SUCCESS, Frame #47
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.api] at command: NJ (60,)
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.api] Command at (b'NJ', b'<')
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x080NJ<'
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x880NJ\x00'
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.api] Frame received: at_response
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.api] at command: AC ()
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.api] Command at (b'AC', b'')
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x081AC'
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x881AC\x00'
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.api] Frame received: at_response
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.api] at command: CB (2,)
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.api] Command at (b'CB', b'\x02')
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x082CB\x02'
2020-09-15 17:58:21 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x882CB\x00'
[... 793 lines cut ...]
2020-09-15 18:00:06 DEBUG (MainThread) [zigpy.zcl] [0x610b:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=54 command_id=Command.Read_Attributes>
2020-09-15 18:00:06 DEBUG (MainThread) [zigpy.zcl] [0x610b:1:0x000a] ZCL request 0x0000: [[0]]
2020-09-15 18:00:06 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request tsn #54: b'183601000000e27693f326'
2020-09-15 18:00:06 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:15:8d:00:02:71:22:d9, 0x610B, 1, 1, 10, 260, 0, 0, b'\x186\x01\x00\x00\x00\xe2v\x93\xf3&')
2020-09-15 18:00:06 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11T\x00\x15\x8d\x00\x02q"\xd9a\x0b\x01\x01\x00\n\x01\x04\x00\x00\x186\x01\x00\x00\x00\xe2v\x93\xf3&'
2020-09-15 18:00:06 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8bTa\x0b\x00\x00\x00'
2020-09-15 18:00:06 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2020-09-15 18:00:06 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0x610b: TXStatus.SUCCESS after 0 tries. Discovery Status: DiscoveryStatus.SUCCESS, Frame #84
2020-09-15 18:00:08 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91`\xa4#\xff\xfe\xaeH}y`\x03\x01\x00\x06\x01\x04\x00\x01(\xfd\x00'
2020-09-15 18:00:08 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-09-15 18:00:08 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (60:a4:23:ff:fe:ae:48:7d, 0x7960, 1, 6, 0, b'0128fd00')
2020-09-15 18:00:08 DEBUG (MainThread) [zigpy.zcl] [0x7960:3:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=40 command_id=253>
2020-09-15 18:00:08 WARNING (MainThread) [zigpy.zcl] [0x7960:3:0x0006] Unknown cluster-specific command 253
2020-09-15 18:00:08 DEBUG (MainThread) [zigpy.zcl] [0x7960:3:0x0006] ZCL request 0x00fd: b'\x00'
2020-09-15 18:00:08 DEBUG (MainThread) [zigpy.zcl] [0x7960:3:0x0006] No handler for cluster command 253
2020-09-15 18:00:08 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7960:3:0x0006]: received '253' command with b'\x00' args on cluster_id '6' tsn '40'
2020-09-15 18:00:19 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\xff\xff\xff\xff\xff\xff\xff\xff\x91M\x01\x01\x01\x01\x01\x04\x00\x18\x03\n\x05\x05#\x00\x00\x05\x00'
2020-09-15 18:00:19 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-09-15 18:00:19 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (ff:ff:ff:ff:ff:ff:ff:ff, 0x914d, 1, 257, 0, b'18030a05052300000500')
2020-09-15 18:00:19 DEBUG (MainThread) [zigpy.zcl] [0x914d:1:0x0101] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Report_Attributes>
2020-09-15 18:00:19 DEBUG (MainThread) [zigpy.zcl] [0x914d:1:0x0101] ZCL request 0x000a: [[Attribute(attrid=1285, value=<TypeValue type=uint32_t, value=327680>)]]
2020-09-15 18:00:19 DEBUG (MainThread) [zigpy.zcl] [0x914d:1:0x0101] Attribute report received: 1285=327680
```

[Zigbee 3 Gang Remote TS0043](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3220)

closed 2021-08-13 by github-actions[bot]

jjarven, 2022-12-03:
TS0043 by _TZ3000_sj7jbgks

- Does not work with the existing TS0043 3-button quirk.
But other sources suggested to copy TS0044 quirk and replace TS0044 references to TS0043, which would enable the device and I can confirm this helps.

Thus, how to add this to device handler?

MattWestb, 2022-12-03:
The original posted device with manufacturer ID "_TZ3000_bi6lpsew" shall working out of the box with ZHA for over one year.

If you posting the device signature we can looking how your device is being done and perhaps getting it working 2.

jjarven, 2022-12-04:
Hope these are enough;

"manufacturer": "_TZ3000_sj7jbgks",
"model": "TS0043",
"name": "_TZ3000_sj7jbgks TS0043",

"signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0006",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0006",
            "0xe000"
          ],
          "out_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "2": {
          "profile_id": 260,
          "device_type": "0x0006",
          "in_clusters": [
            "0x0001",
            "0x0006"
          ],
          "out_clusters": []
        },
        "3": {
          "profile_id": 260,
          "device_type": "0x0006",
          "in_clusters": [
            "0x0001",
            "0x0006"
          ],
          "out_clusters": []
        },
        "4": {
          "profile_id": 260,
          "device_type": "0x0006",
          "in_clusters": [
            "0x0001",
            "0x0006"
          ],
          "out_clusters": []
        }
      }

drudgebg, 2022-12-12:
I have same problem with 3 button device, TS0043 by _TZ3000_sj7jbgks

MacroScooter, 2022-12-20:
Same here! 
Batterylevel exact, Buttons not working

TS0043 by _TZ3000_sj7jbgks 3 button scene switch

Followed: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6213#issue-1315016976

drudgebg, 2022-12-20:
> Same here! Batterylevel exact, Buttons not working
> 
> TS0043 by _TZ3000_sj7jbgks 3 button scene switch
> 
> Followed: [dresden-elektronik/deconz-rest-plugin#6213 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6213#issue-1315016976)

Custom quirk that work for TS0043 by _TZ3000_sj7jbgks:
`"""Tuya TS004F devices."""
from __future__ import annotations

import asyncio
import logging

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    ALT_SHORT_PRESS,
    BUTTON,
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    BUTTON_4,
    CLUSTER_ID,
    COMMAND,
    COMMAND_MOVE,
    COMMAND_MOVE_COLOR_TEMP,
    COMMAND_MOVE_SATURATION,
    COMMAND_OFF,
    COMMAND_ON,
    COMMAND_STEP,
    COMMAND_STOP,
    COMMAND_STOP_MOVE_STEP,
    COMMAND_TOGGLE,
    DEVICE_TYPE,
    DIM_DOWN,
    DIM_UP,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LEFT,
    LONG_PRESS,
    LONG_RELEASE,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PARAMS,
    PROFILE_ID,
    RIGHT,
    ROTATED,
    ROTATED_FAST,
    ROTATED_SLOW,
    SHORT_PRESS,
    TURN_OFF,
    TURN_ON,
)
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaSmartRemoteOnOffCluster,
    TuyaZBOnOffAttributeCluster,
    TuyaZBExternalSwitchTypeCluster,
)

_LOGGER = logging.getLogger(__name__)

class TuyaSmartRemote0043_4(CustomDevice):
    """Tuya 4-button remote control ("manufacturer": "_TZ3000_sj7jbgks")."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6, 57344], output_clusters=[10, 25]))
        # SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=0, device_version=1, input_clusters=[1, 6], output_clusters=[])
        # SizePrefixedSimpleDescriptor(endpoint=3, profile=260, device_type=0, device_version=1, input_clusters=[1, 6], output_clusters=[])
		# SizePrefixedSimpleDescriptor(endpoint=4, profile=260, device_type=0, device_version=1, input_clusters=[1, 6], output_clusters=[])
        MODEL: "TS0043",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
			4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                    TuyaZBE000Cluster.cluster_id
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    PowerConfiguration.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    PowerConfiguration.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
			4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    PowerConfiguration.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
        (SHORT_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: DOUBLE_PRESS},
        (SHORT_PRESS, BUTTON_3): {ENDPOINT_ID: 3, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_3): {ENDPOINT_ID: 3, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_3): {ENDPOINT_ID: 3, COMMAND: DOUBLE_PRESS},
    }`

MacroScooter, 2022-12-20:
Thank you for your help!

Unfortunately I posted in the wrong category while seeking for the DDF- :-(
I`m searching for a Phoscon/Deconz DDF

Sorry