zigpy/zha-device-handlers #493: [Device Support Request] Profalux Covers

glamirand, 2020-09-19:
Hi,

I'm using a ZiGate with homeAssistant to control my Profalux Covers but it seems not woring with the native integration / device discovery.

I'd like to install ZHA Device handler so I can customize some clusters but I didn't find any kind of information on how to just install it on homeassistant?!

Sorry for that stupid question but I'm fairly new to HA and I'm sure I'm just missing "that" point so I can understand how to setup it.

Thanks guys!

pipiche38, 2020-09-19:
I might help, as I have 7 profalux shutter and I'm involved in the Domoticz-Zigpy plugin.
For whom needs information let me know, but you can already find a number of information on my wiki page
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Profalux-corner.md

glamirand, 2020-09-19:
I would really appreciate your help!
First could you clarify to me which tools do I need to control those shutters?
I tried both the native ZHA integration and also doudz plugin.
At the moment it's not clear to me if I need those 2 components or if I have to choose only one.

I just managed to see my shutter as a light and managed to control it's position through the brightness_pct option : "HiHa!"
That's a good first quickwin but how to manage to control it as a real shutter (and having proper UI as well..)?

pipiche38, 2020-09-19:
The profalux must be controlled via cluster 0x0008 (dixit profalux )

The integration work should be provided by zha-device-handler and so should provide the interface between HA and the profalux device

For info the profalux device do not provide Model Name (0x0000/0x0005) nor the manufacturer name (0x0000/0x0004). So you should get the manufacturer code via the node descriptor 
request

Binding on cluster 0x0008 must be done and configure reporting as well.

However when pairing a shutter , 2 devices will join the network the shutter and also the remote . Do not bind the remote otherwise the remote will be disconnected from the shutter.

@glamirand i can help someone developing the quirk but not doing the integration job ( for now )

Cc:
@Adminiuga

glamirand, 2020-09-19:
Thanks for those precisions.
So if I understand well I may not need doudz plugin, only the native ZHA intergration combined with that quirk should be sufficient, right?
I have read the "how to" create a quirk and I think I understood how it works, however I absolutly didn't understood how to include the fils in home assistant!
Are those quirks already included in the ZHA intergration, so a new quirk absolutly requires the "integration job" you mention? I thought I read somewhere I can add a custom quirk locally, am I wrong?
Anyway, could you provide me some guidelines on how to deal with those quirks in my case? I think I may manage to write it correctly but I just didn't understood where and how to do it!

Thx

dmulcahey, 2020-10-10:
please keep all conversation related to supporting this device in #348

closed 2020-10-10 by dmulcahey