zigpy/zha-device-handlers #497: [Device Support Request] Legrand Dimmer switch w/o neutral for sw_build_id=002e

smmoroz, 2020-09-22:
Hi, guy!

I have Legrand Dimmer switch w/o neutral which is not recognized neither as DimmerWithoutNeutral nor as DimmerWithoutNeutral2.

The device runs this firmware:
```
hw_version = 7
stack_version = 66
sw_build_id = 002e
```

In this firmware Legrand changed the device type for endpoint=242 from 0x0061 to 0x0066, and they also added a few more clusters:
```
endpoint=1: out: 0x0006, 0x0005
endpoint=242: in: 0x0021
```

Here is a complete device signature
```
{
  "node_descriptor": "<NodeDescriptor byte1=17 byte2=64 mac_capability_flags=142 manufacturer_code=4129 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=10752 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000f",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0019",
        "0xfc01"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Dimmer switch w/o neutral",
  "class": "zigpy.device.Device"
}
```

I've updated the definition of DimmerWithoutNeutral2 (see below) and now the device is properly recognized.

```
class DimmerWithoutNeutral2(DimmerWithoutNeutral):
    """Dimmer switch w/o neutral 2."""

    signature = {
        #  <SimpleDescriptor endpoint=1 profile=260 device_type=256
        # device_version=1
        # input_clusters=[0, 3, 4, 8, 6, 5, 15, 64513]
        # output_clusters=[0, 64513, 25]>
        MODELS_INFO: [(f" {LEGRAND}", " Dimmer switch w/o neutral")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Scenes.cluster_id,
                    BinaryInput.cluster_id,
                    MANUFACTURER_SPECIFIC_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                    MANUFACTURER_SPECIFIC_CLUSTER_ID,
                    Ota.cluster_id,
                    OnOff.cluster_id,
                    Scenes.cluster_id,
                ],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0066,
                INPUT_CLUSTERS: [0x0021],
                OUTPUT_CLUSTERS: [0x0021],
            },
        },
    }
```

Please add this quirk to master.

pipiche38, 2020-09-23:
Please be aware that you have to use the Manufacturer Specific cluster to enable the Dimmer mode.
In addition you need to have a certain Firmware level to get access to that. (Most likely the fresh device do not have this and required a firmware update).

More information here:
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Legrand-Netatmo-corner.md

smmoroz, 2020-09-23:
Absolutely. I already did that. I tested dimmer and leds. Everything works. I only had to tweak zha api as described here: https://github.com/zigpy/zigpy/issues/325#issuecomment-612678210

smmoroz, 2020-11-03:
@pipiche38 , is there any chance to get this feature in master? I didn't write code for a while, but if needed I can make PR ;)

pipiche38, 2021-08-06:
@smmoroz this is up to zigpy to do that work. Even with a more recent firmware we can configure the ballast (in other term we can configure the MIn and Max of the dimmer)

smmoroz, 2021-08-18:
@pipiche38 we still need custom handler to support Legrand specific clusters. For example, this is needed to enable dimmer mode at all.

Is there any chance to have Legrand specific handlers in master?

pipiche38, 2021-08-18:
I’m not working on zigpy due to different understanding with the dev, but would be please to give you the details which are documented here

https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Corner_Legrand-Netatmo.md#cluster-0xfc01

smmoroz, 2021-08-18:
Yes, I know that. I even have a kind of ugly patch. Should I make PR?