zigpy/zha-device-handlers #499: [BUG]

basnagel, 2020-09-23:
**Describe the bug**
Adding a OSRAM Lightify Switch Mini to my network is not working out after quirk was added some time ago.

Before quirk: I picked up zha_events.

After quirk: The remote switches everything(switched bulbs cover) in my network on and off.

 

**To Reproduce**
Steps to reproduce the behavior:
Removed remote, deletes entries from zha.storage and other HA files I could find the remote.

Device signature:
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4364, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {},
  "manufacturer": "OSRAM",
  "model": "Lightify Switch Mini",
  "class": "zhaquirks.osram.switchmini.OsramSwitchMini"
}

Manage cluster is empty.
[image]

**Expected behavior**
A remote that switches the stuff I want, not everything(via zha_event for my usage)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
[osramminiadd.txt](https://github.com/zigpy/zha-device-handlers/files/5267371/osramminiadd.txt)

basnagel, 2020-09-28:
I tried further and after removing the quirks file for the switch I got a working remote again catching the zha_events.

[image]

Is there a setting to choose not to use a quirks, besides removing the file from the site-packages?

I have added the debug log from this join

[osramswitchmini.txt](https://github.com/zigpy/zha-device-handlers/files/5290934/osramswitchmini.txt)

Adminiuga, 2020-09-28:
@dmulcahey that Quirk indeed doesn't have any `replacement` https://github.com/zigpy/zha-device-handlers/blob/1f0bd91c746afe3d1576954ceefd74ee2e37fc63/zhaquirks/osram/switchmini.py#L45 which would cause the new endpoints to be empty.

Adminiuga, 2020-09-28:
@basnagel can you copy this file into your environment and confirm if it is working as expected?

Adminiuga, 2020-09-28:
https://github.com/Adminiuga/zha-device-handlers/blob/fixes/osram-mini/zhaquirks/osram/switchmini.py

basnagel, 2020-09-28:
After a reboot and a reconfigure the switchmini is working as intended.

I can select device-> shortpress etc as triggers, clusters are present and most import, not everything in my network is switching on and off.

Log from the reconfigure.

[osramswitchminifixed.txt](https://github.com/zigpy/zha-device-handlers/files/5294325/osramswitchminifixed.txt)

PR #506 (2020-09-29): Add replacement info for Osram switch mini quirk
Add missing replacement info for Osram mini quirk.

Fixes #499
closed 2020-09-29 by Adminiuga