zigpy/zha-device-handlers #502: [BUG] IKEA Fyrtur blinds command delay

Jdruwe, 2020-09-24:
**Bug/issue**

I am having delay issues when using the [Tradfri Open/Close Remote](https://zigbee.blakadder.com/Ikea_E1766.html#:~:text=Pairing,will%20stop%20shinning%20when%20done) in combination with the [Fyrtur Block-out Roller Blind](https://zigbee.blakadder.com/I), I have 2 next to each other in my setup. Both the blinds are grouped together in 'cover.bedroom_finn'

[image]

In the group binding (home assistant) options of the Tradfri remote I binded the WindowCovering group to the custom 'cover.bedroom_finn' group mentioned above.

[image]

In the video below you can see both of the start going down almost instantly but when I click the ‘stop’ button the left one take a couple seconds longer to stop. When I want them to both go up, the left one takes almost 4 seconds longer to get going:

https://www.youtube.com/watch?v=pX8AkwHNXos

I am not sure if the delay has something to do we my zigbee network quality as the ikea bulb you see below always reacts instantly.

[image]

In the package from Ikea there was also a signal repeater included, I reset the repeater and have it plugged in an outlet nearby but I am not sure if I need further bindings to ensure it actually repeating. I would assume that’s it just works out of the box when reset as it just needs to repeat the zigbee messages.

**Logging**

WindowCovering -> config_status = bitmap8.2|1

```
2020-09-24 11:13:56 DEBUG (MainThread) [zigpy.device] [0x226a] Extending timeout for 0xdb request
2020-09-24 11:13:56 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 219 under 220 request id, data: b'00db000700'
2020-09-24 11:13:56 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 220, 0, <DeconzAddressEndpoint address_mode=2 address=0x226A endpoint=1>, 260, 258, 1, b'\x00\xdb\x00\x07\x00', 2, 0)
2020-09-24 11:13:56 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 220]
2020-09-24 11:13:56 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-09-24 11:13:56 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-09-24 11:13:56 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 220: 00
2020-09-24 11:13:56 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xdc 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x226a endpoint=1>, status: 0x00
2020-09-24 11:14:00 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.processor_use, old_state=<state sensor.processor_use=1; unit_of_measurement=%, friendly_name=Processor use, icon=mdi:cpu-64-bit @ 2020-09-24T11:13:30.582167+02:00>, new_state=<state sensor.processor_use=2; unit_of_measurement=%, friendly_name=Processor use, icon=mdi:cpu-64-bit @ 2020-09-24T11:14:00.583169+02:00>>
2020-09-24 11:14:02 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-09-24 11:14:02 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-09-24 11:14:02 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x226a>, 1, 260, 258, b'\x08\xdb\x01\x07\x00\x00\x18\x03', 0, 175, 255, 230, 244, 254, 0, -55]
2020-09-24 11:14:02 DEBUG (MainThread) [zigpy.zcl] [0x226a:1:0x0102] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=219 command_id=Command.Read_Attributes_rsp>
2020-09-24 11:14:02 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x226a>, ep: 1, profile: 0x0104, cluster_id: 0x0102, data: b'08db010700001803'
2020-09-24 11:14:02 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x226A:1:0x0102]: Attribute report 'Window Covering'[config_status] = bitmap8.2|1
2020-09-24 11:14:02 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [258] cluster_type: [in] endpoint_id: [1] attribute: [7] manufacturer: [None] response: [bitmap8.2|1] failure: [{}],
```

WindowCovering -> window_covering_mode = 0

```
2020-09-24 11:15:10 DEBUG (MainThread) [zigpy.device] [0x226a] Extending timeout for 0xe7 request
2020-09-24 11:15:10 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 231 under 232 request id, data: b'00e7001700'
2020-09-24 11:15:10 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 232, 0, <DeconzAddressEndpoint address_mode=2 address=0x226A endpoint=1>, 260, 258, 1, b'\x00\xe7\x00\x17\x00', 2, 0)
2020-09-24 11:15:10 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 232]
2020-09-24 11:15:10 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-09-24 11:15:10 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-09-24 11:15:10 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 232: 00
2020-09-24 11:15:10 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xe8 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x226a endpoint=1>, status: 0x00
2020-09-24 11:15:13 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-09-24 11:15:13 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-09-24 11:15:13 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x226a>, 1, 260, 258, b'\x08\xe7\x01\x17\x00\x00\x18\x00', 0, 175, 255, 242, 244, 254, 0, -55]
2020-09-24 11:15:13 DEBUG (MainThread) [zigpy.zcl] [0x226a:1:0x0102] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=231 command_id=Command.Read_Attributes_rsp>
2020-09-24 11:15:13 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x226a>, ep: 1, profile: 0x0104, cluster_id: 0x0102, data: b'08e7011700001800'
2020-09-24 11:15:13 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x226A:1:0x0102]: Attribute report 'Window Covering'[window_covering_mode] = 0
2020-09-24 11:15:13 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [258] cluster_type: [in] endpoint_id: [1] attribute: [23] manufacturer: [None] response: [0] failure: [{}],
```

Closing and making it stop by clicking a button again (while it's going down, not holding down any buttons on the remote):

```
2020-09-24 11:32:06 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-09-24 11:32:06 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-09-24 11:32:06 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [26, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.GROUP address=0x0009>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x5a64>, 1, 260, 258, b'\x01\x0e\x01', 0, 175, 151, 188, 246, 254, 0, -75]
2020-09-24 11:32:06 DEBUG (MainThread) [zigpy.zcl] [0x5a64:1:0x0102] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=14 command_id=1>
[... 634 lines cut ...]
2020-09-24 11:32:19 DEBUG (MainThread) [zigpy.zcl] [0x55f7:1:0x0102] Attribute report received: current_position_lift_percentage=29
[... cut ...]
As you can see in the logs the right blinds is a lot faster in this test, the left blind is setting to position to '1' when the right blind was already at position '25'

_setting position: 25 (right blind)
setting position: 1 (left blind)_

At some point I click the button again to make the blinds stop, this is the end position for both of them, of course the right blind is a 'more closed' than the left blind:

_setting position: 54 (right blind)
setting position: 29 (left blind)_

Is there any way to reduce this delay I am experiencing? Thanks in advance!

blair287, 2020-09-26:
This is the same issue I had for months but now my delay is minimal not sure why it's changed.

Jdruwe, 2020-09-27:
Do you have OTA updates enabled for ikea? Can you share you firmware versions:

- Repeaters: 0x22005631
- Blinds: 0x22009631
- Close/open button: 0x22010631

blair287, 2020-09-27:
> @blair287 Do you have OTA updates enabled for ikea? Can you share you firmware versions? These are mine:
> 
> * Repeaters: 0x22005631
> * Blinds: 0x22009631
> * Close/open button: 0x22010631

I do have OTA updates enabled last time I checked they were up to date but don't know what versions.

Jdruwe, 2020-09-27:
> > @blair287 Do you have OTA updates enabled for ikea? Can you share you firmware versions? These are mine:
> > 
> > * Repeaters: 0x22005631
> > * Blinds: 0x22009631
> > * Close/open button: 0x22010631
> 
> I do have OTA updates enabled last time I checked they were up to date but don't know what versions.
> 
> At work will get versions when home.

Thx, eager to know your versions :D

blair287, 2020-10-01:
> Thx, eager to know your versions :D

[image]

Jdruwe, 2020-10-01:
@blair287 I am on the same version it seems

blair287, 2020-10-09:
> @blair287 I am on the same version it seems

My delay is back 5 seconds between issuing command and it moving.

All I did was update from 114.4 to 116.2.

Jdruwe, 2020-10-10:
@blair287 
> All I did was update from 114.4 to 116.2.

Yep, very annoying. We can't be the only 2 with this issue right?

blair287, 2020-10-10:
> @blair287
> 
> > All I did was update from 114.4 to 116.2.
> 
> Yep, very annoying. We can't be the only 2 with this issue right?

ive found a few people with the delay too not sure why its there its very annoying feels like they dont work properly never had an issue with smartthings and this blind.

if you use the ikea hub it seems battery life is not shown in HA it seems there is no good integration for these blinds into HA shame they are good.

APayerl, 2020-10-30:
I can verify that this is an issue.
I purchased 2 and mounted them side by side in our livingroom. 
First they were both quick (reacted within 1 second after click) but as time went they became slower to react. Now I have changed 1 of them to a new one and this one is again very quick but it is getting slower to react also... 
Not sure if it's a blind thing or a ZHA thing... 
Though I think it's a ZHA thing since they became instantly more slow when I switched from deconz to ZHA!

Adminiuga, 2020-10-31:
Logs? Radio?

Jdruwe, 2020-11-02:
@Adminiuga I posted logs in the original post

Adminiuga, 2020-11-02:
> Logs ? Radio?

@Jdruwe those were meant for @APayerl .
You provided a detailed descriptions, thank you for that. But I don't know how to help you. In the logs I see only attribute reports from the blinds --> blinds are moving.

Per your description (+ logs) it sounds like you bound remote directly to the blind group. In this case ZHA/zigpy_deconz are completely out of the control loop and blinds are reacting to the command sent by the remote and since it is a Group command, this command is sent out as a broadcast.

Have you tried issuing the stop command not from the remote, but from ZHA using `zha.issue_zigbee_group_command` service? Just would be interested to know if there's any difference?

Jdruwe, 2020-11-02:
Hmmm never used `zha.issue_zigbee_group_command` before, worth a try. I see that I need to provide the following:

[image]

Is there an easy way to find these out?

Adminiuga, 2020-11-02:
similar to what Remote sends in the zha_event:
`device_ieee=68:0a:e2:ff:fe:41:94:c3, unique_id=68:0a:e2:ff:fe:41:94:c3:1:0x0102, endpoint_id=1, cluster_id=258, command=down_close, args=[]`

Should be something like
```yaml
group: <your group number, should be able to see it in the groups>
cluster_id: 258
command: 2
```

All commands are in 
[image]

so the stop command id is `0x02` you can find more details about other commands in https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf page 451 section 7.4.2.2

Jdruwe, 2020-11-02:
I just unbind the ikea remote from the windowcovering cluster, is this an issue? 258 for cluster_id doesn't seem to do anything.

Adminiuga, 2020-11-02:
no need to unbind the remote.
Yes, in your case the group id is 0x0009 hex which is still 9 in decimal, so full command to **stop** the move would be:
```yaml
group: 9
cluster_id: 258
command: 2
```
may have to add `args: []` but i don't think that's needed.

Jdruwe, 2020-11-02:
I did a rebind for the remote and used the values you provided:

```
2020-11-02 20:28:22 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=zha, service=issue_zigbee_group_command, service_data=group=9, cluster_id=258, command=2, args=[]>
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee multicast with tsn 41 under 42 request id, data: b'012902'
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (17, 42, 0, <DeconzAddressEndpoint address_mode=ADDRESS_MODE.GROUP address=0x0009 endpoint=None>, 260, 258, 1, b'\x01)\x02', 2, 0)
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 42]
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [26, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.GROUP address=0x0009>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, 260, 258, b'\x01)\x02', 0, 175, 137, 69, 99, 49, 1, 80]
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy.device] Ignoring message (b'012902') on cluster 258: unknown endpoint or cluster id: 'No cluster ID 0x0102 on (00:21:2e:ff:ff:04:f7:fd, 1)'
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, ep: 1, profile: 0x0104, cluster_id: 0x0102, data: b'012902'
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 42: 00
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x2a 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.GROUP address=0x0009 endpoint=None>, status: 0x00
2020-11-02 20:28:22 DEBUG (MainThread) [homeassistant.components.zha.api] Issued group command for: cluster_id: [258] command: [2] args: [] manufacturer: [None] response: [2, <Status.SUCCESS: 0>]
```

Adminiuga, 2020-11-02:
>This one seems a bit weird:
```
2020-11-02 20:28:22 DEBUG (MainThread) [zigpy.device] Ignoring message (b'012902') on cluster 258: unknown endpoint or cluster id: 'No cluster ID 0x0102 on (00:21:2e:ff:ff:04:f7:fd, 1)'
```

You can ignore this one. Essentially ConBee is part of any groups and this is just a loopback from the command you sent out. But coordinator itself does not have this cluster thus the error message

>Also when I trigger it using the remote (which does work) I see this:
```
2020-11-02 20:29:46 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=68:0a:e2:ff:fe:41:9
```
`zha_event` does do translation of the command_id to string, if available, thus the string. I don't remember if we allow "command_names" as the command id, might add it later.

So when you issue this command, do the blind stop moving?

Jdruwe, 2020-11-02:
> So when you issue this command, do the blind stop moving?

No, nothing happens, not sure what I forgot.

MattWestb, 2020-11-02:
Some fundamental with IKEAs blinds. The blinds is one end device and the remot 2 so both can / is sleeping. Then sending one command from the remote the blinds must being awake for receiving it and executing it. IKEA is using the repeater that is first bonded to the remote (with groupe). After that you should binding the blinds to the repeater (with the remote if being in IKEA GW) to the same groupe. If using ZHA is best only opening the repeater for joining then paring the blinds so they is using is as router.
Then its done and you is sending one command to the blind thru the group the repeater is "holding" it and waiting for the blind waking up and fetching the command and executing the command.
If trying sending some commands directly from the remote to the blinds without the repeter you need mutch luck for getting it working without delays and missed commands.
Then you have more devices that is acting as routers in the near it can working without the repeater  but i think its have more unktion in the firmware then only acting a router in the mesh (If its have groupe cluster it should being added to the group of the bonded remote).

TXStatus.NWK_ROUTE_DISCOVERY_FAILED = the coordinator cant finding the route to the device then trying sending the command.
If the blindes is connected to the repeater the coordinator is sending the command to the repeater and the repeater is waiting for the blinds to check in and catching the command.

Adminiuga, 2020-11-02:
yes, but the group command request id was 152 and that one was sent out successfuly
```
2020-11-02 21:10:40 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=zha, service=issue_zigbee_group_command, service_data=group=9, cluster_id=258, command=1>
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee multicast with tsn 151 under 152 request id, data: b'019701'
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (17, 152, 0, <DeconzAddressEndpoint address_mode=ADDRESS_MODE.GROUP address=0x0009 endpoint=None>, 260, 258, 1, b'\x01\x97\x01', 2, 0)
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 152]
2020-11-02 21:10:40 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.processor_use, old_state=<state sensor.processor_use=5; unit_of_measurement=%, friendly_name=Processor use (percent), icon=mdi:cpu-64-bit @ 2020-11-02T21:09:40.617740+01:00>, new_state=<state sensor.processor_use=1; unit_of_measurement=%, friendly_name=Processor use (percent), icon=mdi:cpu-64-bit @ 2020-11-02T21:10:40.590668+01:00>>
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [26, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.GROUP address=0x0009>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, 260, 258, b'\x01\x97\x01', 0, 175, 109, 234, 104, 49, 1, 77]
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy.device] Ignoring message (b'019701') on cluster 258: unknown endpoint or cluster id: 'No cluster ID 0x0102 on (00:21:2e:ff:ff:04:f7:fd, 1)'
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, ep: 1, profile: 0x0104, cluster_id: 0x0102, data: b'019701'
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 152: 00
2020-11-02 21:10:40 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x98 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.GROUP address=0x0009 endpoint=None>, status: 0x00
2020-11-02 21:10:40 DEBUG (MainThread) [homeassistant.components.zha.api] Issued group command for: cluster_id: [258] command: [1] args: [] manufacturer: [None] response: [1, <Status.SUCCESS: 0>]
```

the failed with route discovery confirmations, are from the requests prior to it, not present in this log.
Strange that the request are not working, the group requests are  broadcasts, so should be delivered to the end devices as well.

But you have an interesting point, of joining both blinds and the remote through the same repeater, which would act as the same parent for all these devices.

Adminiuga, 2020-11-02:
actually those two failed requests are related to HUE bulbs, probably state polling.

MattWestb, 2020-11-02:
Sorry for not reading the log.
If using the blinds without coordinator then its one must having one repeater also with more then one blinds and more remotes its must have one repeater or more for working. I can being that its working with normal routers if they is nice behaving.
The "forcing joining router" is great i have doing it with all my xaiomi sensor so they have good routers and not being lost.
Most HUE routers dont like IKEA end devices and normalcy dont updating the child table the children is changing parents. (I have one removed ebd device that is reported in ZHA maps as unknown in zigbee.DB and coming from my HUE bulb and is going away then repower the bulb).

Is the IKEA repeater have group cluster or its only basic cluster ?
Some have saying they is only one router without functions but i think it can being more baked in the firmware like hocking up the manufacturer and device id and acting as one gateway / routers for thesis commands.

Jdruwe, 2020-11-02:
I am not sure what you mean by 'forcing joining router'. Before I discovered the ikea devices (remote, blinds and repeaters) within the ZHA integration I reset everything. I haven't done anything special to create a relationship between remote and repeater or repeater and blinds, I just thought the repeater acted as an extra zigbee repeater.

MattWestb, 2020-11-02:
In ZHA open one device you have:
```
Configure device
add devices via this device
Zigbee signature
manage cluster
remove device
```
Clicking on "add devices via this device" and ZHA is opening this device for joining devices and no other router and the coordinator.

Adminiuga, 2020-11-02:
> I just thought the repeater acted as an extra zigbee repeater

Repeater acts as a repeater, but that won't make existing "battery" operated devices to switch to it, unless their previous parent goes down. 

You would need to reset the device blind/remote and when joining, open the IKEA repeater only for joining via as per https://github.com/zigpy/zha-device-handlers/issues/502#issuecomment-720735023
1. Pick ikea repeater
2. click "add devices via this device"
3. initiate pairing on the device you are re-joining: remote/blind

Do not remove the devices from ZHA, just factory reset them. This would keep all configured customization for the devices intact.

Jdruwe, 2020-11-02:
Ok I'll try that just one small question, by 'just factory reset them' you mean like ikea has included in the manual? Wouldn't I lose for example the max length of my ikea covers?

Adminiuga, 2020-11-02:
the max length will be lost, yes.  Would need recalibration again.

Jdruwe, 2020-11-02:
Ok just start completely clean I removed all devices and factory reset everything (blinds, remote and repeaters) + added the re[eaters first as usual and added blinds and remote using the "add devices via this device" option from the repeater. It seems to work faster now. Will test it the next days, thank you soo much guys for all the help, greatly appreciated!

One last question: I linked the remote to the blinds using the group binding select option in the manage cluster option from the remote, that is still the way to go right? It just now using the repeater behind the scenes because we explicitly asked it to do so?

Adminiuga, 2020-11-02:
> One last question: I linked the remote to the blinds using the group binding select option in the manage cluster option from the remote, that is still the way to go right? It's just now using the repeater behind the scenes because we explicitly asked it to do so?

Yes. The idea is that all devices are connected to the same parent and when repeater receives the group command it is going to keep it and notify both blinds about the command, when the blinds do next checkin. 
Very possible, ikea is doing special handling of the blinds connected to it 🤷 Do the blinds respond to group commands issued from ZHA with issue_zigbee_group_command now?

@MattWestb great idea connecting all blinds to the same parent 👍

MattWestb, 2020-11-03:
Great work @Jdruwe !!
And thanks Adminiuga.

Then starting building the network is best starting with the routers devices so they can build the mesh so you getting good coverage and its can stabilization. The great thing with ZHA is the possibility "forcing" with router to joining devices. I was having problems with 2 Xaiomi sensors that didn't like being paired to mw IKEA ZN3 bulbs but gladly paired with some "chinese ZB3) led rives (HOMA) but after some days they was moving to better routers and the other 2 xaiaomi sensors was happy with "IKEA ZB3" rouers. I was moving one aqara weather in my bath that was connected thru one bulb in living room to one moved outlet in the bath thru open the outlet for joining, power of the "old parent" and the short pressing the pairing / contact button on the sensor and it was in and repower thr "old parent". I have doing the same with all my end devices that have routers in the near and looks working good also with Xiaomi ones ;-)

I think ist possible using the blinds and the remote without repeater and letting "normal routers" making the relaying of the commands but i think its better using the originals and trying forcing the blindes using it as parent. If having routing problems like your Philips HUE is having then its very bad then the messages can being lost or must being rerouted around the mesh. And very important thing: always do real binding of the remote to the roupe and also to the blind then the remote is using broadcast to the groupe thru hole mesh and can going around routing problems if percent in the mesh (but technical getting more traffic). Also groupe binding is working then the coordinator is offline that can saving you  mutch "wife problems" if you SD-card is being corrupted or some other HW and SW problems ;-)) 

By the way IKEA is using nery only broadcasts in there GW (Have sniffing little around) that is good for routing problems but is eating capacity of the mesh very fast then the underlying network of zigbee is limited to around 1.2 broadcasts / seconds for not getting broadcast storms that flooding the mesh but they is not made for building very large networks only little lighting in the apartment ;-))

Jdruwe, 2020-11-03:
> Do the blinds respond to group commands issued from ZHA with issue_zigbee_group_command now?

No that still does not work for some reason.

Jdruwe, 2020-11-03:
@MattWestb 

> And very important thing: always do real binding of the remote to the roupe and also to the blind then the remote is using broadcast to the groupe thru hole mesh and can going around routing [...]

I binded the remote to the blinds group using the WindowCovering, is that what you mean? Because I am not sure what you mean with the ` and also to the blind then the remote is using broadcast to the groupe thru hole mesh and can going around routing problems if percent in the mesh` part

[image]

MattWestb, 2020-11-04:
I think the Window Covering is the one you need to bind.
Perhaps also the groupe cluster is needed but i have not testing it then i dont have any blinds.
Then the blindes is members of the groupe they is getting the command then you is pressing the buttons.

blair287, 2020-12-09:
I use hue bulbs too and this issue is still here.

I did try powering all my hue bulbs off for a day and tried the blind again but still slow to react 

My button and blind are both paired through repeater.

Weirdly mine was instant for about a month then when back to being slow.

MattWestb, 2020-12-09:
@blair287 Install [ZHA maps](https://github.com/zha-ng/zha-map) and one of the maps that is described so you can see which routers the blinds and remote is using. 
(zha-network-visualization-card is only showing the links that is being used and zigzag is showing all possible links that can being used and the quality of them)

The worse is then they is connected to the coordinator then its stop working is its down and the devices must stretching one new parent for sending / receiving commands.

If the HUE bulbs is working OK its better having them in the network but I have one that is not aging out its children then they have changed its parent and that can being very bad then sending commands to the devices and the coordinator cant finding one working route to the device.

If binding the remote to one groupe its sending the command with broadcast and can going around bad routes if having moderate routing problems but its not helping if having severe communication problem in the network.

Still if possible getting the blinds and and the remote using the same router (signal repeater or one other good working one) and the group is right configured its should working without problems also with different routers as long the network is working OK and not being blocked of interferences and bad behaving devices.

dmulcahey, 2020-12-09:
> @blair287 Install [ZHA maps](https://github.com/zha-ng/zha-map) and one of the maps that is described so you can see which routers the blinds and remote is using.
> (zha-network-visualization-card is [...]

There is no need to do this if you wait for the next release or use the beta for HA. It is built in now.

MattWestb, 2020-12-09:
Saint Lucia (Swedish the queens of lights that is bringing the light in the dark on 13/12) is bringing HA 1.0 on 13 december if can waiting :-))

blair287, 2020-12-10:
> There is no need to do this if you wait for the next release or use the beta for HA. It is built in now.

Sounds ace I will wait till the new release.

Just to be clear though I have a delay using the button or just sending down or up command from HA too.

But once it starts moving for about 5 minutes everything is instant then if you leave it along for a while the first command will have a delay of about 5 seconds.

Adminiuga, 2020-12-10:
> My button and blind are both paired through repeater.
> Weirdly mine was instant for about a month then when back to being slow.

Probably picked a new parent, thus slow again

blair287, 2020-12-12:
> > My button and blind are both paired through repeater.
> > Weirdly mine was instant for about a month then when back to being slow.
> 
> Probably picked a new parent, thus slow again

Possibly but I have unpaired and repaired through the repeater multiple times after it started to go slow still no good.

If it is the hue bulbs not a fan of having their hub again.

Will wait for new release see how it's all connected and report back.

MattWestb, 2020-12-12:
I dont have any blinds for testing / sniffing but its looks some more magic has being found.
Its looks like normal group commands is sent to thru NWK 0xfffd that is only being picked up of no sleeping end device with radio on and routers. 
IKEA open/close controller looks sending groupe command to NWK 0xffff and its also being picked up of sleeping end device with radio off.

I dont have seeing the sniff analyse but cold being worth looking at if the signal repeater and the blinds  is treating its special as the 5 button remote is doing with the 0xff09 groupe.

Findings: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3881

blair287, 2020-12-14:
> > My button and blind are both paired through repeater.
> > Weirdly mine was instant for about a month then when back to being slow.
> 
> Probably picked a new parent, thus slow again

Updated to the 12-2020 and used the visualisation and both the button and blind are connected to the tradfri repeater and the repeater is also connected the the conbee.

Adminiuga, 2020-12-14:
then must be https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3881

Adminiuga, 2020-12-15:
I can confirm that remote button sends Group Commands to 0xFFFF nwk address, however both ConBee and EZSP radios sends group commands to 0xFFFD address and there are no ways to change it. 

But it looks like OP was able to solve his problems by pairing blinds and remote to the same repeater.

MattWestb, 2020-12-15:
So 0xffff is one "real" broadcast address in the network layer and 0xfffd is one "ZLL" broadcast/multicast address limit to only routes and not sleeping end device with radio on.

I have no evidences but IKEA was doing many firmware updates for the signal repeater and if it was only one "normal" router that should not being necessarily.
Then sniffing the 5 button remote is the remote setting up the network on the new device and is doing "normal" configuration and then setting setting on IKEA specific command on its cluster before the group is being sended with Touchlink to the new device (The command can only being sent with TL if all is ala Zigbee standard). The "IKEA magic" is then doing binding of all clusters to the added groupe also to the cluster that is not exposed on the endpoints (for the remote the binding of the scene cluster is not working with normal binding but is OK then adding group thru TL).
I think IKEA have doing the same with the signal repeater = Configuring the group with TL and binding the clusters that is used for the blinds (that is not exposed on the endpoint I think but dont remember) and also the special 0xfff group addressing.
Also I think IKEA have one "relay" or proxy service bild in the repeater that is being configured thru TL. 

Then the next part: Then IKEA is using 0xffff I dont believe that the devs have rewritten Silabs zigbee stack for implementing it then its to high risk and costs connected to do that. But very likely it can being that its already implemented in the stack as one "undeockumented future" (That is normally one nice name for bug). Perhaps one question in the Silabs dev forum can bringing light in the problematic ?

It can also being that the signal repeater is listening on bothe 0xffff and 0xfffd and if some commands is coming on 0xff0d its "translating" it to 0xffff and putting it in its relay / proxy for sending it the the blinds in the corresponding groupe of its connected children. Still we cant configuring the groupe without touch link :-(((

One thing with IKEA. They is not selling some thousands of one product (normally one batch is multi millions pieces but its depends of the product / region) and is having one good working quality control system that is working world wide. If one product that have being sold for years like the bookshelf Billy (NOT THE EZSP ONE !!) is start getting reclamations its taking very short time before the alarm bells is sounding and then the time is ticking to finding the problem and elemating it. If one supplier is doing somthing wrong with one serie its being fast pinned out and blocked for selling it world wide. IKEA cant selling one product that 50% is coming back as reclamations (as there chinese TV was doing in EU) then the company is ded in some months / hals year because its so large and costing so much money.

Now to the blinds: The release was drawn back many months because of technical problems (hard / software or both) and is still not being sold on all markets / regions. 
In EU all gods that is being sold to private person must have 24 guarantee and also the customer have the right to reclaiming it in 36 months if its manufacturing fault in it. If all the blinds that is being sold in EU is being returned within 3 years its very much money for the company being lost for the company.

So my conclusion is the signal repeater is not one dumb router without function but have some hidden magic inside but very unlikely not one rewritten zigbee stack.

Its in AT 15€ and the E1743 is 6€, The E1743 is having more components and custom PCB without the standard module = more expensive to produce (minus the USB charger that is included).

Also being very interesting getting one good (with all packages and no lost one) sniff from paring the open/close switch with the repeater and then the blinds and then the commands being transmitted between then in real. But I dont having the money for buying them for the moment.

Little funny / strange is that in some systems for some users its working very good also without the repeater.

blair287, 2020-12-20:
I've tried with and without repeater always a delay on first command.

Pairing the blind to smartthings hub it's instant all the time.

Adminiuga, 2021-01-27:
there's 
[image]

so with EZSP coordinator looks like it is possible to send group commands to sleepy devices, although it is not Zigbee compliant

MattWestb, 2021-01-27:
Then its possible doing one "bellows IKEA" thing as they have done with the Open/Close Switch :-)))
In pretty sure is not possible doing it with all radios (at least very unlikely with the CC-253X ones) but implanting one general function for sending groupe commands with the address 0XFFFF and if the radio is not supporting it then the radio lib (or zigpy) is defaulting to normal grupe broadcasts in ZLL (0XFFFD) way.

Then making one function that can being called in quirks for sending the grupe commands to 0XFFFF for specific command / clusters. This way the the quirk is working if the radio is supporting the function or not.
And its can being user in the future if needed for some more devices that have using the network layer functions that is not implanted in the Zigbee layer.  

I think is worth trying if some users can testing it in real but its not helping if the remote is not working OK then its then behaving in the same way in the network layer as groupe commands from the coordinator but i think its one step in the right direction for getting it working little better.

Is our @puddly knowing if its possible to implanting this function in the new and / or old TI radios stacks ?

Is it some users out there with IKEA blinds on one IKEA GW that can making some sniffs for the devs looking ok ??

Interesting is seeing normal commands being sent from the remote to the signal repeater and then to the blinds if its possible catching at the same time and also the pairing sequence then pairing first the remote with the GW then signal repeater with the remote and then the blinds with the remote (ala IKEAs pairing procedure).

If trying you can  easily getting your network keý by paring one light with pushing the pairing button "under the hud" of the GW and its doing one ZHA pairing with ZHA default key that you can decrypting your network key with :-))

blair287, 2021-01-28:
> Then its possible doing one "bellows IKEA" thing as they have done with the Open/Close Switch :-)))
> In pretty sure is not possible doing it with all radios (at least very unlikely with the CC-253X [...]

id me willing to help no idea what you would need though.

like to point out about 3 months ago i went through a period of about 3 weeks where it worked fine instant all the time but before that period and after is the delay on first command behaviour, i thought it was a HA update but reverted back and issue was still there so no idea why it worked for a few weeks.

closed 2021-08-12 by github-actions[bot]

HemiBob, 2023-06-06:
I see this is a very old thread but having recently bought IKEA blinds I now have this same issue with a pair that used to work instantly together but now the left one has a 5-10sec delay to all commands.
I didn't fully understand what was being said in this thread and didn't see if there was a fix.
I don't use the IKEA hub, just a Sonnoff dongle and HA.

HemiBob, 2023-06-06:
> IKEA have released firmware update for all blind and controller with many bug fixes.
> First try updating the blinds and then the remotes .
> The problem is that the remotes is loosing group bindings and must being bonded / device in the remote if you like having them real bonded (bonding = the remote sending commands to other devices without the host system so its working also then the system is down).

Thanks Matt, yes I have 4 blinds and four controllers and have been trying to do OTA updates for days now.
7 have completed (eventually) and haven't made any difference. But the last one is so frustrating as the battery drains faster than the update and the update then fails. I've just installed my 3rd new 3032 so will try again. It's not unusual to see 10 hours as the expected completion time and the controller I guess doesn't get to sleep, hence the battery drain. I'm only doing one update at a time.

The only blind out of the four that has this delay in part of a pair in a group, the other blind in the group works fine. I'm thinking of removing the group and just have HA control them on their own - hopefully, that might remove the delay.

I don't bind any of my IKEA devices, I want everything to come from HA only and avoid any conflicts.

Good of you to reply, I thought the thread might be too old and neglected :-)

MattWestb, 2023-06-06:
If battery's  is not nearly in the blinds and remotes they is not accepting the OTA request so need have good batteries in all devices and for the controllers its take 2 hours or more is its eating around 50% of one new battery.

Fast batteries darning was made of around 10 bugs and most if fixed in the later 24.X firmware and the best not getting draining is pairing it nera one good router device and keep it there so its not jumping to one new parent and trigger the bugs.

The OnOff Dimmer have the same firmware but different data setting in the device and i and some other user have devices that was not flashing the OTS file and was looping updates and in the end we was flashing it with debugger and it was working OK.

If you like have fast response you shall binding the remote to the blinds (group with old firmware and devices for the latest ones) and you is not loosing functionality in HA all is working OK also reporting of the status of the bling shall being as before only that is local and faster with remotes and you still making normal automatons in HA as before.