zigpy/zha-device-handlers #503: [Device Support Request] icasa Pulse 8S

PaulAntonDeen, 2020-09-25:
**Is your feature request related to a problem? Please describe.**
icasa Pulse 8s is not recognized by zha

**Describe the solution you'd like**
Being able to switch lights with the device

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4644, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "icasa",
  "model": "ICZB-KPD18S",
  "class": "zigpy.device.Device"
}`

**Additional context**
I got working on the quirk by adapting the IKEA 5 button remote but am now stuck:
`
class IcasaPulse8S(CustomDevice):
    """Custom device representing icasa pulse 8s remote control."""

    signature = {
        # input_clusters=[0, 1, 3, 2821]
        # output_clusters=[3, 4, 5, 6, 8, 25, 768, 4096]>
        MODELS_INFO: [(ices, "ICZB-KPD18S")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.SCENE_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    DIAGNOSTICS_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
				#colour control,
                    #Touchlink Commissioning
                ],
            }
        },
    }

[... 123 lines cut ...]
`

Longpress on:

`
2020-09-25 11:38:44 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame command_type=CommandType.AREQ subsystem=Subsystem.AF command_id=129 data=b'\x00\x00\x08\x00D\xe4\x01\x01\x00Q\x00\x7fr\xea\x00\x00\x05\x01I\x05\x002\x06i\x1c' length=25 fcs=203>
2020-09-25 11:38:44 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 8, 'srcaddr': 0xE444, 'srcendpoint': 1, 'dstendpoint': 1, 'wasbroadcast': 0, 'linkquality': 81, 'securityuse': 0, 'timestamp': 15364735, 'transseqnumber': 0, 'len': 5, 'data': b'\x01I\x05\x002'}
2020-09-25 11:38:44 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message incomingMsg
2020-09-25 11:38:44 DEBUG (MainThread) [zigpy.zcl] [0xe444:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=73 command_id=5>
2020-09-25 11:38:44 DEBUG (MainThread) [zigpy.zcl] [0xe444:1:0x0008] ZCL request 0x0005: [0, 50]
2020-09-25 11:38:44 DEBUG (MainThread) [zigpy.zcl] [0xe444:1:0x0008] No handler for cluster command 5
2020-09-25 11:38:44 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=cc:cc:cc:ff:fe:26:a1:48, unique_id=cc:cc:cc:ff:fe:26:a1:48:1:0x0008, endpoint_id=1, cluster_id=8, command=move_with_on_off, args=[0, 50]>
`

and on release:

`
2020-09-25 11:38:49 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame command_type=CommandType.AREQ subsystem=Subsystem.AF command_id=129 data=b'\x00\x00\x08\x00D\xe4\x01\x01\x00T\x00c\xb6\xea\x00\x00\x03\x01J\x07\x06i\x1c' length=23 fcs=45>
2020-09-25 11:38:49 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 8, 'srcaddr': 0xE444, 'srcendpoint': 1, 'dstendpoint': 1, 'wasbroadcast': 0, 'linkquality': 84, 'securityuse': 0, 'timestamp': 15382115, 'transseqnumber': 0, 'len': 3, 'data': b'\x01J\x07'}
2020-09-25 11:38:49 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message incomingMsg
2020-09-25 11:38:49 DEBUG (MainThread) [zigpy.zcl] [0xe444:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=74 command_id=7>
2020-09-25 11:38:49 DEBUG (MainThread) [zigpy.zcl] [0xe444:1:0x0008] ZCL request 0x0007: []
2020-09-25 11:38:49 DEBUG (MainThread) [zigpy.zcl] [0xe444:1:0x0008] No handler for cluster command 7
2020-09-25 11:38:49 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=cc:cc:cc:ff:fe:26:a1:48, unique_id=cc:cc:cc:ff:fe:26:a1:48:1:0x0008, endpoint_id=1, cluster_id=8, command=stop, args=[]>
`

Adminiuga, 2020-09-25:
You already can use it in automation using zha_event differentiating by cluster_id, command and args.
For the device_actions to be supported, you would need to map those combinations of cluster_id, command and args into device action and adding it to quirk class

dmulcahey, 2020-09-27:
A brief description of how device triggers work is here: https://github.com/zigpy/zha-device-handlers/blob/dev/CONTRIBUTING.md#how-device_automation_triggers-work

Try to implement them using other quirks as a guide. Is this all you are stuck on?

PaulAntonDeen, 2020-09-27:
@Adminiuga While it is great that I can make it work by hacking the automations together within HA, I think it would be much nicer if we can get this thing implemented well so that the whole community can benefit. So making the quirk should be better right?

@dmulcahey I have not had the time to try these solutions yet. I was following https://github.com/zigpy/zha-device-handlers/blob/dev/CONTRIBUTING.md so I know I have to write a replacement and some subroutines, but what to write in that section is beyond me.

PaulAntonDeen, 2020-09-28:
I got what I need by using the zha_event, thanks for the tip. It is still worth pursuing the quirk I think.

Bascht74, 2021-01-09:
@PaulAntonDeen Hi Paul, 
can you share an example of your automatisation using the zha_event? This would be great, since I switched over from Homey to HA and I want to use ZHA, as all of my devices are supported by it - except the ICASA PULE8s...

Maybe your findings could be used to generate a blueprint together as it is here for DeCONZ: 
https://community.home-assistant.io/t/deconz-icasa-pulse-s8-keypad-remote/257628

Thanks for your help!

Sebastian

I started working on the blueprint, but I got stuck right now: 
```
blueprint:
  name: icasa PULSE 8S keypad (with ZHA)
  description: 'Control anything using icasa Pulse 8S keypad'
  domain: automation
  input:
    remote:
      name: Keypad
      description: Keypad to use
      selector:
        device: 
          integration: Zigbee Home Automation
          manufacturer: icasa
          model: ICZB-KPD18S
    button_on:
      name: On Button
      description: Action to run on press on On button
      default: []
      selector:
        action: {}
    button_off:
      name: Off Button
      description: Action to run on press on Off button
      default: []
      selector:
        action: {}
    button_1:
      name: Button S1
      description: Action to run on Button S1
      default: []
      selector:
        action: {}
    button_2:
      name: Button S2
      description: Action to run on Button S2
      default: []
      selector:
        action: {}
    button_3:
      name: Button S3
      description: Action to run on Button S3
      default: []
      selector:
        action: {}
    button_4:
      name: Button S4
      description: Action to run on Button S4
      default: []
      selector:
        action: {}
    button_5:
      name: Button S5
      description: Action to run on Button S5
      default: []
      selector:
        action: {}
    button_6:
      name: Button S6
      description: Action to run on Button S6
      default: []
      selector:
        action: {}
  source_url: https://community.home-assistant.io/t/...
mode: restart
max_exceeded: silent
trigger:
- platform: event
  event_type: zha_event
  event_data:
    device_id: !input 'remote'
action:
- choose:
  - conditions:
    - condition: template
        value_template: '{{ trigger.event.data.command == ''off'' }}'
    sequence: !input 'button_off'
  - conditions:
    - condition: template
        value_template: '{{ trigger.event.data.command == ''on'' }}'
    sequence: !input 'button_on'
  - conditions:
    - condition: template
        value_template: '{{ trigger.event.data.command == ''recall'' && trigger.event.data.args == [0,1] }}'
    sequence: !input 'button_1'
  - conditions:
    - condition: template
        value_template: '{{ trigger.event.data.command == ''recall'' && trigger.event.data.args == [0,2] }}'
    sequence: !input 'button_2'
  - conditions:
    - condition: template
        value_template: '{{ trigger.event.data.command == ''recall'' && trigger.event.data.args == [0,3] }}'
    sequence: !input 'button_3'
  - conditions:
    - condition: template
        value_template: '{{ trigger.event.data.command == ''recall'' && trigger.event.data.args == [0,4] }}'
    sequence: !input 'button_4'
  - conditions:
    - condition: template
        value_template: '{{ trigger.event.data.command == ''recall'' && trigger.event.data.args == [0,5] }}'
    sequence: !input 'button_5'
  - conditions:
    - condition: template
        value_template: '{{ trigger.event.data.command == ''recall'' && trigger.event.data.args == [0,6] }}'
    sequence: !input 'button_6'

```

Hedda, 2021-04-27:
@Emrvb FYI, there is a relative PR with button mappings that deCONZ use for iCasa Pulse 2, Pulse 4S and Pulse 8S here:

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4160

Hedda, 2023-02-28:
Should probably be re-open and not closed until at least open PR is merged? -> https://github.com/zigpy/zha-device-handlers/pull/1039

Hedda, 2023-09-02:
Should probably be left open at least until PR https://github.com/zigpy/zha-device-handlers/pull/1039 is closed?

closed 2024-03-07 by github-actions[bot]