zigpy/zha-device-handlers #505: Device support TS0041

oxamafia, 2020-09-26:
Hi,

I bought a wireless one-gang zigbee switch from Zemismart (TS0041/_TZ3400_keyjqthh).

The device is recognized by ZHA but nothing happens when I push the button. The switch stays in off mode in Home Assistant.

I would love to have the same things as Xiaomi Wireless Button for one/double/triple press.

There is no ZHA EVENT as well.

The debug says (when I press the button) :
2020-09-26 16:27:25 WARNING (MainThread) [zigpy.zcl] [0x281e:1:0x0006] Unknown cluster-specific command 253

and ZHA device signature is :
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3400_keyjqthh",
  "model": "TS0041",
  "class": "zigpy.device.Device"
}

Can you please help me? Do you need anything else?

I also want to thank you for all the great job that you do. I was on deconz at first and then I switched to ZHA, that is wayyyyy better for many reasons.

Max

oxamafia, 2020-10-05:
Nobody can help me? :(

meremortals70, 2020-10-06:
This might help 

From the looks you have to watch the event stream and use an automation

https://www.zigbee2mqtt.io/devices/TS0041.html

avee87, 2020-12-12:
This should be fixed by #627

oxamafia, 2020-12-18:
Hello
Thank you for this hard work! You’re giving my Christmas gift in advance! So thank you
Actually I updated HA with the lastest version but the quirk doesn’t seem to apply to my device. 
i don’t understand why?
Is there a way to force the quirk?
I have tried to remove the device (and restart HA), then relink it... like many times... and it still doesn’t take the quirk 🙁

Informations sur l’appareil
TS0041
par _TZ3400_keyjqthh
dresden elektronik ConBee II

avee87, 2020-12-19:
Hm, that is weird - signature seems to be correct...
Could you check ZHA logs when you pair the device? It prints some information about trying to match various quirks, that could probably help answer why this quirk doesn't apply.

oxamafia, 2020-12-19:
Yes sure : here is the whole logs when pairing. Is it what you need?

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Sending 'zdo_leave_req' failed: 
[0x119d:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=71 command_id=Command.Report_Attributes>
[0x119d:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=CharacterString, value=!�!�C!0>)]]
[0x119d:1:0x0000] Attribute report received: 65281=!�!�C!0
[0x5b53:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=101 command_id=Command.Report_Attributes>
[0x5b53:1:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=8451>)]]
[0x5b53:1:0x0400] Attribute report received: measured_value=8451
Device 0x73a7 (bc:33:ac:ff:fe:3b:ff:2d) joined the network
[0x73a7:zdo] ZDO request ZDOCmd.Device_annce: [0x73A7, bc:33:ac:ff:fe:3b:ff:2d, 128]
[0x73a7] Requesting 'Node Descriptor'
Tries remaining: 2
[0x73a7] Extending timeout for 0x8a request
[0x73a7] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0x73a7] Discovering endpoints
Tries remaining: 3
[0x73a7] Extending timeout for 0x8c request
[0x73a7] Discovered endpoints: [1]
[0x73a7:1] Discovering endpoint information
Tries remaining: 3
[0x73a7] Extending timeout for 0x8e request
light.lampe_rouge: polling current state
[0xe82f:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=144 command_id=Command.Read_Attributes_rsp>
[0xe82f:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=146 command_id=Command.Read_Attributes_rsp>
[0xE82F:1:0x0008]: received attribute: 0 update with value: 254
[0xe82f:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=148 command_id=Command.Read_Attributes_rsp>
[0x73a7:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
[0x73a7] Extending timeout for 0x96 request
[0x73a7:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=150 command_id=Command.Read_Attributes_rsp>
[0x73a7:1] Manufacturer: _TZ3400_keyjqthh
[0x73a7:1] Model: TS0041
Checking quirks for _TZ3400_keyjqthh TS0041 (bc:33:ac:ff:fe:3b:ff:2d)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0x73A7:bc:33:ac:ff:fe:3b:ff:2d entering async_device_initialized - is_new_join: True
device - 0x73A7:bc:33:ac:ff:fe:3b:ff:2d has joined the ZHA zigbee network
[0x73A7](TS0041): started configuration
[0x73A7:ZDO](TS0041): 'async_configure' stage succeeded
[0x73a7] Extending timeout for 0x98 request
[0x73a7] Extending timeout for 0x9a request
[0x73a7] Extending timeout for 0x9c request
[0x73A7:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x73A7:1:0x0000]: finished channel configuration
[0x73a7] Extending timeout for 0x9e request
[... 44 lines cut ...]
[0x73A7:1:0x0019]: 'async_configure' stage succeeded
[0x73A7](TS0041): completed configuration
[... cut ...]
[0x73A7:1:0x0006]: failed to get attributes '['on_off']' on 'on_off' cluster: 
[0x73A7:1:0x0000]: 'async_initialize' stage succeeded
[0x73A7:1:0x0001]: 'async_initialize' stage succeeded
[0x73A7:1:0x0006]: 'async_initialize' stage succeeded
[0x73A7:1:0x0019]: 'async_initialize' stage succeeded
[0x73A7](TS0041): power source: Battery or Unknown
[0x73A7](TS0041): completed initialization
[0x73A7:1:0x0006]: attempting to update onoff state - from cache: True
[0x343F:1:0x0500]: Updated alarm state: 0
[0x6CF2:1:0x0500]: Updated alarm state: 0

oxamafia, 2020-12-19:
Thank you in advance for your precious help 😊

avee87, 2020-12-19:
That is weird, it doesn't even try to match TS0041 quirks. Manufacturer and model seem to match the quirk exactly... Are you sure you use 2020.12 version of HA?
@tube0013, do you know why this could fail to pick up correct quirk?

oxamafia, 2020-12-19:
Yes I updated supervisor and HA to the lastest versions.

tube0013, 2020-12-19:
Will take a look in a bit

dmulcahey, 2020-12-19:
Debug logs will tell you exactly why

oxamafia, 2020-12-19:
Ok so you need something else?

tube0013, 2020-12-19:
so I don't see anything not matching...  

quirk signiture:

```
    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
        MODELS_INFO: [("_TZ3000_tk3s5tyg", "TS0041"), ("_TZ3400_keyjqthh", "TS0041")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
        },
    }
```

From the logs:
```
[0x73a7:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[0, 1, 6], output_clusters=[25, 10])
[0x73a7] Extending timeout for 0x96 request
[0x73a7:1:0x0000] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=150 command_id=Command.Read_Attributes_rsp>
[0x73a7:1] Manufacturer: _TZ3400_keyjqthh
[0x73a7:1] Model: TS0041
Checking quirks for _TZ3400_keyjqthh TS0041 (bc:33:ac:ff:fe:3b:ff:2d)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
```

Says it's failing on device_type mismatch. but signature is reporting device_type=0 which is which is a the On_Off_Switch - <https://github.com/zigpy/zigpy/blob/7e05720e57d371177c4543cee8838b8e056fbf99/zigpy/profiles/zha.py#L8>

oxamafia, 2020-12-19:
So what does that mean?
I cannot force by any way to apply the quirk you developed?

avee87, 2020-12-19:
@tube0013, I'm pretty sure that particular log relates to `bellows.zigbee.application.EZSPCoordinator` quirk, not the one in question.
From what I guess (knowing literally nothing about zha quirks), the list of quirks to match is formed taking into account `MODELS_INFO` property - i.e., if quirk contains device manufacturer/model or has none specified at all. So since manufacturer/model should match this device I assume the quirk is somehow missing in @oxamafia 's install for some reason.

@oxamafia , would it be too hard for you to follow the steps [here](https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install) to check if the quirk is actually present? I.e., you would need to type something like this in portainer console:
`ls /usr/local/lib/python3.8/site-packages/zhaquirks/tuya/`

tube0013, 2020-12-19:
the log shows what is used to match the quirk. I pasted that and the quirk code there to show that I didn't see anything that didn't match, so I'm not sure what is going on. Hoping someone else can spot something off.

tube0013, 2020-12-19:
I pasted the wrong line.. lol

@oxamafia what version of HA are you running. this quirk was only just included with 2020.12.0

oxamafia, 2020-12-20:
I was using the 12.0 before and I didn’t work so I updated then to : Home Assistant 2020.12.1 (copy and paste from Configuration > Info)
Sorry for giving you so much troubles 🙁

closed 2021-08-12 by github-actions[bot]