zigpy/zha-device-handlers #507: [BUG] Xiaomi Aqara LLKZMK11LM 2nd channel not working

duncanhoggan, 2020-09-28:
**Describe the bug**
I got a load of these sensors and the pair great and the first channel works fine, but the second channel never switches.
Physical hardware is fine as the switch lines on the device seem to be all good

**To Reproduce**
Steps to reproduce the behavior:
1. Add device to ZHA devices
2. Toggle channel 1 it works
3. Toggle channel 2 it doesn't works, sends command but then fails and switches back
4. See errors in logs

`[homeassistant.components.zha.entity] light.lumi_lumi_relay_c2acn01_on_off_2: turned on: {'on_off': TimeoutError()}`

**Expected behavior**
Both channels should work

duncanhoggan, 2020-09-28:
I read the packet from the logs and resent it, it works fine.
`echo -en "\xfe\r$\x01\x16\xdf\x01\x01\x06\x00'\x00\x1e\x03\x01'\x01\xfa" > /dev/ttyACM0`
So I decided to use `zigpy-cc ` to generate the binary frame and then push it down to the router so I can test some changes.

After adjusting the src and destination points from 1,1 to 1,2 respectively I managed to get the second channel to switch.
`payload2_on = {'dstaddr': 57110, 'destendpoint': 2, 'srcendpoint': 1, 'clusterid': 6, 'transid': 39, 'options': 0, 'radius': 30, 'len': 3, 'data': b"\x01t\x01"}`

I believe this should just be a change in the device quirks? Not sure what the process is to make this change or where is should be done.
If you need more info please ask, i really want to help get this alive.

Adminiuga, 2020-09-29:
it is a bug in zigpy-cc https://github.com/zigpy/zigpy-cc/issues/57 https://github.com/zigpy/zigpy-cc/issues/57
the 0.116 beta which will be released tomorrow would contain `zigpy-znp` switch your integration to use `zigpy-znp` radio type instead  and the problem should be solved.

Adminiuga, 2020-09-29:
Duplicate of https://github.com/zigpy/zigpy-cc/issues/56
Closing this one to keep everything in one place.

closed 2020-09-29 by Adminiuga

duncanhoggan, 2020-09-29:
@Adminiuga You are the best, thanks!