zigpy/zha-device-handlers #508: [Device Support Request] PEQ (Centralite) 3300-P

majorsl, 2020-09-29:
**Is your feature request related to a problem? Please describe.**
These sensors "fall off" my network often, sometimes all at once, with ZHA.  They were mostly stable with my old Vera unit. They seem a variant of the existing (supported) 3300 modules, but are not using the quirk for those.

**Describe the solution you'd like**
Possibly a slight update to the exist 3300 quirk to add support.  Willing to test/help out/modify the existing one myself if someone can get me started.

**Device signature**
{
  "node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=49887 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "CentraLite",
  "model": "3300",
  "class": "zigpy.device.Device"
}

**Additional context**
Item is here: https://www.amazon.com/3300-P-Door-Window-Sensor-White/dp/B00PDD958W

Adminiuga, 2020-10-22:
What radio?
What firmware version?

majorsl, 2020-10-22:
Thanks for the reply, I was using a Sonoff Zigbee bridge flashed with Tasmota: https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

However, that wasn't working out for me and I switched to a Conbee II with deCONZ.  This sensor wasn't supported with them, but I provided the info they needed and am just waiting for the merge and update over there.

I suppose this can be closed unless it helps add support with the info I provided for others trying to use these.

Adminiuga, 2020-10-22:
someone told me that Ezsp firmware based on 6.7.6 version was kicking direct children. If you don't have any other routers on the network, they may not rejoin after that. 
Won't be able to tell without debug logs, but if it works with deCONZ then i'll close the issue.

closed 2020-10-22 by Adminiuga