zigpy/zha-device-handlers #509: [Device Support Request]

mikosoft83, 2020-09-30:
**Is your feature request related to a problem? Please describe.**
I've seen these TRVs made by Tuya apparently but they don't stick to ZHA.

**Describe the solution you'd like**
Adding a quirk would be nice

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be I don't own them yet but here's a link to deconz thread:
[](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3110)

**Additional context**
Add any other context or screenshots about the feature request here.

dmulcahey, 2020-10-10:
Please fill out all of the required information

mikosoft83, 2020-10-10:
Here's a missing [link to deconz thread](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3110), is this enough to work with?

thumDer, 2020-11-16:
I would very like to have this as well!

MattWestb, 2020-11-16:
Some devs have starting doing the ground for the tuya comand decoding and some for the TRVs https://github.com/zigpy/zha-device-handlers/pull/538#issue-509500699.

Device support request is posted for them https://github.com/zigpy/zha-device-handlers/issues/581#issue-740866517.

If making one working then its only adding the model id and its working and if little different its only applying different options from the tuya palette of functions (look at Z2M how they is adding new ones its similar as for Zigpy).

[image]
I have some twins (different model number) and its being supported in the dev branch today in deCONZ ;-))

HA dont have support for all functions but more is in the pipe.

closed 2021-08-12 by github-actions[bot]