zigpy/zha-device-handlers #511: [Device Support Request] Heiman HS1RC Smart Remote Control

meremortals70, 2020-10-02:
I purchased a couple of these remotes to add to my network.

[image]

The system recognises them and I get two entities. One which is battery.

[image]

I was expecting to have to watch events for the buttons however when I start listening for events and start pressing buttons I don't see any events.

When I do the same for a Philips Dimmer, I do see zha_events.

My question is therefore are the buttons actually entities and I need a quick please?

Device Signature is

{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0x0501"
      ]
    }
  },
  "manufacturer": "HEIMAN",
  "model": "RC-EM",
  "class": "zigpy.device.Device"
}

If so would someone please help me create a quirk. I know that the docs were recently updated however if I can go through an existing quirk with someone and comment it line by line it may also help others.

pipiche38, 2020-10-02:
I just integrate it on Zigate for Domoticz and here are my findings so far

1/ IAS Enrollement as per Cluster 0x0500

2/ The remote is sending commands to gateway on cluster 0x0502
- "01" Arm Day (Home Zones Only) - Command Arm 0x00 - Payload 0x01
- "02" Emergency - Command Emergency 0x02
- "03" Arm All Zones - Command Arm 0x00 - Payload Arm all Zone 0x03
- "04" Disarm - Command 0x00 - Payload Disarm 0x00

meremortals70, 2020-11-28:
Hi everyone, any chance someone would like to tackle this one for me?

lusit86, 2021-03-25:
Hi,
I purchased this remote and had the same disappointment with ZHA events.
I’m not a Python developper but I made this code based on Philips Hue dimmer.
The script generate a ZHA event for each buttons : 
- top : arm_home
- bottom : emergency
- left : arm_all_zones
- right : disarm

It's a dirty solution but seems to work enough to wait for a better integration.

Files :
- __init__.py : https://pastebin.com/ec93fhgS
- samrtremote.py : https://pastebin.com/fg19JyNJ

Hedda, 2021-04-27:
FYI, dmulcahey has submitted a PR for alarm control panel support in ZHA to Home Assistant core -> https://github.com/home-assistant/core/pull/49080

dmulcahey development branch for it is found here -> https://github.com/dmulcahey/home-assistant/tree/dm/zha-ias-ace-support

( Comments if testing that branch before the PR is merged into HA should go into the pull request -> home-assistant/core#49080 )

Btw, see zha-device-handlers discussion about Centralite 3400-G which is a similar or even the same rebranded keypad -> https://github.com/zigpy/zha-device-handlers/issues/838

Also see the discussion about Centralite 3405-L / Lowes Iris Keypad V2 which too is similar to it as well -> https://github.com/zigpy/zha-device-handlers/issues/87

PS: There is also a broader zigpy discussion about Zigbee IAS and ACE (for "Alarm Control Panel" support) here -> https://github.com/zigpy/zigpy/issues/419

Hedda, 2021-04-30:
FYI; Home Assistant 2021.5 (planned for release May 5, 2021) Beta release notes: "**_ZHA now has support for alarm control panels_**".

https://rc.home-assistant.io/blog/2021/04/28/release-20215/ -> https://github.com/home-assistant/core/pull/49080

closed 2021-11-03 by github-actions[bot]

mycanaletto, 2025-09-08:
It's possible, but not easy : https://www.canaletto.fr/home-assistant-and-remote-alarm/

gillenoarbe, 2025-09-08:
> It's possible, but not easy : https://www.canaletto.fr/home-assistant-and-remote-alarm/

Thanks mycanaletto, I read your post that I saw posted in HA before this... indeed, possible but too complicated in this case.

Thanks for your hard work and publication.