zigpy/zha-device-handlers #514: Creating new quirk following guide, pycache not recreated

b1g1an, 2020-10-06:
Hi, as above, I'm following this guide...

https://github.com/zigpy/zha-device-handlers

to create a quirk for my Osram rgb GU10s which I previously used with zigbee2mqtt. After deleting pycache and rebooting it doesn't get replaced as it suggests it should? I've tried just placing the test quirk directly into the 'zhaquirks' folder and deleting the pychache at that level and also placing it the 'Osram' folder and deleting the pycache in there, no joy either way.

dmulcahey, 2020-10-06:
When HA is started and ZHA inits there are logs that detail the quirk matching process. They will explain why it is not matching. Please post them here.

b1g1an, 2020-10-06:
Sorry, thought I might just be missing something from the technique rather than it actually going wrong so didn't check, here they be...

2020-10-06 11:57:25 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO nodeDescRsp {'srcaddr': 0xCDA5}
2020-10-06 11:57:26 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO nodeDescRsp {'srcaddr': 0x2F31}
2020-10-06 11:57:46 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO bindRsp {'srcaddr': 0x8F08}
2020-10-06 11:57:46 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO bindRsp {'srcaddr': 0x8F08}
2020-10-06 11:57:54 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO nodeDescRsp {'srcaddr': 0xCDA5}
2020-10-06 11:57:55 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO nodeDescRsp {'srcaddr': 0x2F31}
2020-10-06 11:58:01 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO bindRsp {'srcaddr': 0x8F08}
2020-10-06 11:58:01 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO bindRsp {'srcaddr': 0x8F08}
2020-10-06 11:58:12 WARNING (MainThread) [zigpy.device] [0xcda5] Requesting Node Descriptor failed
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 86, in get_node_descriptor
    status, _, node_desc = await self.zdo.Node_Desc_req(
  File "/usr/local/lib/python3.8/site-packages/zigpy/util.py", line 110, in retry
    r = await func()
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 214, in request
    result = await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
2020-10-06 11:58:13 WARNING (MainThread) [zigpy.device] [0x2f31] Requesting Node Descriptor failed
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 86, in get_node_descriptor
    status, _, node_desc = await self.zdo.Node_Desc_req(
  File "/usr/local/lib/python3.8/site-packages/zigpy/util.py", line 110, in retry
    r = await func()
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 214, in request
    result = await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
2020-10-06 11:58:22 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO activeEpRsp {'srcaddr': 0xCDA5}
2020-10-06 11:58:23 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO activeEpRsp {'srcaddr': 0x2F31}
2020-10-06 11:58:25 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO nodeDescRsp {'srcaddr': 0xCDA5}
2020-10-06 11:58:25 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO nodeDescRsp {'srcaddr': 0x2F31}
2020-10-06 11:58:53 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO activeEpRsp {'srcaddr': 0x2F31}
2020-10-06 11:58:53 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO simpleDescRsp {'srcaddr': 0xCDA5}
2020-10-06 11:58:54 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO nodeDescRsp {'srcaddr': 0xCDA5}
2020-10-06 11:58:54 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO nodeDescRsp {'srcaddr': 0x2F31}
2020-10-06 11:59:12 WARNING (MainThread) [zigpy.device] [0xcda5] Requesting Node Descriptor failed
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 86, in get_node_descriptor
    status, _, node_desc = await self.zdo.Node_Desc_req(
  File "/usr/local/lib/python3.8/site-packages/zigpy/util.py", line 110, in retry
    r = await func()
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 214, in request
    result = await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
2020-10-06 11:59:12 WARNING (MainThread) [zigpy.device] [0x2f31] Requesting Node Descriptor failed
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 86, in get_node_descriptor
    status, _, node_desc = await self.zdo.Node_Desc_req(
  File "/usr/local/lib/python3.8/site-packages/zigpy/util.py", line 110, in retry
    r = await func()
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 214, in request
    result = await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
2020-10-06 11:59:18 WARNING (MainThread) [homeassistant.config_entries] Config entry for kodi not ready yet. Retrying in 80 seconds
2020-10-06 11:59:23 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO activeEpRsp {'srcaddr': 0x2F31}
2020-10-06 11:59:23 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO simpleDescRsp {'srcaddr': 0xCDA5}
2020-10-06 11:59:24 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO nodeDescRsp {'srcaddr': 0x2F31}
2020-10-06 11:59:25 WARNING (MainThread) [zigpy_cc.api] No response for: AREQ ZDO nodeDescRsp {'srcaddr': 0xCDA5}
2020-10-06 11:59:41 ERROR (MainThread) [zigpy.device] Failed ZDO request during device initialization
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 109, in _initialize
[... 19 lines cut ...]
[... cut ...]
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 86, in get_node_descriptor
    status, _, node_desc = await self.zdo.Node_Desc_req(
  File "/usr/local/lib/python3.8/site-packages/zigpy/util.py", line 110, in retry
    r = await func()
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 214, in request
    result = await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError

b1g1an, 2020-10-06:
Sonoff ZBBridge turned up this afternoon so removed integration and readded for that, works same as previously but now all I get in the log is one occurrence of...

2020-10-06 19:15:21 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: zha

and still no new pycache.

Adminiuga, 2020-10-06:
check log prior that message for exceptions

b1g1an, 2020-10-06:
> check log prior that message for exceptions

This is everything down to that point...

2020-10-06 19:14:16 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.
2020-10-06 19:14:16 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for hive which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.
2020-10-06 19:14:16 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for zigbee2mqtt_networkmap which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.
2020-10-06 19:14:16 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for alexa_media which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.
2020-10-06 19:15:21 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: zha

Adminiuga, 2020-10-06:
does bootstrap tries to setup any components for ZHA? 
that's too little of the log.  
DOes it start if you revert the changes?

b1g1an, 2020-10-07:
> does bootstrap tries to setup any components for ZHA?
> that's too little of the log.
> DOes it start if you revert the changes?

Not sure what you mean. ZHA does work, I've added a Xiaomi button, Hue bulb and dimmer since last post and they seem to work fine, just no recreation of pycache as per OP.

b1g1an, 2020-10-07:
Turned on all the logging as per the integration page, hope that's what you meant, and did a restart, fingers crossed the attached has something helpful in it.

[Log.txt](https://github.com/zigpy/zha-device-handlers/files/5339046/Log.txt)

dmulcahey, 2020-10-10:
The logs that start with “Checking quirks for” are where the quirk matching process starts. I don’t see the Osram bulbs you mentioned in the output. The only osram device mentioned is the switch mini. Are the bulbs in question paired?

b1g1an, 2020-10-10:
The Osram bulbs were pairing as 'unk_manufacturer unk_model' and hence why I was trying to create a new quirk, I had to do similar for zigbee2mqtt so wasn't surprised that they weren't working correctly. I've seen several posts from people having the same problem with them.

Home Assistant then had an update and this overwrote the component as expected and pycache came back as a result. I then removed the Osram bulbs and paired with the ZBBridge, they immediately showed up as 'OSRAM PAR 16 50 RGBW - LIGHTIFY' which is their correct name and using the default 'zigpy.device.Device'. All functions work perfectly.

As a result, it appears the issue with non recognition is down to using a CC2531 and all the posts I'd seen with the problem were from people using one of these. Prime example here...

https://github.com/zigpy/zha-device-handlers/pull/500

@Goldwing1973 also has the same bulbs I do and gets the same results as he did for the ones he wrote the new quirk for. All this suggests the quirk actually covers up a problem with the CC2531 rather than the bulbs actually need anything more than the defaults. I replied to his post on the HA forum where this was discussed but have had no reaction as yet.

Obviously all this means I don't need to do any work on a custom quirk now but doesn't go any way to explain what I originally posted about which was pycache not recreating itself :)

dmulcahey, 2020-10-10:
so we're good to close this then? the pycache thing has nothing to do w/ quirks.

Goldwing1973, 2020-10-10:
Yes, it looks like a C2531 (which I was using)  problem, so it can be closed.

b1g1an, 2020-10-10:
> so we're good to close this then? the pycache thing has nothing to do w/ quirks.

Depends how you look at it, I was following the quirks instructions to create one and it went wrong at deleting pycache, that hasn't been resolved but regards my setup I'm happy for now 👍

closed 2021-08-12 by github-actions[bot]