zigpy/zha-device-handlers #516: [Device Support Request] Total Power Consumption Xiaomi Plug EU [lumi.plug.mmeu01]

Draluak, 2020-10-06:
**Problem**
The EU Version [lumi.plug.mmeu01] of Xiaomi Mi Smart Plug creates a sensor for total consumption received (smartenergy_metering). The value has state "unknown" and never gets updated.

**Target**
Sensor for total consumption (smartenergy_metering) shows correct value when using lumi.plug.mmeu01 with ZHA.
 
**Additional Information**
Device signature shown in HA is missing the right endpoint in my opinion. There should be two at least two endpoints with a key "000C", one containing the value for current consumption, the other one containing total consumption (also see screenshot from deconz). Maybe because of the existing [quirk](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/plug_mmeu01.py), but i guess it is related to the fact that these devices can report wrong clusters at initial discovery process. This issue was described [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2583#issuecomment-633282976) for deconz. I think they added some delay which solved the problem.

**Example from deconz/phoscon ui with correct endpoints**
[image]

**Device signature (possibly incorrect)**
```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.mmeu01",
  "class": "zhaquirks.xiaomi.aqara.plug_mmeu01.Plug"
}
```
**Pairing Log**
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Ignoring message (b'1c5f11060afc001000') on cluster 64704: unknown endpoint or cluster id: 'No cluster ID 0xfcc0 on (04:cf:8c:df:3c:75:a2:e7, 1)'
[0x36CF:1:0x0b04]: async_update
[0xAC0F:1:0x0b04]: async_update
[0x1396:1:0x0b04]: async_update
[0x3837:1:0x0b04]: async_update
[35985:1:0x0b04]: async_update
[... 90 lines cut ...]
[0x2a73:21:0x000c] Attribute report received: present_value=0.0
[0x2a73:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=12 command_id=Command.Report_Attributes>
[0x2a73:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>), Attribute(attrid=245, value=<TypeValue type=uint32_t, value=53113600>)]]
[0x2a73:1:0x0006] Attribute report received: on_off=1, 245=53113600
[0x2a73:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=13 command_id=Command.Report_Attributes>
[0x2a73:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>), Attribute(attrid=245, value=<TypeValue type=uint32_t, value=53113600>)]]
[0x2a73:1:0x0006] Attribute report received: on_off=0, 245=53113600
[0x2a73:21:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=14 command_id=Command.Report_Attributes>
[0x2a73:21:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=0.0>)]]
[0x2a73:21:0x000c] Attribute report received: present_value=0.0
[0x2a73:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=15 command_id=Command.Report_Attributes>
[0x2a73:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>), Attribute(attrid=245, value=<TypeValue type=uint32_t, value=53113600>)]]
[0x2a73:1:0x0006] Attribute report received: on_off=1, 245=53113600
[0x2a73:21:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=16 command_id=Command.Report_Attributes>
[0x2a73:21:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=0.0>)]]
[0x2a73:21:0x000c] Attribute report received: present_value=0.0
[0x2a73:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=17 command_id=Command.Report_Attributes>
[0x2a73:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>), Attribute(attrid=245, value=<TypeValue type=uint32_t, value=53113600>)]]
[0x2a73:1:0x0006] Attribute report received: on_off=0, 245=53113600
light.livingroom2: polling current state - from cache: False
[0x0ebe:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=179 command_id=Command.Read_Attributes_rsp>
[0x0ebe:11:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=181 command_id=Command.Read_Attributes_rsp>
[0x0EBE:11:0x0008]: received attribute: 0 update with value: 180
[0x0ebe:11:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=183 command_id=Command.Read_Attributes_rsp>
```

Okosotthon, 2020-10-29:
I also have a problem with this plug. 
Tasmota Zigbee bridge (Tasmota 8.5.1, Hassio 0.117.0) not even recognizing the consumption, even tho during the pairing process it receives the attributum (Wattage), but not recognizing it.

`[0xb03f:21:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=16.386999130249023>)]]
[0xb03f:21:0x000c] Attribute report received: present_value=16.386999130249023`

It is not showing up in zha_events, but it seems like it is reporting the attributum constantly.
Non EU Aqara plugs consumption functions are working.

zandadoum, 2020-11-11:
I also just received these plugs from amazon (listed there as "Xiaomi XM500008") and i can't get the powermonitoring to work either through ZHA
 iam a newb but i'd like to help in anything i can to get this fixed.

Adminiuga, 2020-11-11:
@Okosotthon post your Zigbee device signature

Adminiuga, 2020-11-11:
@zandadoum post your device signature.
Enable debug logging. Do you get reports from your device? Like  
```
ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=16.386999130249023>)]]
[0xb03f:21:0x000c] Attribute report received: present_value=16.386999130249023
```
Where the 1st hex number in `0xb03f:21:0x000c` matches the nwk address of your device

zandadoum, 2020-11-11:
here is the signature. i'll check your other question in a few minutes

`{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "22": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.mmeu01",
  "class": "zigpy.device.Device"
}`

zandadoum, 2020-11-11:
this is awkward, i can't find a way to enable debug logging. can you help with that?

TheJulianJES, 2020-11-11:
@zandadoum Check this: https://www.home-assistant.io/integrations/zha/#debug-logging

zandadoum, 2020-11-11:
@Adminiuga 
ok, so the signature is posted in my above post.

with debug logs on, i do have reports:

```
2020-11-11 22:34:37 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=166), 255, -35, 0x92f3, 255, 255, b'\x18\x0b\x0b\x00\x00']
2020-11-11 22:34:37 DEBUG (MainThread) [zigpy.zcl] [0x92f3:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=11 command_id=Command.Default_Response>
2020-11-11 22:34:37 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x92F3:1:0x0006]: executed 'off' command with args: '()' kwargs: '{}' result: [0, <Status.SUCCESS: 0>]
2020-11-11 22:34:37 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=switch.xiaomi_xm500008_01_sw, old_state=<state switch.xiaomi_xm500008_01_sw=on; friendly_name=Xiaomi XM500008_01 on_off @ 2020-11-11T22:33:58.712620+01:00>, new_state=<state switch.xiaomi_xm500008_01_sw=off; friendly_name=Xiaomi XM500008_01 on_off @ 2020-11-11T22:34:37.836296+01:00>>
2020-11-11 22:34:37 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010600010100010000a7ffdcf392ffff0e18490a00001000f500230b000007'
2020-11-11 22:34:37 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=167), 255, -36, 0x92f3, 255, 255, b'\x18I\n\x00\x00\x10\x00\xf5\x00#\x0b\x00\x00\x07']
2020-11-11 22:34:37 DEBUG (MainThread) [zigpy.zcl] [0x92f3:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=73 command_id=Command.Report_Attributes>
2020-11-11 22:34:37 DEBUG (MainThread) [zigpy.zcl] [0x92f3:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>), Attribute(attrid=245, value=<TypeValue type=uint32_t, value=117440523>)]]
2020-11-11 22:34:37 DEBUG (MainThread) [zigpy.zcl] [0x92f3:1:0x0006] Attribute report received: on_off=0, 245=117440523
2020-11-11 22:34:37 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010c00150100010000a8ffdcf392ffff0a184a0a55003900000000'
2020-11-11 22:34:37 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=12, sourceEndpoint=21, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=168), 255, -36, 0x92f3, 255, 255, b'\x18J\nU\x009\x00\x00\x00\x00']
2020-11-11 22:34:37 DEBUG (MainThread) [zigpy.zcl] [0x92f3:21:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=74 command_id=Command.Report_Attributes>
2020-11-11 22:34:37 DEBUG (MainThread) [zigpy.zcl] [0x92f3:21:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=0.0>)]]
2020-11-11 22:34:37 DEBUG (MainThread) [zigpy.zcl] [0x92f3:21:0x000c] Attribute report received: present_value=0.0
2020-11-11 22:34:40 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=switch, service=turn_on, service_data=entity_id=switch.xiaomi_xm500008_01_sw>
2020-11-11 22:34:40 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x92F3, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=13), 14, b'\x01\r\x01')
2020-11-11 22:34:40 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'00f9'
2020-11-11 22:34:40 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'00f39204010600010140010000f90e0000'
2020-11-11 22:34:40 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 37619, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=249), 14, <EmberStatus.SUCCESS: 0>, b'']
2020-11-11 22:34:40 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010600010100010000a9ffdcf392ffff05180d0b0100'
2020-11-11 22:34:40 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=169), 255, -36, 0x92f3, 255, 255, b'\x18\r\x0b\x01\x00']
[... cut ...]
2020-11-11 22:34:40 DEBUG (MainThread) [zigpy.zcl] [0x92f3:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=75 command_id=Command.Report_Attributes>
2020-11-11 22:34:40 DEBUG (MainThread) [zigpy.zcl] [0x92f3:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>), Attribute(attrid=245, value=<TypeValue type=uint32_t, value=117440525>)]]
2020-11-11 22:34:40 DEBUG (MainThread) [zigpy.zcl] [0x92f3:1:0x0006] Attribute report received: on_off=1, 245=117440525
```

zandadoum, 2020-11-13:
was this info helpful or can i provide anything else?

Adminiuga, 2020-11-14:
try changes in https://github.com/zigpy/zha-device-handlers/pull/587/files

TheJulianJES, 2020-11-14:
electrical_measurement (which shows the current consumption) still works fine.
smartenergy_metering (which is apparently supposed to show the total consumption) still doesn't work for me.
I tried to "reconfigure" the device and also re-paired it, but neither steps didn't have any effect.
Perhaps it just takes some time to show up in Hass?

Adminiuga, 2020-11-14:
it does not have smartenergy_metering cluster

TheJulianJES, 2020-11-14:
Sorry, yeah. The description of the initial bug report: 
> **Problem**
> The EU Version [lumi.plug.mmeu01] of Xiaomi Mi Smart Plug creates a sensor for total consumption received (smartenergy_metering). The value has state "unknown" and never gets updated.
made it sound like the issue that you're trying to fix. But, I guess other people have the issue of electrical_measurement not working at all for them (which did work fine for me even without your changes).

zandadoum, 2020-11-14:
> try changes in https://github.com/zigpy/zha-device-handlers/pull/587/files

i'd like to give this a try, but i am extremely new to all this and i have no idea how i would implement that change

TheJulianJES, 2020-11-14:
> > try changes in https://github.com/zigpy/zha-device-handlers/pull/587/files
> 
> i'd like to give this a try, but i am extremely new to all this and i have no idea how i would implement that change

If you're running HassOS, this should do the job: https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install
You basically need to get the __init.py__ file from the xiaomi folder and override your current one. Then restart Hass. (Note: these changes will get lost with the next Home Assistant upgrade)

Adminiuga, 2020-11-14:
Apparently Zandadums have a different device version, even though the model name is the same, his device is missing the `smart_energy` cluster entirely. 
@zandadoum if you still have debug logging enabled, can you do `grep 2.:0x000c home-assistant.log` and post output here. I'd like to see if there's anything reported on the endpoint id 22

Adminiuga, 2020-11-14:
ouch, if the device reports different signatures, I'm not sure we'll be able to handle that, but per https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2583#issuecomment-633875378 looks like is the consumption, but quirk is masking it. Do you get any incoming traffic on endpoint 22?

zandadoum, 2020-11-15:
@Adminiuga 
i don't have debug logging on right now, so current logs don't show much.
but there is certainly traffic on the endpoints because i can see the changes if i go to 
ZHA->clusters->select a cluster and then click on "get zigbee attribute"
endpoint 22 is not the one showing the watts BTW, the correct one is endpoint 21

i followed a tutorial to get a sensor working by reading the zigbee.db
this is working for me to show current watts usage. i havent figured out yet if there are any other sensors that would be useful to me, but this one certainly works

`
# xiaomi_zncz04lm_01_watts 
- platform: sql
  db_url: sqlite:////config/zigbee.db
  scan_interval: 15
  queries:
    - name: xiaomi_zncz04lm_01_watts
      query: "SELECT round(value,2) as value FROM attributes where ieee = '04:cf:8c:df:3c:80:2a:b8' and endpoint_id = 21 and cluster = 12 and attrid = 85"
      column: "value"
      unit_of_measurement: "W"
`

does this help?

Adminiuga, 2020-11-15:
I wanted to see what if there's traffic on endpoint 22. I know there's traffic on endpoint 21, question is if there's data on endpoint 22 like deCONZ issue a suggests

zandadoum, 2020-11-15:
i'll enable debug mode tomorrow morning and post the home-assistant.log
i am not using that plug much yet, so i need to think of something to actually plug into it during the day :D

TheJulianJES, 2020-11-15:
I just tried adding endpoint 22, so I have the following now:
``
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [AnalogInputCluster],
                OUTPUT_CLUSTERS: [AnalogInput.cluster_id, Groups.cluster_id],
            },
            22: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [AnalogInputCluster],
                OUTPUT_CLUSTERS: [AnalogInput.cluster_id, Groups.cluster_id],
            },
``
Fetching on endpoint 21 still gives the current consumption (27 to 29 in my case).
Reading present_value on endpoint 22 gives me 13 which might be about half (but it doesn't vary much unlike the current power consumption).
This doesn't look like it's the total consumption, but it's at least something.
However, I guess due to the custom ElectricalMeasurementCluster replacement, the values from the endpoint 22 now display as the power_consumption entity. (The smartenergy entity still shows as unavailable.)

Adminiuga, 2020-11-15:
Out of all attributes on 0x0702 'smartenergy_metering" cluster, do you read any attributes?

Adminiuga, 2020-11-15:
some interesting copy'n'paste. I don't see that cluster in the SimpleDescriptor
the attributes from cluster FCC0:
```
ZigBee Cluster Library Frame, Mfr: Unknown (0x115f), Command: Discover Attributes Response, Seq: 157
    Frame Control Field: Profile-wide (0x1c)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .1.. = Manufacturer Specific: True
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Manufacturer Code: Unknown (0x115f)
    Sequence Number: 157
    Command: Discover Attributes Response (0x0d)
    Attribute Status Record
        Attribute: 0x00f3
        Data Type: 8-Bit Unsigned Integer (0x20)
    Attribute Status Record
        Attribute: 0x00f5
        Data Type: 32-Bit Unsigned Integer (0x23)
    Attribute Status Record
        Attribute: 0x00f6
        Data Type: 16-Bit Unsigned Integer (0x21)
    Attribute Status Record
        Attribute: 0x00fa
        Data Type: Boolean (0x10)
    Attribute Status Record
        Attribute: 0x00fc
        Data Type: Boolean (0x10)
    Attribute Status Record
        Attribute: 0x00fe
        Data Type: Character String (0x42)
    Attribute Status Record
        Attribute: 0x00ff
        Data Type: Octet String (0x41)
    Attribute Status Record
        Attribute: 0x0100
        Data Type: 8-Bit Unsigned Integer (0x20)
    Attribute Status Record
        Attribute: 0x0200
        Data Type: 8-Bit Unsigned Integer (0x20)
    Attribute Status Record
        Attribute: 0x0201
        Data Type: Boolean (0x10)
    Attribute Status Record
        Attribute: 0x0202
        Data Type: Boolean (0x10)
    Attribute Status Record
        Attribute: 0x0203
        Data Type: Boolean (0x10)
    Attribute Status Record
        Attribute: 0x0204
        Data Type: Single Precision Floating Point (0x39)
    Attribute Status Record
        Attribute: 0x0205
        Data Type: 8-Bit Unsigned Integer (0x20)
    Attribute Status Record
        Attribute: 0x0206
        Data Type: Single Precision Floating Point (0x39)
    Attribute Status Record
        Attribute: 0x0207
        Data Type: Boolean (0x10)
    Attribute Status Record
        Attribute: 0x020b
        Data Type: Single Precision Floating Point (0x39)
    Attribute Status Record
        Attribute: 0xf000
        Data Type: 8-Bit Unsigned Integer (0x20)
    Attribute Status Record
        Attribute: 0xfffd
        Data Type: 16-Bit Unsigned Integer (0x21)

0000   1c 5f 11 9d 0d 01 f3 00 20 f5 00 23 f6 00 21 fa   ._...... ..#..!.
0010   00 10 fc 00 10 fe 00 42 ff 00 41 00 01 20 00 02   .......B..A.. ..
0020   20 01 02 10 02 02 10 03 02 10 04 02 39 05 02 20    ...........9.. 
0030   06 02 39 07 02 10 0b 02 39 00 f0 20 fd ff 21      ..9.....9.. ..!
```
And some investigation to find out what they are for:

```
<attribute id="0x00f3" name="Unknown" type="u8" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x00f5" name="Unknown" type="u32" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x00f6" name="Reporting Interval" type="u16" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x00fa" name="Unknown" type="bool" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x00fc" name="Unknown" type="bool" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x00fe" name="Serial Number" type="cstring" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x00ff" name="Unknown (write only)" type="ostring" mfcode="0x115f" access="w" required="m"> </attribute>
<attribute id="0x0100" name="Unknown" type="u8" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x0200" name="Unknown" type="u8" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x0201" name="Restore Power on Outage" type="bool" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x0202" name="Auto-off after 20m below 2W" type="bool" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x0203" name="Device LED off" type="bool" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x0204" name="Unknown" type="float" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x0205" name="Unknown" type="u8" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x0206" name="Unknown" type="float" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0x0207" name="Consumer connected" type="bool" mfcode="0x115f" access="r" required="m"> </attribute>
<attribute id="0x020b" name="Max. Load exceeded at (in W)???" type="float" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0xf000" name="Unknown" type="u8" mfcode="0x115f" access="rw" required="m"> </attribute>
<attribute id="0xfffd" name="Unknown" type="u16" mfcode="0x115f" access="rw" required="m"> </attribute>
```

Adminiuga, 2020-11-15:
```
Decoding of manufacturer specific attribute reporting (attribute 0x00F7. cluster FCC0, Endpoint 1, first byte is always the length)

ZigBee Cluster Library Frame, Mfr: Unknown (0x115f), Command: Report Attributes, Seq: 45
    Frame Control Field: Profile-wide (0x1c)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .1.. = Manufacturer Specific: True
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Manufacturer Code: Unknown (0x115f)
    Sequence Number: 45
    Command: Report Attributes (0x0a)
    Attribute Field, Octets: 64:10:00:03:28:17:98:39:00:00:00:00:95:39:1a:a4:…
        Attribute: 0x00f7
        Data Type: Octet String (0x41)
        Octet String: 64:10:00:03:28:17:98:39:00:00:00:00:95:39:1a:a4:…

0000   3d 64 10 00 03 28 17 98 39 00 00 00 00 95 39 1a
0010   a4 ec 3d 96 39 00 60 0b 45 97 39 00 00 00 00 05
0020   21 02 00 9a 20 00 08 21 16 01 07 27 00 00 00 00
0030   00 00 00 00 09 21 00 01 0b 20 00 9b 10 01
03 28 18				(Zcl8BitInt)     Device temperature
05 21 02 00				(Zcl16BitUint)   Unknown
07 27 00 00 00 00 00 00 00 00		(Zcl64BitUint)   Unknown
08 21 16 01				(Zcl16BitUint)   Unknown
09 21 00 01				(Zcl16BitUint)   Unknown
0b 20 00				(Zcl8BitUint)    Unknown
64 10 01				(ZclBoolean)     on/off
95 39 e0 7c e5 3d			(ZclSingleFloat) Consumption
96 39 00 60 0b 45			(ZclSingleFloat) Voltage
97 39 49 5f 53 41			(ZclSingleFloat) Current
98 39 44 8b 3c 40			(ZclSingleFloat) Power
9a 20 00				(Zcl8BitUint)    Unknown
9b 10 01				(ZclBoolean)     Unknown
```

Adminiuga, 2020-11-15:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2572/files

zandadoum, 2020-11-15:
@Adminiuga 
ok, turned debug on this morning 09:30am (CET) and let it run 2h with a 25w lightbulb on (reported as 24.71w)

the logfile has a total of 8000+ entries in this 2 hours, of which only  these 11 have the criteria you told me last night (0x000c)

if i can provide anything else, let me know.

`2020-11-15 09:27:53 DEBUG (MainThread) [zigpy.appdb] [0x823b:22:0x000c] Attribute id: 256 value: 720896
2020-11-15 09:27:53 DEBUG (MainThread) [zigpy.appdb] [0x823b:22:0x000c] Attribute id: 81 value: 0
2020-11-15 09:27:53 DEBUG (MainThread) [zigpy.appdb] [0x823b:21:0x000c] Attribute id: 111 value: 0
2020-11-15 09:27:53 DEBUG (MainThread) [zigpy.appdb] [0x823b:22:0x000c] Attribute id: 85 value: 0.326031357049942
2020-11-15 09:27:53 DEBUG (MainThread) [zigpy.appdb] [0x823b:21:0x000c] Attribute id: 85 value: 0.0
2020-11-15 09:40:37 DEBUG (MainThread) [zigpy.zcl] [0x823b:21:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=90 command_id=Command.Report_Attributes>
2020-11-15 09:40:37 DEBUG (MainThread) [zigpy.zcl] [0x823b:21:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=0.0>)]]
2020-11-15 09:40:37 DEBUG (MainThread) [zigpy.zcl] [0x823b:21:0x000c] Attribute report received: present_value=0.0
2020-11-15 09:40:41 DEBUG (MainThread) [zigpy.zcl] [0x823b:21:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=92 command_id=Command.Report_Attributes>
2020-11-15 09:40:41 DEBUG (MainThread) [zigpy.zcl] [0x823b:21:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=24.714000701904297>)]]
2020-11-15 09:40:41 DEBUG (MainThread) [zigpy.zcl] [0x823b:21:0x000c] Attribute report received: present_value=24.714000701904297`

Adminiuga, 2020-11-15:
> `2020-11-15 09:27:53 DEBUG (MainThread) [zigpy.appdb] [0x823b:22:0x000c] Attribute id: 85 value: 0.326031357049942`

So this appears to be the summation then?

it appears we're talking about two different issues:
1. @zandadoum devices signature does not have smart energy cluster and does not have measurement cluster and also does not match quirk. @dmulcahey should we could add the analog sensor back, till there's a quirk to convert summation into "smart_energy" cluster updats
2. Everyone else has quirk and homeautomation.electric_measrement  cluster works fine, but people are asking to support the `smart_energy` cluster. For this to happen need three things:
1. what actual attributes work on `smart_energy` cluster and contain values
2. make ZHA to support multiple entities per single cluster
3. create a quirks to convert updates from analog sensor on Endpoint id 22 to smart energy summation attribute?

hmonteiro, 2020-11-17:
I've been facing the exact same issue on some of my smart plugs. In the end, while ZHA doesn't support multiple attributes per cluster, I have redirected the values of the ones that are used by the plug to publish the values, to the attributes that ZHA hardcodes. For example, I have a plug that publishes the total consumption in attribute 0x000 of the Metering cluster, but ZHA only reads attribute 0x0400 from this cluster so I just created a quirk to redirect attribute 0x0000 to 0x0400 and now I have the value available in HA.

I have another plug that produces both instantaneous and cumulative consumption values in the same Metering cluster and in this case it's harder to fix. I am building another quirk for this that routes the second attribute of the cluster to the ElectricalMeasurement cluster so that I can have 2 sensors.

At the moment, until point 4 of @Adminiuga post is ready, AFAIK we have to resort to weird solutions like this.

TheJulianJES, 2020-12-06:
Here's another signature for the LUMI plug: https://pastebin.com/raw/ziigC1vL (from https://www.reddit.com/r/homeassistant/comments/k6n6ye/xiaomi_mi_power_plug_zha_not_getting_power/).
It also reports endpoint 22. (The ones I have don't report endpoint 22, but at least the total consumption works just fine with the existing quirk)
The workaround using analog_input added in Hass 1.0.0 beta seems to work fine though.

Adminiuga, 2020-12-06:
hrm, this one seem to be ZB3, probably why the signature doesn't match. But without a quirk it should work in 1.0.0 beta, although the question remains:
1. Some LUMI plugs report on Endpoint 21 and 22 and others report only on 21 (if 22 is summation, maybe just need to wait longer with larger load for it to start reporting?) 
2. If we create entity for analog cluster which never reports, would user complain?

TheJulianJES, 2020-12-06:
I don't think users would "complain" really if there is a "useless entity". However, some people might be confused. (The current Home Assistant beta might have the power consumption duplicated which also really isn't an issue.)
The (non "Z3") OSRAM plugs I have also randomly report "electrical_measurement" with 400W or 4000W. Just disabling the entity is ok and would probably good enough of a solution for most people with the LUMI plugs.

Also, my LUMI plug (which only reports endpoint 21 and works just fine with the quirk) reports "zcl_version": 3. (Signature: signature: https://paste.ubuntu.com/p/93kSrRzT5F/)

zandadoum, 2020-12-16:
updated today, see the 2 new entities. 1 seems to be for current wattage usage and the other for summation of kw? i will test this for some days and come back if anything is "weird" but i think the problem is solved for now :)

bcutter, 2021-01-23:
> 
> 
> updated today, see the 2 new entities. 1 seems to be for current wattage usage and the other for summation of kw? i will test this for some days and come back if anything is "weird" but i think the problem is solved for now :)

Update after 5 weeks of testing?

closed 2021-08-12 by github-actions[bot]