zigpy/zha-device-handlers #518: [Device Support Request] Heiman HS1SA-E smoke detector

nikopaulanne, 2020-10-07:
**Is your feature request related to a problem? Please describe.**
Problem: 
I suppose I have the same problem as already corrected in zigbee2mqtt (https://github.com/Koenkk/zigbee2mqtt/issues/3773):
Heiman HS1SA-E smoke detector have trouble to work properly. Pairing is ok and the battery and linkquality messages too. But if I press the test button or activate the alarm with some smoke no messages or log entries are generated.

The main problem is as in zigbee2mqtt case?
- The device label on a battery slot means it is the model HS1SA-E, but it sends a "SmokeSensor-N-3.0’ so the HA detects it (wrong?) as a HS3SA.

**Describe the solution you'd like**
I would love to see same correction as zigbee2mqtt already to did to be also done in ZHA. 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
Checking quirks for HEIMAN SmokeSensor-EM (00:0d:6f:00:14:b9:d0:eb)
[52631](SmokeSensor-EM): started configuration
[52631:ZDO](SmokeSensor-EM): 'async_configure' stage succeeded
[52631](SmokeSensor-EM): completed configuration
[52631](SmokeSensor-EM): stored in registry: ZhaDeviceEntry(name='HEIMAN SmokeSensor-EM', ieee='00:0d:6f:00:14:b9:d0:eb', [52631](SmokeSensor-EM): started initialization
[52631:ZDO](SmokeSensor-EM): 'async_initialize' stage succeeded
[52631](SmokeSensor-EM): power source: Battery or Unknown
[52631](SmokeSensor-EM): completed initialization

**Additional context**
Add any other context or screenshots about the feature request here.

nikopaulanne, 2020-10-24:
Is there something else I could do, provide some logs etc? I would like to integrate these smoke alarms covering multiple stories properly to Hass with ZHA.

nikopaulanne, 2020-11-24:
As seems like this is not the kind of issue which is "pop" and will be patched anytime soon, how one should enter to adding code this to quirks etc on the ZHA-codebase? Where to start? Can anyone give a pointer what example to look at then code the relevant fixes by myself?

ministryofsillywalks, 2020-12-02:
Any news on this? I just ordered a bunch as they were on offer dead cheap.

nikopaulanne, 2020-12-03:
> Any news on this? I just ordered a bunch as they were on offer dead cheap.
> Happy to test and debug if I can be of any assistance.

No dice and no hopes up on this issue on ZHA. You're better off just dumping ZHA and going for Zigbee2Mqtt. Sad.

dmulcahey, 2020-12-03:
Comments like that won’t be tolerated. It has been hidden. Folks contribute their free time to implement this for the community.

dmulcahey, 2020-12-04:
ZHA shouldn’t need this fix as we enroll IAS every time the device is joined. I’ll look at your log when I have a moment

dmulcahey, 2020-12-04:
Wait... there is no log from you. Please remove the device, then re-join it and provide the entire pairing log.

ministryofsillywalks, 2020-12-04:
> 
> 
> Wait... there is no log from you. Please remove the device, then re-join it and provide the entire pairing log.

Mine should be arriving next week. Will provide logs once I get them and join them.

ministryofsillywalks, 2020-12-17:
Finally got the devices (postage was slow) and paired the first one up.
Here are the logs from pairing @dmulcahey 
`[0x0C69:3:0x0b04]: async_update
[0x7FC6:3:0x0b04]: async_update
[0x3163:1:0xef00] Unknown cluster-specific command 36
Device 0x0fcf (00:50:43:c9:03:35:29:3a) joined the network
[0x0fcf] Requesting 'Node Descriptor'
Device 0x0fcf (00:50:43:c9:03:35:29:3a) joined the network
[0x0fcf] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=132, manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=3)
[0x0fcf] Discovering endpoints
[0x0fcf] Discovered endpoints: [1]
[0x0fcf:1] Discovering endpoint information
[0x0fcf:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1026, device_version=0, input_clusters=[0, 3, 1280, 1, 9, 1282], output_clusters=[25])
device - 0x0FCF:00:50:43:c9:03:35:29:3a entering async_device_initialized - is_new_join: True
device - 0x0FCF:00:50:43:c9:03:35:29:3a has joined the ZHA zigbee network
[0x0FCF](SMOK_YDLV10): started configuration
[0x0FCF:ZDO](SMOK_YDLV10): 'async_configure' stage succeeded
[0x0FCF:1:0x0500]: started IASZoneChannel configuration
[0x0FCF:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x0FCF:1:0x0000]: finished channel configuration
[0x0FCF:1:0x0502]: bound 'ias_wd' cluster: Status.SUCCESS
[0x0FCF:1:0x0502]: finished channel configuration
[0x0C69:3:0x0b04]: async_update
[0x7FC6:3:0x0b04]: async_update
[0x0FCF:1:0x0500]: bound 'ias_zone' cluster: Status.SUCCESS
[0x0FCF:1:0x0000]: initializing channel: from_cache: False
[0x0FCF:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0x0FCF:1:0x0019]: finished channel configuration
[0x0FCF:1:0x0500]: wrote cie_addr: 60:a4:23:ff:fe:42:58:96 to 'ias_zone' cluster: [WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]
[0x0FCF:1:0x0500]: Sending pro-active IAS enroll response
[0x0FCF:1:0x0500]: finished IASZoneChannel configuration
[0x0FCF:1:0x0000]: 'async_configure' stage succeeded
[0x0FCF:1:0x0500]: 'async_configure' stage succeeded
[0x0FCF:1:0x0502]: 'async_configure' stage succeeded
[0x0FCF:1:0x0019]: 'async_configure' stage succeeded
[0x0FCF](SMOK_YDLV10): completed configuration
[0x0FCF](SMOK_YDLV10): stored in registry: ZhaDeviceEntry(name='Heiman SMOK_YDLV10', ieee='00:50:43:c9:03:35:29:3a', last_seen=1608235231.8112361)
light.zoe_hue_licht_level_light_color_on_off: polling current state
[0x8CF3:11:0x0008]: received attribute: 0 update with value: 200
[0x0FCF:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.UNSUP_CLUSTER_COMMAND: 129>]
[0x0FCF](SMOK_YDLV10): started initialization
[0x0FCF:ZDO](SMOK_YDLV10): 'async_initialize' stage succeeded
[0x0FCF:1:0x0502]: initializing channel: from_cache: False
[0x0FCF:1:0x0019]: initializing channel: from_cache: False
[0x0FCF:1:0x0000]: initializing channel: from_cache: False
[0x0FCF:1:0x0500]: initializing channel: from_cache: False
[0x0FCF:1:0x0000]: 'async_initialize' stage succeeded
[0x0FCF:1:0x0500]: 'async_initialize' stage succeeded
[0x0FCF:1:0x0502]: 'async_initialize' stage succeeded
[0x0FCF:1:0x0019]: 'async_initialize' stage succeeded
[0x0FCF](SMOK_YDLV10): power source: Mains
[0x0FCF](SMOK_YDLV10): completed initialization
[0x0C69:3:0x0b04]: async_update
[0x7FC6:3:0x0b04]: async_update`

Here is the device signature:
`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=132, manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=3)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0500",
        "0x0502"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Heiman",
  "model": "SMOK_YDLV10",
  "class": "zigpy.device.Device"
}`

I will tcause an alarm tomorrow and see what the logs give me. (not happening today or I will wake everyone up ;) )

gitawayu, 2021-02-05:
Hi David,
I have a Heiman HS1SA-E smoke detector now, and although I am able to pair it in Home Assistant in the ZHA integration using a Conbee 2 stick, I do not seem to be able to trigger anything off it when I press the test button or when I light a fire under it(!).
If you have time to look at this at some point, please let me know if there are any logs or other info I can provide.
Many thanks!
Tim.

P.S.  The Zigbee device signature has the following at the bottom... 

`  "manufacturer": "HEIMAN",`
`  "model": "SmokeSensor-EF-3.0",`
`  "class": "zigpy.device.Device"`

It's [this](http://www.heimantech.com/product/?type=detail&id=3) (battery operated) device.  The label on the box it came in says "HS1SA-E".

hvddrift, 2021-06-30:
@dmulcahey and @gitawayu 

I have this exact issue. I am also happy to provide any logs etc. The importance for me is I would like to have the unit if it detects smoke trigger the other devices in the house.

Thanks

Hans

hvddrift, 2021-07-01:
I tried to reconfigure the device and this was the result I got... Not sure if it helps

[image]

closed 2022-01-05 by github-actions[bot]

oneiropagides, 2022-07-10:
Hi, has anyone manage to solve this?
It seems the issue is "closed" but I cannot understand what the solution is.

frittes, 2023-05-24:
Hi, I have the same problems. I can join the sensor to the network and it is identified as SmokeSensor-N-3.0 with that quirk https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/heiman/smoke.py.

The Sensor stays online after joining for a few hours and then it is offline. No chance to get it online again.

Any ideas to get it work?

jochenruben, 2023-11-02:
Hi, I have the same issue. Sensor is in the network . The are no entities, no battery status and hier is the log:

Logger: homeassistant
Source: components/zha/core/cluster_handlers/__init__.py:75
First occurred: 11:53:37 (15 occurrences)
Last logged: 16:36:17

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 64, in wrap_zigpy_exceptions
    yield
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 84, in wrapper
    return await RETRYABLE_REQUEST_DECORATOR(func)(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/util.py", line 132, in retry
    return await func()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 377, in request
    return await self._endpoint.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 253, in request
    return await self.device.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 293, in request
    await self._application.request(
  File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 828, in request
    await self.send_packet(
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/zigbee/application.py", line 1202, in send_packet
    raise DeliveryError(
zigpy.exceptions.DeliveryError: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 640, in async_device_initialized
    await self._async_device_joined(zha_device)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 656, in _async_device_joined
    await zha_device.async_configure()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 578, in async_configure
    await self.identify_ch.trigger_effect(
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 83, in wrapper
    with wrap_zigpy_exceptions():
  File "/usr/local/lib/python3.11/contextlib.py", line 155, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 75, in wrap_zigpy_exceptions
    raise HomeAssistantError(message) from exc
homeassistant.exceptions.HomeAssistantError: Failed to send request: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>

snevs, 2025-03-27:
Closed without a solution. Having the same issue.