zigpy/zha-device-handlers #523: [BUG]

krue81, 2020-10-11:
**Describe the bug**
I have 3 Mijia/Xiaomi Motion Sensors (lumi.sensor_motion) working in HA via a Sonoff Zigbee Bridge. I did the hack for reporting motion every 5 seconds (https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806), but after that, I always get a fake/ghost trigger after the real motion trigger. Sometimes it happens 3 o 4 seconds after the real trigger, sometimes it is 1 minute later, but the ghost trigger happens ALWAYS after a real trigger.

**To Reproduce**
Hack the motion sensor with a bridge between TP4 and the switch.
[image]

Use a Sonoff Zigbee Bridge flashed with Tasmota and connect it to Home Assistant with ZHA

Add/Pair the motion sensor to HA

The motion sensor reports every 120/180 seconds... with the hack and a Node Red reset flow I can read the sensor every 5 seconds: Use an API node with POST to the path states/binary_sensor.sensor_name and data {"state":"off"}

After the sensor state is reset, It will always report a false/ghost trigger after a real trigger.

**Expected behavior**
A single motion trigger after motion is detected.

Adminiuga, 2020-10-11:
Debug logs?

krue81, 2020-10-11:
> Debug logs?

Sorry, here they are (the first motion is the real one, the second one, more than 60s later, was false (no one was in the room):
This screenshot is from node red, for the reset function mentioned earlier:
[image]

`
2020-10-11 13:08:04 DEBUG (MainThread) [bellows.zigbee.application] EZSP Counters: COUNTER_MAC_RX_BROADCAST: 0,COUNTER_MAC_TX_BROADCAST: 79,COUNTER_MAC_RX_UNICAST: 19,COUNTER_MAC_TX_UNICAST_SUCCESS: 0,COUNTER_MAC_TX_UNICAST_RETRY: 0,COUNTER_MAC_TX_UNICAST_FAILED: 0,COUNTER_APS_DATA_RX_BROADCAST: 2,COUNTER_APS_DATA_TX_BROADCAST: 2,COUNTER_APS_DATA_RX_UNICAST: 18,COUNTER_APS_DATA_TX_UNICAST_SUCCESS: 0,COUNTER_APS_DATA_TX_UNICAST_RETRY: 0,COUNTER_APS_DATA_TX_UNICAST_FAILED: 0,COUNTER_ROUTE_DISCOVERY_INITIATED: 0,COUNTER_NEIGHBOR_ADDED: 0,COUNTER_NEIGHBOR_REMOVED: 0,COUNTER_NEIGHBOR_STALE: 0,COUNTER_JOIN_INDICATION: 0,COUNTER_CHILD_REMOVED: 0,COUNTER_ASH_OVERFLOW_ERROR: 0,COUNTER_ASH_FRAMING_ERROR: 0,COUNTER_ASH_OVERRUN_ERROR: 0,COUNTER_NWK_FRAME_COUNTER_FAILURE: 0,COUNTER_APS_FRAME_COUNTER_FAILURE: 0,COUNTER_UTILITY: 0,COUNTER_APS_LINK_KEY_NOT_AUTHORIZED: 0,COUNTER_NWK_DECRYPTION_FAILURE: 0,COUNTER_APS_DECRYPTION_FAILURE: 0,COUNTER_ALLOCATE_PACKET_BUFFER_FAILURE: 0,COUNTER_RELAYED_UNICAST: 0,COUNTER_PHY_TO_MAC_QUEUE_LIMIT_REACHED: 0,COUNTER_PACKET_VALIDATE_LIBRARY_DROPPED_COUNT: 0,COUNTER_TYPE_NWK_RETRY_OVERFLOW: 0,COUNTER_PHY_CCA_FAIL_COUNT: 0,COUNTER_BROADCAST_TABLE_FULL: 0,COUNTER_PTA_LO_PRI_REQUESTED: 0,COUNTER_PTA_HI_PRI_REQUESTED: 0,COUNTER_PTA_LO_PRI_DENIED: 0,COUNTER_PTA_HI_PRI_DENIED: 0,COUNTER_PTA_LO_PRI_TX_ABORTED: 0,COUNTER_PTA_HI_PRI_TX_ABORTED: 0,COUNTER_ADDRESS_CONFLICT_SENT: 0
2020-10-11 13:08:05 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040106040101000100002ab0c81bb6ffff0718570a00001801'
2020-10-11 13:08:05 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1030, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=42), 176, -56, 0xb61b, 255, 255, b'\x18W\n\x00\x00\x18\x01']
2020-10-11 13:08:05 DEBUG (MainThread) [zigpy.zcl] [0xb61b:1:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=87 command_id=Command.Report_Attributes>
2020-10-11 13:08:05 DEBUG (MainThread) [zigpy.zcl] [0xb61b:1:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.1>)]]
2020-10-11 13:08:05 DEBUG (MainThread) [zigpy.zcl] [0xb61b:1:0x0406] Attribute report received: occupancy=1
2020-10-11 13:08:05 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB61B:1:0x0500]: Updated alarm state: 1
2020-10-11 13:08:05 DEBUG (MainThread) [zhaquirks] 00:15:8d:00:04:ab:22:5d - Received motion event message
2020-10-11 13:08:05 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=binary_sensor.sala_mvto, old_state=<state binary_sensor.sala_mvto=off; friendly_name=Movimiento Sala, device_class=motion @ 2020-10-11T08:05:51.560667-05:00>, new_state=<state binary_sensor.sala_mvto=on; friendly_name=Movimiento Sala, device_class=motion @ 2020-10-11T08:08:05.133610-05:00>>
2020-10-11 13:08:10 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=binary_sensor.sala_mvto, old_state=<state binary_sensor.sala_mvto=on; friendly_name=Movimiento Sala, device_class=motion @ 2020-10-11T08:08:05.133610-05:00>, new_state=<state binary_sensor.sala_mvto=off @ 2020-10-11T08:08:10.239617-05:00>>
2020-10-11 13:08:14 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2020-10-11 13:08:14 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 241 (readCounters) received: b'00004f00140000000000000002000200130000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'
2020-10-11 13:08:14 DEBUG (MainThread) [bellows.zigbee.application] EZSP Counters: COUNTER_MAC_RX_BROADCAST: 0,COUNTER_MAC_TX_BROADCAST: 79,COUNTER_MAC_RX_UNICAST: 20,COUNTER_MAC_TX_UNICAST_SUCCESS: 0,COUNTER_MAC_TX_UNICAST_RETRY: 0,COUNTER_MAC_TX_UNICAST_FAILED: 0,COUNTER_APS_DATA_RX_BROADCAST: 2,COUNTER_APS_DATA_TX_BROADCAST: 2,COUNTER_APS_DATA_RX_UNICAST: 19,COUNTER_APS_DATA_TX_UNICAST_SUCCESS: 0,COUNTER_APS_DATA_TX_UNICAST_RETRY: 0,COUNTER_APS_DATA_TX_UNICAST_FAILED: 0,COUNTER_ROUTE_DISCOVERY_INITIATED: 0,COUNTER_NEIGHBOR_ADDED: 0,COUNTER_NEIGHBOR_REMOVED: 0,COUNTER_NEIGHBOR_STALE: 0,COUNTER_JOIN_INDICATION: 0,COUNTER_CHILD_REMOVED: 0,COUNTER_ASH_OVERFLOW_ERROR: 0,COUNTER_ASH_FRAMING_ERROR: 0,COUNTER_ASH_OVERRUN_ERROR: 0,COUNTER_NWK_FRAME_COUNTER_FAILURE: 0,COUNTER_APS_FRAME_COUNTER_FAILURE: 0,COUNTER_UTILITY: 0,COUNTER_APS_LINK_KEY_NOT_AUTHORIZED: 0,COUNTER_NWK_DECRYPTION_FAILURE: 0,COUNTER_APS_DECRYPTION_FAILURE: 0,COUNTER_ALLOCATE_PACKET_BUFFER_FAILURE: 0,COUNTER_RELAYED_UNICAST: 0,COUNTER_PHY_TO_MAC_QUEUE_LIMIT_REACHED: 0,COUNTER_PACKET_VALIDATE_LIBRARY_DROPPED_COUNT: 0,COUNTER_TYPE_NWK_RETRY_OVERFLOW: 0,COUNTER_PHY_CCA_FAIL_COUNT: 0,COUNTER_BROADCAST_TABLE_FULL: 0,COUNTER_PTA_LO_PRI_REQUESTED: 0,COUNTER_PTA_HI_PRI_REQUESTED: 0,COUNTER_PTA_LO_PRI_DENIED: 0,COUNTER_PTA_HI_PRI_DENIED: 0,COUNTER_PTA_LO_PRI_TX_ABORTED: 0,COUNTER_PTA_HI_PRI_TX_ABORT [... cut ...] ECRYPTION_FAILURE: 0,COUNTER_ALLOCATE_PACKET_BUFFER_FAILURE: 0,COUNTER_RELAYED_UNICAST: 0,COUNTER_PHY_TO_MAC_QUEUE_LIMIT_REACHED: 0,COUNTER_PACKET_VALIDATE_LIBRARY_DROPPED_COUNT: 0,COUNTER_TYPE_NWK_RETRY_OVERFLOW: 0,COUNTER_PHY_CCA_FAIL_COUNT: 0,COUNTER_BROADCAST_TABLE_FULL: 0,COUNTER_PTA_LO_PRI_REQUESTED: 0,COUNTER_PTA_HI_PRI_REQUESTED: 0,COUNTER_PTA_LO_PRI_DENIED: 0,COUNTER_PTA_HI_PRI_DENIED: 0,COUNTER_PTA_LO_PRI_TX_ABORTED: 0,COUNTER_PTA_HI_PRI_TX_ABORTED: 0,COUNTER_ADDRESS_CONFLICT_SENT: 0
`

Adminiuga, 2020-10-12:
yeah, it does look like it is switching state without any traffic. 
What was the last version this worked properly?

krue81, 2020-10-12:
I just got the sensors last week and they were behaving this way since then. I was on 0.115.

Thanks for looking into this!

Routout, 2020-10-16:
Zigbee2mqtt behaves similar with modded sensor. I have 30sec timeout. 
It triggered false movement 17:55:50 and resetted it 17:55:50 without movement.
In the logs is only "occupancy":false, at that time.

```
Zigbee2MQTT:debug 2020-10-16 17:54:20: Received Zigbee message from '0x00158d000314a5cd', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-16 17:54:20: MQTT publish: topic 'zigbee2mqtt/0x00158d000314a5cd', payload '{"battery":91,"illuminance":0,"illuminance_lux":0,"linkquality":78,"occupancy":false,"voltage":2985}'
Zigbee2MQTT:debug 2020-10-16 17:54:20: Received Zigbee message from '0x00158d000314a5cd', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-16 17:54:20: MQTT publish: topic 'zigbee2mqtt/0x00158d000314a5cd', payload '{"battery":91,"illuminance":0,"illuminance_lux":0,"linkquality":78,"occupancy":true,"voltage":2985}'
Zigbee2MQTT:info  2020-10-16 17:55:50: MQTT publish: topic 'zigbee2mqtt/0x00158d000314a5cd', payload '{"battery":91,"illuminance":0,"illuminance_lux":0,"linkquality":78,"occupancy":false,"voltage":2985}'

```

cigas4, 2020-10-27:
Never had this issue while using Zigbee2MQTT (CC2531 zigbee sniffer usb stick). Got all my motion sensors soldered and adjusted timeout to 10s.
Now after moving to Sonoff Zigbee bridge and ZHA integration I have started to face this false motion issue. I am using same hacked Xiaomi sensors as before and resetting motion after 10s over node-red as in the first post (API HTTP POST {"state":"off"}).

MirkoBianchetti, 2020-11-21:
I have this behaviour  on HA 0.115 and ZHA with the regular sensor, without the hack. Instead of the hack, I've set an automation that after a certain period (5sec in my case) force the state of the sensor to off. I've used the python script set_state.py that is available on several forums.
It works, but I'have, as mentioned before, 2-3 ghost trigger after the real trigger occurs.

rspaargaren, 2020-11-22:
Same here some ghost triggers after modding the sensor. Also a reset of the sensor via the python script.

adrez99, 2020-12-14:
I've been looking at the code to search for a fix, and I think that this is not a bug in zigpy's source code.
The motion sensor in ZHA has an internal state that should always be in sync with the state that HA stores, so when ZHA changes the state then it sends it to HA (on motion detected and on no motion). But every so often HA polls ZHA state too to avoid missed events. This is where the ghost switching occurs and to fix it, we don't have to change HA's state but ZHA's.
This is the yaml to change ZHA state to off (replace IEEE with the sensor address):

service: zha.set_zigbee_cluster_attribute
data:
  ieee: '00:15:8d...'
  endpoint_id: 1
  cluster_id: 1280
  cluster_type: in
  attribute: 2
  value: 0

Adminiuga, 2020-12-14:
Wanna try something else?
Modify your [zhaquirks/__init__.py](https://github.com/zigpy/zha-device-handlers/blob/64a65c1959d161ac3e312872e88665c66584a25d/zhaquirks/__init__.py#L262) file and after line #262 insert `self._update_attribute(ZONE_STATE, ON)`

krue81, 2020-12-14:
> I've been looking at the code to search for a fix, and I think that this is not a bug in zigpy's source code.
> The motion sensor in ZHA has an internal state that should always be in sync with the [...]

Just tried this solution with one sensor, and it is working perfectly! I'll spend a little longer checking for those ghost triggers before updating all my sensors code to this.

Thank you for your solution!

dmulcahey, 2020-12-14:
> Wanna try something else?
> Modify your [`zhaquirks/__init__.py`](https://github.com/zigpy/zha-device-handlers/blob/64a65c1959d161ac3e312872e88665c66584a25d/zhaquirks/__init__.py#L262) file and after line `#262` insert `self._update_attribute(ZONE_STATE, ON)`

Ah, cache not getting updated?

Adminiuga, 2020-12-14:
yep. we only update it on reset,

adrez99, 2020-12-18:
> 
> 
> Wanna try something else?
> Modify your [`zhaquirks/__init__.py`](https://github.com/zigpy/zha-device-handlers/blob/64a65c1959d161ac3e312872e88665c66584a25d/zhaquirks/__init__.py#L262) file and after line `#262` insert `self._update_attribute(ZONE_STATE, ON)`

I've tried adding the line but it doesn't seem to make any difference: there are false motion reports using HA API to reset the sensor, but using zha.set_zigbee_cluster_attribute service there are no more false motion reports

PR #658 (2020-12-21): Small cleanups in several classes
This PR:

- cleans up IASZone implementations that use constant attributes to set zone type
- changes the Xiaomi motion cool down to 70 seconds
- cleans up the Aqara vibration sensor motion class
- changes the Aqara vibration sensor cool down to 70 seconds
- adds an event to publish the reported vibration strength

~~I still have to test the motion cool down after the device vacates test mode...~~ I can confirm that the device does in fact trigger about every 60s like this says:  https://community.smartthings.com/t/xiaomi-motion-sensor-aqara-how-to-test/170752 [... cut]
closed 2020-12-21 by dmulcahey