zigpy/zha-device-handlers #524: Tuya TS0218

meremortals70, 2020-10-11:
New Device Support Tuya TS0218

[image]

The device joins the Zigbee network and presents a binary_sensor and a sensor for battery power.

I am gathering we will need to look at what events it presents when the button is pressed.

{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0401",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x0019",
        "0x0501"
      ]
    }
  },
  "manufacturer": "_TYZB01_trqoesc6",
  "model": "TS0218",
  "class": "zigpy.device.Device"
}

https://www.zigbee2mqtt.io/devices/TS0218.html

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3368

https://github.com/ioBroker/ioBroker.zigbee/issues/730

meremortals70, 2020-10-17:
Other people are also asking for some help with this device.

https://community.home-assistant.io/t/ts0218-sos-button-unsupported-cluster-0x0501-iaszone-state-perpetually-off/236334

meremortals70, 2020-10-17:
https://community.home-assistant.io/t/ts0218-sos-button-unsupported-cluster-0x0501-iaszone-state-perpetually-off/236334

meremortals70, 2020-11-28:
Hey folks, would anyone like to tackle this please?

alexanderfitu, 2021-01-29:
+1 looking for support for this if possible please.

gowildchild, 2021-03-15:
+1 looking for support for this button. Thanks.

closed 2021-09-18 by github-actions[bot]