zigpy/zha-device-handlers #526: [Device Support Request] [Enhancement] Sercomm SZ-ESW01-AU Plug

meremortals70, 2020-10-14:
Would it be possible to please add this to the list of switches instead of identifying simply as lights?

Please let me know if you need any additional information.

Adminiuga, 2020-10-14:
You can override entity type via zha configuration. Look for example on the forum, it was posted there a while back

meremortals70, 2020-10-14:
Thanks, which forum I searched the HA forum and the issues here before adding the request?

Adminiuga, 2020-10-14:
https://community.home-assistant.io/t/zha-zigbee-tested-devices-please-add-your-device-results/17718/384?u=adminiuga

The device section. Usually it is the ieee address + endpoint id, so in your case it should end in `-1`

meremortals70, 2020-10-14:
That's awesome, can we get that added to documentation please?

meremortals70, 2020-10-31:
Hi folks

Are we able to add the additional clusters for this device please?

closed 2021-08-12 by github-actions[bot]