zigpy/zha-device-handlers #527: [BUG] CentraLite 3130 missing switch entity

joejulian, 2020-10-14:
**Describe the bug**
Opening the zigbee logs and pressing the buttons, the events are recorded. The device is discovered but it only adds the battery entity.

**To Reproduce**
Steps to reproduce the behavior:
- In home assistant, add a new device
- Activate the pairing for the switch (hold both buttons until the led turns blue)
The device is discovered and displayed.
- Exit the discovery page and open the new `OSRAM LIGHTIFY Dimming Switch` in Devices
Only the `OSRAM LIGHTIFY Dimming Switch <id> power` entity is listed.

**Expected behavior**
I expected both the power entity and a switch entity

**Screenshots**
[image]

**Additional context**

Adminiuga, 2020-10-14:
remotes are stateless and don't have entities. Remotes only do `zha_events`.

closed 2020-10-14 by Adminiuga

joejulian, 2020-10-15:
Well that's a bummer. When I used them in home assistant a couple years ago I could use them. In between I've been using them with smartthings. So they're trash now?

dmulcahey, 2020-10-15:
That isn’t remotely what he said. Stateless devices emit events in HA. Use device triggers or event triggers to leverage these in HA.

joejulian, 2020-10-15:
Ah, sorry, I didn't know that. hass has changed a bit since I used it last. Thanks for the helpful comment!