zigpy/zha-device-handlers #529: [Device Support Request] Salus SP600

hmonteiro, 2020-10-17:
I am using a [Salus SP600 smart plug](https://zigbee.blakadder.com/Salus_SP600.html) and, although it correctly shows as a smart plug with energy monitoring capablities in HA, there are some problems:
- The temperature reported is always 2x the actual value
- The instantaneous demand reported is always 1/10 of the actual value
- The total consumption (attribute `current_summ_delivered` attribute from the metering cluster) is not reported

It would be great if these values were correctly reported by the system.

The device signature is below. If you need any more informations let me know.

**Device signature**
```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4216, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=0, maximum_outgoing_transfer_size=255, descriptor_capability_field=0)",
  "endpoints": {
    "9": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0402",
        "0x0702",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Computime",
  "model": "SP600",
  "class": "zigpy.device.Device"
}
```

davet2001, 2021-01-25:
I have the SP600, and it works ok.
After making the switch from zigbee2mqtt to zha, it still works, but now reads power 10x too high.
Could it be the same as this issue:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/897

I see that in zigbee2mqtt, there is a specific handler for a certain datecode:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/d751f07919cc1122da3f0ac75053a7ba0cfe137b/converters/fromZigbee.js#L1650-L1669

I wonder whether this needs handling in a similar way in ZHA?

There are different versions of firmware for this device, as described here:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/915

They behave differently.  Can similar behaviour be accommodated in ZHA?

davet2001, 2021-08-06:
I still have problems with my sp600 
-over reading power by 10x
-not reporting energy consumption (just live power and temperature)

Attached: dishwasher is not really using 25kW!
[image]

davet2001, 2021-10-06:
Agreed, the additional kWh sensors are reporting values now for this device using the development HA version.

But the 10x problem remains on these SP600 devices. 

Generally I think scaling errors on kWh sensors will become much more of an issue to users now that those sensors are being used to visualize the split of total energy consumption in a house.  E.g. "House used 10kWh today made up of 1kWh lights and 20kWh dishwasher" ??!?!.

I am considering updating my PR #1051 to make the behavior correct just for the newest firmware, then add ZHA FW update capability for Salus to resolve the /10 issue for legacy users.  Currently [ZHA can't select quirk behaviour based on FW datecode](https://github.com/zigpy/zha-device-handlers/pull/1051#issuecomment-932654834).

Adminiuga, 2021-10-06:
Contact salus and ask for the firmware update where they fixed the scaling?

Adminiuga, 2021-10-06:
Would be interesting to know how many non updated devices nare there. 

You can update firmware in zha too, if you have the file.  See https://www.reddit.com/r/homeassistant/comments/fak430/how_to_update_your_ikea_or_ledevance_firmware for hint, all you really need to do is to put the ota file into a dolder

davet2001, 2022-02-12:
@hmonteiro The FW update for this device is now supported in HA (not GUI but in the YAML config). Info here:

https://www.home-assistant.io/integrations/zha/#ota-firmware-updates

Please could try this to see if it solves your issues?