zigpy/zha-device-handlers #533: [Device Support Request]  zhaquirks.smartthings.tag_v4.SmartThingsTagV4 Battery Support

JMan7777, 2020-10-23:
Hello,

I migrated from Samsung SmartThings to Home Assistant and was able to integrate all my existing sensors including the Samsung Presence sensors.

However, only for the "zhaquirks.smartthings.tag_v4.SmartThingsTagV4" sensors I'm not getting any power entities created.
I added each sensor via the "Zigbee Home Automation" integration and it created just one "device_tracker." entity but no power "sensor." like for my other SmartThings battery operated sensors.

The funny thing is the "device_tracker." entity shows in it's settings on the bottom the battery level. E.g.:

[image]

So my question is why there is no power entity as it looks like that the battery level is available.

Thx a lot.

roquemore92, 2021-02-05:
I'll add to this, looking at the ZHA logs when pairing a new device, I noticed this come through

`[0xcf0b:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=105 command_id=Command.Report_Attributes>
[0xcf0b:1:0x0001] ZCL request 0x000a: [[Attribute(attrid=32, value=<TypeValue type=uint8_t, value=18>)]]
[0xcf0b:1:0x0001] Attribute report received: battery_voltage=18
[0xcf0b:1:0x0001] Voltage [RAW]:18 [Max]:2.8 [Min]:1.5, Battery Percent: 23.0
device_tracker.{sensor_name}: battery_percentage_remaining updated: 46    `

So it would seem it is possible.

Related to this, and I can open a new topic if that's better, but the newer model of the SmartThings Arrival Sensor has the model PGC410 and it can pair with ZHA just fine, but shows no entities. I'm assuming that's because it can't find a quirk for it. And these for sure should support battery reporting from what I understand.

Here's what it shows when pairing a PGC410 with ZHA

`Device 0xe29e (d0:52:a8:00:ca:80:00:05) joined the network
[0xe29e] Requesting 'Node Descriptor'
Tries remaining: 2
[0xe29e] Extending timeout for 0x3f request
Received frame on uninitialized device d0:52:a8:00:ca:80:00:05 (Status.NEW) for endpoint: 2
[0xe29e] Received ZCL while uninitialized on endpoint id 2, cluster 0xfc05 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4362 tsn=48 command_id=1>, payload: b''
Device 0xe29e (d0:52:a8:00:ca:80:00:05) joined the network
Skip initialization for existing device d0:52:a8:00:ca:80:00:05
[0xe29e:zdo] ZDO request ZDOCmd.Device_annce: [0xE29E, d0:52:a8:00:ca:80:00:05, 128]
[0xe29e] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4362, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0xe29e] Discovering endpoints
Tries remaining: 3
[0xe29e] Extending timeout for 0x41 request
[0xe29e] Discovered endpoints: [1, 2]
[0xe29e:1] Discovering endpoint information
Tries remaining: 3
[0xe29e] Extending timeout for 0x43 request
[0xe29e:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=410, device_version=0, input_clusters=[0], output_clusters=[])
[0xe29e] Extending timeout for 0x45 request
Received frame on uninitialized device d0:52:a8:00:ca:80:00:05 endpoint 2 for cluster: 64517
[0xe29e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=69 command_id=Command.Read_Attributes_rsp>
[0xe29e:1] Manufacturer: SmartThings
[0xe29e:1] Model: PGC410
[0xe29e:2] Discovering endpoint information
Tries remaining: 3
[0xe29e] Extending timeout for 0x47 request
[0xe29e:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=64513, device_type=410, device_version=0, input_clusters=[], output_clusters=[])
Checking quirks for SmartThings PGC410 (d0:52:a8:00:ca:80:00:05)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2}
device - 0xE29E:d0:52:a8:00:ca:80:00:05 entering async_device_initialized - is_new_join: True
device - 0xE29E:d0:52:a8:00:ca:80:00:05 has joined the ZHA zigbee network
[0xE29E](PGC410): started configuration
[0xE29E:ZDO](PGC410): 'async_configure' stage succeeded
[0xe29e] Extending timeout for 0x49 request
[0xE29E:1:0x0000]: bound 'basic' cluster: Status.NOT_SUPPORTED
[0xE29E:1:0x0000]: finished channel configuration
[0xE29E:1:0x0000]: 'async_configure' stage succeeded
[0xE29E](PGC410): completed configuration
[0xE29E](PGC410): stored in registry: ZhaDeviceEntry(name='SmartThings PGC410', ieee='d0:52:a8:00:ca:80:00:05', last_seen=1612487793.3279982)
[0xE29E](PGC410): started initialization
[0xE29E:ZDO](PGC410): 'async_initialize' stage succeeded
[0xE29E:1:0x0000]: initializing channel: from_cache: False
[0xE29E:1:0x0000]: finished channel configuration
[0xE29E:1:0x0000]: 'async_initialize' stage succeeded
[0xE29E](PGC410): power source: Battery or Unknown
[0xE29E](PGC410): completed initialization
`

roquemore92, 2021-08-06:
Just checked in the latest version of Home Assistant and  both items are still the same. Tagv4 shows a device tracker entity, but not power, and the PGC410 shows no entities.

JMan7777, 2021-08-06:
The issue still exists on my tagv4 as well (tested with supervisor-2021.06.8).
Please do not close this issue.

JMan7777, 2022-02-03:
The issue still exists on my tagv4 as well (tested with supervisor-2021.12.2).
Please do not close this issue.

closed 2022-08-09 by github-actions[bot]

cipnt, 2023-02-26:
I can confirm Home Assistant 2023.2.5 still not showing any entities for Arrival/Presence Sensor PGC410/PGC410EU by SmartThings

alandrumo, 2024-02-26:
Still the same issue with 2024.2.3
The PGC410 will pair but no entities and isn't updating after pairing.