zigpy/zha-device-handlers #535: [Device Support Request] Lonsonho X711A _TZ3000_9hpxg80k TS0011

christovic, 2020-10-23:
**Is your feature request related to a problem? Please describe.**
I'm having strange issues with this No Neutral switch I have. I've managed to get it working by editing a file in this repository, so I'd like some clarification on *why* it works, and whether it would be possible to create a pull request so that it works for people going forward.

**Describe the solution you'd like**
Tell me what I'm doing wrong/right!

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_9hpxg80k",
  "model": "TS0011",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Add any other context or screenshots about the feature request here.

So this is where it gets weird (for me at least):

I've added this to https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/singleswitch.py:

```
class TuyaNoNeutralSwitch(TuyaSwitch):
    """Tuya single channel switch device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098,
        # maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        # maximum_outgoing_transfer_size=82, descriptor_capability_field=0)"
        # device_version=1
        # input_clusters=[0x0000,0x0004, 0x0005,0x0006,]
        # output_clusters=[0x000a, 0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=100 device_version=1 input_clusters=[0, 4, 5, 6,] output_clusters=[10,25]>
        MODELS_INFO: [("_TZ3000_9hpxg80k", "TS0011")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        }
    }
```
With this in place, the switch works as expected, without it, the switch turns off by itself after two minutes exactly. 

The main difference I can find in the logs is that with the quirk, I get this when turning it on/off:
`[zigpy.zcl] [0x08de:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]`
and
`[zigpy.zcl] [0x08de:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]`

and without the quirk I USED to see:

`[zigpy.zcl] [0x08de:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]`
and
`[zigpy.zcl] [0x08de:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>), Attribute(attrid=16385, value=<TypeValue type=uint16_t, value=0>), Attribute(attrid=16386, value=<TypeValue type=uint16_t, value=0>), Attribute(attrid=32769, value=<TypeValue type=enum8, value=enum8.undefined_0x01>), Attribute(attrid=32770, value=<TypeValue type=enum8, value=enum8.undefined_0x00>)]]`

The only difference I see between the signature and the replacement is `Time.cluster_id`. This is my first ZigBee switch, I have absolutely no idea what's going on. Can someone point me in the right direction please?

christovic, 2020-10-23:
Okay actually screw it, I went back to the config I thought worked and it doesn't. I am completely lost.

christovic, 2020-10-23:
Repairing seems to stop it from turning off after two minutes? I'll leave this open until I can find a pattern for sure, might be barking up the wrong tree.

christovic, 2020-10-25:
Going to write this here for anyone that finds it:

It seems like there's two versions of this switch coming from AliExpress. The TS0011 appears to turn off by itself after two minutes after being paired for the first time with ZHA. To get it to function correctly after pairing the first time, press and hold the switch for 10s to enter pairing mode and then press 'Add Devices Via This Device' (don't remove the deivce!) in your Zigbee Coordinator in Home Assistant. After that, it should function correctly.

closed 2020-10-25 by christovic

PerJarlemark, 2020-12-06:
Wow, Thank you for this, I just bought 7pcs of these devices and they all acted this way, in my case I had to re-pair the device again via another Coordinator (powerplug) than the device I added it the first time for the problem to go away.

these switches are sold with two different chipset and it is the new one after september 2020 that seems to have this problem.