zigpy/zha-device-handlers #536: [BUG] Mismatched Quirk: A19TunableWhite without model info (and Osram RGBW Outdoor Flex)

TheJulianJES, 2020-10-24:
**Describe the bug**
Someone mentioned on the Hass Discord in #zigbee that a OSRAM Lightify Outdoor Flex RGBW always got a wrong quirk: A19TunableWhite. This quirk removes the colorwheel (since #488) (which is needed for a RGBW led strip).
This happens because the quirk (https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/osram/a19twhite.py) never included any "MODELS_INFO" and the signature matches exactly between those two devices.
It's also possible that further OSRAM devices also match and thus don't show the color wheel.
A solution to this seems like to add the proper model (and manufacturer info) to the quirk.
(Or is there a reason this quirk doesn't include any model info at all? If not, perhaps a test that every quirk has at least one model in MODELS_INFO would be a good idea?)

**To Reproduce**
Steps to reproduce the behavior:
1. Bind a OSRAM Lightify Outdoor FLEX RGBW to ZHA
2. Try to change the color using Hass
3. See color wheel missing

**Expected behavior**
The OSRAM A19TunableWhite quirk doesn't apply for the RGBW light.

**Additional context**
(The user on Discord with this issue wanted to submit an issue, but it seems like that didn't happen after multiple days.)

Adminiuga, 2020-10-24:
It is better if it included model and manufacturer

TheJulianJES, 2020-10-25:
@danielbrunt57 Do you have the device signature of the A19TunableWhite OSRAM light?

danielbrunt57, 2020-10-25:
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=3)",
  "endpoints": {
    "3": {
      "profile_id": 260,
      "device_type": "0x0102",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x0b04",
        "0xfc0f"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "OSRAM",
  "model": "LIGHTIFY A19 Tunable White",
  "class": "zhaquirks.osram.a19twhite.A19TunableWhite"
}

Thanks,
Daniel

TheJulianJES, 2020-10-25:
Created PR #543 that just adds the model info, so the A19TunableWhite quirk isn't applied to other Osram RGBW products (thus not removing the color wheel for them anymore).

danielbrunt57, 2020-10-26:
My apologies for introducing a bug with PR #488 ...

TheJulianJES, 2020-10-26:
@danielbrunt57 The bug wasn’t introduced in your PR. It just so happens that now everyone notices that the quirk was missing the MODELS_INFO, as the RGB Color wheel was gone.

PR #543 (2020-10-29): Add model info to Osram A19TunableWhite quirk (Fixes #536)
Fixes #536 by also including MODELS_INFO for the Osram A19TunableWhite quirk (based on the device signature given here: https://github.com/zigpy/zha-device-handlers/issues/536#issuecomment-716198388)
commit 4df1f079 (2020-10-29): Add model info to Osram A19TunableWhite quirk (Fixes #536) (#543)

* Add model info to Osram A19TunableWhite quirk

* Renamed A19TunableWhite quirk to OsramTunableWhite, added 'LIGHTIFY RT Tunable White' model info
closed 2020-10-29 by Adminiuga