zigpy/zha-device-handlers #541: Support for 378RT

AlexMarent, 2020-10-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

**Additional context**
Add any other context or screenshots about the feature request here.

FrnchFrgg, 2020-10-26:
If you provide the correct signature (by copy-pasting the debug log when making the device join your network), I will probably be able to make it work. It seems it is another instance of tuya-based thermostats, as far as I can tell from the deconz support, which means #538 would be trivially extended to handle it.

AlexMarent, 2020-10-27:
I hope this is what you're looking for: 

```
Device 0x4614 (84:2e:14:ff:fe:f3:0b:8d) joined the network
[0x4614:zdo] ZDO request ZDOCmd.Device_annce: [0x4614, 84:2e:14:ff:fe:f3:0b:8d, 128]
[0x4614] Requesting 'Node Descriptor'
Tries remaining: 2
[0x4614] Extending timeout for 0x16 request
[0x4614] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0x4614] Discovering endpoints
Tries remaining: 3
[0x4614] Extending timeout for 0x18 request
[0x4614] Discovered endpoints: [1]
[0x4614:1] Discovering endpoint information
Tries remaining: 3
[0x4614] Extending timeout for 0x1a request
[0x4614:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x4614] Extending timeout for 0x1c request
[0x4614:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=28 command_id=Command.Read_Attributes_rsp>
[0x4614:1] Manufacturer: _TZE200_ckud7u2l
[0x4614:1] Model: TS0601
Checking quirks for _TZE200_ckud7u2l TS0601 (84:2e:14:ff:fe:f3:0b:8d)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0x4614:84:2e:14:ff:fe:f3:0b:8d entering async_device_initialized - is_new_join: True
device - 0x4614:84:2e:14:ff:fe:f3:0b:8d has joined the ZHA zigbee network

FrnchFrgg, 2020-10-29:
@AlexMarent That's exactly what I needed. It seems your device is exactly the same as the Moes HY368. At least it advertises itself with the same manufacturer and model ids, and the exact same cluster list too.

If you replace your zha quirks installation with the checkout of the PR #538 it should work already. If you can test it, that would be great, in particular the « child lock » function. Also if you can enable debugging on zhaquirks and zhaquirks.tuya then play with the functions of the valve, that would be great. I would be interested in the debug log and the list of what you tried.

Since I have none of the devices, I need that kind of help to ensure the support is correct.

AlexMarent, 2020-11-02:
@FrnchFrgg 
Sorry for that stupid question, but where I can find the quirks folder in the installation of home assistant?

FrnchFrgg, 2020-11-21:
@AlexMarent See https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install

PR #538 (2020-11-27): Thermostat
Implement support for the Siterwell GS361 and Shojzj 378RT (fixes: #420, fixes: #541).
Also implement support for Moes HY638
Support for Revolt NX4914-944 (fixes: #581)
Also support electric heating thermostat Moes BHT-002GCLZB (fixes #560)
closed 2020-11-27 by dmulcahey