zigpy/zha-device-handlers #542: [BUG] Philips RWL021 - delayed action on button press

bartek-w, 2020-10-25:
**Describe the bug**
Button press action for Philips RWL021 is significantly delayed after adding support for multiple clicks. 
(https://github.com/zigpy/zha-device-handlers/pull/501).
It adds 300ms (initially change adds 500ms) to every action. This changes perception of 'instant action' to 'slow response'.
I think this is great implementation for many usecases, but unfortunately destroys usability for basic 'single-click'.
I use a multiple Hue RWL021 with multiple rules and the usability after a change is low. 
I did a test recording with camera (240fps) and initial click-to-turn-light-on time was 60-80ms. After a change it is 380-400ms.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup automation rule: on single click, toggle hue light.
2. Click the button.

**Expected behavior**
Light turn on almost instantly. 

Is there option to make a 'delay threshold' configurable, so I could configure to 0? I'm happy to contribute a change, but I don't see any configurable quirk that I could use as reference.

dmulcahey, 2020-10-25:
1/3 of a second is a significant delay? Device specific configuration isn’t really possible yet. Change the quirk locally if the issue is that bad for you at the moment. I’ll look for a way to make this configurable when time permits.

Adminiuga, 2020-10-25:
I think it's a combination of the network latency and 300ms process delay gives this perception.
Maybe should do two different events for the 1st click? One without delay and always going to be there and 2nd one the counted press? E.g. `button_pressed` & `single_press`, `button_pressed` & `double_press`?
So there's a pair of events slightly off of each other, but user can pick and use one without the delay or if he wants counted clicks, then he has to use the delayed one?

Adminiuga, 2020-11-04:
Although I've just tested with RWL020 connected directly to ConBee and I can't say i feel the delay.

closed 2021-08-12 by github-actions[bot]