zigpy/zha-device-handlers #553: [BUG] IKEA 5 button remote, long turn on button press not recognised

SimonWilkinson, 2020-10-31:
**Describe the bug**

With the IKEA 5 button remote E1524, a long press of the on/off button isn't correctly recognised by the fivebtnremotezha.py device handler.

**To Reproduce**

Steps to reproduce the behavior:

Connect a 5 button remote to Home Assistant, and create an automation using the (LONG PRESS, TURN_ON) device action. This automation will not fire.

Observing the zha_event for the button gives:

`{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0d:6f:ff:fe:21:8b:75",
        "unique_id": "00:0d:6f:ff:fe:21:8b:75:1:0x0005",
        "endpoint_id": 1,
        "cluster_id": 5,
        "command": "release",
        "args": [
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-10-31T15:39:08.257617+00:00",
    "context": {
        "id": "37098ab81b8f11ebbadca97acff5559c",
        "parent_id": null,
        "user_id": null
    }
}`

**Expected behavior**

The automation should fire

**Additional context**

Replacing ARGS: [] with ARGS: [0] in the device_automation_trigger and restarting home assistant makes the device work correctly. Is this a bug with this device handler?

Signature of this particular device is:

`
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc7c"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI remote control",
  "class": "zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote"
}`

Adminiuga, 2020-10-31:
update firmware on the button

SimonWilkinson, 2020-10-31:
I think the button is running the newest Firmware. It was connected onto a Tradfri Gateway until very recently and updated before it was moved. It reports firmware 0x23014631 via Zigbee, which seems to match the buttons which are still on that Gateway (and claim to be up to date) which have 2.3.014.

There doesn't immediately seem to be a newer firmware version available in the version.json that lists all of the IKEA firmware, and https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html lists v2.3.0.14 as the newest firmware for it.

MattWestb, 2020-11-01:
Then the remote is one light steering device for string lights is sending commands (on/off/toggle/dim up/dim down) to do so that and not HA codes (short press/release/long press/longpres release/double/triple/quad). The power button is sending command "toggle" to the light groupe and is log pressing its sending one IKEA special soft reset / sync to the group for getting the powersate, brightness and color to being synced through the groupe.
So the remote is, cant and dont sending anny "long press" or "long press release" only that some creative ones like trying imprinting lights commands as that but its not true and only making problems if you is using it with lights that is made to do (You is soft resetting the lights state in the groupe then hope getting one command triggered for something else like changing wath is displaying on your TV).
The soft reset / sync funktion is documented in the [manual](https://www.ikea.com/at/de/manuals/tradfri-remote-control-kit-white-spectrum__AA-2143433-1_pub.pdf) for the remote / kitts with the remote.
For HA commands us HA switches that is sending pure commands and not messing with light switches that is sending light steering commands that is made for sending command direct to lights.
Its the same if trying using HA switches that is sending triple pres to one bulb and the bulb is not understanding it then its expecting getting one dimm down command. Going around and "translating" commands is possible but it can being very bad (not working if the SD-card is corrupted or after one power out and so on).

TheJulianJES, 2020-11-01:
Might be related to the issue described in this PR: https://github.com/zigpy/zha-device-handlers/pull/318

MattWestb, 2020-11-01:
Also some undocumented futures in the remote :-)))
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2618#issuecomment-649822696

closed 2021-08-12 by github-actions[bot]