zigpy/zha-device-handlers #555: [Device Support Request] CentraLite Contact Sensor-A

RubberDuQQy, 2020-10-31:
**TL;DR**: The battery entity for the `CentraLite Contact Sensor-A` always reports `Unavailable`.  It's very similar in signature to the CentraLite 3320-L sensor, so I'm hoping it just needs a device-specific quirk.

**Is your feature request related to a problem? Please describe.**
I have some Sylvania SMART+ Door/Window sensors that are detected as `CentraLite Contact Sensor-A`.  They do not have a device-specific quirk; the ZHA Network Card reports them using zha.device.Device.

The open/close and temp entities currently work fine in HA, but the battery level entity always shows `Unavailable`.  There is battery information available in the UI via the entity: battery size is reported as `Other`, battery quantity is reported as `1` and battery voltage is reported as a float like `2.7` or `2.6`.

I also have some other sensors that are very similar both in terms of device signature and physical attributes.  They are Lowes Iris Window/Door sensors and are detected as `Centralite 3320-L`.  These use the quirk zhaquirks.centralite.ias.CentraLiteIASSensor.

The only differences in the device signatures for these two devices (other than the detected model, of course) are: 

- The manufacturer_code entry in the node descriptor is different - I’m assuming this is how the different device model is detected.
- On the Contact Sensor-A Endpoint 2 has an extra in cluster with id 0x0001
- The Contact Sensor-A uses the default Zigbee device quirk, but the 3320-L uses a CentraLite-specific one.

**Describe the solution you'd like**
I'd like for the battery entity in HA to show the remaining percentage, similar to the entities for all my other sensor batteries.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4174, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 49887,
      "device_type": "0x000c",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05",
        "0xfc0f"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "CentraLite",
  "model": "Contact Sensor-A",
  "class": "zigpy.device.Device"
}
```

Adminiuga, 2020-11-04:
to me it looks exactly like signature of `CentraLiteIASSensor` quirk. 

Can you try modifying `zhaquirks/centralite/ias.py` and after https://github.com/zigpy/zha-device-handlers/blob/6682592d6b3935e495503ab20177312b8dd6b55a/zhaquirks/centralite/ias.py#L40 add another line with `(CENTRALITE, "Contact Sensor-A"),` ?

RubberDuQQy, 2020-11-04:
> Can you try modifying `zhaquirks/centralite/ias.py`

Still a bit of a rookie here... how do I go about doing that?

I'm running HA Supervised on a Linux (Ubuntu) box.  I found 3 copies of a file with that name in `/var/lib/docker/overlay2/`, one in a `merged` dir and two in `diff` dirs.  The relative path after he `merged` dir is `/usr/local/lib/python3.8/site-packages/zhaquirks/centralite/ias.py`.  I'm guessing that's the path inside the docker container.

Is it a matter of shutting down HA, modifying one of those files and then restarting HA?  Or is there a cleaner/better way to do it?

Adminiuga, 2020-11-04:
I don't think editing the merged directly is a good idea, but I've never tried.

Just run `docker exec -ti homeassistant pip3 install -U git+https://github.com/Adminiuga/zha-device-handlers@ac/centralite-contact-a#zha-quirks==0.0.46`

RubberDuQQy, 2020-11-04:
> Just run docker exec -ti homeassistant pip3 install -U git+https://github.com/Adminiuga/zha-device-handlers@ac/centralite-contact-a#zha-quirks==0.0.46

Ok, I can do that, but before I do... what's the process for reverting back if there's a problem?  

And assuming it does work, when the next release of HA comes along, will it automagically pull in the version of this package that's included with the new HA release (instead of this locally-modified version)? 

Thanks for your patience - since this is my production system and I have the WAF to consider, I want to make sure I understand how things go from here.  I'm learning about docker, but not quite comfortable yet.

Adminiuga, 2020-11-04:
There's a supervisor command to rebuild the container. Need to look it up

Adminiuga, 2020-11-06:
any luck?
to rebuild the container, you could do `ha core rebuild`

RubberDuQQy, 2020-11-07:
@Adminiuga: I updated the ZHA quirks as you mentioned above, restarted Home Assistant, and I now see the `CentraLiteIASSensor` quirk listed for both the `3320-L` sensors and the `Contact Sensor-A` sensors. 

However, the battery level still shows up as Unavailable for all 6 `Contact Sensor-A` sensors.  Further, the `Battery voltage` entry on the entity info dialog is now missing - only `Battery size` and `Battery quantity` are present now.  I've confirmed in the ZHA network info card that the quirks are listed and the devices have checked in (valid RSSI and LQI values).  I also tried clicking on `Reconfigure Device` in the HA UI, and that didn't seem to do anything.  Do I need to remove and re-pair the devices to get them to work properly with the new quirk?  Anything else to try?

Adminiuga, 2020-11-07:
Read battery voltage and batt % remaining attributes. A few times. Do you get any values? 

How long was it running?

RubberDuQQy, 2020-11-07:
It's been running about 20 minutes now.  Here are some examples, as reported by the States tab in Developer Tools:

**Contact Sensor-A:**

Entity: `sensor.living_room_window_south_battery_level`
State: `unavailable`

State attributes:
```
battery_size: Other
battery_quantity: 1
unit_of_measurement: '%'
friendly_name: Living Room Window South Battery Level
device_class: battery
```

**3320-L**

Entity: `sensor.living_room_window_north_battery_level`
State: `100`

State attributes:
```
battery_size: Other
battery_quantity: 1
unit_of_measurement: '%'
friendly_name: Living Room Window North Battery Level
device_class: battery
```

Is there a different way to read the attributes, since voltage and % are missing from the list?

Adminiuga, 2020-11-07:
Integrations -> zha -> devices -> pick your device -> manage clusters -> power configuration cluster -> pick attribute -> get attribute value button & wait

RubberDuQQy, 2020-11-07:
Ah ha!  I hadn't waited long enough for the sensors to report a battery value.  When navigating to a device per your above suggestion, I saw that the battery percentage is now being reported on the device page.  I checked the other 5 `Contact Sensor-A` sensors, and they all now have battery percentage data.  Awesome!

Thanks for the quick response and detailed suggestions, @Adminiuga!

Ship It 👍

PR #569 (2020-11-08): Add Centralite contact sensor a
Fixes #555
closed 2020-11-08 by dmulcahey