zigpy/zha-device-handlers #557: [Device Support Request]  Ikea Tradfri Wireless Dimmer Model ICTC-G-1

MattWestb, 2020-11-01:
**Is your feature request related to a problem? Please describe.**
https://github.com/home-assistant/core/issues/42672#issue-733729408

**Describe the solution you'd like**
Battery is not working OK without quirk (have some one pointed out).
Binding to groups is working and dimming lights but not the HA part for automations then its not implanted without quirk.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

logs while pairing:

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Tries remaining: 1
[0x8a5f] Extending timeout for 0x25 request
[0x8a5f] Delivery error for seq # 0x25, on endpoint id 0 cluster 0x0004: message send failure
[0x8a5f:1] Failed to discover endpoint_id 1
Device not found for removal: 00:0b:57:ff:fe:97:d2:cd
Device 0xdf77 (00:0b:57:ff:fe:97:d2:cd) joined the network
[0xdf77] Requesting 'Node Descriptor'
Tries remaining: 2
[0xdf77] Extending timeout for 0x27 request
Device 0xdf77 (00:0b:57:ff:fe:97:d2:cd) joined the network
Skip initialization for existing device 00:0b:57:ff:fe:97:d2:cd
[0xdf77:zdo] ZDO request ZDOCmd.Device_annce: [0xDF77, 00:0b:57:ff:fe:97:d2:cd, 128]
[0xdf77] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=10752, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0xdf77] Discovering endpoints
Tries remaining: 3
[0xdf77] Extending timeout for 0x29 request
[0xdf77] Discovered endpoints: [1]
[0xdf77:1] Discovering endpoint information
Tries remaining: 3
[0xdf77] Extending timeout for 0x2b request
[0xdf77:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=49246, device_type=2064, device_version=2, input_clusters=[0, 1, 3, 9, 2821, 4096], output_clusters=[3, 4, 6, 8, 25, 4096])
[0xdf77] Extending timeout for 0x2d request
[0xdf77:1:0x0000] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=45 command_id=Command.Read_Attributes_rsp>
[0xdf77:1] Manufacturer: IKEA of Sweden
[0xdf77:1] Model: TRADFRI wireless dimmer
Checking quirks for IKEA of Sweden TRADFRI wireless dimmer (00:0b:57:ff:fe:97:d2:cd)
Considering <class 'zhaquirks.ikea.tradfriplug.TradfriPlug'>
Fail because endpoint list mismatch: {1, 2, 242} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because profile_id mismatch on at least one endpoint
device - 0xDF77:00:0b:57:ff:fe:97:d2:cd entering async_device_initialized - is_new_join: True
device - 0xDF77:00:0b:57:ff:fe:97:d2:cd has joined the ZHA zigbee network
[0xDF77](TRADFRI wireless dimmer): started configuration
[0xDF77:ZDO](TRADFRI wireless dimmer): 'async_configure' stage succeeded
[0xdf77] Extending timeout for 0x2f request
[0xdf77] Extending timeout for 0x31 request
[0xdf77] Extending timeout for 0x33 request
[0xDF77:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0xdf77] Extending timeout for 0x35 request
[0xDF77:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xDF77:1:0x0000]: finished channel configuration
[0xdf77] Extending timeout for 0x37 request
[0xDF77:1:0x1000]: bound 'lightlink' cluster: Status.SUCCESS
[0xDF77:1:0x1000]: finished channel configuration
[0xdf77] Extending timeout for 0x39 request
[0xdf77:1:0x0001] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=53 command_id=Command.Configure_Reporting_rsp>
[0xdf77:1:0x0000] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=55 command_id=Command.Read_Attributes_rsp>
[0xDF77:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[ConfigureReportingResponseRecord(status=0)]]'
[0xdf77] Extending timeout for 0x3b request
[0xDF77:1:0x0000]: initializing channel: from_cache: False
[0xdf77] Extending timeout for 0x3d request
[0xDF77:1:0x0008]: bound 'level' cluster: Status.SUCCESS
[0xDF77:1:0x0008]: finished channel configuration
[0xdf77] Extending timeout for 0x3f request
[0xdf77:1:0x0001] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=59 command_id=Command.Configure_Reporting_rsp>
[0xDF77:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[[ConfigureReportingResponseRecord(status=0)]]'
[0xDF77:1:0x0001]: finished channel configuration
[0xDF77:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xDF77:1:0x0006]: finished channel configuration
[0xDF77:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0xDF77:1:0x0019]: finished channel configuration
[0xDF77:1:0x0001]: 'async_configure' stage succeeded
[0xDF77:1:0x0000]: 'async_configure' stage succeeded
[0xDF77:1:0x1000]: 'async_configure' stage succeeded
[0xDF77:1:0x0008]: 'async_configure' stage succeeded
[0xDF77:1:0x0006]: 'async_configure' stage succeeded
[0xDF77:1:0x0019]: 'async_configure' stage succeeded
[0xDF77](TRADFRI wireless dimmer): completed configuration
[0xDF77](TRADFRI wireless dimmer): stored in registry: ZhaDeviceEntry(name='IKEA of Sweden TRADFRI wireless dimmer', ieee='00:0b:57:ff:fe:97:d2:cd', last_seen=1604152683.289528)
[0xdf77] Extending timeout for 0x41 request
[0xdf77:1:0x0003] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=65 command_id=Command.Default_Response>
[0xDF77:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.SUCCESS: 0>]
[0xDF77](TRADFRI wireless dimmer): started initialization
[0xDF77:ZDO](TRADFRI wireless dimmer): 'async_initialize' stage succeeded
[0xdf77] Extending timeout for 0x43 request
[0xdf77] Extending timeout for 0x45 request
[0xDF77:1:0x1000]: initializing channel: from_cache: False
[0xDF77:1:0x0008]: initializing channel: from_cache: False
[0xDF77:1:0x0006]: initializing channel: from_cache: False
[0xDF77:1:0x0019]: initializing channel: from_cache: False
[0xdf77:1:0x0001] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=67 command_id=Command.Read_Attributes_rsp>
[0xDF77:1:0x0001]: initializing channel: from_cache: False
[0xdf77] Extending timeout for 0x47 request
[0xdf77:1:0x0000] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=69 command_id=Command.Read_Attributes_rsp>
[0xDF77:1:0x0000]: initializing channel: from_cache: False
[0xdf77:1:0x0001] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=71 command_id=Command.Read_Attributes_rsp>
[0xDF77:1:0x0001]: 'async_initialize' stage succeeded
[0xDF77:1:0x0000]: 'async_initialize' stage succeeded
[0xDF77:1:0x1000]: 'async_initialize' stage succeeded
[0xDF77:1:0x0008]: 'async_initialize' stage succeeded
[0xDF77:1:0x0006]: 'async_initialize' stage succeeded
[0xDF77:1:0x0019]: 'async_initialize' stage succeeded
[0xDF77](TRADFRI wireless dimmer): power source: Battery or Unknown
[0xDF77](TRADFRI wireless dimmer): completed initialization

**Additional context**

The logs its copied from @springstan issue posted on HA git and the credit is going to hi !!

MattWestb, 2020-12-30:
Hey @TheJulianJES can you please taking one look if its possible doing one quirk for the old wireless remote that is having flace battery reporting.
I think its no need converting the ZLL commands being send to ZHA commands then its pretty useless as remote  for other things then lights (and also very tricky for steering lights) but its nearly one retro device from IKEA (the hardware is one of the most advanced / expensive but was little overkill then is not working so well with the electronic gyros).

Its no high prio but being nice to getting the battery reporting OK :-))

Thanks in advance !!

TheJulianJES, 2020-12-30:
You can try this: https://github.com/TheJulianJES/zha-device-handlers/blob/ikea_old_dimmer/zhaquirks/ikea/dimmer.py
This should only double the percentage of the battery (like it's done for every other IKEA device).
This won't include any device automations. If this works, I'll PR it.

MattWestb, 2020-12-30:
Thanks very much!!

Very strange problem that the cluster on the endpoint is not the same as my dimmer is having.
Also the device type is wrong (its technical no color dimmer controlling device).
Perhaps the original poster was running in old ZLL firmware and the current is ZB3.
Was trying "patching" your quirk but it dont like loading then i changed cluster and device type so i am doing somthing wrong :-(((
Was comparing with the E1743 and trying adopting it but with no luck.

Can you pleas making one more try to fixing it ?

Here is the init strings then pairing it with HA 2020.12.0:
```` yaml
Device 0xc300 (00:0b:57:ff:fe:27:04:b3) joined the network
[0xc300] Requesting 'Node Descriptor'
Device 0xc300 (00:0b:57:ff:fe:27:04:b3) joined the network
Device 0xc300 (00:0b:57:ff:fe:27:04:b3) joined the network
[0xc300] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0xc300] Discovering endpoints
[0xc300] Discovered endpoints: [1]
[0xc300:1] Discovering endpoint information
[0xc300:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=2080, device_version=1, input_clusters=[0, 1, 3, 32, 4096], output_clusters=[3, 4, 6, 8, 25, 4096])
.
.
2020-12-30 18:52:42 DEBUG (MainThread) [zigpy.zcl] [0x94bf:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=242 command_id=Command.Read_Attributes_rsp>
2020-12-30 18:52:42 DEBUG (MainThread) [zigpy.endpoint] [0x94bf:1] Manufacturer: IKEA of Sweden
2020-12-30 18:52:42 DEBUG (MainThread) [zigpy.endpoint] [0x94bf:1] Model: TRADFRI wireless dimmer
2020-12-30 18:52:42 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for IKEA of Sweden TRADFRI wireless dimmer (00:0b:57:ff:fe:27:04:b3)
2020-12-30 18:52:42 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.dimmer.IkeaDimmer'>
2020-12-30 18:52:42 DEBUG (MainThread) [zigpy.quirks.registry] Fail because profile_id mismatch on at least one endpoint
````
And one screenshot from the cluster panel:
[image]

If you need more information please give my one sign.

TheJulianJES, 2020-12-30:
Updated it to the new ZB3 version: https://github.com/TheJulianJES/zha-device-handlers/blob/ikea_old_dimmer/zhaquirks/ikea/dimmer.py
This is now similar to the SYMFONISK quirk. (Perhaps the ``device_automation_triggers`` could just be copied over?)

MattWestb, 2020-12-30:
Its working great !!!!

Now i have 100% new battery instead of the old 50% on one new battery and 60% of the old battery :-)))

Was trying implanting the SYMFONISK commands but its not triggering then its little different:
```
Rotate left:
2020-12-30 22:02:13 DEBUG (MainThread) [zigpy.zcl] [0x94bf:1:0x0008] ZCL request 0x0001: [1, 195]
2020-12-30 22:02:13 DEBUG (MainThread) [zigpy.zcl] [0x94bf:1:0x0008] No handler for cluster command 1

Ritate right:
2020-12-30 22:14:12 DEBUG (MainThread) [zigpy.zcl] [0x94bf:1:0x0008] ZCL request 0x0005: [0, 195]
2020-12-30 22:14:12 DEBUG (MainThread) [zigpy.zcl] [0x94bf:1:0x0008] No handler for cluster command 5
```

Its sending command 1,4,5 and 7 with some different parameters on cluster 0x0008.

Thanks very much !!!!

MattWestb, 2020-12-30:
Its firing ZHA events but its not so easy for some that like using it for automations then its different light levels commands with values:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:27:04:b3",
        "unique_id": "00:0b:57:ff:fe:27:04:b3:1:0x0008",
        "device_id": "5a8fbe69058511eb9ddadbb100ad3bbf",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move",
        "args": [
            1,
            195
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-12-30T21:52:44.098715+00:00",
    "context": {
        "id": "1e072caa86709421e93af64842dbe29f",
        "parent_id": null,
        "user_id": null
    }
}
```
For my its more then enough for straining one groupe of lights and if some one like triggering HA events they can doing little more digging ;-))

TheJulianJES, 2020-12-30:
The device triggers looks like the same from symfonisk.py:
``
    device_automation_triggers = {
        (SHORT_PRESS, TURN_ON): {
            COMMAND: COMMAND_TOGGLE,
            CLUSTER_ID: 6,
            ENDPOINT_ID: 1,
        },
        (ROTATED, RIGHT): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 195],
        },
        (ROTATED, LEFT): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 195],
        },
        (DOUBLE_PRESS, TURN_ON): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 1, 0],
        },
        (TRIPLE_PRESS, TURN_ON): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 1, 0],
        },
    }
``
Can you perhaps confirm if they are all the same? (They could be included in the PR then (even if you don't use them))

MattWestb, 2020-12-30:
I was trying but only right and left turn can being triggered (dont having the switch that symfonisk is having or clicks) and the move to xxx is with or without off is not working and i think it's to much work digging all usebal parameters its sending for doing this ZHA commands.
If some like using then its fired as raw ZHA-events and can being triggered in automations.
Dont looks so nice but can being done.

TheJulianJES, 2020-12-30:
``
        (ROTATED, RIGHT): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 195],
        },
        (ROTATED, LEFT): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 195],
        },
``
So basically only these two work. The log from the one you posted should be "rotated left" then.
I'll add these two triggers and PR it.

MattWestb, 2020-12-30:
You need defining the `ROTATED = "device_rotated"` also or its not loading the quirk.

TheJulianJES, 2020-12-30:
Yep, updated code is here: https://github.com/TheJulianJES/zha-device-handlers/blob/ikea_old_dimmer/zhaquirks/ikea/dimmer.py
I'll PR it in a couple of seconds, but it *should* all work.
(Thought about putting the ``ROTATED = "device_rotated"`` in the IKEA ``__init__.py`` file, but I'm unsure if this will make things unclear/if it's required)

TheJulianJES, 2020-12-30:
Opened PR here now: https://github.com/zigpy/zha-device-handlers/pull/685

closed 2020-12-30 by Adminiuga