zigpy/zha-device-handlers #559: Aqara Vibration - No battery detected

wes1993, 2020-11-02:
**Is your feature request related to a problem? Please describe.**
When i try to pair my Aqara Vibrations sensor (DJT11LM) i don't have the battery details.

**Describe the solution you'd like**
Have the battery details

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
Device 0xc2b5 (00:15:8d:00:02:b1:5a:7a) left the network
Tries remaining: 2
[0xc2b5] Extending timeout for 0x05 request
Device 0x7056 (00:15:8d:00:02:b1:5a:7a) joined the network
Device 00:15:8d:00:02:b1:5a:7a changed id (0xc2b5 => 0x7056)
Canceling old initialize call
[0x7056] Requesting 'Node Descriptor'
Tries remaining: 2
[0x7056] Extending timeout for 0x06 request
[0x7056:zdo] ZDO request ZDOCmd.Device_annce: [0x7056, 00:15:8d:00:02:b1:5a:7a, 128]
Ignoring message (b'18000a050042126c756d692e766962726174696f6e2e61713101002008') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'1c5f11010a01ff422e01219b0a0328170421a8010521710206240100000000082108030a21000098212800992100009a25bffe1801cb03') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18020a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
[0x7056] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)
[0x7056] Discovering endpoints
Tries remaining: 3
[0x7056] Extending timeout for 0x07 request
Ignoring message (b'18030a5500210100') on cluster 257: unknown endpoint or cluster id: 1
Ignoring message (b'18040a5500210300') on cluster 257: unknown endpoint or cluster id: 1
Tries remaining: 2
[0x7056] Extending timeout for 0x08 request
Tries remaining: 1
[0x7056] Extending timeout for 0x09 request
Ignoring message (b'18050a0805253000f3ffe104') on cluster 257: unknown endpoint or cluster id: 1
Ignoring message (b'18060a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
[0x7056] Failed to discover active endpoints
Ignoring message (b'18070a5500210300') on cluster 257: unknown endpoint or cluster id: 1
Ignoring message (b'18080a0805253400efffdf04') on cluster 257: unknown endpoint or cluster id: 1
[0x7053:1:0x0500]: Updated alarm state: 0
Ignoring message (b'18090a5500210300') on cluster 257: unknown endpoint or cluster id: 1
Ignoring message (b'180a0a5500210100') on cluster 257: unknown endpoint or cluster id: 1
Ignoring message (b'180b0a0805253000f2ffde04') on cluster 257: unknown endpoint or cluster id: 1
Ignoring message (b'180c0a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'180d0a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'180e0a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'180f0a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18100a5500210300') on cluster 257: unknown endpoint or cluster id: 1
Ignoring message (b'18110a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18120a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18130a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18140a5500210300') on cluster 257: unknown endpoint or cluster id: 1
Ignoring message (b'18150a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18160a5500210100') on cluster 257: unknown endpoint or cluster id: 1
Device 0x7056 (00:15:8d:00:02:b1:5a:7a) left the network
Device 0xec7e (00:15:8d:00:02:b1:5a:7a) joined the network
Device 00:15:8d:00:02:b1:5a:7a changed id (0x7056 => 0xec7e)
[0xec7e] Requesting 'Node Descriptor'
Tries remaining: 2
[0xec7e] Extending timeout for 0x0b request
[0xec7e:zdo] ZDO request ZDOCmd.Device_annce: [0xEC7E, 00:15:8d:00:02:b1:5a:7a, 128]
Ignoring message (b'18000a050042126c756d692e766962726174696f6e2e61713101002008') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'1c5f11010a01ff422e0121910a0328190421a8010521710206240100000000082108030a21000098212800992100009a252f002b02d403') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18020a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
[0xec7e] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)
[0xec7e] Discovering endpoints
Tries remaining: 3
[0xec7e] Extending timeout for 0x0c request
Ignoring message (b'18030a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 1
[0xec7e] Discovered endpoints: [1, 2]
[0xec7e:1] Discovering endpoint information
Tries remaining: 3
[0xec7e] Extending timeout for 0x0d request
Ignoring message (b'18040a050042126c756d692e766962726174696f6e2e617131') on cluster 0: unknown endpoint or cluster id: 'No cluster ID 0x0000 on (00:15:8d:00:02:b1:5a:7a, 1)'
[... 42 lines cut ...]
[0xEC7E:1:0x0005]: 'async_initialize' stage succeeded
[0xEC7E:2:0x0005]: 'async_initialize' stage succeeded
[0xEC7E](lumi.vibration.aq1): power source: Battery or Unknown
[0xEC7E](lumi.vibration.aq1): completed initialization
[0xec7e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=7 command_id=Command.Report_Attributes>
[0xec7e:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=5, value=<TypeValue type=CharacterString, value=lumi.vibration.aq1>)]]
[0xec7e:1:0x0000] Attribute report received: model=lumi.vibration.aq1
[0xec7e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=8 command_id=Command.Report_Attributes>
[0xec7e:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=5, value=<TypeValue type=CharacterString, value=lumi.vibration.aq1>)]]
[0xec7e:1:0x0000] Attribute report received: model=lumi.vibration.aq1
[0xec7e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=9 command_id=Command.Report_Attributes>
[0xec7e:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=5, value=<TypeValue type=CharacterString, value=lumi.vibration.aq1>)]]
[0xec7e:1:0x0000] Attribute report received: model=lumi.vibration.aq1
[0xec7e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=10 command_id=Command.Report_Attributes>
[0xec7e:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=5, value=<TypeValue type=CharacterString, value=lumi.vibration.aq1>)]]
[0xec7e:1:0x0000] Attribute report received: model=lumi.vibration.aq1
**Additional context**
Add any other context or screenshots about the feature request here.

Operating System
Debian GNU/Linux 10 (buster)
Docker version
19.03.13

Home Assistant 0.117.2

Adminiuga, 2020-11-06:
It applied the quirk and if it is still on the network should report the battery within an hour. Did it report it?

wes1993, 2020-12-15:
OK now i can see the battery

closed 2020-12-15 by wes1993