zigpy/zha-device-handlers #561: [Device Support Request] Trying getting the IKEA 5 button remotes (IKEA E1524/E1810) to work with color / TC control

MattWestb, 2020-11-03:
**Is your feature request related to a problem? Please describe.**
Very mutch posted all around but no have getting it working !!

**Describe the solution you'd like**
IKEA is using some undocumented commands for changing / walking thru color scenes in their bulbs that is working if doing one firmware upgrade of bulbs but is deleted then resetting the bulbs :-((

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

**Additional context**
I have reading logs and testing in deCONZ and the firmware is loading 3 scenes (0,1 and 2 with groupe 0xff09) in the bulbs then doing firmware upgrade. deCONZ is very tricky and many times deleting them and not adding the groupe stored in the bulb then initiating it. Some time its configuring one light groupe 0xff09 without controlling device but oft not.
Then having the groupe 0xff09 or making on in the DB and restarting it its is possible adding scenes to groupe 0xff09 in the bulb and its working selecting / walking thru them with the remot.
The tricky thing is that the remote is bounded to one  normal light group that that cant being changed (in deCONZ) and its not the same as the 0xff09 one and its can being conflicts if having more sending command to the same groupe. If adding the scenes with the "normal" light groupe the scenes is stored in the bulb but dot reacting on the remotes command being sended.
IKEA is very likely setting up the "0XFF09" group with touchlink that wireshark is not so good decoding. If can finding the commands send to the remote for setting up the "0XFF09" group bindings its open for using normal group binding and adding normal scenes that is individual configured on group basis  with multiple remotes in the system.

One user have doing very good sniffs from different scenarios and mutch info is in them like how the GW is setting up on normal groupe and adding the scenes but i cant getting how the GW is configuring the "0XFF09" group in the remote (can being its made with touch link only).https://github.com/Koenkk/zigbee2mqtt/issues/4414#issue-705096988
I have posted some of  my findings of how adding more the scenes in the bulbs there.
I have trying sniffing paring one remote with the GW but its normally dont like paring with the GW then i have the sniffer setted up :-(((

I have using deCONZ for working with scenes (with buggy problems and limitations) then it have it  implanted in the GUI then I have not working with Z2M then its have luck of support for EZSP and ZHA have not yet implanted the "upper layer" for scenes.

Then I knowing Adminiuga (and some more here around) have learning how to use his IQ very well for solving problems i kindly asking for help looking in the posted logs and trying finding the hidden secrets how IKEA is doing there magick with the remote then pairing it.

I still trying doing more sniffs from remote pairing with the GW and also trying then touch linking remote to remote if its can helping.

Thanks in advance for spending time and offers for doing ZHA being one great system for our zigbee devices around the world !!!

MattWestb, 2020-11-04:
## Findings for @Adminiuga 

Have finding one interesting writing about group Addressing in ZLL. [UG103.9: ZLL Fundamentals](https://www.silabs.com/documents/public/user-guides/ug103-09-fundamentals-zll.pdf) UG103.9: ZLL Fundamentals.
```
The group identifiers need to be unique in the network and their range is (0x0001…0xFEFF). 
Group identifier 0x0000 is reserved for the default group in the ZCL scene cluster, 
and group identifiers in the range (0xFF00…0xFFFF) are reserved.
```
So the groupe 0x0000 is reserved so sould not being used as default binding groupe for the coordinator then its reserved as default group for scenes in the ZCL.

I think its for the moment no problems then ZHA is not using scenes but deCONZ have silent changing there default binding groupe from 0x0000 to 0xfff0.

My finding in this case is that IKEA is using 0xff09 for their scenes stored in the bulbs (after firmware update) and the controlling devices have it as the default allocated group then not being setted up with touchlink (classical paired controlling devices). 

Only for info if getting problems later on the road :-)))

Adminiuga, 2020-11-04:
Group 0x0000 is reserved for the scenes commands. Coordinator is assigned to it, not bound only because Ikea remotes send commands to that group if not bound to any other group

MattWestb, 2020-11-04:
That is awaring way its possible binding the remotes to one group without unbinding it from the coordinator as was the thinking with the Philips HUE dimmer switch :-)))

deCONZ is for the moment binding to the 0xfff0 then adding one device to it and i think its because problem if not getting the group binding working then pairing one remote its steering all the devices then most devices having 0x0000 as default groupe after being reseted.

It was only for info ;-)

MattWestb, 2020-11-06:
I doing the sniffing analyce in 2 parts for not messing all things up.
The first easy part is paring one GU10 WS with one remote that is already paired with the IKEA GW and the network key is known (sniffed then doing one classical pairing pf one bulb to the GW).

Its use little of IKEA specific commands and touchlink commands for transferring network information then joining the network but its looks that its possible doing the same in deCONZ after one normal joining.

I have only looking how IKEA is adding the scenes to the bulb so the other setup is skipped.

```
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
    Sequence Number: 32
    Command: Remove all Groups (0x04)

ZigBee Cluster Library Frame, Command: Default Response, Seq: 32
    Frame Control Field: Profile-wide (0x08)
    Sequence Number: 32
    Command: Default Response (0x0b)
    Response to Command: 0x04
    Status: Success (0x00)

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
    Sequence Number: 33
    Command: Add Group (0x00)
    Payload, String: default
        Group ID: 0x002a
        Length: 7
        String: default

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x19)
    Sequence Number: 33
    Command: Add Group Response (0x00)
    Payload
        Group Status: Success (0x00)
        Group ID: 0x002a

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 33
    Command: Get Group Membership (0x02)
    Payload
        Group Count: 0

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x19)
    Sequence Number: 33
    Command: Get Group Membership Response (0x02)
    Payload
        Group Capacity: 255
        Group Count: 1
        Group List
            Group ID: 0x002a

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 34
    Command: Get Scene Membership (0x06)
    Payload
        Group ID: 0x002a

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x19)
    Sequence Number: 34
    Command: Get Scene Membership Response (0x06)
    Payload
        Scenes Status: Success (0x00)
        Scene Capacity: 61
        Group ID: 0x002a
        Scene Count: 0
[... 211 lines cut ...]

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x19)
    Sequence Number: 68
    Command: View Scene Response (0x01)
    Payload, String: 
        Scenes Status: Success (0x00)
        Group ID: 0x002a
        Scene ID: 0x03
        Transition Time: 1 seconds
        Length: 0
        String: 
        Extension field set 1
            Cluster: On/Off (0x0006)
            On/Off: 1
        Extension field set 2
            Cluster: Level Control (0x0008)
            Level: 132
        Extension field set 3
            Cluster: Color Control (0x0300)
            Color X: 0,3790
            Color Y: 0,3753

```
That was the easy part if the remote is sending the "previous and next scene" command to the group that was configured to the scene . For the moment i can only getting the remote sending this commands to groupe =0xff09 but its working very well in deCONZ.

MattWestb, 2020-11-06:
One thing is then looking how the scenes is added is it with colour, On/Off and level configured.
Some magic is here because changing the default scene with the remote its only doing the colour and not the other setted attributes of the scene. Then manual adding scenes in deCONZ its using all attributes so also the brightness is being forced then changing scene.

MattWestb, 2020-11-06:
Now to the tricky thing:
The IKEA GW is tecnikely one "Control Bridge" and not one coordinator:

2.3.5 Control Bridge
The Control Bridge device is typically used in nodes that relay ZLL control commands
issued from another network, e.g. an Internet router with a ZLL interface.
Its Device ID is 0x0840.
The clusters supported by this device are as follows:

Server (Input) Side Clusters Client (Output) Side Clusters
- Basic S
- Identify C
- Groups C
- Scenes C
- On/Off C
- Level Control C
- Colour Control C
- ZLL Commissioning S  ZLL Commissioning C
Table 15: Clusters for Control Bridge Device

The remote is one "Non-Colour Controller":

2.3.3 Non-Colour Controller
The Non-Colour Controller device is typically used in nodes that issue ZLL control
commands that are not related to colour, e.g. a remote control unit for monochrome
lamps.
Its Device ID is 0x0820.
The clusters supported by this device are as follows:
Server (Input) Side Clusters Client (Output) Side Clusters
- Basic S
- Identify C
- Groups C
- On/Off C
- Level Control C
- ZLL Commissioning ZLL S  Commissioning C
Table 13: Clusters for Non-Colour Controller Device

But its soul being one " Non-Colour Scene Controller":

2.3.4 Non-Colour Scene Controller
The Non-Colour Scene Controller device is typically used in nodes that issue ZLL
control commands that are not related to colour and that support ‘scenes’, e.g. a
remote control unit for monochrome lamps.
Its Device ID is 0x0830.
The clusters supported by this device are as follows:
Server (Input) Side Clusters Client (Output) Side Clusters
- Basic S
- Identify C
- Groups C
- Scenes C
- On/Off C
- Level Control C
- ZLL Commissioning S  ZLL Commissioning C
Table 14: Clusters for Non-Colour Scene Controller Device

(from NXP JN-UG-3091.pdf)

In TL and also in ZB3 with touch link its the inistator of the network giving all neseerly network parameters to the new device. Then forming the network its the GW that is doing that for the first controlling device. Its also giving the controller device one address and one unique group plus one address range for giving new joined devices. Then one devices is touch link joined its getting one address from the controller device (in the range of its address range) and the group its having. Controller devices can't normally not changing grupes (in the IKEA world) but lights can.

The good thing its very easy cloning one controller device with touchlink. Reset the "new" device and touchlink it with one aktive device and its getting the network parameters and one new address plus the group from the aktive device. Then both is controlling the same groupe and have different address that the GW is sniffing and allocating it in the system.

Now we is having 2 devices that is working good in touchlink joined but having problem then being joined to one TC in classical joining mode.

The first is the one is the TRADFRI signal repeater.
The problem is that is having basic, identity, alarm, diagnostic, touchlink, IKEA, OTA and pull control cluster but no grope or window closing cluster.
The being touchlinked its getting one group that is bonded to the device and is working like one realty for commands from the open/close switch to the blinds that is bonded to the group.
The doing one clasicat joining its not possible binding the window closing cluster then its dont haven it but both the on/off switch and the blinds is having it. Also its not having any grope cluster for binding and configuring the device. I think its not working as intended in one TC network as is doing in the IKEA network then its having the group from the touchlink pairing.
The signal repeater is more or less "floating around" in the network.

The second device is the remote control.
The problem its not having any scene cluster and its sending scene command to its grope.
Then touchlinking paring its getting the group and its sending the scene commands to the groupe and also on/off and light level commands. Then being classical paired its falling back to its default group its having it being reseted (0xff09). If adding one group and bonding the clusters its sending on/off and light level to the "new" group but scene commands its sent to the 0xff09 group.

I have testing adding scenes to IKEA GU10 WS bulbs with groupe 0xff09 in deCONZ and i can controlling / sending scenes changing commands to them with deCONZ and the remote without problems. The problem is then having more remotes in the system then all lights is changing the scene then all scenes is in the same scene group. If having more groups for the lights its possible steering the on/off and light level for the binded group but the scenes is controlled for al remotes then all have the same group bounded for scenes commands.

I have not finding any solution for this problem but more deep diving in sniffs can giving more input.

For the moment its gets 2 alternatives.
1: Kindly asking IKEA implanting scene cluster on the remote so its possible bonding it to one groupe or doing it with the first groupe storad (after delete all group command) and not the "default" groupe and the same with the signal repeater with window closing cluster (and implanting group cluster).
2: Configuring the groupe thru touchlink (joining / configuring the device thru touchlink in zb3 mode that is possible but many saying its not).

Next is coming more touchlink sniffing :-)))

MattWestb, 2020-11-06:
## Wireshark Zigbee sniffing problems.

Then pairing one IKEA remotes with touchlink its requesting one "Get Group Identifiers Request (0x41)" in the beginning for getting the network parameters and getting one Get Group Identifiers Request (0x41). Then its getting the endpoints and cluster config / binding and reporting.
For the remote its its then requesting one more time "Get Group Identifiers Request (0x41)" for getting the groupe from the initiator and getting one response and doing the rest of the confing.
```
Frame 176: 113 bytes on wire (904 bits), 113 bytes captured (904 bits) on interface \Device\NPF_Loopback, id 0
Null/Loopback
Internet Protocol Version 4, Src: 127.0.0.1, Dst: 127.0.0.1
User Datagram Protocol, Src Port: 17754, Dst Port: 17754
ZigBee Encapsulation Protocol, Channel: 20, Length: 49
    Protocol ID String: EX
    Protocol Version: 2
    Type: Data (1)
    Channel ID: 20
    Device ID: 38048
    LQI/CRC Mode: LQI
    Link Quality Indication: 244
    Timestamp: Nov  3, 2020 13:49:43.165999999 W. Europe Standard Time
    Sequence Number: 251
    .011 0001 = Length: 49 bytes
IEEE 802.15.4 Data, Dst: 0x0304, Src: 0x9bd1
    Frame Control Field: 0x8861, Frame Type: Data, Acknowledge Request, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
    Sequence Number: 43
    Destination PAN: 0xe758
    Destination: 0x0304
    Source: 0x9bd1
    [Extended Source: Ember_ff:fe:51:43:5c (00:0d:6f:ff:fe:51:43:5c)]
    [Origin: 145]
    TI CC24xx-format metadata: FCS OK
ZigBee Network Layer Data, Dst: 0x0304, Src: 0x9bd1
    Frame Control Field: 0x2248, Frame Type: Data, Discover Route: Enable, Security, End Device Initiator Data
    Destination: 0x0304
    Source: 0x9bd1
    Radius: 30
    Sequence Number: 125
    [Extended Source: Ember_ff:fe:51:43:5c (00:0d:6f:ff:fe:51:43:5c)]
    [Origin: 145]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x40)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .1.. .... = Acknowledgement Request: True
        0... .... = Extended Header: False
    Destination Endpoint: 1
    Cluster: ZLL Commissioning (0x1000)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 241
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...0 .... = Disable Default Response: False
    Sequence Number: 0
    Command: Get Group Identifiers Request (0x41)
[Malformed Packet: ZCL Touchlink]
    [Expert Info (Error/Malformed): Malformed Packet (Exception occurred)]
        [Malformed Packet (Exception occurred)]
        [Severity level: Error]
        [Group: Malformed]
```` 
Request.
````
Frame 186: 118 bytes on wire (944 bits), 118 bytes captured (944 bits) on interface \Device\NPF_Loopback, id 0
Null/Loopback
Internet Protocol Version 4, Src: 127.0.0.1, Dst: 127.0.0.1
User Datagram Protocol, Src Port: 17754, Dst Port: 17754
[... 66 lines cut ...]
    Frame Control Field: Cluster-specific (0x19)
    Sequence Number: 59
    Command: Get Group Membership Response (0x02)
    Payload
        Group Capacity: 255
        Group Count: 1
        Group List
            Group ID: 0x0003
````

So the group is 0x0003 and the group id is 0x01 then its only one 0x01 or 0x0001 that is missing.

The the groupe information is in the message then the decryption must being OK but the message wrongly parsed because of bug or some have paying for not doing it right,

I know that the touchlink part is not complete implanted in wireshark and its can also being bugs in the sniffer firmware. I have using one EZSP with 6.7.6.0 for the sniffing.

### @Adminiuga  Do you have one clue ??
Do you knowing some other program that can do the zigbee decrypting  and message decoding of the touchlink frames ?
I have all the normal keys also the network key for my IKEA GW.
With my mathematics knowledge its not so good that i doing it by hand :-(((
Can it being sniffer hardware problem ?
I have 2 CC-2531 that i can flashing with sniffing firmware but i think its useless if the problem is in wireshark.

For the remote problem this sniff analyse verifying that configuring the groupe of IKEA controlling devices thru touchlink is doing the "hidden" scene cluster" binding working. Then is the large question if its possible doing the same with EZSP with one coordinator with TC. 
Senario: ZHA_customcomponent Touchlink paring. Adding the group ID and clicking execute. Touchlink joining the device is the network and have getting the network config and the groupe then continuing doing the "normal" pairing so the reporting and bindings its being OK.

MattWestb, 2020-11-08:
## deCONZ / HA is working

If adding groupe 0xff09 to deCONZ (old web app) and then adding scenes to the group for one IKEA GU10 WS its possible controlling them thru the remote and HA.

[image]
(the TRADFRI remote control scenes is the normal and cant being controlled from the remote)

The new scenes is absolute = its saved states of the bulbs status with all attribute.
The "factory" one is only saved with colour attributes so only changing color with the remote for those scenes. I have not finding how configuring enchanted scenes with only some of the attributes (normal scenes dont have transition time engaged have that).

MattWestb, 2020-11-09:
Have sending one long  e-mail  in swedish to the head of the technical office of tradfri products at IKEA with kindly request of implanting work around for  ZB3 without touchlink configuration off default group for the remote and the signal repeater that is not working as expected then being paired thru classical joining.
Hoping getting some creative response from him :-))

closed 2021-08-12 by github-actions[bot]