zigpy/zha-device-handlers #565: [Device Support Request] Linkind 1 Key Remote Control (ZBT-DIMSwitch-D0001)

MrYutz, 2020-11-05:
**Is your feature request related to a problem? Please describe.**
The device successfully joins the Zigbee network but no entities are created on HA.

They are $10 buttons with short, double, and long presses.

Please consider adding.  I am willing to give it a go, but I am new to Zigbee.  Python 7 out of 10.  Also willing to test.

**Deconz Config**
```
static const Sensor::ButtonMap linkind1keyMap[] = {
//    mode                          ep    cluster cmd   param button                                       name
    { Sensor::ModeScenes,           0x01, 0x0006, 0x01, 0,    S_BUTTON_1 + S_BUTTON_ACTION_SHORT_RELEASED, "On" },
    { Sensor::ModeScenes,           0x01, 0x0006, 0x00, 0,    S_BUTTON_1 + S_BUTTON_ACTION_DOUBLE_PRESS, "Off" },
    { Sensor::ModeScenes,           0x01, 0x0006, 0x00, 0,    S_BUTTON_2 + S_BUTTON_ACTION_DOUBLE_PRESS, "Off" },
    { Sensor::ModeScenes,           0x01, 0x0008, 0x01, 0,    S_BUTTON_1 + S_BUTTON_ACTION_HOLD, "Move up" },
    { Sensor::ModeScenes,           0x01, 0x0008, 0x03, 0,    S_BUTTON_1 + S_BUTTON_ACTION_LONG_RELEASED, "Stop" },
    { Sensor::ModeScenes,           0x01, 0x0008, 0x01, 1,    S_BUTTON_2 + S_BUTTON_ACTION_HOLD, "Move down" },
    { Sensor::ModeScenes,           0x01, 0x0008, 0x03, 0,    S_BUTTON_2 + S_BUTTON_ACTION_LONG_RELEASED, "Stop" },
    // end
    { Sensor::ModeNone,             0x00, 0x0000, 0x00, 0,    0,                                           nullptr }
};
```

**Log File:**
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0xa4cb:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=153 command_id=Command.Report_Attributes>
[0xa4cb:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
[0xa4cb:1:0x0006] Attribute report received: on_off=0
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0x1237 (cc:cc:cc:ff:fe:ab:0c:aa) joined the network
[0x1237:zdo] ZDO request ZDOCmd.Device_annce: [0x1237, cc:cc:cc:ff:fe:ab:0c:aa, 128]
[0x1237] Requesting 'Node Descriptor'
Tries remaining: 2
[0x1237] Extending timeout for 0xdd request
[0x1237] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4456, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0x1237] Discovering endpoints
Tries remaining: 3
[0x1237] Extending timeout for 0xdf request
[0x1237] Discovered endpoints: [1]
[0x1237:1] Discovering endpoint information
Tries remaining: 3
[0x1237] Extending timeout for 0xe1 request
[0x1237:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=2080, device_version=1, input_clusters=[0, 1, 3, 2821, 4096, 64769], output_clusters=[3, 4, 6, 8, 25, 4096])
[0x1237] Extending timeout for 0xe3 request
[0x1237:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=227 command_id=Command.Read_Attributes_rsp>
[0x1237:1] Manufacturer: lk
[0x1237:1] Model: ZBT-DIMSwitch-D0001
Checking quirks for lk ZBT-DIMSwitch-D0001 (cc:cc:cc:ff:fe:ab:0c:aa)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
[... 51 lines cut ...]
[0x1237] Extending timeout for 0xfb request
[0x1237:1:0x0008]: initializing channel: from_cache: False
[0x1237:1:0x0006]: initializing channel: from_cache: False
[0x1237:1:0x0019]: initializing channel: from_cache: False
[0x1237:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=249 command_id=Command.Read_Attributes_rsp>
[0x1237:1:0x0001]: initializing channel: from_cache: False
[0x1237] Extending timeout for 0xfd request
[0x1237:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=251 command_id=Command.Read_Attributes_rsp>
[0x1237:1:0x0000]: initializing channel: from_cache: False
[0x1237:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=253 command_id=Command.Read_Attributes_rsp>
[0x1237:1:0x0001]: 'async_initialize' stage succeeded
[0x1237:1:0x1000]: 'async_initialize' stage succeeded
[0x1237:1:0x0000]: 'async_initialize' stage succeeded
[0x1237:1:0x0008]: 'async_initialize' stage succeeded
[0x1237:1:0x0006]: 'async_initialize' stage succeeded
[0x1237:1:0x0019]: 'async_initialize' stage succeeded
[0x1237](ZBT-DIMSwitch-D0001): power source: Battery or Unknown
[0x1237](ZBT-DIMSwitch-D0001): completed initialization
[0x1237:zdo] ZDO request ZDOCmd.Match_Desc_req: [0xFFFD, 260, [25], []]
[0x1237:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=2 command_id=1>
[0x1237:1:0x0019] ZCL request 0x0001: [0, 4456, 789, 571473926, None]
[0x1237:1:0x0019] OTA query_next_image handler for 'lk ZBT-DIMSwitch-D0001': field_control=0, manufacture_id=4456, image_type=789, current_file_version=571473926, hardware_version=None
[0x1237:1:0x0019] No OTA image is available

```

Adminiuga, 2020-11-06:
show log for what device is sending for each button press

MrYutz, 2020-11-06:
For the sequence:

1. Single press and release
2. Single press, hold and then release
3. Double press

**homeassistant.log:**

```
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xa4cb>, 1, 260, 6, b'\x08Q\n\x00\x00\x10\x00', 0, 175, 255, 157, 42, 255, 255, -46]
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy.zcl] [0xa4cb:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=81 command_id=Command.Report_Attributes>
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy.zcl] [0xa4cb:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy.zcl] [0xa4cb:1:0x0006] Attribute report received: on_off=0
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 81 under 245 request id, data: b'18510b0a00'
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xa4cb>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'08510a00001000'
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 245, 0, <DeconzAddressEndpoint address_mode=2 address=0xA4CB endpoint=1>, 260, 6, 1, b'\x18Q\x0b\n\x00', 2, 0)
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 245]
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 245: 00
2020-11-06 07:23:04 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xf5 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xa4cb endpoint=1>, status: 0x00
2020-11-06 07:23:47 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-11-06 07:23:47 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-11-06 07:23:47 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [26, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.GROUP address=0x400a>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x1237>, 1, 260, 6, b'\x01P\x00', 0, 175, 255, 160, 42, 255, 255, -52]
2020-11-06 07:23:47 DEBUG (MainThread) [zigpy.zcl] [0x1237:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=80 command_id=0>
2020-11-06 07:23:47 DEBUG (MainThread) [zigpy.zcl] [0x1237:1:0x0006] ZCL request 0x0000: []
2020-11-06 07:23:47 DEBUG (MainThread) [zigpy.zcl] [0x1237:1:0x0006] No handler for cluster command 0
2020-11-06 07:23:47 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=cc:cc:cc:ff:fe:ab:0c:aa, unique_id=cc:cc:cc:ff:fe:ab:0c:aa:1:0x0006, endpoint_id=1, cluster_id=6, command=off, args=[]>
2020-11-06 07:23:47 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x1237>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'015000'
2020-11-06 07:23:50 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-11-06 07:23:50 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-11-06 07:23:50 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.GROUP address=0x400a>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x1237>, 1, 260, 8, b'\x01Q\x01\x00\x7f\x00\x00', 0, 175, 255, 162, 42, 255, 255, -39]
2020-11-06 07:23:50 DEBUG (MainThread) [zigpy.zcl] [0x1237:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=81 command_id=1>
2020-11-06 07:23:50 WARNING (MainThread) [zigpy.zcl] [0x1237:1:0x0008] Data remains after deserializing ZCL frame
2020-11-06 07:23:50 DEBUG (MainThread) [zigpy.zcl] [0x1237:1:0x0008] ZCL request 0x0001: [0, 127]
2020-11-06 07:23:50 DEBUG (MainThread) [zigpy.zcl] [0x1237:1:0x0008] No handler for cluster command 1
[... cut ...]
        "user_id": null
    }
}
Event 56 fired 7:26 AM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "cc:cc:cc:ff:fe:ab:0c:aa",
        "unique_id": "cc:cc:cc:ff:fe:ab:0c:aa:1:0x0006",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-11-06T13:26:37.588370+00:00",
    "context": {
        "id": "5d82f308df1f2ab96e3da16ea2bb60d5",
        "parent_id": null,
        "user_id": null
    }
}
```

dmulcahey, 2020-11-08:
So, this is a remote and remotes do not have entities. It seems you are already getting the events. The only thing a quirk would give you here is battery support and device automation trigger support in HA.

brushj, 2020-11-28:
@MrYutz Curious how you paired your device? It may just be that you are using deconz and I am using ZHA, I am getting the following:

```[0x9b27:1] Manufacturer: lk
[0x9b27:1] Model: ZBT-DIMSwitch-D0001
Checking quirks for lk ZBT-DIMSwitch-D0001 
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>

MrYutz, 2020-11-29:
@brushj I went back and tried it a couple of ways to give you a definitive answer.

It only worked on my dev box (Pi4 8GB with HassOS 4.16).  It did not work on my production Ubuntu Server setup (Pi 4 4GB).

I took the opportunity to upgrade my production box to a supported platform (HassOS 4.17 with Supervisor 2020.11.5) and it worked right out of the box.

Just held the reset button for 10 seconds and it joined via the closest JASCO dimmer.

Go0oSer, 2020-12-10:
Using ZHA the device pairs, but no zha_events appear when the button is pressed.

Adminiuga, 2020-12-10:
Yeah, device does send the commands over the group addressing. ConBee would accept messages for any group, EZSP needs to be subscribed to specific groups. 
If you want to experiment, copy `homeassistant/components/zha` folder from the current dev tree to your `/config_directory/custom_components/zha`, restart HA, remove the device and re-pair it. It should add EZSP to the device group and you should be receiving events too

HutchMD, 2020-12-11:
I don't know if I'm doing this correctly, but I downloaded the home assistant core dev branch, navigated to this folder
> X:\Users\User\Downloads\core-dev.zip\core-dev\homeassistant\components\zha

and used sambashare to try to copy it to

> \\homeassistant\config\custom_components\
which then became

> \\homeassistant\config\custom_components\zha

I restarted HA, paired the button-device, and I get 2 'devices':
One is the battery, the other is the button I think? 
When I try to save the button, I get 

> Error saving area: required key not provided @data['device_id']. Got None

Go0oSer, 2020-12-11:
> Yeah, device does send the commands over the group addressing. ConBee would accept messages for any group, EZSP needs to be subscribed to specific groups.
> If you want to experiment, copy [...]

That worked thank you.  Any idea when this will make it to master?

closed 2021-08-12 by github-actions[bot]

k37chup, 2021-09-21:
I know this is an old post but can i resurrect it?