zigpy/zha-device-handlers #572: [Device Support Request] CentraLite 3323-G

ryantiger658, 2020-11-08:
**Is your feature request related to a problem? Please describe.**
I purchased a new [CentraLite multisensor from Amazon ](https://www.amazon.com/gp/product/B071YQCWQM/)that supposedly has a humidity sensor, however, the quark that Home Assistant uses doesn't seem to pull through the humidity data. 

**Describe the solution you'd like**
It would be great if someone could add the Humidity cluster to the CentraLite quark, or to help me understand how to work with the device signatures so I can put in my own PR for it.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4174, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 49887,
      "device_type": "0x000c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "CentraLite",
  "model": "3323-G",
  "class": "zhaquirks.centralite.3300S.CentraLite3300S"
}`

**Additional context**
Add any other context or screenshots about the feature request here.
Thanks in advance for your help!

Adminiuga, 2020-11-08:
Is this a door sensor? Does not have humidity, only temp.

dmulcahey, 2020-11-09:
are you referring to this:

[image]

If so, that isn't capabilities for the device. That is tolerance for shipping / storage of the device.

ryantiger658, 2020-11-09:
@Adminiuga, Ya the door sensor, I snagged the outlet too for another purpose though.

Ultimately, looking through the documentation I don't see anything from the manufacturer, but I was looking at some of the comments. 

[image]

[image]

[image]

Perhaps since they are from 2018 the sensor has changed. I have been looking for a reliable (and cheep) humidity sensor for a while now and have struggled to find one. 

Thanks for looking into this for me!

Adminiuga, 2020-11-09:
Konke temp & humidity sensor.

ryantiger658, 2020-11-09:
I have a couple [Aqara climate sensors](https://www.amazon.com/Aqara-WSDCGQ11LM-Temperature-Humidity-Sensor/dp/B07D37FKGY/ref=sr_1_12?dchild=1&keywords=Konke+humidity+zigbee&qid=1604937798&sr=8-12), but I have a problem where they loose connection and stop functioning about 24 hours after I set them up. There are zigbee repeaters in the same room that work fine so there shouldn't be range issues.

Adminiuga, 2020-11-09:
and that's why the recommendation was for Konke and not Xiaomi.

Adminiuga, 2020-11-09:
Won't fix. Working as designed.

Adminiuga, 2020-11-09:
Check pinned messages of #Zigbee channel on HA discord server to see the answer to
>so there shouldn't be range issues.

closed 2020-11-09 by Adminiuga

ryantiger658, 2020-11-09:
Thanks again!