zigpy/zha-device-handlers #590: [Device Support Request] Aqara B1 Curtains

k3lvar, 2020-11-17:
I'm trying to add support for the Aqara B1 Curtains motor but have encountered some issues.

1) i have added a handle_cluster_request but it is not getting called when i use the open or close in home assistant. What do i need to add to catch the cluster command events?

2) How do i initial a read on a cluster attribute? I tried to execute read_attributes_raw but looks like it is not returning.

signature = {
        MODELS_INFO: [(LUMI, "lumi.curtain.hagl04")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=514
            # device_version=1
            # input_clusters=[0, 1, 3, 13, 19, 258]
            # output_clusters=[3, 10]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    AnalogOutput.cluster_id,
                    Basic.cluster_id,
                    Identify.cluster_id,
                    MultistateOutput.cluster_id,
                    PowerConfiguration.cluster_id,
                    WindowCovering.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Time.cluster_id, ],
            },
        },
    }
}
    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    AnalogOutputCluster,
                    Basic.cluster_id,
                    Identify.cluster_id,
                    MultistateOutput.cluster_id,
                    PowerConfiguration.cluster_id,
                    WindowCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Time.cluster_id, ],
            },
        },
    }

closed 2020-11-24 by k3lvar

disrupted, 2020-12-18:
@k3lvar hey, I was just having a look whether this device is supported and came across your issue. Since you closed it I hope you don't mind me asking is this curtain motor now supported?

k3lvar, 2020-12-19:
@disrupted No, it still isn't supported but i closed it as there was no response and i figured out the commands to open and close the curtains.

mikemike404, 2021-02-03:
@k3lvar Could you share the commands to open and close the curtains? Thanks for advance!

lukasz-burak, 2023-11-20:
@k3lvar can you share your custom quirk to make this motor working in ZHA?
I've tried for few days without luck.