zigpy/zha-device-handlers #592: [Device Debug Request] Xiaomi Aqara D1 Remote quirk stopped working

ronaldtveen, 2020-11-21:
When I connect the Xiaomi Aqara D1 Remote (wireless switch, double rocker) to ZHA the pairing goes without issue. But I'm left with only a battery entity: `sensor.lumi_lumi_remote_b286acn02_e8[..]04_power`

This is unexpected looking through the code, since the `b286acn02` model is actually included in the quirk ([remote_b286acn01.py#L97](https://github.com/zigpy/zha-device-handlers/blob/1980fdf840fccfec5800671686c45590b42cea80/zhaquirks/xiaomi/aqara/remote_b286acn01.py#L97))

Reading around on the community forms [it is suggested](https://community.home-assistant.io/t/xiaomi-aqara-wall-switch-d1/183802/10) that the D1 is fully functional. The code suggests that too, including short/long/double presses of the rockers.

Anyone any idea why it only shows the battery entity for me and I am not able to use the remote? Any pointers on how to debug are welcome.

Adminiuga, 2020-11-22:
Remotes don't have entities.

ronaldtveen, 2020-11-22:
That makes sense... "Go easy, I've never done this before".
Just added an automation in the UI and just turned on a light via this remote. So all is good.

Probably not the time and place to ask for more and I understand if you don't want to answer but a little nudge in the right direction would be much appreciated;
I do my automations by hand, not via the UI. Normally I've always used entity_id's as triggers, not devices. This remote has a long undescriptive `device_id`. This never changes so I can use those in my own yaml code for automations? Or is this only meant to be done through the Home Assistant UI?

Adminiuga, 2020-11-22:
Remotes would generate `zha_event`. You need to use `event_trigger` in automation https://www.home-assistant.io/docs/automation/trigger/#event-trigger How much of the event data to match in the trigger is up to you. 
To see what "events" are generated, go to Developer -> Events -> type in "zha_event" and start listening and then press the buttons.

e.g.
```yaml
  trigger:
    - platform: event
      event_type: zha_event
      event_data:
        unique_id: "cc:cc:cc:ff:ff:12:34:45:1:0x0008"
        command: "move"
```

Adminiuga, 2020-11-22:
`unique_id` of the zha_event matches uniquely to a device + endpoint_id + cluster_id, but again it is up to you how much and how precisely you want to match events from a specific device.

ronaldtveen, 2020-11-23:
Thanks a lot man, that clears things up quite a bit.

> To see what "events" are generated, go to Developer -> Events -> type in "zha_event" and start listening and then press the buttons.

Especially this, now I don’t have to create a faux automation in the GUI first every time.
Thanks again for the (extra) help.

closed 2020-11-23 by ronaldtveen