zigpy/zha-device-handlers #593: [BUG] Xiaomi switch/button recognized only as "battery indicator" after pairing (Zigate)

csacre, 2020-11-21:
**Describe the bug**
I am moving from the plugin of Doudz for Zigate to ZHA as recomanded.
The pairing of the Xiaomi Switch is difficult but I can do it (close to Zigate key, pressing several time buttion etc...)
Only the "Battery" entity is recognized, not ideal for a button.
I am using this button for 1 year via  @doudz plugin for Zigate and the device worked perfectly,supporting click, multiple-click. etc...

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Configuration/ZHA
2. Click on '+ button to start a paring session' suing method described in [https://github.com/home-assistant/core/issues/37947](url)
3. Only a "battery entity is created"

**Expected behavior**
Have click/multiple-click entities

**Screenshots**
[image]

What I did to have logs:
- Remove the device/entity with battery only
- Activate log 
>   logs:
    homeassistant.components.shell_command: debug
    homeassistant.components.zha: debug
    zigpy_zigate: debug
    zhaquirks: debug

- Rebooted HA 
- Pairing again (very close to the Zigate)
- After device with only the battery appeared, I did press once, then double click then triple click several times

Zigbee Info in ZHA screen after pairing

> IEEE: 00:15:8d:00:03:60:fd:a3
Nwk: 0x0289
Device Type: EndDevice
LQI: 180
RSSI: Unknown
Last Seen: 2020-11-21T17:37:26
Power Source: Battery or Unknown
Quirk: zhaquirks.xiaomi.mija.sensor_switch.MijaButton

Log coming from the ZHA screen during & after pairing:
> Device 0x0000 (00:15:8d:00:03:60:fd:a3) left the network
Device 0x1209 (00:15:8d:00:03:60:fd:a3) joined the network
[0x1209] Requesting 'Node Descriptor'
Tries remaining: 2
[0x1209] Extending timeout for 0x0c request
[0x1209:zdo] ZDO request ZDOCmd.Device_annce: [0x1209, 00:15:8d:00:03:60:fd:a3, 128]
Ignoring message (b'18140a00001000') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18140a00001000') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18150a050042126c756d692e73656e736f725f737769746368') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18150a050042126c756d692e73656e736f725f737769746368') on cluster 0: unknown endpoint or cluster id: 1
[0x1209] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=0)
[0x1209] Discovering endpoints
Tries remaining: 3
[0x1209] Extending timeout for 0x0d request
Ignoring message (b'18160a00001000') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18160a00001000') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18170a00001000') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18170a00001000') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18180a00001001') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18180a00001001') on cluster 6: unknown endpoint or cluster id: 1
Device 0x0000 (00:15:8d:00:03:60:fd:a3) left the network
Device 0x0289 (00:15:8d:00:03:60:fd:a3) joined the network
Device 00:15:8d:00:03:60:fd:a3 changed id (0x1209 => 0x0289)
Canceling old initialize call
[0x0289] Requesting 'Node Descriptor'
Tries remaining: 2
[0x0289] Extending timeout for 0x0e request
[... 1246 lines cut ...]
2020-11-21 17:39:00 DEBUG (MainThread) [homeassistant.components.zha.entity] light.0017880103324e32_level_light_color_on_off: polling current state - from cache: False
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x52A7, 260, 6, 1, 11, 18, b'\x00\x12\x00\x00\x00', True, False)
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'0252a7010b000601040000050012000000'
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x11\xcd\x02R\xa7\x01\x0b\x00\x06\x01\x04\x00\x00\x05\x00\x12\x00\x00\x00'
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x11\xcd\x02\x12R\xa7\x02\x11\x02\x1b\x02\x10\x02\x16\x02\x11\x02\x14\x02\x10\x02\x10\x02\x15\x02\x10\x12\x02\x10\x02\x10\x02\x10'
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000007f300410530000000
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'004105300000' LQI:0
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 65, 1328, b'\x00\x00']
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800200162800010400060b010252a702000018120100000010015a
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00010400060b010252a70200001812010000001001' LQI:90
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 6, 11, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x52a7>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18\x12\x01\x00\x00\x00\x10\x01']
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 800200162800010400060b010252a702000018120100000010015a
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00010400060b010252a70200001812010000001001' LQI:90
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 6, 11, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x52a7>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18\x12\x01\x00\x00\x00\x10\x01']
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0x52A7, 260, 8, 1, 11, 19, b'\x00\x13\x00\x00\x00', True, False)
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'0252a7010b000801040000050013000000'
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x11\xc2\x02R\xa7\x01\x0b\x00\x08\x01\x04\x00\x00\x05\x00\x13\x00\x00\x00'
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x11\xc2\x02\x12R\xa7\x02\x11\x02\x1b\x02\x10\x02\x18\x02\x11\x02\x14\x02\x10\x02\x10\x02\x15\x02\x10\x13\x02\x10\x02\x10\x02\x10'
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000007f000420530000000
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'004205300000' LQI:0
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 66, 1328, b'\x00\x00']
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80020016e800010400080b010252a702000018130100000020fe5a
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'00010400080b010252a702000018130100000020fe' LQI:90
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 8, 11, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x52a7>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18\x13\x01\x00\x00\x00 \xfe']
2020-11-21 17:39:00 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80020016e800010400080b010252a702000018130100000020

csacre, 2020-11-21:
Here is the corresponding signature I found in the previous zigate file of Doudz

>         {
            "addr": "664e",
            "discovery": "auto-discovered",
            "endpoints": [
                {
                    "clusters": [
                        {
                            "attributes": [
                                {
                                    "attribute": 5,
                                    "data": "lumi.sensor_switch",
                                    "name": "type",
                                    "type": "str",
                                    "value": "lumi.sensor_switch"
                                },
                                {
                                    "attribute": 1,
                                    "data": 10,
                                    "name": "application_version",
                                    "value": 10
                                },
                                {
                                    "attribute": 65282,
                                    "data": "100121c40b21a813240a00000104217c022060",
                                    "name": "battery_voltage",
                                    "type": "float",
                                    "unit": "V",
                                    "value": 3.012
                                }
                            ],
                            "cluster": 0
                        },
                        {
                            "attributes": [
                                {
                                    "attribute": 0,
                                    "data": true,
                                    "name": "onoff",
                                    "type": "bool",
                                    "value": true
                                },
                                {
                                    "attribute": 32768,
                                    "data": 0,
                                    "expire": 2,
                                    "name": "multiclick",
                                    "type": "int",
                                    "value": 0
                                }
                            ],
                            "cluster": 6
                        }
                    ],
                    "device": 0,
                    "endpoint": 1,
                    "in_clusters": [],
                    "out_clusters": [],
                    "profile": 0
                }
            ],
            "generictype": "",
            "info": {
                "addr": "664e",
                "ieee": "00158d0004081a14",
                "last_seen": "2020-11-20 17:22:19",
                "lqi": 66,
                "mac_capability": "10000000"
            },
[... 117 lines cut ...]
                                    "name": "multiclick",
                                    "type": "int",
                                    "value": 0
                                }
                            ],
                            "cluster": 6
                        }
                    ],
                    "device": 0,
                    "endpoint": 1,
                    "in_clusters": [],
                    "out_clusters": [],
                    "profile": 0
                }
            ],
            "generictype": "",
            "info": {
                "addr": "a679",
                "ieee": "00158d000360fda3",
                "last_seen": "2020-11-20 17:29:19",
                "lqi": 72,
                "mac_capability": "10000000"
            },
            "name": ""
        }

GoSpursGoNL, 2020-11-21:
I'm not sure if the above behaviour is intended. However, I did notice that you can create automations in HA. You can select the device (although indeed only a battery sensor added as entity) and then as trigger you can select the single click, double, triple and quadruple click.
So you can actually base automations on it.
I did notice though that the 'triple click' trigger seems broken, it gets never triggered.

But the triple click is correctly detected. I use node-red for my automations. There you can listen to the zha-event events. The number of clicks is under msg.payload.event.args.value and the device (no human readably name though) is under payload.event.device_ieee

csacre, 2020-11-21:
I tried the basic click and it worked!
You made my day!

It does not solve what is an issue but provide a viable work-around!

Thank you!

csacre, 2020-11-21:
The work-around works from single click up to quadruple click.  😁😁😁

Adminiuga, 2020-11-22:
Remotes in HA are stateless devices and don't have entities. You can either use zha_events in events trigger or use device automation triggers.

csacre, 2020-11-22:
@Adminiuga  I understand. I am using HA for more than 1 year now and this is the first time I have to use directly a "device" object.  For me an entity abstract the concept of device into a more generic one to build on more standardized object less dependant of any particular device. This is how I understand the definition of an entity (see [)](https://developers.home-assistant.io/docs/core/entity/)). 

Using the device works in this particular case. Should I consider this as an acceptable work-around or a standard approach ?

Adminiuga, 2020-11-22:
This is a standard approach mandated by ha architecture.

csacre, 2020-11-22:
Thank you for feedback, I am closing the issue then ;-)

closed 2020-11-22 by csacre