zigpy/zha-device-handlers #595: Support for Tuya based LoraTap 3 button remote

tube0013, 2020-11-22:
**Is your feature request related to a problem? Please describe.**
New Device Support for LoraTap 3 Button remote (Tuya based). - https://zigbee.blakadder.com/LoraTap_SS600ZB.html
(https://www.aliexpress.com/item/1005001298730723.html)

**Describe the solution you'd like**
Events/ Device Automations for button presses. Maybe block the creation of the switch entities in HA as it is a remote.

If there is an existing quirk that could be adapted let me know and I can take a stab at it, I looked around and didn't see anything that stood out to me.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0x000a"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_bi6lpsew",
  "model": "TS0043",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Add any other context or screenshots about the feature request here.

Screen shot of device in HA:

[image]

Log Data from button Presses.

Button 1 single press:

```
2020-11-22 11:41:54 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=25 command_id=253>
2020-11-22 11:41:54 WARNING (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] Unknown cluster-specific command 253
2020-11-22 11:41:54 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL request 0x00fd: b'\x00'
2020-11-22 11:41:54 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] No handler for cluster command 253
2020-11-22 11:41:55 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=25 command_id=253>
2020-11-22 11:41:55 WARNING (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] Unknown cluster-specific command 253
2020-11-22 11:41:55 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL request 0x00fd: b'\x00'
2020-11-22 11:41:55 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] No handler for cluster command 253
```

Button 1 double press:

```
2020-11-22 11:42:26 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=26 command_id=253>
2020-11-22 11:42:26 WARNING (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] Unknown cluster-specific command 253
2020-11-22 11:42:26 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL request 0x00fd: b'\x01'
2020-11-22 11:42:26 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] No handler for cluster command 253
2020-11-22 11:42:27 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=26 command_id=253>
2020-11-22 11:42:27 WARNING (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] Unknown cluster-specific command 253
2020-11-22 11:42:27 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL request 0x00fd: b'\x01'
2020-11-22 11:42:27 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] No handler for cluster command 253
```

Button 1 Hold action:
```
2020-11-22 11:42:50 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=27 command_id=253>
2020-11-22 11:42:50 WARNING (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] Unknown cluster-specific command 253
2020-11-22 11:42:50 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL request 0x00fd: b'\x02'
2020-11-22 11:42:50 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] No handler for cluster command 253
2020-11-22 11:42:51 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=27 command_id=253>
2020-11-22 11:42:51 WARNING (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] Unknown cluster-specific command 253
2020-11-22 11:42:51 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] ZCL request 0x00fd: b'\x02'
2020-11-22 11:42:51 DEBUG (MainThread) [zigpy.zcl] [0x86c6:1:0x0006] No handler for cluster command 253
```

Button 2/3 are the same on corresponding endpoints.

ie Button 2 single press:

```
2020-11-22 11:44:09 DEBUG (MainThread) [zigpy.zcl] [0x86c6:2:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=28 command_id=253>
2020-11-22 11:44:09 WARNING (MainThread) [zigpy.zcl] [0x86c6:2:0x0006] Unknown cluster-specific command 253
2020-11-22 11:44:09 DEBUG (MainThread) [zigpy.zcl] [0x86c6:2:0x0006] ZCL request 0x00fd: b'\x00'
2020-11-22 11:44:09 DEBUG (MainThread) [zigpy.zcl] [0x86c6:2:0x0006] No handler for cluster command 253
2020-11-22 11:44:10 DEBUG (MainThread) [zigpy.zcl] [0x86c6:2:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=28 command_id=253>
2020-11-22 11:44:10 WARNING (MainThread) [zigpy.zcl] [0x86c6:2:0x0006] Unknown cluster-specific command 253
2020-11-22 11:44:10 DEBUG (MainThread) [zigpy.zcl] [0x86c6:2:0x0006] ZCL request 0x00fd: b'\x00'
2020-11-22 11:44:10 DEBUG (MainThread) [zigpy.zcl] [0x86c6:2:0x0006] No handler for cluster command 253
```

tube0013, 2020-11-22:
Looks like this is similar to #589, #525, #505 and is exactly the same device as #489 but in a different enclosure.

tube0013, 2020-12-03:
Doesn't this still require a quirk, as the pr just added the custom class for the tuya remotes, it still needs to be implemented in an actual quirk. I was going to look at trying to do that soon. Or am I missing something?

Adminiuga, 2020-12-03:
Doh, yea it does. 🤦

tube0013, 2020-12-03:
I did my best to knock this out.. working for me... 🎉

commit cc0d3c91 (2020-12-03): Add quirks for ts0041, ts0043 and ts0044 (Tuya 1, 3 and 4 button Remotes) (#611)

* Add ts0043 (Tuya 3 Button Remote)

Add Quirk for Tuya 3 button remote
Make TuyaSmartRemote Class a bit more generic

solves #595

* Fix test issues

* fix isort

* Add quirk varients for 1 & 4 button remotes

Add quirks for 1 (ts0041) and 4 (ts0044) button tuya remotes.

so this shoudl fix:
#595
#589
#505
#489

* Fix Missing import and add MODELS_INFO

* black

* isort

getting there.
closed 2020-12-03 by dmulcahey