zigpy/zha-device-handlers #596: [Device Support Request] Livolo Switch

Kalypox, 2020-11-22:
**Describe the solution you'd like**
I see Livolo switch are not implemented, somebody could do it :) ?
I have a lot of informations about it !

This switch works only on channel 26, but i found a tips ! 
With a software (TestGUI_ZiGate_NXP-4 for me because i use ZiGate) i change the channel to 26, then i pair the device en ZHA and finally i back to channel 11.
I found this tips here:
https://zigate.fr/documentation/changer-le-canal-zigbee-pour-les-appareils-livolo/

There is also informations for DomoticZ here:
https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Livolo-corner.md

Informations about switch:
Zigbee info:
IEEE: 00:12:4b:00:21:74:d7:08
Nwk: 0x6cc0
Device Type: EndDevice
LQI: Inconnu
RSSI: Inconnu
Dernière vue: 2020-11-22T22:00:31
Source d'énergie: Battery or Unknown

Informations:
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=0)",
  "endpoints": {
    "6": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0006"
      ]
    }
  },
  "manufacturer": "LIVOLO",
  "model": "TI0001          ",
  "class": "zigpy.device.Device"
}
```

I think the problem is cluster 0x0006, because i see on the domoticz github that Livolo switch uses 0x0008 !
[image]

If you have question don't hesitate !

Thanks !

Kalypox, 2020-11-28:
Hi,

I tried to write a quirk:
[image]
But when i pair my switch, it doesn't use my quirk, dou you know why ?
```
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LIVOLO TI0001           (00:12:4b:00:21:74:d7:08)
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {3} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {6}
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
2020-11-28 01:11:02 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {6}
```

I read during 2/3 hours the doc about zigbee cluster, and i find "DeviceType.DIMMER_SWITCH" and  "LevelControl.cluster_id", I think i'm on the right path

Kalypox, 2020-12-01:
Well, I give up, it's not my level I don't understand anything ^^

I hope someone better than me can help me !

Thank you !

nofateg, 2021-02-05:
channels: [26]
extended_pan_id: "21:75:8D:19:00:4B:12:00"

livolo.py:
```
import copy
import time
import binascii
import asyncio

import zigpy.types as t
import zigpy.zdo.types as zdo_t

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Identify, OnOff, LevelControl
from zigpy.zcl import foundation

from typing import Optional, Union

from zhaquirks.const import (
    CLUSTER_ID,
    COMMAND,
    COMMAND_OFF,
    COMMAND_ON,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
    TURN_OFF,
    TURN_ON,
)

from . import LIVOLO

LIVOLO_CLUSTER_ID = 0x0001
APS_REPLY_TIMEOUT = 5

LEVELCONTROL_COMMAND_MOVE_TO_LEVEL_WITH_ON_OFF = 0x0004
ON_OFF_COMMAND_TOGGLE = 0x0002
LEVELCONTROL_ATTRIBUTE_CURRENT_LEVEL = 0x0000
LEVELCONTROL_ATTRIBUTE_ON_OFF_TRANSITION_TIME = 0x0010

class LivoloSwitch (CustomDevice) :
    """TI0001 device."""

    def __init__(self, application, ieee, nwk, replaces):
        super().__init__(application, ieee, nwk, replaces)
        self.debug("LivoloSwitch#init: ieee: %s, nwk: %s, replaces: %s, application: %s", 
            ieee, nwk, replaces, application)

    def make_attribute(self, attrid, value):
        attr = foundation.Attribute()
        attr.attrid = attrid
        attr.value = foundation.TypeValue()
        attr.value.value = value
        return attr

    def to_status(self, val):
        if val is None:
            return "Unknown"
        if val > 0:
            return "On"
        return "Off"

    def handle_message(
[... 164 lines cut ...]
                    OnOff,
                    LivoloOnOff,
                ],
                OUTPUT_CLUSTERS: [
                    LevelControl,
                    OnOff
                ],
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic,
                    Identify,
                    OnOff,
                    LivoloOnOff,
                ],
                OUTPUT_CLUSTERS: [
                    LevelControl
                ],
            }
        },
    }

```

MiguelAngelLV, 2021-08-06:
Livolo devices not yet.

ZHA find the device but not get any entity

MiguelAngelLV, 2021-08-08:
@nofateg How I can use you script? I use Home Assistant OS.

Any way to install "easily" and permanente between updates?

nofategm, 2021-10-08:
> @nofateg How I can use you script? I use Home Assistant OS.
> 
> Any way to install "easily" and permanente between updates?

docker container:

nofategm, 2021-10-08:
> @nofateg How I can use you script? I use Home Assistant OS.
> 
> Any way to install "easily" and permanente between updates?

1. easy:

nofategm, 2021-10-08:
> @nofateg How I can use you script? I use Home Assistant OS.
> 
> Any way to install "easily" and permanente between updates?
1: "easy" no permanent:
bash script:
```
container_id=$(docker ps | grep raspberrypi4-homeassistant | awk '{print $1}')
echo "container_id = ${container_id}"
docker cp livolo ${container_id}:/usr/local/lib/python3.9/site-packages/zhaquirks

```

2. automation:
```
  automation:
  - id: permit_join
    alias: permit_join

    trigger:
      platform: time_pattern
      minutes: '/1'
    action:
      - service: zha.permit
        data:
          duration: 60

```

MattWestb, 2021-10-08:
Can some one that **can** coding explaining what this automation is making with the Zigbee network ??
```
 automation:
  automation:
  - id: permit_join
    alias: permit_join

    trigger:
      platform: time_pattern
      minutes: '/1'
    action:
      - service: zha.permit
        data:
          duration: 60
```
And how long ??

MattWestb, 2021-10-08:
Pleas @nofateg can you explaining for us "normal" user what your automation is doing with the Zigbee network or do you not knowing that ?

MattWestb, 2021-10-08:
@MiguelAngelLV if you is configuring local quirks in ZHA https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and making the quirk file in the folder is being stored out of the HA container and is not deleted then you is updating HA .

I cant saying if the quirk is working OK or not but is looks like doing somthing with the data sent from the device and is harmless is not working so worse trying it.

### Then the automation you **SHALL NOT INSTALLING !!!**

Its sending permit joining every minute minute for 60 seconds = is holding your Zigbee network open 24/7 for joining so its very dangerous like leaving your home door key in the lock on the outside on the street and all can going in your home 24/7.

And then the person is not replaying i thinking that hi is very well knowing what is doing and dont like explaining it for some reason that i dont knowing  !!

MattWestb, 2021-10-08:
@Adminiuga can you taking one look on the automation (point 2) in https://github.com/zigpy/zha-device-handlers/issues/596#issuecomment-939064721 i think it shall being deleted !!!

Sorry for addressing you directly but i think its very bad if user is using the automation and not knowing what is doing with there system.

Thanks in advance.

MattWestb, 2021-10-11:
Still "open for sync" (= opening the network for joining devices) is still have your network open for all who like joining it and can getting all your network information inclusive you private network key.

If you have the device i recommending contacting the manufacture / seller and demanding that they is fixing the problem in the device firmware or returning them to the seller then its not behaving as Zigbee standard and cant staying in one "normal"zigbee network (with normal zigbee 3 security).

Its up to you but i wold putting them in the trash can or on the shelf until i have getting them working OK with new firmware.

MiguelAngelLV, 2021-12-27:
When I toggle the livolo switch from the device, ZHA don't show the change. Only work from ZHA to Livolo, not vicebersa.

MiguelAngelLV, 2022-02-03:
@nofategm it works bidirectional for you?

nofategm, 2022-02-03:
HA configuration:
1. configuration.yaml:
zha:
  custom_quirks_path: "/config/zha"

```
pi@raspberrypi:/usr/share/hassio/homeassistant/zha/livolo $ ls -la
total 28
drwxr-xr-x 3 pi   pi   4096 Jan 21 14:08 .
drwxr-xr-x 3 root root 4096 Jan 21 15:38 ..
-rw-r--r-- 1 pi   pi     53 Feb 10  2021 __init__.py
-rw-r--r-- 1 pi   pi   9191 Feb 10  2021 livolo.py
drwxr-xr-x 2 root root 4096 Jan 21 14:08 __pycache__
```

2. My Zigbee adapter: Sonoff ZHABridge ( tasmota 10.1.0 )
3. ZHA works more stable than z2m
4. I have 8 switches. 2 switches always need to be manual rejoin after a power failure.
5. @MiguelAngelLV:

quikote, 2022-02-05:
It doesn't work for me.
I have a rpi4 running hassio and ZHA with several sensors already configured (mostly AQARA)

I have included this in my configuration.yaml:

zha:
  custom_quirks_path: "/config/zha"
  zigpy_config:
    network:
      channel: 26  
      channels: [26]
      extended_pan_id: "21:75:8D:19:00:4B:12:00"

The I have created the zha/livolo path with the following files inside with your scripts:

livolo.py
init.py

After restart I tried to find my 1-gang livolo switches but no luck.

Some tips from your side????

nofategm, 2022-02-05:
oops, init.py must be renamed to __init__.py

quikote, 2022-02-05:
> oops, init.py must be renamed to __init__.py

Renamed, but same result :-(

quikote, 2022-02-05:
Hi again... now it works :-) but the trick was erase de NVRAM of my stick so it gets reseted at all. Then I plug it again in the raspberry and let it to form the network again from the scracth with the right configuration. I hope that the others sensors I have around make the channel migration by themselves.

Anyway, I have another problem now and I would like to check if your behaivour is the same. I can switch on and off the livolo from the HA dashboard, but if I switch on or off the livolo manually it doesn't send the new state to HA. So, it seems that the comunication is HA to livolo but not backwards. It happend the same in your system?

Thanks a lot in advance for your help  ;-)

nofategm, 2022-02-07:
please add a full log

MiguelAngelLV, 2022-02-07:
> Anyway, I have another problem now and I would like to check if your behaivour is the same. I can switch on and off the livolo from the HA dashboard, but if I switch on or off the livolo manually it [...]

Some here.

quikote, 2022-02-26:
> please add a full log

Sorry for my late answer...
How can I get the full log you need?

quikote, 2022-02-26:
@MiguelAngelLV Hola Miguel Ángel. ¿Has conseguido ver el estado de los interruptores en home assistant?. No sé cómo hacer para que al accionarlos manualmente envíen el nuevo estado. 
Un saludo

MiguelAngelLV, 2022-03-15:
@quikote No :(

Solo me funciona de HA a Livolo, no de Livolo a HA.

He encargado un Sonoff USB Doggle a ver si con ese funciona, pero seguramente me tardará en llegar.

quikote, 2022-03-15:
> @quikote No :(
> 
> Solo me funciona de HA a Livolo, no de Livolo a HA.
> 
> He encargado un Sonoff USB Doggle a ver si con ese funciona, pero seguramente me tardará en llegar.

Pues no creo que te funcione. Yo también pillé uno y nada. Lo dejado por imposible, me he cambiado a zigbee2mqtt y mano de santo. Siguiendo las instrucciones todos los livolo pareados a la primera y funcionando perfectamente con un coordinador (sonoff usb dongle plus) y dos routers (un zzh y otro sonoff usb dongle). El resto de mis dispositivos zigbee que son aqara, hue y Ikea pareados sin cambiar el canal ni nada y funcionando. 
Me gustaba ZHA pero z2m le da mil vueltas y lo del mqtt si no lo usas para nada más no molesta.

MiguelAngelLV, 2022-03-26:
I've been doing tests and got nothing.

It seems that there is some way to request the status of the device using the clusters, but I don't know very well how. In Z2M they have a pool every few seconds checking the status.

MiguelAngelLV, 2022-03-27:
I have two issues:

### Sockets are recognized just like switches, but do nothing when their state changes. It seems they need a different order.

This is the code of livolo socket in Z2M

```js
 key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            if (typeof value !== 'string') {
                return;
            }

            const state = value.toLowerCase();
            let oldstate = 1;
            if (state === 'on') {
                oldstate = 108;
            }
            let channel = 1.0;
            const postfix = meta.endpoint_name || 'left';
            await entity.command('genOnOff', 'toggle', {}, {transactionSequenceNumber: 0});
            const payloadOn = {0x0001: {value: Buffer.from([1, 0, 0, 0, 0, 0, 0, 0]), type: 1}};
            const payloadOff = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 1}};
            const payloadOnRight = {0x0001: {value: Buffer.from([2, 0, 0, 0, 0, 0, 0, 0]), type: 2}};
            const payloadOffRight = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 2}};
            const payloadOnBottomLeft = {0x0001: {value: Buffer.from([4, 0, 0, 0, 0, 0, 0, 0]), type: 4}};
            const payloadOffBottomLeft = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 4}};
            const payloadOnBottomRight = {0x0001: {value: Buffer.from([8, 0, 0, 0, 0, 0, 0, 0]), type: 136}};
            const payloadOffBottomRight = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 136}};
            if (postfix === 'left') {
                await entity.command('genLevelCtrl', 'moveToLevelWithOnOff', {level: oldstate, transtime: channel});
                await entity.write('genPowerCfg', (state === 'on') ? payloadOn : payloadOff,
                    {
                        manufacturerCode: 0x1ad2, disableDefaultResponse: true, disableResponse: true,
                        reservedBits: 3, direction: 1, transactionSequenceNumber: 0xe9,
                    });
                return {state: {state_left: value.toUpperCase()}, readAfterWriteTime: 250};
            } else if (postfix === 'right') {
                channel = 2.0;
                await entity.command('genLevelCtrl', 'moveToLevelWithOnOff', {level: oldstate, transtime: channel});
                await entity.write('genPowerCfg', (state === 'on') ? payloadOnRight : payloadOffRight,
                    {
                        manufacturerCode: 0x1ad2, disableDefaultResponse: true, disableResponse: true,
                        reservedBits: 3, direction: 1, transactionSequenceNumber: 0xe9,
                    });
                return {state: {state_right: value.toUpperCase()}, readAfterWriteTime: 250};
            } else if (postfix === 'bottom_right') {
                await entity.write('genPowerCfg', (state === 'on') ? payloadOnBottomRight : payloadOffBottomRight,
                    {
                        manufacturerCode: 0x1ad2, disableDefaultResponse: true, disableResponse: true,
                        reservedBits: 3, direction: 1, transactionSequenceNumber: 0xe9,
                    });
                return {state: {state_bottom_right: value.toUpperCase()}, readAfterWriteTime: 250};
            } else if (postfix === 'bottom_left') {
                await entity.write('genPowerCfg', (state === 'on') ? payloadOnBottomLeft : payloadOffBottomLeft,
                    {
                        manufacturerCode: 0x1ad2, disableDefaultResponse: true, disableResponse: true,
                        reservedBits: 3, direction: 1, transactionSequenceNumber: 0xe9,
                    });
                return {state: {state_bottom_left: value.toUpperCase()}, readAfterWriteTime: 250};
            }
            return {state: {state: value.toUpperCase()}, readAfterWriteTime: 250};
        },
        convertGet: async (entity, key, meta) => {
            await entity.command('genOnOff', 'toggle', {}, {transactionSequenceNumber: 0});
        },

```

And this for Livolo Switch
[... 282 lines cut ...]
                    LightLink.cluster_id,
                    LevelControl.cluster_id

                ],
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                     Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    LightLink.cluster_id,
                    LevelControl.cluster_id
                ],
            }
        },
    }

```

MiguelAngelLV, 2022-04-07:
@nofategm with the last HA update livolo not works anymore.

I try with you script and my custom script, nothing.

I get this error:

[image]

TheJulianJES, 2022-04-17:
Can you see if there's anything in the (debug) logs when starting up Home Assistant or trying to control the device when using the custom quirk?
It's possible that it needs to be updated due to ZCLv7/v8 changes in zigpy. Some logs could help with this.

dmulcahey, 2022-04-17:
Can you attach the diagnostics data from the ZHA tile on the devices and services screen in HA please?

MiguelAngelLV, 2022-04-18:
Sorry, I need these devices so I change to Z2M again.

But, before change, I did some test and remember the problem is the variable schema, it's type is not iterable. The class name was ZCL or something.

The line that fail is:

data = hdr.serialize() + t.serialize([], schema)

When try serialize scheme.

MiguelAngelLV, 2022-04-18:
Um... I have another Pi and CC2531. Tonight I will make another HA installation and get loggers.

nofategm, 2022-05-13:
I'm sorry guys, I'm from Ukraine.
After the last changes, livolo doesn't work for me now.

nofategm, 2022-05-13:
so I'll start from the beginning

nofategm, 2022-05-17:
```
import copy
import time
import binascii
import asyncio

import zigpy.types as t
import zigpy.zdo.types as zdo_t

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Identify, OnOff, LevelControl
from zigpy.zcl import foundation

from typing import Optional, Union

from zhaquirks.const import (
    CLUSTER_ID,
    COMMAND,
    COMMAND_OFF,
    COMMAND_ON,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
    TURN_OFF,
    TURN_ON,
)

from . import LIVOLO

LIVOLO_CLUSTER_ID = 0x0001
APS_REPLY_TIMEOUT = 5

LEVELCONTROL_COMMAND_MOVE_TO_LEVEL_WITH_ON_OFF = 0x0004
ON_OFF_COMMAND_TOGGLE = 0x0002
LEVELCONTROL_ATTRIBUTE_CURRENT_LEVEL = 0x0000
LEVELCONTROL_ATTRIBUTE_ON_OFF_TRANSITION_TIME = 0x0010

class LivoloSwitch (CustomDevice) :
    """TI0001 device."""

    def __init__(self, application, ieee, nwk, replaces):
        super().__init__(application, ieee, nwk, replaces)
        self.debug("LivoloSwitch#init: ieee: %s, nwk: %s, replaces: %s, application: %s",
            ieee, nwk, replaces, application)

    def make_attribute(self, attrid, value):
        attr = foundation.Attribute()
        attr.attrid = attrid
        attr.value = foundation.TypeValue(t.uint8_t, value);
        return attr

    def to_status(self, val):
        if val is None:
            return "Unknown"
        if val > 0:
            return "On"
        return "Off"

    def handle_message(
        self,
        profile: int,
        cluster: int,
        src_ep: int,
        dst_ep: int,
[... 163 lines cut ...]
                    OnOff,
                    LivoloOnOff
                ],
                OUTPUT_CLUSTERS: [
                    LevelControl,
                    OnOff
                ],
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic,
                    Identify,
                    OnOff,
                    LivoloOnOff,
                ],
                OUTPUT_CLUSTERS: [
                    LevelControl
                ],
            }
        },
    }

```

ciscomax111, 2022-08-04:
Still unresolved? Got a livolo switch 3 buttons and be damed if i'll buy their hub. (ZHA on raspberry pi with sonoff plus dongle)

MiguelAngelLV, 2022-11-20:
Still not worked correctly. You can send data from HA to Livolo, but not read changes from device,

MiguelAngelLV, 2023-01-16:
Any advance? It the only requirement for migrate from z2m to zha.

ciscomax111, 2023-01-17:
raspi 3 with zha and z2mqtt (2 sonoff dongle plus) . z2mqtt configured only for livolo(channel 26).works good, only problem is with power loss on raspi(need re-pair with controller every time but it keeps the config).

MiguelAngelLV, 2023-01-17:
> raspi 3 with zha and z2mqtt (2 sonoff dongle plus) . z2mqtt configured only for livolo(channel 26).works good, only problem is with power loss on raspi(need re-pair with controller every time but it keeps the config).

I have the same configuration and it works, but I want use ZHA.

For reparing I have a automation to enable pair mode after HA boot and it works most times.

ciscomax111, 2023-01-17:
Got a replacement for livolo 3 gang......now in testing mode . Moes 3 gang , on zha , it looks almost like livolo but dont have the red light in stand by(thats a bummer) and no sound click when turned on/off. Sent me the link for automation to test it pls.

ciscomax111, 2023-02-27:
Found a new one from aliexpress AVATTO switch. Tested one gang switch TS0011. Looks similar with moes but have click and blue light in standby.  Adoption with zha works perfectly. It really looks like this is the one 🥇

MiguelAngelLV, 2023-02-28:
I have a another very similar livolo switch in aliexpress: bseed

MiguelAngelLV, 2023-06-26:
@nofateg bidirectional? livolo -> zha, zha -> livolo?

nofategm, 2023-06-26:
yes, and nothing is lost

nofategm, 2023-06-26:
zigbee controller - sonoff zbbridge ( tasmota )

closed 2023-12-30 by github-actions[bot]

nofategm, 2023-12-30:
https://github.com/Koenkk/zigbee2mqtt/issues/15299
this is a big problem with z2m

I purchased zb pro
it works on a different protocol and the previous code does not start reading the status. Livolo uses the battery channel to detect the status of the switch, so there is a conflict with ha.

nofategm, 2023-12-30:
I tried several alternatives for livolo... and all bad.

nofategm, 2023-12-30:
I'm hoping to find a way to disable protocol checking.