zigpy/zha-device-handlers #597: [BUG] Xiaomi Mija sensor switch reports a single click in addition to long press events

samueltardieu, 2020-11-24:
When long pressing a Mija sensor switch, it reports a single click before reporting the long press and the end of long press. This does not happen with the phoscon integration, where long press and end of long press are independent events.

Here is an event log for a long press, we can see three events instead of two (events are in reversed order), the single click event should not have been generated:

```
 Événement 2 déclenché 21:30:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:15:8d:00:04:18:59:aa",
        "unique_id": "00:15:8d:00:04:18:59:aa:1:0x0006",
        "device_id": "0b7324457bbc259b3e1085d9a072acd3",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "attribute_updated",
        "args": {
            "attribute_id": 0,
            "attribute_name": "on_off",
            "value": false
        }
    },
    "origin": "LOCAL",
    "time_fired": "2020-11-24T20:30:19.559430+00:00",
    "context": {
        "id": "728f8ec3ab9480a2abcd7d3d84b1c8f0",
        "parent_id": null,
        "user_id": null
    }
}

Événement 1 déclenché 21:30:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:15:8d:00:04:18:59:aa",
        "unique_id": "00:15:8d:00:04:18:59:aa:1:0x0006",
        "device_id": "0b7324457bbc259b3e1085d9a072acd3",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "attribute_updated",
        "args": {
            "attribute_id": 0,
            "attribute_name": "on_off",
            "value": true
        }
    },
    "origin": "LOCAL",
    "time_fired": "2020-11-24T20:30:17.352194+00:00",
    "context": {
        "id": "6ddc789fc0a179e6fd28f8f9f458b67d",
        "parent_id": null,
        "user_id": null
    }
}

Événement 0 déclenché 21:30:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:15:8d:00:04:18:59:aa",
        "unique_id": "00:15:8d:00:04:18:59:aa:1:0x0006",
        "device_id": "0b7324457bbc259b3e1085d9a072acd3",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "click",
        "args": {
            "click_type": "single"
        }
    },
    "origin": "LOCAL",
    "time_fired": "2020-11-24T20:30:17.349971+00:00",
    "context": {
        "id": "43d523129578c85dfa216fb8d25b2519",
        "parent_id": null,
        "user_id": null
    }
}
```

Adminiuga, 2020-11-24:
- You do need to fill in the template
- There're no long presses with this button -- but you can implement those using automations
- it supports only "short-press", double/triple/quadruple/quintuple presses
- the other two events are by design and totally expected

samueltardieu, 2020-11-25:
>     * You do need to fill in the template

Done
 
>     * There're no long presses with this button -- but you can implement those using automations

So what are those two events, which trigger when I start pressing the button and when I release it? If you don't like the term "long press", replace it with "press and hold".

>     * it supports only "short-press", double/triple/quadruple/quintuple presses

The Phoscon plugin supports those in addition to press and hold detection, and does not send a single click event first. The setup I had was (and that I'd like to reproduce with ZHA, same hardware):

- single click toggles the light
- press event starts changing the intensity up then down then up then down…
- until release event stops the automation

Right now, with ZHA, the extra single click makes it more complex to implement those kind of automation in HomeAssistant than with Phoscon.

>     * the other two events are by design and totally expected

My problem is not with the "other two events" which indeed are what I want to keep (start pressing, and release). It is with the extra "single click" event.

Adminiuga, 2020-11-25:
> So what are those two events, which trigger when I start pressing the button and when I release it? If you don't like the term "long press", replace it with "press and hold".

Those two events are from presses, will be present for long or for short presses.

closed 2021-08-12 by github-actions[bot]