zigpy/zha-device-handlers #599: [Device Pre Support Request] Lidl Home smart home (HAs Home Sweet Home) IKEA killer ?

MattWestb, 2020-11-25:
**Is your feature request related to a problem? Please describe.**

Its looks like after IKEA and Xiaomi have making our devs making bunches of quirks now the next large wave is coming !!!

PM: Lidl is one large german discount supermarket that have stores in most european countries.

**Describe the solution you'd like**

In Austria the release is [30/11](https://www.lidl.at/de/c/smart-home/c2889/w2) and germany [3/12](https://www.lidl.de/de/home-smart-home-ab-03-12/c29863)and more is following.
Some center europe countries looks have already getting them between the ginger cake, christmas ham and milk the last weeks.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

In one week new request is posted for the USB-control outlet liste then having gettin my hands on some of them !!

IKEA have lowering the price on ther GW with 1/4 for some month in most of europe.
I expecting motion sensor and control outlets is following very sooon.

Great that we have some devs that is implanting tuya protocol and can helping getting the devices supported in ZHA with not so much work.
Some devices have being added in deCONZ and Z2M and more requests is coming every day.

**Additional context**

This post is more informative and can being deleted of the devs without hard feelings !!

Jeltel, 2020-11-26:
I would be very interested in having these properly added to ZHA.
I hope to receive the christmas lights and the colour bulbs tomorrow. And I'll share the logs on both.

MattWestb, 2020-11-26:
ZHA have the ground for the tuya protocol in place. The second part is for TRV and thermostats is ready for revue and then coming in the dev branch. Then its much easier adding support for more tuya devices like the motion, window/door and the 3 way outlet socket with USB  and the RGBWW LED stripes. The more normal devices like bulbs and the outlet plug should working as normal zigbee 3 devices and working out the box.

I looking for OpenWRT firmware for the GW so can getting one nice EZSP with ethernet and done need using WiFI. Its looks some work is done for the [RTL8196E](https://forum.openwrt.org/t/openwrt-on-rtl8196e-totolink-ex200-comfast-cf-wr302s/52030/17) but is not verified if its working.

The christmas lights looks great but i dont have any three this year because its not possible buying one before christmas then its lockdown ;-(((

Jeltel, 2020-11-26:
Well I'm willing to do the work on the Christmas lights then.

Jeltel, 2020-11-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
The Christmas lights of Lidl can be coupled. But only with an on/off switch and a brightness. Switching the lights on and off work. Doing something with the brightness results in the lights going to a shade of white, eventhough a colour was set previously

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Full control of the Christmas lights, colour selector wheel and different effects avaiable:
[Name] [Options]
Steady (color)
Snow (color, speed)
Rainbow (speed)
Snake (color list up to 6, speed)
Twinkle (color list up to 2, speed)
Fireworks (color, speed)
Horizontal Flag (color list up to 3, speed)
Waves (color list up to 3, speed)
Updown (color list up to 2, speed
Vintage (color, speed)
Fading (color, speed)
Collide (color, speed)
Strobe (color list up to 5, speed)
Sparkles (color list up to 3, speed)
Carnival (color list up to 6, speed)
Glow (color list up to 6, speed)

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x67a9:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=166 command_id=Command.Read_Attributes_rsp>
[0x67a9:11:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=168 command_id=Command.Read_Attributes_rsp>
[0x67A9:11:0x0008]: received attribute: 0 update with value: 176
[0x67a9:11:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=170 command_id=Command.Read_Attributes_rsp>
Device 0x85cf (bc:33:ac:ff:fe:77:55:94) joined the network
[0x85cf:zdo] ZDO request ZDOCmd.Device_annce: [0x85CF, bc:33:ac:ff:fe:77:55:94, 142]
[0x85cf] Requesting 'Node Descriptor'
Tries remaining: 2
[0x85cf] Extending timeout for 0xac request
[0x85cf] Node Descriptor: NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0x85cf] Discovering endpoints
Tries remaining: 3
[0x85cf] Discovered endpoints: [1, 242]
[0x85cf:1] Discovering endpoint information
Tries remaining: 3
[0x85cf:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=258, device_version=1, input_clusters=[0, 4, 5, 3, 6, 8, 768, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x85cf:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=178 command_id=Command.Read_Attributes_rsp>
[0x85cf:1] Manufacturer: _TZE200_s8gkrkxk
[0x85cf:1] Model: TS0601
[0x85cf:242] Discovering endpoint information
Tries remaining: 3
[0x85cf:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=258, device_version=0, input_clusters=[], output_clusters=[33])
Checking quirks for _TZE200_s8gkrkxk TS0601 (bc:33:ac:ff:fe:77:55:94)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 242}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1, 242}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1, 242}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 242}
[... 90 lines cut ...]
        "0x0300",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0102",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_s8gkrkxk",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Add any other context or screenshots about the feature request here.

Hedda, 2020-12-02:
These new Zigbee devices seem to use the new Tuya custom "manuSpecificTuya" cluster in their UART protocol and it sounds as if that is used more by rebranded OEM devices that use internal hardware from Tuya. Because of this the zigbee2mqtt / zigbee-herdsman developers have now even written a specific guide to help add support additional new Tuya devices here:

https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html

FYI, looks like zigbee-herdsman (and zigbee2mqtt) devs figured out some things about Tuya UART protocol in these discussions:

Silvercrest HG06467 issue discussion: https://github.com/Koenkk/zigbee-herdsman-converters/issues/1792

and

Silvercrest HG06467 pull request: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1799

As noted, zigbee-herdsman-converters is zigbee2mqtt equivalent to zha-device-handlers for zigpy.

Hedda, 2020-12-02:
Tip from zigbee2mqtt developers regarding contacting the manufacturer for help on how to communicate with their devices: 

When contacting a manufacturer of Tuya compatible device, DO NOT ask for Zigbee protocol of the device, they usually have no idea how the Tuya radio that they bought communicates over Zigbee. Instead, ask for the UART protocol for their device, this should give you better cooperation. You can also ask them about DPIDs and data formats for their functions.

Hedda, 2020-12-02:
> I looking for OpenWRT firmware for the GW so can getting one nice EZSP with ethernet and dont need using WiFI. Its looks some work is done for the [RTL8196E](https://forum.openwrt.org/t/openwrt-on-rtl8196e-totolink-ex200-comfast-cf-wr302s/52030/17) but is not verified if its working.

You might then also want to check out the OpenLumi (Open Lumi Gateway) project -> https://github.com/openlumi

OpenLumi is an OpenWRT based Zigbee gateway/bridge made currently for other SoCs but I think that their concept is the same?

closed 2021-08-12 by github-actions[bot]