zigpy/zha-device-handlers #604: [Device Support Request] Lidl Christmas lights

Jeltel, 2020-11-27:
**Is your feature request related to a problem? Please describe.**
The Christmas lights of Lidl are recognized. But only with an on/off switch and a brightness. Switching the lights on and off work. Doing something with the brightness results in the lights going to a shade of white, eventhough a colour was set previously

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Full control of the Christmas lights, colour selector wheel and different effects avaiable:
[Name] [Options]
Steady (color)
Snow (color, speed)
Rainbow (speed)
Snake (color list up to 6, speed)
Twinkle (color list up to 2, speed)
Fireworks (color, speed)
Horizontal Flag (color list up to 3, speed)
Waves (color list up to 3, speed)
Updown (color list up to 2, speed
Vintage (color, speed)
Fading (color, speed)
Collide (color, speed)
Strobe (color list up to 5, speed)
Sparkles (color list up to 3, speed)
Carnival (color list up to 6, speed)
Glow (color list up to 6, speed)

[Manuals](https://www.lidl-service.com/cps/rde/xchg/SID-1FC8C7D5-504DC4E3/lsp/hs.xsl/product.html?id=5019460647&title=LED-verlichtingsketting&count=1)

**Device signature**

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x67a9:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=166 command_id=Command.Read_Attributes_rsp>
[0x67a9:11:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=168 command_id=Command.Read_Attributes_rsp>
[0x67A9:11:0x0008]: received attribute: 0 update with value: 176
[0x67a9:11:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=170 command_id=Command.Read_Attributes_rsp>
Device 0x85cf (bc:33:ac:ff:fe:77:55:94) joined the network
[0x85cf:zdo] ZDO request ZDOCmd.Device_annce: [0x85CF, bc:33:ac:ff:fe:77:55:94, 142]
[0x85cf] Requesting 'Node Descriptor'
Tries remaining: 2
[0x85cf] Extending timeout for 0xac request
[0x85cf] Node Descriptor: NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0x85cf] Discovering endpoints
Tries remaining: 3
[0x85cf] Discovered endpoints: [1, 242]
[0x85cf:1] Discovering endpoint information
Tries remaining: 3
[0x85cf:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=258, device_version=1, input_clusters=[0, 4, 5, 3, 6, 8, 768, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x85cf:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=178 command_id=Command.Read_Attributes_rsp>
[0x85cf:1] Manufacturer: _TZE200_s8gkrkxk
[0x85cf:1] Model: TS0601
[0x85cf:242] Discovering endpoint information
Tries remaining: 3
[0x85cf:242] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=258, device_version=0, input_clusters=[], output_clusters=[33])
Checking quirks for _TZE200_s8gkrkxk TS0601 (bc:33:ac:ff:fe:77:55:94)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 242}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1, 242}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1, 242}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
[... 91 lines cut ...]
        "0x0300",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0102",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_s8gkrkxk",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

**Additional context**
[image]

Jeltel, 2020-11-27:
What else can I do to get this implemented sooner? :)

Jeltel, 2020-11-28:
Scrap that. Going to return this and bought a twinkly.

GH2user, 2020-11-29:
Maybe the HG06467 protocol is useful to you guys, [https://pastebin.com/pcgWVjCp](url) as it describes all options for the Xmas lights.  The guys over at z2m have sniffed it and it use classic tuya command, can use the tuya code without problem it seems.

brantje, 2020-11-29:
> 
> 
> Scrap that. Going to return this and bought a twinkly.
> 
> I'll leave the code here, for future implementation.

May i ask why you're gonna go for a twinkly? I've got the same lights, i think they could be useful with a update of ZHA.

Jeltel, 2020-11-29:
> > Scrap that. Going to return this and bought a twinkly.
> > I'll leave the code here, for future implementation.
> 
> May i ask why you're gonna go for a twinkly? I've got the same lights, i think they could be useful with a update of ZHA.

The better app, you can calibrate you setup with a camera in the app. And wifi based, so integration in Home Assistant is easier. There is supposed to be a native HA implementation from 0.119. Don't know if it will include the scenes, but that would be very cool.

Yes the lidl lights will become more usefull with a proper zigpy integration. Hope that works :)

Adminiuga, 2020-12-01:
```
== Protocol== :
Sequence	The sequence number of the transmitted data, ranging from 0 to 65535, and returned to 0 after reaching 65535
Function Code	Power: 01 01
		Mode: 02 04
		In White: 03 02
		In Color: 05 03
		In Scene: 06 03
Length		effective data length of the transmission; single-frame data length does not exceed 64 bytes
Data		Valid Data Transferred
 
== Power Function ==
Fc		SEQ	CMD	LENGTH	DATA
Turn OFF	.. ..	01 01	00 01	00
Turn ON		.. .. 	01 01	00 01	01
 
== Mode Function ==
Fc		SEQ	CMD	LENGTH	DATA
White Mode	.. ..	02 04	00 01	00
Color Mode	.. .. 	02 04	00 01	01
Scene Mode	.. .. 	02 04	00 01 	02
 
== White mode function ==
Fc		SEQ	CMD	LENGTH	DATA
Set brightness	.. ..	03 02	00 04	00 00 xx xx
 
xx xx should be a number between 0 and 1000. 
 
 
== Color mode function == 
Fc		SEQ	CMD	LENGTH	DATA
Set Color	.. ..	05 03	00 0c	hh hh hh ss ss ss ss ll ll ll ll
 
hh hh hh hh, ss ss ss ss and ll ll ll ll should be a number between
30 30 30 61 and 30 33 65 38
 
== Scene mode function ==
Fc		SEQ	CMD	LENGTH	DATA
Set scene	.. ..	06 03	nn nn	fc fc ss ss rr rr gg gg bb bb
 
ss ss (speed) should be between 30 30 (12336) and 36 34 (13876)
rr rr gg gg bb bb (colors) should be between 30 30 (12336) and 66 66 (26214)
 
nn nn is not a fixed length, depends on data.
 
Depending on the amount of colors, multiple colors can be passed in the data, sequential after each other (so it becomes fc fc ss ss rr rr gg gg bb bb rr rr gg gg bb bb etc.)
 
Function codes (fc fc) are: 
Scene		FC	# colors
Steady		30 30	1
Snow		30 31	1
Rainbow		30 32	0
Snake		30 33	6
Twinkle		30 34	2
Firework	30 35	2
Horizontal Flag	30 36	3
Waves		30 37	3
Updown		30 38	2
Vintage		30 39	1
Fading		30 61	1
Collide		30 62	1
Strobe		30 63	5
Sparkles	30 64	3
Carnaval	30 65	6
Glow		30 66	6
```

Hedda, 2020-12-02:
Looks like zigbee-herdsman (and zigbee2mqtt) devs figured out some things about the Tuya UART protocol in these discussions:

https://github.com/Koenkk/zigbee-herdsman-converters/issues/1792

and

https://github.com/Koenkk/zigbee-herdsman-converters/pull/1799

These new Zigbee devices seem to use the new Tuya custom "manuSpecificTuya" cluster in their UART protocol and it sounds as if that is used more by rebranded OEM devices that use internal hardware from Tuya. Because of this the zigbee2mqtt / zigbee-herdsman developers have now even written a specific guide to help add support additional new Tuya devices here:

https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html

As noted, zigbee-herdsman-converters is zigbee2mqtt equivalent to zha-device-handlers for zigpy.

Hedda, 2020-12-02:
> > May i ask why you're gonna go for a twinkly? I've got the same lights, i think they could be useful with a update of ZHA.
> 
> The better app, **you can calibrate you setup with a camera in the app**. 

While off-topic here that idea of using [AR function with the app using the camera of your phone](https://www.twinkly.com/knowledge/map-your-lights-2/) sounds like that sounds awesome!

Hedda, 2020-12-02:
Tip from zigbee2mqtt developers regarding contacting the manufacturer for help on how to communicate with their devices: 

When contacting a manufacturer of Tuya compatible device, DO NOT ask for Zigbee protocol of the device, they usually have no idea how the Tuya radio that they bought communicates over Zigbee. Instead, ask for the UART protocol for their device, this should give you better cooperation. You can also ask them about DPIDs and data formats for their functions.

dmulcahey, 2020-12-03:
is there any way to purchase these in the US?

MattWestb, 2020-12-03:
East coast have stores https://www.lidl.com/stores and more is coming.
The bad thing is no sign for "Home Smart Home" on the zite ;-((

Hedda, 2020-12-04:
> is there any way to purchase these in the US?

Sounds like most if not all of Lidl's Zigbee products are rebranded OEM versions of Tuya or HEIMAN devices, and if so it might be easiest to find the same product but different brand from a Chinese online store and order it to the USA directly from China?

An alternative could be for a community member to buy the Lidl product in Europe and ship it to the USA but that is expensive.

One thing to check before getting products for the European marked shipped to North America is if the product also supports ~110v AC-power at 60Hz because all of western Europe runs on ~230v AC-power at 50Hz, and not all products support both.

Hedda, 2020-12-04:
> Sounds like most if not all of Lidl's Zigbee products are rebranded OEM versions of Tuya or HEIMAN devices

FYI, just a few other companies who are known for rebranding and selling at least a couple OEM or while-label Zigbee devices (and WiFi devices) from Tuya and HEIMAN are Zemismart, Alecto, Bandi, Binthen, Blitzwolf, Malmbergs, Gosund, Eachen, Moes, Sonoff, Teckin, Earda, LEELKI, Larkkey, Lonsonho, LoraTa, Norklmes, SHOJZJ, Samotech, Smart9, Somgoms, TUYATEC, UseeLink, Vrey, Woox, Yushun, and CR Smart Home

Many of those OEM or while-label Zigbee devices manufactured by Tuya can by the way be found in these lists:

https://www.zigbee2mqtt.io/information/supported_devices.html

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices

Hedda, 2020-12-04:
> Looks like zigbee-herdsman (and zigbee2mqtt) devs figured out some things about the Tuya UART protocol in these discussions:
> 
> Silvercrest HG06467 issue discussion: [Koenkk/zigbee-herdsman-converters#1792](https://github.com/Koenkk/zigbee-herdsman-converters/issues/1792)
> 
> and
> 
> Silvercrest HG06467 pull request: [Koenkk/zigbee-herdsman-converters#1799](https://github.com/Koenkk/zigbee-herdsman-converters/pull/1799)
> 
> These new Zigbee devices seem to use the new Tuya custom "manuSpecificTuya" cluster in their UART protocol and it sounds as if that is used more by rebranded OEM devices that use internal hardware from Tuya. Because of this the zigbee2mqtt / zigbee-herdsman developers have now even written a specific guide to help add support additional new Tuya devices here:
> 
> https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html
> 
> As noted, zigbee-herdsman-converters is zigbee2mqtt equivalent to zha-device-handlers for zigpy.

Zigbee2mqtt now have a webpage for these Melinera / Silvercrest HG06467 Christmas lights from Lidl with Zigbee ID "TS0601":

https://www.zigbee2mqtt.io/devices/HG06467.html

For reference, here is also the older Discourse discussion about Zigbee2mqtt support for these:

https://zigbee2mqtt.discourse.group/t/lidl-christmas-lights/1788

Hedda, 2020-12-04:
> Sounds like most if not all of Lidl's Zigbee products are rebranded OEM or while-label versions of Tuya or HEIMAN devices, and if so it might be easiest to find the same product but different brand from a Chinese online store and order it to the USA directly from China?

Does anyone have better pictures of the Christmas lights themselves and their control/power-adapters to help a search in China?

https://www.lidl.de/de/melinera-lichterkette-zigbee-smart-home/p360021

https://www.lidl.nl/nl/p/smart-home/200-rgb-led-verlichtingsketting/p43246

Also, it does not look like Lidl is selling this Christmas lights product in all countries in Europe or at least is not marketing them.

Hedda, 2020-12-04:
deconz development discussion about these "Melinera Smart LED string lights" from Lidl can be found here:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3716

Hedda, 2020-12-04:
> deconz development discussion about these "Melinera Smart LED string lights" from Lidl can be found here:
> 
> [dresden-elektronik/deconz-rest-plugin#3716](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3716)
> 
> > ## Device
> > 
> > * Product name: The device name as shown on the product or package.
> > * Manufacturer: _TZE200_s8gkrkxk
> > * Model identifier: TS0601
> > * Device type : Light

@ebaauw has now also submitted a PR to deconz for REST API support for the LIDL Melinera Smart LED strip (a.k.a. Xmas lights):

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3776

+a follow-up PR with enhancements and bug fixes for that 

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3789

MattWestb, 2020-12-04:
I have the 3 way outlet with USB charging and the RGBWS LED stripe and all have EC marking but no FCC marking and no of the Home Smart Home products is applied for it (under name Lidl US, LLC) at FCC-ID.
But likely more suppliers is making it from tuya for the next christmas.

By the way the led controller have the same EFR32 as SonOff Zigbee Bridge and pretty easy souldring the pins for flashing it :-)))
[image]

And easy modding for external antenna if needed and putting force boot loader switch.

Hedda, 2020-12-04:
@MattWestb did you also buy the Lidl Christmas lights as well and if so would you mind terribly to fully opening it up as well?

Thinking it might help find the matching OEM or white-label product in China if know all the markings on the circuit board?

MattWestb, 2020-12-04:
Was not getting it. In Austria they was only having half of the products in the east part and the west part the other (the GW and the motion sensor was for all). 
So no Zigbee Christmas Tree Lights here ;-((

Hedda, 2020-12-04:
> Was not getting it. In Austria they was only having half of the products in the east part and the west part the other (the GW and the motion sensor was for all).
> So no Zigbee Christmas Tree Lights here ;-((

Same in Sweden as no Zigbee Christmas at Lidl brick-and-mortar stores here and there is not even an Swedish Lidl online-store.

MattWestb, 2020-12-04:
I think LIDL dont want killing the tower of Kungens Kurva ;-)
But its sad for the HA community that not releasing it  in all stores !
I is on the hunt for the ceiling lamps but cant going to Innsbruck then its lockdown here and also IKEA is closed so no köttbullar. löksill and lussekatter for christmas this year !!

rangulvers, 2020-12-07:
is there any chance of this being picked up and supported?

dmulcahey, 2020-12-07:
Yes, will just take time. Especially because we can’t get them.

ebaauw, 2020-12-07:
The LIDL Xmas light strip is supported in v2.7.0, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3716#issuecomment-735467996.  Be sure to (re-)pair it under that version.

rangulvers, 2020-12-07:
I'm guessing the PR by @ebaauw  is nothing that can be reused by the ZHA integration right away? @dmulcahey  anything that I can do to help out? Would be really great to have this before xmas time. I do have the light strip and the gateway. But I rather not use the gateway and have the controls with HA

brantje, 2020-12-07:
If needed, i can pick them up here at the local lidl, and ship them to you guys. 
Plug will be EU (230V), not sure if that can be switched easily. 

Would it be possible to have the different modes in HA/ZHA?

Hedda, 2020-12-08:
> Plug will be EU (230V), not sure if that can be switched easily.

FYI, @Somberland posted some pictures of the lights + control-box + the transformer/plug power-supply combo (JT-DC24V24W by Jutai/CZJutai) in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3716 and it looks to use one of those standard AC-to-DC 24 Volt 24W capacity transformer with an IP44 (outdoor) plug that is commonly used for strings / light-chain of Christmas-style LED-lights. So should be relatively easy to find a replacement transformer capable of outputting at least 24 volts @ 0.85A so a 1 Amp / 24 Watt 24 volts DC-transformer is probably recommended.

https://wap.jutai.com/product/25W-SERIES-LED-DRIVERS.html

Hedda, 2020-12-08:
Tuya "TY-REV002" Smart LED String Lights looks to be the original white-label / OEM product that Lidl has rebranded for Europe:

https://zigbeealliance.org/zigbee_products/smart-led-string-lights/

Zigbee Alliance certification report for Tuya TY-REV002 :

https://api.knack.com/v1/applications/54e658034b4f44e42fb18201/download/asset/5f50ddfbe759b10015534119/pics.zip

That fact has not helped me narrow down what the equivalent Zigbee product version(s) is known as in Chinese online stores or where to order it from in China. Tuya list a few similar products on their website but the string lights and lightstrips there are Bluetooth and WiFi-based or white-color only and not RGB:

https://expo.tuya.com/smart/lighting/CT9cn91vklm6mj

mjdegraaff, 2020-12-09:
> The LIDL Xmas light strip is supported in v2.7.0, see [dresden-elektronik/deconz-rest-plugin#3716 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3716#issuecomment-735467996). Be sure to (re-)pair it under that version.

Little bit off topic.

Currently I'm using the zha device handler and switching to deconz interferes with zha so I have to move all my devices to deconz.. or is there a way both integrations can work at the same time?

ijzel, 2020-12-09:
> Little bit off topic.
>
>
>
> Currently I'm using the zha device handler and switching to deconz interferes with zha so I have to move all my devices to deconz.. or is there a way both integrations can [...]

I'm new with ZHA and using it with a flashed zbbridge from Sonoff. I repaired my christmas lights in Home Assistant 2 days ago, but it's not working. Should I wait for an update of HA? And is compatibility in Deconz equal to ZHA?
(Sorry for the n00b questions, just started with this)

ijzel, 2020-12-09:
Found this so it should work I guess:

https://zigbee.blakadder.com/Lidl_HG06467.html

Hedda, 2020-12-10:
> Currently I'm using the zha device handler and switching to deconz interferes with zha so I have to move all my devices to deconz.. or is there a way both integrations can work at the same time?

You can only use both integrations at the same time if you buy/have two separate Zigbee USB adapters. That is, you can only use any Zigbee USB adapter exclusivly with one Zigbee integration, and any Zigbee integration can in turn only have one adapter.

> I'm new with ZHA and using it with a flashed zbbridge from Sonoff. I repaired my christmas lights in Home Assistant 2 days ago, but it's not working. Should I wait for an update of HA? And is compatibility in Deconz equal to ZHA?

Sorry but that link was only for reference for the developers. Support is deCONZ does not equal support in ZHA. ZHA and deCONZ are two different Zigbee gateway products. From an end-users point-of-view they have nothing to do with each other but developers from different can learn from each others success and failures so the information provided there could possibly help ZHA developers come closer to a solution for ZHA, however, it is no guarantee and no time-frame can be given based on that.

> Found this so it should work I guess:
> 
> https://zigbee.blakadder.com/Lidl_HG06467.html
> 
> But I repaired it 5 min ago and it's not seeing it properly

Sorry but that information on that website is just wrong in regards to ZHA compatibility with Melinera LED String Lights from Lidl. 

Strange though as website code does not list ZHA https://github.com/blakadder/zigbee/blob/master/_zigbee/Lidl_HG06467.md

ijzel, 2020-12-10:
> You can only use both integrations at the same time if you buy/have two separate Zigbee USB adapters. That is, you can only use any Zigbee USB adapter exclusivly with one Zigbee integration, and any [...]

Thank you for your detailed answer. If the blakadder website is not correct, then that explains why it is not working.
I've tried to figure out how to make a quirk on my own, but it's just too complicated for me as I'm not very familiar with ZHA and zigbee. Hopefully someone will step up and make a working quirk.

brantje, 2020-12-10:
Seems the site is displaying it wrong. The [MD](https://github.com/blakadder/zigbee/blob/master/_zigbee/Lidl_HG06467.md) file shows only z2m

MattWestb, 2020-12-10:
Hey @blakadder Its somthing wrong with your metadata or so of some devices.
The HG06467 is only supported by Z2M and deCONZ (partly but working) and is not implanted in ZHA then we have not finding one way getting one device to our devs in US for implanting it.

Thanks in advance.

blakadder, 2020-12-10:
website was going on the statement that zha will work with all zigbee lights. I removed that part from the code now

MattWestb, 2020-12-10:
Thanks @blakadder !
That is true for "normal" lights that is reasonable working  with some zigbee standards.
The on/off is standard zigbee but all other "tuya" commands must being sniffed and and implanting for working with the system to getting the christmas tree lighting to working.

Adminiuga, 2020-12-10:
>  that zha will work with all zigbee lights

Should have been with all "Standard" Zigbee lights, i.e. devices with server clusters: 6, 8 and 300
Tuya devices a hardly standard Zigbee Lights.  :) Unless for regular users we also need to put a disclaimer that ZHA won't work with Bluetooth, Zwave and WiFi lights.

Hedda, 2020-12-10:
> website was going on the statement that zha will work with all zigbee lights. I removed that part from the code now

> Should have been with all "Standard" Zigbee lights, i.e. devices with server clusters: 6, 8 and 300
> Tuya devices a hardly standard Zigbee Lights. :) Unless for regular users we also need to put a [...]

@blakadder  FYI, I think that the same should go for Zigbee2MQTT and Tasmota2Zigbee (Tasmota 2 Zigbee).

MattWestb, 2020-12-10:
For deCONZ and T2Z its true but for Z2M all "normal" devices need being added in there database for working out of the box.

blakadder, 2020-12-10:
You can adapt most devices in Tasmota since it usually operates with endpoints and codes from ZCL

@Hedda same what?

Hedda, 2020-12-11:
> website was going on the statement that zha will work with all zigbee lights. I removed that part from the code now

> @Hedda same what?

Suggest that you reconsider if your website should automatically list all Zigbee lights as supported by any gateway product. You can always assume but you can not know for sure that all Zigbee lights until someone have tested the device with that solution.

rangulvers, 2020-12-11:
the only thing that is not correct on @blakadder  site is the color temperature. 

Brightness, color xy and on/off work as they should. 

Now we are just waiting for ZHA to also support the effect option

MattWestb, 2020-12-11:
The normal Zigbee cluster is working but the tuya commands is not implanted yet :-))

Hope to see your christmas tree then you have getting it Zigbeesized !!!

blakadder, 2020-12-11:
> Suggest that you reconsider if your website should automatically list all Zigbee lights as supported by any gateway product. You can always assume but you can not know for sure that all Zigbee [...]

All lights that are automatically supported is if the developer of the said gateway solution stated so.

ijzel, 2020-12-11:
> the only thing that is not correct on @blakadder  site is the color temperature. 
> 
> 
> 
> Brightness, color xy and on/off work as they should. 
> 
> 
> 
> Now we are just waiting for ZHA to also support the effect option

I have tested it, It turns on and off, but color pick or brightness are working a bit iffy. Colors not always match and brightness is very unpredictable. 
Fingers crossed for good support soon

brantje, 2020-12-18:
Is we could send custom states with a ZHA service, then it should work.

For example, with Z2M i'm using this:
```
type: button
tap_action:
  action: call-service
  service: mqtt.publish
  service_data:
    topic: zigbee2mqtt/christmas_lights_1/set
    payload: >-
      {"effect":{"effect":"glow","speed":100,"colors":[{"r":164,"g":168,"b":50},{"r":168,"g":50,"b":153},{"r":102,"g":50,"b":168},{"r":50,"g":155,"b":168},{"r":50,"g":168,"b":72},{"r":168,"g":88,"b":50}]}}
name: Glow
``

cmitz, 2020-12-30:
I gave it a shot, building a quirk, but I'm way out of my comfort zone and not getting anywhere...
[image]

closed 2021-08-12 by github-actions[bot]

Raapale, 2021-12-07:
I set it up in IDE as ZigBee RGBW Bulb and I'm able to switch it on/off, set the color (hue and saturation) and also brightness.

Hedda, 2021-12-08:
> Has anyone managed to build a smartthings device handler for Lidl christmas lights yet ("smart led string lights")

If you are referring to "Hub-connected Device Handlers for the SmartThings Classics platform" (as in for Samsung SmartThings) and not to "ZHA Device Handler" as handers for zigpy and Home Assistant ZHA then you need refer to SmartThings community instead.

https://www.home-assistant.io/integrations/zha#zha-exception-and-deviation-handling

https://community.smartthings.com/t/how-do-i-add-a-custom-device-handler/120454

d3nnis0, 2021-12-21:
> I set it up in IDE as ZigBee RGBW Bulb and I'm able to switch it on/off, set the color (hue and saturation) and also brightness.

I currently try to implement a custom quirk for the Christmas lights and also support the effects - Do you already have a config to share?

(As I'm  not very experienced with this I'm currently not sure if I'm able to implement the scene handling etc. but let's see ;-) )

ijzel, 2021-12-21:
> I currently try to implement a custom quirk for the Christmas lights and also support the effects - Do you already have a config to share?
>
>
>
> (As I'm  not very experienced with this I'm currently [...]

Is there some tutorial about installing/using this IDE? Like to play with it myself

Raapale, 2021-12-22:
Sorry, I wrote on a wrong thread. I meant to post it on the smartthings group. My posts are related to smartthings platform. Sorry for confusion.

erkr, 2022-11-17:
Please reopen this issue. Christmas is nearing. Lidl still had these lights for sale.