zigpy/zha-device-handlers #605: [Device Support Request] BlitzWolf BW-SHP13 smart plug (aka Tuya TS0121)

jmoransalama, 2020-11-27:
**Is your feature request related to a problem? Please describe.**
BlitzWolf BW-SHP13 smart plug 16A EU version, same device as TuYa TS0121 smart plug.
[image]

Device joins zigbee network and it's recognized by home-assistant. 

Entities are created for:
- On/Off switch: It works
- Electrical power measurement (W): Values updated correctly.
- Energy metering (kWh): Entity is created but values are never updated.

This entities are missing:
- Current (A).
- Voltage (V).

**Describe the solution you'd like**
Please consider adding support for the missing values and fixing the electrical power measurement value.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_3ooaz3ng",
  "model": "TS0121",
  "class": "zigpy.device.Device"
}
```

**Additional context**
This device doesn't support reporting of electrical measurement values, so it needs to be polled. It's [supported on zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt/issues/3812), they fixed this by polling the values every 10 seconds.

Voltage value is correctly reported if I manually get the attribute value:
[image]

CharlieBailly, 2020-12-22:
Hi, yes it would be great if it was fully supported as you describ it.
I also have the same device, but with a slightly different signature : (the manufacturer part differ)
```json
 {
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_g5xawfcq",
  "model": "TS0121",
  "class": "zigpy.device.Device"
}
```

TheJulianJES, 2021-01-03:
Z2M polls them every 60 seconds: https://github.com/Koenkk/zigbee-herdsman-converters/blob/c9ed6bb75a9682cf60b41e4e7647c653480f9a6c/devices.js#L1490-L1503 because they don't support attribute reports apparently (or at least don't do it properly).
It doesn't look like that zha-quirks has something similar yet. So I'm not sure how something like this would even be implemented correctly.

dmulcahey, 2021-01-03:
ZHA already polls the electrical measurement cluster. Is this not an EM cluster?

TheJulianJES, 2021-01-03:
It should be. Perhaps it's missing these constant attributes (https://github.com/Koenkk/zigbee-herdsman-converters/blob/c9ed6bb75a9682cf60b41e4e7647c653480f9a6c/devices.js#L1479-L1483) (or the binding above that)? (Just guessing at this point though).
The Xiaomi quirk does something similar: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/__init__.py#L410-L415
(Although the Xiaomi Smart Plug (EU) with power reporting also worked before doing that for me)

TheJulianJES, 2021-01-04:
I just re-read the issue. It looks like the electrical power measurement (W) is working correctly, right? Only voltage and current do not have entities created for them. The total "smart" (?) energy meter isn't updated.
That looks like the same "issue" basically all power measuring plugs have (at least the Xiaomi Plugs) (?)

dmulcahey, 2021-01-04:
Yes, if that’s the case. We currently only support 1 sensor per cluster. This is currently a limitation in ZHA.

TheJulianJES, 2021-01-18:
Got my Blitzwolf plugs today. They seem to work well so far. (On/off + current consumption).
With Z2M you can also set the initial power on state: https://www.zigbee2mqtt.io/devices/TS0121_plug.html#power_outage_memory-enum
I guess that's something ZHA could also implement. Since that shouldn't be hard, I'll see if I get around to it in the next couple of days.

Hedda, 2021-01-28:
> Yes, if that’s the case. We currently only support 1 sensor per cluster. This is currently a limitation in ZHA.

Any plans to extend ZHA to support more than one sensor per cluster? 

> BlitzWolf BW-SHP13 smart plug 16A EU version, same device as TuYa TS0121 smart plug.
> [image]
>
> Device joins zigbee network and it's recognized by home-assistant.
>
> Entities are created for:
>
> * On/Off [...]

Hope to see full support for these Tuya smart plugs as there are few inexpensive Zigbee plugs featuring power/energy-metering.

Lonsonho also has Tuya TS0121 branded variant which also features support. for power-metering and energy-metering

https://zigbee.blakadder.com/Lonsonho_TS0121.html 

https://www.aliexpress.com/item/4001118546894.html

Really like that and BlitzWolf BW-SHP13 because power/energy-metering and fact that they are both inexpensive and very small.

[image]

[image]

[image]

[image]

Hedda, 2021-01-28:
And this one from "Rehentele" looks to be yet another Tuya TS0121 variant with power-metering and energy-metering support:

https://www.aliexpress.com/item/4000449939329.html

Note that Rehentele seem to sell a TS0121 version with power-metering as well as a TS0111 version without that looks the same.

[image]

[image]

[image]

vpomax, 2021-02-12:
It shows consumption in kWh (energy metering) as "unknow"... any solution about?

sakara, 2021-02-12:
Please, add the missing entities so i can move from z2m to ZHA.

McLP11, 2021-02-12:
same request from my side, this plug is perfect as size and power capability, so having the chance to see the consumption in ZHA will be the perfect match. I really hope this feature can be implemented, thanks.

Gamester17, 2021-02-15:
> That looks like the same "issue" basically all power measuring plugs have (at least the Xiaomi Plugs) (?)

> For all other issues (voltage, current not displaying), ZHA would need to be "changed".

What exactly needs to be "changed" in the ZHA integration to overcome the limit of 1 sensor per cluster?

How many sensors per cluster does other Zigbee implementations like example Zigbee2MQTT support?

manouallou, 2021-02-18:
It would be nice to know if there is a roadmap for this "fix" (more than 1 sensor per cluster) or else we should make a decision to steer away from Zigbee power plugs.

rklomp, 2021-03-18:
+1. Bought a Tuya TS0121 from aliexpress to measure consumption of different devices.

On/Off switch works and Electrical power measurement (W) updated correctly.

I can retrieve the total usage from the device in the Metering cluster using the current_summ_delivered attribute, but unfortunately there is no sensor for this.

flabbamann, 2021-04-03:
I did some research with my BW-SHP13 in Home Assistant via ZHA:

The current_summ_delivered attribute @rklomp mentioned gets written to zigbee.db. You could use a sql sensor to read the value from there.

You have to multiply that value by 10 to get Wh or by 0.01 to get kWh.

```
- platform: sql
  db_url: sqlite:////config/zigbee.db
  scan_interval: 30
  queries:
  - name: plug_metering
    query: "SELECT value * 0.01 as value FROM attributes WHERE ieee = 'YOUR_DEVICEID' AND cluster = 1794 AND attrid = 0"
    column: "value" 
    unit_of_measurement: kWh
   ```

So, the data is there. What needs to be done to show that via the device sensor?

_Sidenote_:
I cross checked my Blitzwolf with an old dumb metering plug a have and the Blitzwolf always shows about 50% higher values. The Blitzwolf shows 120W while my other plug shows 80W (power measurement). So in that case I would multiply by 0.0066 instead of 0.01 to get a more accurate value, but I'm unsure which device I should trust more. In the end both are very cheap and no high-precision metering devices 😉

Hedda, 2021-05-19:
> > I just re-read the issue. It looks like the electrical power measurement (W) is working correctly, right? Only voltage and current do not have entities created for them. The total "smart" (?) energy meter isn't updated.
> > That looks like the same "issue" basically all power measuring plugs have (at least the Xiaomi Plugs) (?)
> 
> Yes, if that’s the case. We currently only support 1 sensor per cluster. This is currently a limitation in ZHA.

FYI, there is a developers/development discussion about this "issue" in this related PR by @stattin42 -> https://github.com/home-assistant/core/pull/50471

stattin42, 2021-05-19:
While sql sensors or events for attribute changes could be a way to handle multiple attributes per cluster, I am still struggling a little with the rationale behind selecting an optional attribute as the base entity/sensor for a cluster or channel. Wouldn't it be more useful to use a mandatory attribute for the sensor? 

Now that things are what they are, I have some sympathy for not breaking things by changing existing behaviour. I cannot help thinking, though, that there is a sense of urgency finding a solution. Hopefully a solution which does not require templating for commonly used mandatory attributes. Support is currently broken for many compliant devices unless they support a certain optional attribute.

flonair, 2021-08-05:
> You are correct, current_summ_delivered in in zigbee db, BUT it updates there very rarely ( once per day)
> I have several different plugs and only one plug, BlitzWolf TS0121, updates correctly, every minute.

Same for me. Two plugs update regular, but four other plugs update only sporadically (sometimes after 5 minutes, sometimes after 30 minutes).

pipiche38, 2021-08-05:
my understanding is that then current_summ_delivered is sent by the device when it reach a certain level of current delivered. So I would suggest to put something consuming more current to see if you see the update arriving more often.

flonair, 2021-08-05:
> my understanding is that then current_summ_delivered is sent by the device when it reach a certain level of current delivered. So I would suggest to put something consuming more current to see if you see the update arriving more often.

For current_summ I think this makes sense. But I also have the update problem with current_power. And there something like that would be absolutely pointless. For example, I want to be notified when the washing machine is ready and not 30 or even 60 minutes later.

pipiche38, 2021-08-05:
For instant power, my understanding is that there is no reporting for  as the Blitzwolf TS0121 are concerned, and you must poll the information by doing a read attribute as per the time cycle you want

juhanjuku, 2021-08-05:
> For instant power, my understanding is that there is no reporting for as the Blitzwolf TS0121 are concerned, and you must poll the information by doing a read attribute as per the time cycle you want

How do you read attribute with code? I can do this only from Manage Clusters dialog manually.

flonair, 2021-08-06:
Maybe, but actually ZHA/Z2M polls independently. Otherwise no values would arrive, because I have not set this anywhere. Or am I misunderstanding this?
If I use the node "poll state" in nodered, then I get only the state that HA knows. That means: if the plug polls 100W and then 30 minutes no more (although the consumer is off), then the node polls always the 100W. 

So I would have to set a poll interval directly in ZHA or what do you mean?

flonair, 2021-08-06:
> For instant power, my understanding is that there is no reporting for as the Blitzwolf TS0121 are concerned, and you must poll the information by doing a read attribute as per the time cycle you want

Maybe, but actually ZHA/Z2M polls independently. Otherwise no values would arrive, because I have not set this anywhere. Or am I misunderstanding this?
If I use the node "poll state" in nodered, then I get only the state that HA knows. That means: if the plug polls 100W and then 30 minutes no more (although the consumer is off), then the node polls always the 100W.

So I would have to set a poll interval directly in ZHA or what do you mean?

pipiche38, 2021-08-06:
I'm sorry but I don't know how zigpy and zha works. But the way we are doing on other plugins not zigpy related) is to poll every 30s or every 60s , so you can catch if there is something consuming power or not.
Then as mentioned for the current_summation this is a based on some trigger on the device itself.

Blitzwolf is a Tuya based device, and you get what you pay for. If you want a full working plug with instant and summation you can go for Salus SP600 which works very well, with reporting and they are also other brands like Legrand, but the price is somehow more expensive :-(

jordyals, 2021-08-18:
Hey all, since the new energy dashboard, is this also the topic to see the opportunities to integrate this plug inside that? I'm now trying to do something with a template sensor to get things inside the energy dashboard.

Hedda, 2021-10-05:
Looks like this might be working soon, see:

https://github.com/home-assistant/core/pull/56909

https://github.com/home-assistant/core/pull/56666

MattWestb, 2021-10-05:
For all users of TS0121 device is ZHA adding the device to summation_delivered channel but no TS0121 user have testing is it working with there devices.

One user have doing it with TS011F and the device is sending values with wrong multiplier and hi is fixing it on the device quirk that is ready for merging in the dev.

If like getting the TS0121 working OK its need testing and feedback from the user for getting it working OK and perhaps updating the quirk for the devices.

Hedda, 2021-10-06:
Also note the last comment from Adminiuga  in https://github.com/home-assistant/core/pull/56909

"**_Support for other attributes could be extended later, as I couldn't find too many devices other than Sinopé RM3250ZB Electrical Load Controller – 50A – Zigbee https://www.amazon.com/dp/B07CGJ6XH2/ reporting anything interesting on the electrical measurement cluster._**"

Celer21, 2021-10-08:
> > Hi, I have 4 of BW-SHP13 and after new HA version, they get new sensor and it shows data. But it show kWH unit, but it number is actually in Wh.
> 
> Are you sure the value is Wh and not "10Wh"? On my devices it's "10Wh" so we must divide by 100 and not by 1000. Can you confirm that?

Hi, 
maybee it is in 10Wh. I am only shure, it is not in kWh. Because it is connectect to whashing mashine and showing 16 323,0 kWh. And that is too much after 1/2 year of using it...

juhanjuku, 2021-10-08:
I have plugs from Schwaiger, Innr SP 120, BlitzWolf, Lonsonho and Salus.
Now they all show smartenergy_metering_summation_delivered, but values are not updating in Lovelace.
electrical measurement utdates frequently but summation_delivery shows reading after restart.
Only Lonsonho plug updates also summation_delivery. I have made restarts, nothing helps.

[image]
.

stattin42, 2021-10-08:
Drilling a bit deeper into the WARNING above, it appears that in `configure_reporting_multiple` in `zigpy/zcl/__init__.py` it is expected that the return value from `_configure_reporting_multiple` may be a list.

However, in the case of the WARNINGs I see, `_configure_reporting_multiple` returns a one element nested list with `[[ConfigureReportingResponseRecord(status=0)]]` which does not go very well with the condition:

```
if (
            isinstance(res, list)
            and not (res[0].status == foundation.Status.SUCCESS and len(res) == 1)
            and len(res) >= 0
        ):

@Adminiuga, is this specific to my devices or more general? Should a separate issue be created for this?
```

Adminiuga, 2021-10-12:
Do you have same log, but as instead of zigpy_deconz.uart include logging for zigpy_deconz.zigbee.application and zigpy_deconz.api

Celer21, 2021-11-05:
I update HA to 2021.11. The BW-SPH13 has now another sensors - great!
But i think, that sensors electrical_measurement_rms_current and total comsumption has to be divided by 1000.

It show, tahat my Washingmachine consume 2.400 W (correct ) and 10.000 A (that is is not correct).
Correct is 10 A.

[image]

[image]

juhanjuku, 2021-11-05:
Is your energy sensor kWh updating correctly?
My BlitzWolf plug updates kWh sensor only once when restarted and is freezed after that.

Celer21, 2021-11-05:
It's updatete with no problem.
This is chart of my Washing Mashine cycle.
[image]

If there is small load, it change after every 1 or 10 W or so. The kWh shoud be divided by 1000 or 100 (I have no energymeter to test it)

[image]

flabbamann, 2021-11-05:
@Celer21 please try if reconfiguring your plug fixes total consumption 😃

mbo18, 2021-11-05:
Works as expected since last HA update! Consumption (kWh) seems still not reliable

juhanjuku, 2021-11-05:
I can confirm now BlitzWolf plug kWh is updating, but Shwaiger and Innr SP120 are still freezed.
[image]
For Innr SP120 reconfigure fails
[image]
[logInnrSP120.txt](https://github.com/zigpy/zha-device-handlers/files/7487642/logInnrSP120.txt)

flabbamann, 2021-11-08:
#1131 adds the correct divisor for rms_current.
The devices likely need to be reconfigured to display the correct values.

davet2001, 2021-11-08:
> #1131 adds the correct divisor for rms_current. The devices likely need to be reconfigured to display the correct values.
Yes, for a different device I had to re-set up the zigbee device in HA to get the new scaling.

Celer21, 2021-11-08:
In which standard version of HA will it be distributed? I only have one instance of HA and cannot use dev or beta channel.
Thanks!

Hedda, 2021-11-08:
> In which standard version of HA will it be distributed? I only have one instance of HA and cannot use dev or beta channel. 

No one can tell until a new release and it has been bumped in HA so keep an eye on https://github.com/zigpy/zha-device-handlers/releases in combination with "Bump ZHA quirks version" in https://github.com/home-assistant/core/pulls?q=zha

Celer21, 2021-11-08:
@Hedda Thank you for links.

pfriedrich84, 2022-01-24:
Hi everyone,
i think this is not fixed yet? The current is showing wrong values.

Here yu have an example:
[image]

So in fact the current requires a divider of appr. 1.000 - in this case 224V multiplied with 662A would mean 148,29kw - I am not charing an electric car here ;-)
With the correction it would be 148Watts so way more reasonable and especially the consumption would be better.

cheers

zhaquirks.tuya.ts0121_plug.Plug
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_g5xawfcq",
  "model": "TS0121",
  "class": "zhaquirks.tuya.ts0121_plug.Plug"
}`

jmoransalama, 2022-01-24:
> Hi everyone, i think this is not fixed yet? The current is showing wrong values.
> 
> Here yu have an example: [image]
> 
> So in fact the current requires a divider of appr. 1.000 - in this case 224V multiplied with 662A would mean 148,29kw - I am not charing an electric car here ;-) With the correction it would be 148Watts so way more reasonable and especially the consumption would be better.
> 
> cheers
> 
> zhaquirks.tuya.ts0121_plug.Plug `{ "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)", "endpoints": { "1": { "profile_id": 260, "device_type": "0x0051", "in_clusters": [ "0x0000", "0x0004", "0x0005", "0x0006", "0x0702", "0x0b04" ], "out_clusters": [ "0x000a", "0x0019" ] } }, "manufacturer": "_TZ3000_g5xawfcq", "model": "TS0121", "class": "zhaquirks.tuya.ts0121_plug.Plug" }`
> 
> Update: I fix it with a simple template right now but maybe someone wants to fix the zhaqurik: [image]

Have you tried reconfiguring the switch in the device page? I did that with the switches that I had previosly paired and it fixed all the wrong values.

Celer21, 2022-01-25:
My plug shows wrong current value too.
I am running HA 2021.12.10 so maybe the patch will be included in next release?
Is it enough to reconfiguring device? Or is it necessary to remove from ZHA and repair it again?

pfriedrich84, 2022-01-25:
> > Hi everyone, i think this is not fixed yet? The current is showing wrong values.
> > Here yu have an example: [image]
> > So in fact the current requires a divider of appr. 1.000 - in this case 224V multiplied with 662A would mean 148,29kw - I am not charing an electric car here ;-) With the correction it would be 148Watts so way more reasonable and especially the consumption would be better.
> > cheers
> > zhaquirks.tuya.ts0121_plug.Plug `{ "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)", "endpoints": { "1": { "profile_id": 260, "device_type": "0x0051", "in_clusters": [ "0x0000", "0x0004", "0x0005", "0x0006", "0x0702", "0x0b04" ], "out_clusters": [ "0x000a", "0x0019" ] } }, "manufacturer": "_TZ3000_g5xawfcq", "model": "TS0121", "class": "zhaquirks.tuya.ts0121_plug.Plug" }`
> > Update: I fix it with a simple template right now but maybe someone wants to fix the zhaqurik: [image]
> 
> Have you tried reconfiguring the switch in the device page? I did that with the switches that I had previously paired and it fixed all the wrong values.

sure multiple times. its still showing crazy currents and wrong consumption - real consumption is about 0,33kwh ... ;-)
[image]

jmoransalama, 2022-01-25:
> > Have you tried reconfiguring the switch in the device page? I did that with the switches that I had previously paired and it fixed all the wrong values.
> 
> sure multiple times. its still showing crazy currents and wrong consumption - real consumption is about 0,33kwh ... ;-) There is just my work laptop running on this switch

What values do you get for attributes ac_current_divider and ac_current_multiplier in the TuyaZBElectricalMeasurement cluster?

[image]
[image]

MattWestb, 2022-01-25:
I think repairing its needed or reading the "new" added (in the quirk) attribute then its `null ` if not being populated and then ZHA is not using them.

pfriedrich84, 2022-01-26:
Hi all,
the cluster attributes where Null so i selected the right divisor/multiplier and then it changed to a reasonable value:
[image]

Thanks a lot!

rkruisselbrink, 2022-02-04:
> > In which standard version of HA will it be distributed? I only have one instance of HA and cannot use dev or beta channel.
> 
> No one can tell until a new release and it has been bumped in HA so keep an eye on https://github.com/zigpy/zha-device-handlers/releases in combination with "Bump ZHA quirks version" in https://github.com/home-assistant/core/pulls?q=zha then wait for it to appear under "All changes" in https://rc.home-assistant.io/latest-release-notes/

FYI: The new version of zha quirks (0.0.66) is included in the HA 2022.02 release :)

rkruisselbrink, 2022-02-04:
My device total usage readings (metering_summation_delivered) still appear to be wrong in HA (after reconfigure). I don't know what steps to take to fix this?

Amperage, voltage and wattage readings all seem fine, but summation shows a lot of kWh's which should be Wh i think. So probably a wrong multiplier somewhere?

metering_summation_delivered:
[image]

Energy dashboard - individual device:
[image]

jmoransalama, 2022-02-04:
> My device total usage readings (metering_summation_delivered) still appear to be wrong in HA (after reconfigure). I don't know what steps to take to fix this?

Try setting the values of the attributes divisor (0x0603) and multiplier (0x0602) in the TuyaZBMeteringCluster cluster. Divisor should be 100 and multiplier should be 1.

rkruisselbrink, 2022-02-04:
> > My device total usage readings (metering_summation_delivered) still appear to be wrong in HA (after reconfigure). I don't know what steps to take to fix this?
> 
> Try setting the values of the attributes divisor (0x0302) and multiplier (0x0301) in the TuyaZBMeteringCluster cluster. Divisor should be set to 100 and multiplier should be set to 1.

Those are already set to 100 (divider) and 1 (multiplier), but I set them both anyway and now the values seem much more reasonable! Thanks! 😊

ChefkeGremmen, 2022-02-06:
> Those are already set to 100 (divider) and 1 (multiplier), but I set them both anyway and now the values seem much more reasonable! Thanks! 😊

I tried the same and it also fixed the issue for me completely!

radek-foltyn, 2022-04-08:
Hi, can I ask you. Is it possible to delete sum consumption (power)? I have Zigbee2mqtt.

Hedda, 2022-05-10:
> Hi, can I ask you. Is it possible to delete sum consumption (power)? I have Zigbee2mqtt.

@radek-foltyn  zha-device-handlers (zha-quirks) is for [Home Assistant ZHA](https://www.home-assistant.io/integrations/zha), not for Zigbee2MQTT. For Zigbee2MQTT see instead:

https://github.com/Koenkk/zigbee2mqtt/issues/

https://github.com/Koenkk/zigbee2mqtt/discussions

closed 2022-11-13 by github-actions[bot]