zigpy/zha-device-handlers #610: [Device Support Request] Busch-Jaeger Zigbee Light Link control devices (especially 6737)

docgalaxyblock, 2020-12-02:
[Busch Jaeger Productsite](https://www.busch-jaeger.de/en/online-catalogue/product-overview?tx_nlbjproducts_catalog%5Baction%5D=list&tx_nlbjproducts_catalog%5BcatDokument%5D=1217&tx_nlbjproducts_catalog%5Bcontroller%5D=CatBjeProdukt&cHash=cc7e44a09ccd15aedebc0963cf33f033)
Already supported at Z2M: https://www.zigbee2mqtt.io/devices/6735_6736_6737.html
And deconz

**Is your feature request related to a problem? Please describe.**
My device (6737-*XX*) recognizes as RM01
I am able to switch it
it also shows a brigthness slider but the connected module is only a switch

**Describe the solution you'd like**
- [x] switchable
- [x] dimmable (I guess that is working but I only have the relais for testing) 
- [ ] switch state should report back
- [ ] button actions should report back
- [ ] buttons should be selectable at the automations wizard
- [ ] distinguish whether a relay module or a dimmer module is connected

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
[Device signature and pairing log at my dev VM](https://pastebin.com/85iaZPLQ)

**Additional context**
I am nut shure but it could be that Z2M pools the state of the switch because if I use the device buttons it recognizes the change 5s later

docgalaxyblock, 2020-12-04:
Signature
```
{
  "node_descriptor": "NodeDescriptor(byte1=17, byte2=64, mac_capability_flags=142, manufacturer_code=4398, maximum_buffer_size=89, maximum_incoming_transfer_size=61, server_mask=0, maximum_outgoing_transfer_size=61, descriptor_capability_field=0)",
  "endpoints": {
    "10": {
      "profile_id": 49246,
      "device_type": "0x0810",
      "in_clusters": [
        "0x0000",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "11": {
      "profile_id": 49246,
      "device_type": "0x0810",
      "in_clusters": [
        "0x0000",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "12": {
      "profile_id": 49246,
      "device_type": "0x0810",
      "in_clusters": [
        "0x0000",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "13": {
      "profile_id": 49246,
      "device_type": "0x0810",
      "in_clusters": [
        "0x0000",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "18": {
      "profile_id": 49246,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "Busch-Jaeger",
  "model": "RM01",
  "class": "zigpy.device.Device"
}

docgalaxyblock, 2021-08-06:
Comment for the bot 😉

ornago, 2021-10-09:
Not really getting what you want to say with your first sentence. Should I pair the device in a different way or configure it with cluster settings?

Here is the cluster signature:
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4398, maximum_buffer_size=89, maximum_incoming_transfer_size=61, server_mask=0, maximum_outgoing_transfer_size=61, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "10": {
      "profile_id": 49246,
      "device_type": "0x0810",
      "in_clusters": [
        "0x0000",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "18": {
      "profile_id": 49246,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "Busch-Jaeger",
  "model": "RM01",
  "class": "zigpy.device.Device"
}`

Also tried to reconnect the device, but I am not sure if the device is doing anything right now. Normally I have to press both buttons to get it into programming mode. After blinking alternately the LEDs will be illuminated. Stop pressing the rocker and briefly press both sides of it again. Unfortunately can't reproduce this anymore. Hope I didn't broke that thing?!

No matter what, here ist the log output while trying to connect:
`[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x7577:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=71 command_id=Command.Report_Attributes>
[0x7577:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=10552>)]]
[0x7577:2:0x0400] Attribute report received: measured_value=10552
[0xc0c3:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=171 command_id=Command.Report_Attributes>
[0xc0c3:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.1>)]]
[0xc0c3:2:0x0406] Attribute report received: occupancy=1
[0xc0c3:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=172 command_id=Command.Report_Attributes>
[0xc0c3:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=7325>)]]
[0xc0c3:2:0x0400] Attribute report received: measured_value=7325
[0xBED8:1:0x0b04]: async_update
[0xbed8:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=79 command_id=Command.Read_Attributes_rsp>
[0xc0c3:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=173 command_id=Command.Report_Attributes>
[0xc0c3:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.0>)]]
[0xc0c3:2:0x0406] Attribute report received: occupancy=0
[0xc0c3:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=175 command_id=Command.Report_Attributes>
[0xc0c3:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.1>)]]
[0xc0c3:2:0x0406] Attribute report received: occupancy=1
[0xc0c3:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=176 command_id=Command.Report_Attributes>
[0xc0c3:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=5986>)]]
[0xc0c3:2:0x0400] Attribute report received: measured_value=5986
[0xc0c3:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=177 command_id=Command.Report_Attributes>
[0xc0c3:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.0>)]]
[0xc0c3:2:0x0406] Attribute report received: occupancy=0
[... 9 lines cut ...]
[0xc0c3:2:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=178 command_id=Command.Report_Attribut [... cut ...]
[0xbed8:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=96 command_id=Command.Read_Attributes_rsp>
[0xc0c3:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=184 command_id=Command.Report_Attributes>
[0xc0c3:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.0>)]]
[0xc0c3:2:0x0406] Attribute report received: occupancy=0`

MattWestb, 2021-10-09:
Now its possible reading nicely :-))

But only attribute reports from your sensors in the log.

What i was saying is that Zigbee light swishes can configuring one light group then there is being reseted and joined one network.
ZHA can reading this group then its pairing the device and adding it as one group with one strange name and the switch is sending all commands to that group.
They is looking like that:
```
Group Information
Name: No name group 0x9CED

Group Id: 0x9ced
```
Some light  switches is not doing it at all and other is making it wrong or using the same group for all switches of the same model so it not working with more switches.

If you is having one light group like this you shall putting you lights in it and the switch is controlling them and the status of the group is the status of the lights in the group (switches dont have any status they is only sending commands)

ornago, 2021-10-10:
I am not sure if you know this 6735-84 (84 is just the color code) device. It is sitting on a 6715 U device in the wall outlet. I don't know exactly but i think the rocker of 6735 is controlling the light of 6715 directly? Or is it in one zigbee group you described? Maybe i have to get the light controller (6715) and the rocker of the switch (6735) into one group to see a state change in my Home-Assistant App?

Got my switch working again and made a new connection to my zigbee network.
This is the log dump:
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0xa3c5:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=138 command_id=Command.Report_Attributes>
[0xa3c5:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.1>)]]
[0xa3c5:2:0x0406] Attribute report received: occupancy=1
[0xa3c5:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=139 command_id=Command.Report_Attributes>
[0xa3c5:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=5337>)]]
[0xa3c5:2:0x0400] Attribute report received: measured_value=5337
[0xBED8:1:0x0b04]: async_update
[0xbed8:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=229 command_id=Command.Read_Attributes_rsp>
[0xa3c5:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=140 command_id=Command.Report_Attributes>
[0xa3c5:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.0>)]]
[0xa3c5:2:0x0406] Attribute report received: occupancy=0
[0x3e82:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, ec:1b:bd:ff:fe:93:ed:67, 0x3E82, None, None, []]
[0x3e82:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x3e82:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, ec:1b:bd:ff:fe:93:ed:67, 0x3E82, None, None, []]
[0x3e82:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
[0x7577:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=73 command_id=Command.Report_Attributes>
[0x7577:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.1>)]]
[0x7577:2:0x0406] Attribute report received: occupancy=1
[0x7577:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=74 command_id=Command.Report_Attributes>
[0x7577:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=12464>)]]
[0x7577:2:0x0400] Attribute report received: measured_value=12464
Device 0x6384 (d8:5d:ef:11:a1:00:46:30) joined the network
Device d8:5d:ef:11:a1:00:46:30 changed id (0x6384 => 0x6384)
[0x6384] Skipping initialization, device is fully initialized
Device is initialized <Device model='RM01' manuf='Busch-Jaeger' nwk=0x6384 ieee=d8:5d:ef:11:a1:00:46:30 is_initialized=True>
Checking quirks for Busch-Jaeger RM01 (d8:5d:ef:11:a1:00:46:30)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {18, 10}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {18, 10}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {18, 10}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {18, 10}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {18, 10}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {18, 10}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {18, 10}
[0x6384:zdo] ZDO request ZDOCmd.Device_annce: [0x6384, d8:5d:ef:11:a1:00:46:30, 142]
device - 25476:d8:5d:ef:11:a1:00:46:30 entering async_device_initialized - is_new_join: False
device - 25476:d8:5d:ef:11:a1:00:46:30 has been reset and re-added or its nwk address changed
skipping discovery for previously discovered device - 25476:d8:5d:ef:11:a1:00:46:30
[25476](RM01): started configuration
[25476:ZDO](RM01): 'async_configure' stage succeeded
[25476:10:0x0000]: finished channel configuration
[0x6384] Delivery error for seq # 0xec, on endpoint id 0 cluster 0x0021: message send failure
[25476:10:0x0008]: Failed to bind 'level' cluster: [0x6384:0:0x0021]: Message send failure
[25476:10:0x0008]: finished channel configuration
[25476:10:0x0019]: finished channel configuration
[0x6384] Delivery error for seq # 0xee, on endpoint id 0 cluster 0x0021: message send failure
[25476:10:0x0006]: Failed to bind 'on_off' cluster: [0x6384:0:0x0021]: Message send failure
[25476:10:0x0006]: finished channel configuration
[... cut ...]
[0xbed8:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=13 command_id=Command.Read_Attributes_rsp>
[0x7577:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=78 command_id=Command.Report_Attributes>
[0x7577:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=11287>)]]
[0x7577:2:0x0400] Attribute report received: measured_value=11287
```

MattWestb, 2021-10-10:
In ZHA is light switches is one relay that is switching on / off (perhaps dimming) one light (or somthing else) and you is getting one light entry in ZHA.

The "controller" (some saying remote) that is switching the light on/off is normally controlling the relay direct and do not getting and entry in ZHA if its main powered (if battery you is getting one battery entry).

It can being that the controller is not sending and commands to other devices and only working local or its also sending the commands (on/of dim up/down) for steering light to one group that is random after reset (then ZHA is reading it and adding it in groups) or if its possible group that is added to the device with zigbee commands (if the device is supporting it).

Now to your device its have 2 endpoints:
EP 10 = "device_type": "0x0810", = COLOR_SCENE_CONTROLLER
And **out** cluster "0x0004", is the group cluster.

EP 18 = "device_type": = "0x0100", DIMMABLE_LIGHT
And **in** cluster "0x0004", is the group cluster.

So you can adding groups to the light part on EP 10 on in cluster 0x0004 (is normally being made in the group panele in ZHA).

The controller part is EP 18 with in group cluster 0x0004  you is normally binding the controller to one group so the controller is sending all light commands to groups.
Normal you need binding (made from the device card manage cluster) on/off (0x0006) and light level cluster (0x0008) and also group cluster (0x0004) (option color cluster 0x0300) to one group you have making and adding light to it.

The work flow is making one group if not having one and adding lights to the group.
Binding the controller to the group (out cluster on EP 10) so it sending commands to the lights in the group.
ZHA is getting the status from the light is reporting there status to the coordinator.
The controller is only sending the commands to the group and the coordinator can catching them then listening to the group and you can see them in events log under `zha_evens` and using them in automatons.

The binding is making the light and controller is working autonomous also then the coordinator / host system is not online.

ornago, 2021-10-15:
Hey, thanks for your long post. I did try to do your work flow but didn't got any further. I do have some groups in the ZHA Configuration Tab. One is called 'Default Lightlink Group', so I added the deconz stick and the devices to it. I also made a new group for each switch in combination with deconz.
In the device configuration i tried to bind the EP 10 clusters to the group and also made a 'device reconfiguration'
[image]
[image]
But still no state change in the frontend while triggering with the rocker.
I do get a change of state when asking an cluster attribute.
[image]
What am I missing?

docgalaxyblock, 2021-10-16:
@ornago Regarding the incorrect report to ZHA that the switch is a dimmable light I found a solution.
 https://www.home-assistant.io/integrations/zha#modifying-the-device-type
 
 And yes my one relais (bathroom vent) does not report the state back...

ornago, 2021-10-20:
Thanks for your information, I'm not sure if the naming is wrong. The endpoint 10 is the rocker or you could call it a switch. It's naming is 'COLOR_SCENE_CONTROLLER'. There is nothing with color changing but the rest fits.
And then there is the endpoint 18 'DIMMABLE_LIGHT' which is the electronical device dimming my light bulb.
Can't say if a naming would help, but please tell me if it does.

OngiiDE, 2022-02-08:
I am also desperately needing ZHA support to the Busch 673x family.
How can I support?

docgalaxyblock, 2022-08-07:
.

spabinger, 2022-09-01:
Hi,

thanks for creating the ticket. I'm facing the same issue with my Busch-Jaeger 6737, which allows me to control the associated light but I cannot assign automations or trigger events to the remaining 6 buttons.

It would be great if somebody could post a solution.

Thanks,
Stephan

Dabbadoeber, 2022-12-16:
2 years later........... after leaving deconz for Z2M and not happy with the result of Z2M, i wanted to switch to ZHA but unfortunately the switch RB01 is not supported (yes the top rocker works but the bottom one does not)
is there anybody how got this  Busch-Jaeger stuff working in ZHA ?

lepatrick, 2023-02-14:
Still no support for this switches?

aitchofman, 2023-02-26:
Hi, how can I help make this device work with ZHA? I have got an updated 6737/01-8 on firmware 1.2 that works on all 8 buttons with deconz. I can post the DDF file here - will it help?

weakfl, 2023-05-06:
I've stumbled upon this because I'm thinking about trying out HA (currently using OpenHAB), but wanted to make sure that my BJ dimmers/relays work before even giving it a try. So maybe I can shed some light on some of the issues described here.

> @gclever but status is not reported back

Unfortunately the device completely ignores whatever reporting configuration you set and does not report changes of the relay/dimmer state. It only reports like once an hour.

I've mitigated this issue by setting a polling period of 60 seconds, so you have to rely on polling to get state updates in reasonable time.

**Is it possible to configure the polling interval in HA?**

> @gclever but have no access to the lower rockers

The binding table of these devices has only room for four entries (I only have the 2-rocker variants, maybe the other devices have more).

If the endpoints are not consolidated properly the rockers won't work. You might end up with a binding table that's missing one of the endpoints:

```
> bindtable D85DEF11A1003142
Binding table for node 37485 [D85DEF11A1003142]
Src Address          | Dest Address         | Group | Mode    | Cluster
D85DEF11A1003142/18  | 0022A300002921FD/1   |       | Address | 0006:ON_OFF
D85DEF11A1003142/10  | 0022A300002921FD/1   |       | Address | 0008:LEVEL_CONTROL
D85DEF11A1003142/10  | 0022A300002921FD/1   |       | Address | 0006:ON_OFF
D85DEF11A1003142/18  | 0022A300002921FD/1   |       | Address | 0008:LEVEL_CONTROL
```

Here it's missing endpoint for the 2nd rocker, which would be 11:

```
node D85DEF11A1003142
IEEE Address     : D85DEF11A1003142
Network Address  : 37485
Node Descriptor  : NodeDescriptor [apsFlags=0, bufferSize=89, complexDescriptorAvailable=false, manufacturerCode=4398, logicalType=ROUTER, serverCapabilities=[], incomingTransferSize=61, outgoingTransferSize=61, userDescriptorAvailable=true, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[MAINS_POWER, RECEIVER_ON_WHEN_IDLE, FULL_FUNCTION_DEVICE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=0]
Power Descriptor : RECEIVER_ON_IDLE, [DISPOSABLE_BATTERY, MAINS, RECHARGABLE_BATTERY], MAINS, FULL
Associations     : []
Endpoints        : 
            10   : Profile     C05E ZIGBEE_LIGHT_LINK
                 : Device Type 0000 ON_OFF_SWITCH
                   -> BASIC
                   -> TOUCHLINK
                   <- TOUCHLINK
                   <- COLOR_CONTROL
                   <- IDENTIFY
                   <- GROUPS
                   <- SCENES
                   <- ON_OFF
                   <- LEVEL_CONTROL
                   <- OTA_UPGRADE
            11   : Profile     C05E ZIGBEE_LIGHT_LINK
                 : Device Type 0000 ON_OFF_SWITCH
                   -> BASIC
                   -> TOUCHLINK
                   <- TOUCHLINK
                   <- COLOR_CONTROL
                   <- IDENTIFY
                   <- GROUPS
                   <- SCENES
                   <- ON_OFF
                   <- LEVEL_CONTROL
            18   : Profile     C05E ZIGBEE_LIGHT_LINK
                 : Device Type 0000 ON_OFF_SWITCH
                   -> BASIC
                   -> IDENTIFY
                   -> GROUPS
                   -> SCENES
                   -> ON_OFF
                   -> LEVEL_CONTROL
```                   

So you have to manually unbind endpoint 10 and bind endpoint 11 to make it work.

I obviously have no idea if that's the problem here, but it might be worth checking out.

E-ASa, 2023-08-14:
I also have the same issue. I moved from an old Rasbee / Deconz setup to Home Assistant Yellow and the integrated Zigbee module with ZHA. I didn't realize that my Busch & Jaeger switches would stop working. Any help in getting these switches working in ZHA would be greatly appreciated!

JohnDoep, 2023-11-04:
I also had the problem of getting the Busch-Jaeger 2gang working with ZHA (which was working perfectly before with Deconz). I fiddled around with a quirk I found for another switch. And after some back and forth, I finally managed to get the switch working with ZHA. Using the quirk appended to this message, I can join the device and get events for all four buttons. Probably the quirk can be adapted to the 4gang version quite easily. Since I had not worked with quirks before and had no time to dig deeper into the topic, my implementation might be mediocre. But I hope it might be a good starting point for someone who knows how to do it properly (or someone who desperately needs support for the switch).

```"""Busch-Jaeger RB01 Remote"""

from zigpy.profiles import zha, zll
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, OnOff, Identify, Ota, LevelControl, PowerConfiguration, Time, Groups, Scenes
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    MODEL,
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    BUTTON_4,
    ARGS,
    COMMAND,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    CLUSTER_ID,
    LONG_PRESS,
    LONG_RELEASE,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
    COMMAND_ON,
    COMMAND_OFF,
    COMMAND_STEP,
    COMMAND_MOVE,
    COMMAND_STOP,
    COMMAND_STEP_ON_OFF
)

class BJRB01ButtonTriggers:
    """Busch-Jaeger RB01 4-button triggers."""

    device_automation_triggers = {
            (SHORT_PRESS, BUTTON_1): {COMMAND: COMMAND_OFF, CLUSTER_ID: 6, ENDPOINT_ID: 10, ARGS: []},
            (SHORT_PRESS, BUTTON_2): {COMMAND: COMMAND_ON, CLUSTER_ID: 6, ENDPOINT_ID: 10, ARGS: []},
            (SHORT_PRESS, BUTTON_3): {COMMAND: COMMAND_OFF, CLUSTER_ID: 6, ENDPOINT_ID: 11, ARGS: []},
            (SHORT_PRESS, BUTTON_4): {COMMAND: COMMAND_ON, CLUSTER_ID: 6, ENDPOINT_ID: 11, ARGS: []},
            (LONG_PRESS, BUTTON_2): {
                COMMAND: COMMAND_STEP_ON_OFF,
                CLUSTER_ID: 8,
                ENDPOINT_ID: 10,
                ARGS: [0, 254, 50],
            },		
            (LONG_RELEASE, BUTTON_2): {COMMAND: COMMAND_STOP, CLUSTER_ID: 8, ENDPOINT_ID: 10, ARGS: []},       
    }

class BJRB01Remote(CustomDevice, BJRB01ButtonTriggers):
    """Busch-Jaeger 4-button remote device."""

    signature = {
        MODELS_INFO: [("Busch-Jaeger", "RB01")],
        ENDPOINTS: {
            10: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.COLOR_SCENE_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                ],
            },
            11: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.COLOR_SCENE_CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                ],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            10: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                ],
            },
            11: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.CONTROLLER,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                ],
            },
        },
    }
```

Dabbadoeber, 2023-11-10:
I have same switch's which are working, but I use Zigbee2mqtt, also large networks cause the devices to become unreliable so I am running 3 networks at the moment. I think max number of devices is limited to around 20

Ra72xx, 2023-12-06:
> device_automation_triggers = {
>             (SHORT_PRESS, BUTTON_1): {COMMAND: COMMAND_OFF, CLUSTER_ID: 6, ENDPOINT_ID: 10, ARGS: []},
>             (SHORT_PRESS, BUTTON_2): {COMMAND: COMMAND_ON, [...]

Is there any reason why the long press actions are only defined for button 2?

MattWestb, 2023-12-06:
Look if its sending light level command from EP 11 and if yes you can adding it like its done for EP 10.
Use `zha_event` for listening to events and you can see all commands its sending.

ullmonde, 2023-12-21:
Same here! Please, please add Busch Jaeger Support for Wall switches e.g. RB01 and RM01 | RB01. Thank u so much!!!

Chrissi2812, 2024-01-07:
I have a [dimmer unit (6715 U)](https://www.busch-jaeger.de/online-katalog/detail/2CKA006710A0003) in connection with the [2-way switch (6736-84)](https://www.busch-jaeger.de/online-katalog/detail/2CKA006730A0074)

<details><summary>Device Signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4398, maximum_buffer_size=89, maximum_incoming_transfer_size=61, server_mask=0, maximum_outgoing_transfer_size=61, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "10": {
      "profile_id": "0xc05e",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "11": {
      "profile_id": "0xc05e",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "18": {
      "profile_id": "0xc05e",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Busch-Jaeger",
  "model": "RM01",
  "class": "zigpy.device.Device"
}
```

</details> 

I got the bottom row working with this custom quirk.
<details><summary><b>Custom Quirk</b></summary>

```python
from zigpy.profiles import zha, zll
[... 105 lines cut ...]
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                ],
            }
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_4): {COMMAND: COMMAND_ON, CLUSTER_ID: 6, ENDPOINT_ID: 11, ARGS: []},
        (SHORT_PRESS, BUTTON_3): {COMMAND: COMMAND_OFF, CLUSTER_ID: 6, ENDPOINT_ID: 11, ARGS: []},
    }
```

</details> 

Whats not working:

* Dimmer status is not updated when button on the device is used (Zigbee2Mqtt seems to use this workaround somehow: https://github.com/Koenkk/zigbee-herdsman-converters/blob/22c4673533b9380dc4d8abb115c70069e607ee13/src/devices/busch_jaeger.ts#L124-L135) don't know how to do the same in ZHA.
* Long Press doesn't seem to work for me. I only get `on` & `off` zha_events logged...
* Double press doesn't seem to trigger any zha_event

Ra72xx, 2024-01-08:
> * Double press doesn't seem to trigger any zha_event

Does this device support double presses? I never had that with Deconz, and Z2M does not support them, too.

weakfl, 2024-01-08:
> @Chrissi2812 Dimmer status is not updated when button on the device is used 

As mentioned in the link you shared the device doesn't report state changes. I'm also using a very short polling interval to work a round it (on openHAB).

I've contacted BJ numerous times reporting the issue but they never fixed it. Given how long it's been since the firmware has been updated I'm not expecting them to fix this anymore.

Chrissi2812, 2024-01-09:
I still had the `Scenes.cluster_id` in the replacement part. And that prevented the Long_Press from working. 

```
replacement = {
    ENDPOINTS: {
       11: {
            PROFILE_ID: zll.PROFILE_ID,
            DEVICE_TYPE: zha.DeviceType.COLOR_SCENE_CONTROLLER ,
            INPUT_CLUSTERS: [
                Basic.cluster_id,
                LightLink.cluster_id,
            ],
            OUTPUT_CLUSTERS: [
                Identify.cluster_id,
                Groups.cluster_id,
                # Scenes.cluster_id, # This had to be removed for the Long_Press to work
                OnOff.cluster_id,
                LevelControl.cluster_id,
                Color.cluster_id,
                LightLink.cluster_id,
            ],
        }
    }
}
```

Has anybody an idea how to poll the `OnOff` periodically (Like every 5 seconds or so). I'm not that into Python but want to try to get it working inside the quirk itself and not with an automation in Homeassistant.

rulerise, 2024-01-16:
I am also interested in finding a solution. It is currently happening. The following: If an RM01 is only plugged onto a power supply, it is recognised by the zigpy.appdb as "Considering <class 'BJ-RM01.BJRM01Remote". This "Remote" can then also trigger up to four switches, depending on the version. The 8 switch version then only works with 4 switches. At the moment when a switch or dimmer is used as a wall module, only light is recognised regardless of the control module, whether 2, 4 or 8 switches. The switches that are then controlled similarly to the remote do not work. I have already created a quirk. However, this is not activated because the zigpy.appdb is queried first and apparently has priority over the quirks. I am happy to contribute something for everyone but need the support of a "sighted" person.  Please let me know how I can help.

theincogtion, 2024-02-19:
> I am not sure if you know this 6735-84 (84 is just the color code) device. It is sitting on a 6715 U device in the wall outlet. I don't know exactly but i think the rocker of 6735 is controlling the light of 6715 directly? Or is it in one zigbee group you described? Maybe i have to get the light controller (6715) and the rocker of the switch (6735) into one group to see a state change in my Home-Assistant App? Unfortunately I can only see and connect the rocker into my network. The light is an entity of that switch device.
> 
> Got my switch working again and made a new connection to my zigbee network. This is the log dump:
> 
> ```
> [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
> [0xa3c5:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=138 command_id=Command.Report_Attributes>
> [0xa3c5:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.1>)]]
> [0xa3c5:2:0x0406] Attribute report received: occupancy=1
> [0xa3c5:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=139 command_id=Command.Report_Attributes>
> [0xa3c5:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=5337>)]]
> [0xa3c5:2:0x0400] Attribute report received: measured_value=5337
> [0xBED8:1:0x0b04]: async_update
> [0xbed8:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=229 command_id=Command.Read_Attributes_rsp>
> [0xa3c5:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=140 command_id=Command.Report_Attributes>
> [0xa3c5:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.0>)]]
> [0xa3c5:2:0x0406] Attribute report received: occupancy=0
> [0x3e82:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, ec:1b:bd:ff:fe:93:ed:67, 0x3E82, None, None, []]
> [0x3e82:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
> [0x3e82:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, ec:1b:bd:ff:fe:93:ed:67, 0x3E82, None, None, []]
> [0x3e82:zdo] Unsupported ZDO request:ZDOCmd.IEEE_addr_rsp
> [0x7577:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=73 command_id=Command.Report_Attributes>
> [0x7577:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.1>)]]
> [0x7577:2:0x0406] Attribute report received: occupancy=1
> [0x7577:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=74 command_id=Command.Report_Attributes>
> [0x7577:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=12464>)]]
> [0x7577:2:0x0400] Attribute report received: measured_value=12464
> Device 0x6384 (d8:5d:ef:11:a1:00:46:30) joined the network
> Device d8:5d:ef:11:a1:00:46:30 changed id (0x6384 => 0x6384)
> [0x6384] Skipping initialization, device is fully initialized
> Device is initialized <Device model='RM01' manuf='Busch-Jaeger' nwk=0x6384 ieee=d8:5d:ef:11:a1:00:46:30 is_initialized=True>
> Checking quirks for Busch-Jaeger RM01 (d8:5d:ef:11:a1:00:46:30)
> Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
> Fail because endpoint list mismatch: {232, 230} {18, 10}
> Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
> Fail because endpoint list mismatch: {232, 230} {18, 10}
> Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
> Fail because endpoint list mismatch: {1} {18, 10}
> Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
> Fail because endpoint list mismatch: {1} {18, 10}
> Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
> Fail because endpoint list mismatch: {1} {18, 10}
> Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
> Fail because endpoint list mismatch: {11, 13} {18, 10}
> Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
> Fail because endpoint list mismatch: {1} {18, 10}
> [0x6384:zdo] ZDO request ZDOCmd.Device_annce: [0x6384, d8:5d:ef:11:a1:00:46:30, 142]
> device - 25476:d8:5d:ef:11:a1:00:46:30 entering async_device_initialized - is_new_join: False
> device - 25476:d8:5d:ef:11:a1:00:46:30 has been reset and re-added or its nwk address changed
> skipping discovery for previously discovered device - 25476:d8:5d:ef:11:a1:00:46:30
> [25476](RM01): started configuration
> [25476:ZDO](RM01): 'async_configure' stage succeeded
> [25476:10:0x0000]: finished channel configuration
> [0x6384] Delivery error for seq # 0xec, on endpoint id 0 cluster 0x0021: message send failure
> [25476:10:0x0008]: Failed to bind 'level' cluster: [0x6384:0:0x0021]: Message send failure
> [25476:10:0x0008]: finished channel configuration
> [25476:10:0x0019]: finished channel configuration
[... cut ...] p>
> [0x7577:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=78 command_id=Command.Report_Attributes>
> [0x7577:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=11287>)]]
> [0x7577:2:0x0400] Attribute report received: measured_value=11287
> ```

I am thinking of buying a 6715U. Did you already got it to work with ZHA?

TeeJayR, 2024-05-21:
I have the 8-button control unit (6737) mounted both to a switch (6711 U) and a dimmer (6715 U), respectively. Both have different signatures of course:
<details><summary>Signature 6737 on 6711 U</summary>
<p>
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4398, maximum_buffer_size=89, maximum_incoming_transfer_size=61, server_mask=0, maximum_outgoing_transfer_size=61, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "10": {
      "profile_id": "0xc05e",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "11": {
      "profile_id": "0xc05e",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "12": {
      "profile_id": "0xc05e",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "13": {
      "profile_id": "0xc05e",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
[... 355 lines cut ...]
            CLUSTER_ID: 5,
            ENDPOINT_ID: 12,
            ARGS: [2, 6],
        },
        (SHORT_PRESS, BUTTON_5): {
            COMMAND: COMMAND_RECALL,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 13,
            ARGS: [2, 7],
        },
        (SHORT_PRESS, BUTTON_6): {
            COMMAND: COMMAND_RECALL,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 13,
            ARGS: [2, 8],
        },
    }
</p>
</details> 

Of course I could simply add a custom quirk per unit with matching group IDs, respectively, but this would not be anything I could contribute to the repo so it would directly work for others (with potentially different group IDs).

I see two directions from here to explore:
1. Write some custom logic in the quirks to drop the group ID on the events. Not sure if that is even possible though.
2. Drop the group assignment from the devices. But so far all my attempts to do this failed. Does anyone have an idea how to remove the assigned group from the devices?

TeeJayR, 2024-05-26:
The arguments of the recall events (and probably even the whole events themself) of the two bottom rows seem to be controlled by the "Scenes" cluster, since when I comment out the cluster in all output clusters of the endpoints in the replacement there are no events sent anymore at all.

TeeJayR, 2024-05-29:
Diving deeper into the whole ZigBee protocol specification topic things start to make sense somehow. From the [ZigBee Cluster Library Specification](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf):

> In most cases scenes are associated with a particular group ID. Scenes MAY also exist without a group, in which case the value 0x0000 replaces the group ID. Note that extra care is required in these cases to avoid a scene ID collision, and that commands related to scenes without a group MAY only be unicast, i.e., they MAY not be multicast or broadcast.

However, the value of this associated group ID seems not to be controllable really. While I would have assumed that saving a new scene directly through the device's programming mode would also assign a new group ID, I found that in my current setup I cannot program the device like described in the [manual](https://library.e.abb.com/public/2407dc82ba5d494989da69d06514dab8/MBA_2CKA000073B9204_6735-36-37.pdf) (section 2) at all. When long-pressing a button to enter programming mode (required pushing for at least 15 seconds) the button sends a `stop` event after ~10 seconds and that is the end of the story - no entering of the programming mode possible.

I start to believe that the `device_automation_triggers` section of the custom quirk would need a wildcard value definition for some of the arguments. This way one could match a trigger with any group ID but only a particular scene ID.

closed 2024-12-02 by github-actions[bot]

Wyphorn, 2024-12-31:
Reopen please.

ullmonde, 2025-01-02:
> Reopen please.

Same here, these devices are in use! Please reopen.

Ypsilon88, 2026-10-09:
## ✅ Busch-Jaeger RB01: both rows working in ZHA – no quirk needed

> [!TIP]
> **TL;DR:** The RB01's binding table has only **4 slots**, and after pairing endpoint 10 (top row) occupies all of them. Unbind *Scenes* from EP 10, bind again → endpoint 11 (bottom row) gets a slot and fires `zha_event`s.

**Tested with:** Busch-Jaeger **RB01** (battery radio transmitter, 2 rows) · Home Assistant 2026.10 · zha 2.3.0 / zigpy 2.3.0 · SONOFF ZBDongle-E (EZSP) · [zha_toolkit](https://github.com/mdeweerd/zha-toolkit)

---

### 🔍 Root cause

Only the top row (EP 10) emits events. Binding EP 11 is acknowledged by the device but never stored – reading the binding table shows why: it's full.

| src_ep | cluster | after pairing | after fix | needed for `zha_event` |
|:---:|---|:---:|:---:|:---:|
| 10 | `0x0006` OnOff | ✅ | ✅ | ✅ short press |
| 10 | `0x0008` LevelControl | ✅ | ✅ | ✅ long press |
| 10 | `0x0005` Scenes | ✅ | ➖ removed | ❌ |
| 10 | `0x0300` Color | ✅ | ✅ | ❌ |
| **11** | **`0x0006` OnOff** | – | ✅ 🎉 | ✅ short press |

---

### 🛠️ Fix

> [!IMPORTANT]
> The RB01 sleeps almost immediately. Hold **both top-row buttons** until the LEDs stay lit (config mode), then press a **bottom-row** button every 2–3 s while running the steps below.

Placeholders: `xx:…` = IEEE of your RB01, `yy:…` = IEEE of your coordinator.

#### 1️⃣ Read the binding table

```yaml
action: zha_toolkit.binds_get
data:
  ieee: "xx:xx:xx:xx:xx:xx:xx:xx"   # RB01
  tries: 10
```

#### 2️⃣ Free a slot (unbind Scenes / Color from EP 10)

```yaml
action: zha_toolkit.unbind_coordinator
data:
  ieee: "xx:xx:xx:xx:xx:xx:xx:xx"   # RB01
  endpoint: 10
  cluster: [5, 768]   # 0x0005 Scenes, 0x0300 Color
  tries: 20
```

In my case only Scenes was removed, Color stayed. One free slot is enough.

#### 3️⃣ Bind to the coordinator again

> [!NOTE]
> `zha_toolkit.bind_ieee` with `endpoint: 11` did **nothing** for me (returned instantly, no bind attempted). ZHA's own bind command works.

Run this in the browser console (F12) on your HA page:

```js
document.querySelector("home-assistant").hass.callWS({
  type: "zha/devices/bind",
  source_ieee: "xx:xx:xx:xx:xx:xx:xx:xx", // RB01
  target_ieee: "yy:yy:yy:yy:yy:yy:yy:yy"  // coordinator
}).then(() => console.log("bind sent"));
```

#### 4️⃣ Verify

Run `binds_get` (step 1) again – EP 11 / `0x0006` should now be listed (see table above). The bottom row fires `zha_event` with `endpoint_id: 11` and `command: on` / `off`.

---

### ⚠️ Gotcha: every press is sent twice

> [!WARNING]
> The RB01 delivers each command **twice, ~30 ms apart**. With `toggle` actions the light goes on and immediately off again. Use `mode: single` plus a short delay to swallow the duplicate, and filter by `endpoint_id`.

<details>
<summary><b>📋 Example automation (click to expand)</b></summary>

```yaml
alias: BJ RB01 buttons
mode: single
max_exceeded: silent
triggers:
  - trigger: event
    event_type: zha_event
    event_data:
      device_ieee: "xx:xx:xx:xx:xx:xx:xx:xx"
      endpoint_id: 10
      command: "off"
    id: top_left
  - trigger: event
    event_type: zha_event
    event_data:
      device_ieee: "xx:xx:xx:xx:xx:xx:xx:xx"
      endpoint_id: 10
      command: "on"
    id: top_right
  - trigger: event
    event_type: zha_event
    event_data:
      device_ieee: "xx:xx:xx:xx:xx:xx:xx:xx"
      endpoint_id: 11
      command: "off"
    id: bottom_left
  - trigger: event
    event_type: zha_event
    event_data:
      device_ieee: "xx:xx:xx:xx:xx:xx:xx:xx"
      endpoint_id: 11
      command: "on"
    id: bottom_right
actions:
  - choose:
      - conditions:
          - condition: trigger
            id: top_left
        sequence:
          - action: light.toggle
            target:
              entity_id: light.your_light_1
      - conditions:
          - condition: trigger
            id: top_right
        sequence:
          - action: light.toggle
            target:
              entity_id: light.your_light_2
      - conditions:
          - condition: trigger
            id: bottom_left
        sequence:
          - action: light.toggle
            target:
              entity_id: light.your_light_3
      - conditions:
          - condition: trigger
            id: bottom_right
        sequence:
          - action: light.toggle
            target:
              entity_id: light.your_light_4
  - delay:
      milliseconds: 700   # swallows the duplicate
```

</details>

---

### 📌 Limitations

- **Only 4 slots:** OnOff + LevelControl for both rows *and* Color don't fit together. For long-press dimming on row 2, also unbind Color (`0x0300`) on EP 10 and bind EP 11 LevelControl instead.
- **4-gang (6737):** the same approach should work, but 4 rows × OnOff already fill the table, so long press won't fit. *Not tested.*