zigpy/zha-device-handlers #613: [Device Support Request] LIVARNO LUX/LIDL Remote Control Dimmer   TS1001 (Tuya based)

docgalaxyblock, 2020-12-03:
https://zigbee.blakadder.com/Lidl_HG06323.html
Already supported at Z2M: https://www.zigbee2mqtt.io/devices/FB20-002.html

**Describe the solution you'd like**
- [ ] Automations wizard button support
- [ ] Battery support

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
[Device signature and pairing log at my dev VM](https://pastebin.com/CRTbeguc)

**Additional context**
I am nut shure but it could be that Z2M pools the state of the switch because if I use the device buttons it recognizes the change 5s later

Adminiuga, 2020-12-04:
Signature

```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TYZB01_bngwdjsr",
  "model": "TS1001",
  "class": "zigpy.device.Device"
}
```

Adminiuga, 2020-12-04:
Log excerpt
```
Device 0xcf04 (bc:33:ac:ff:fe:1a:85:80) joined the network
[0xcf04] Requesting 'Node Descriptor'
[0xcf04] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
[0xcf04] Discovering endpoints
[0xcf04] Discovered endpoints: [1]
[0xcf04:1] Discovering endpoint information
Ignoring message (b'01350201330a00') on cluster 8: unknown endpoint or cluster id: 'No cluster ID 0x0008 on (bc:33:ac:ff:fe:1a:85:80, 1)'
Ignoring message (b'01360201330a00') on cluster 8: unknown endpoint or cluster id: 'No cluster ID 0x0008 on (bc:33:ac:ff:fe:1a:85:80, 1)'
Ignoring message (b'01370201330a00') on cluster 8: unknown endpoint or cluster id: 'No cluster ID 0x0008 on (bc:33:ac:ff:fe:1a:85:80, 1)'
[0xcf04:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=260, device_version=1, input_clusters=[0, 1, 3, 4, 4096], output_clusters=[25, 10, 3, 4, 5, 6, 8, 4096])
[0xcf04:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=12 command_id=Command.Read_Attributes_rsp>
[0xcf04:1] Manufacturer: _TYZB01_bngwdjsr
[0xcf04:1] Model: TS1001
Checking quirks for _TYZB01_bngwdjsr TS1001 (bc:33:ac:ff:fe:1a:85:80)
...
device - 0xCF04:bc:33:ac:ff:fe:1a:85:80 entering async_device_initialized - is_new_join: True
device - 0xCF04:bc:33:ac:ff:fe:1a:85:80 has joined the ZHA zigbee network
[0xCF04](TS1001): started configuration
[0xCF04:ZDO](TS1001): 'async_configure' stage succeeded
[0xCF04:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xCF04:1:0x0000]: finished channel configuration
[0xCF04:1:0x1000]: bound 'lightlink' cluster: Status.SUCCESS
[0xCF04:1:0x1000]: finished channel configuration
[0xcf04] Delivery error for seq # 0x10, on endpoint id 1 cluster 0x0000: Invalid response status
[0xCF04:1:0x0000]: initializing channel: from_cache: False
[0xcf04:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=56 command_id=2>
[0xcf04:1:0x0008] ZCL request 0x0002: [1, 51, 10]
[0xcf04:1:0x0008] No handler for cluster command 2
[0xCF04:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0xCF04:1:0x0008]: bound 'level' cluster: Status.SUCCESS
[0xCF04:1:0x0008]: finished channel configuration
[0xCF04:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xCF04:1:0x0006]: finished channel configuration
[0xCF04:1:0x0001]: failed to set reporting for 'battery_voltage' attr on 'power' cluster: 
[0xcf04] Delivery error for seq # 0x16, on endpoint id 1 cluster 0x0001: Invalid response status
[0xCF04:1:0x0001]: failed to set reporting for 'battery_percentage_remaining' attr on 'power' cluster: [0xcf04:1:0x0001]: Message send failure
[0xCF04:1:0x0001]: finished channel configuration
[0xcf04:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=57 command_id=2>
[0xcf04:1:0x0008] ZCL request 0x0002: [1, 51, 10]
[0xcf04:1:0x0008] No handler for cluster command 2
[0xCF04:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0xCF04:1:0x0019]: finished channel configuration
[0xCF04:1:0x0005]: Failed to bind 'scenes' cluster: 
[0xCF04:1:0x0005]: finished channel configuration
[0xCF04:1:0x0000]: 'async_configure' stage succeeded
[0xCF04:1:0x1000]: 'async_configure' stage succeeded
[0xCF04:1:0x0001]: 'async_configure' stage succeeded
[0xCF04:1:0x0008]: 'async_configure' stage succeeded
[0xCF04:1:0x0006]: 'async_configure' stage succeeded
[0xCF04:1:0x0019]: 'async_configure' stage succeeded
[0xCF04:1:0x0005]: 'async_configure' stage succeeded
[0xCF04](TS1001): completed configuration
[0xCF04](TS1001): stored in registry: ZhaDeviceEntry(name='_TYZB01_bngwdjsr TS1001', ieee='bc:33:ac:ff:fe:1a:85:80', last_seen=1607034537.9454782)
[0xcf04:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=58 command_id=2>
[0xcf04:1:0x0008] ZCL request 0x0002: [1, 51, 10]
[0xcf04:1:0x0008] No handler for cluster command 2
[0xcf04:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=23 command_id=Command.Default_Response>
[0xCF04:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.UNSUP_CLUSTER_COMMAND: 129>]
[0xCF04](TS1001): started initialization
[0xCF04:ZDO](TS1001): 'async_initialize' stage succeeded
[0xCF04:1:0x1000]: initializing channel: from_cache: False
[0xCF04:1:0x0008]: initializing channel: from_cache: False
[0xCF04:1:0x0006]: initializing channel: from_cache: False
[0xCF04:1:0x0019]: initializing channel: from_cache: False
[0xCF04:1:0x0005]: initializing channel: from_cache: False
[0xcf04:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=24 command_id=Command.Read_Attributes_rsp>
[... cut ...]
[0xCF04:1:0x0001]: failed to get attributes '['battery_voltage', 'battery_percentage_remaining']' on 'power' cluster: 
[0xCF04:1:0x0000]: 'async_initialize' stage succeeded
[0xCF04:1:0x1000]: 'async_initialize' stage succeeded
[0xCF04:1:0x0001]: 'async_initialize' stage succeeded
[0xCF04:1:0x0008]: 'async_initialize' stage succeeded
[0xCF04:1:0x0006]: 'async_initialize' stage succeeded
[0xCF04:1:0x0019]: 'async_initialize' stage succeeded
[0xCF04:1:0x0005]: 'async_initialize' stage succeeded
[0xCF04](TS1001): power source: Battery or Unknown
[0xCF04](TS1001): completed initialization
```

Adminiuga, 2020-12-04:
enable debug logging as per zha docs (cause we need to see the payload too) and check if you get anything from the device (0xFC04) when you press the buttons on the remote.
If you do, post a log here and annotate each section with what button was pressed and was it a short or long press

docgalaxyblock, 2020-12-04:
@Adminiuga I have updated the paste, hope that is now all you need
Should I update #610 also?
And should I do that also with the motion sensor and led panel? (issues not created yet)

docgalaxyblock, 2020-12-11:
@Adminiuga Sorry I had a typo at the znp debug logger so it was off...
now it is corrected and the pastebin has the new log

gijsje, 2020-12-11:
Maybe this info helps out
This happens in the zha_event.
If i press the off button.
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "bc:33:ac:ff:fe:1a:81:39",
        "unique_id": "bc:33:ac:ff:fe:1a:81:39:1:0x0006",
        "device_id": "d20769c3dddb277b26b07008823082f8",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "off",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-12-11T22:16:26.333459+00:00",
    "context": {
        "id": "595f4d4519e64170ce7d0a62b6aeadd0",
        "parent_id": null,
        "user_id": null
    }
}
```

If i press the on button
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "bc:33:ac:ff:fe:1a:81:39",
        "unique_id": "bc:33:ac:ff:fe:1a:81:39:1:0x0006",
        "device_id": "d20769c3dddb277b26b07008823082f8",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-12-11T22:17:47.635653+00:00",
    "context": {
        "id": "700b160d6ff113af17533e6f3792487c",
        "parent_id": null,
        "user_id": null
    }
}
```
If i press the increas button
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "bc:33:ac:ff:fe:1a:81:39",
        "unique_id": "bc:33:ac:ff:fe:1a:81:39:1:0x0008",
        "device_id": "d20769c3dddb277b26b07008823082f8",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            0,
            51,
            10
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-12-11T22:18:23.175146+00:00",
    "context": {
        "id": "cd23eec38040b768341eb834e0da566e",
        "parent_id": null,
        "user_id": null
    }
}
```
If i press the decrease button
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "bc:33:ac:ff:fe:1a:81:39",
        "unique_id": "bc:33:ac:ff:fe:1a:81:39:1:0x0008",
        "device_id": "d20769c3dddb277b26b07008823082f8",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            1,
            51,
            10
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-12-11T22:18:48.103407+00:00",
    "context": {
        "id": "088d5748c1dfc527291531e5945573e0",
        "parent_id": null,
        "user_id": null
    }
}
```

docgalaxyblock, 2020-12-12:
@gijsje This is already at my pastebin ^^
I screwed the paste a bit up but now the unessecarry part is removed

gadgetchnnel, 2020-12-19:
I have this remote control working in Home Assistant in ZHA, but when pressing the "on" button twice (without pushing "off" button first) no **zha_event** event gets fired and the following appears in the log:

```text
2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=122 command_id=253>
2020-12-19 15:38:32 WARNING (MainThread) [zigpy.zcl] [0x7399:1:0x0006] Unknown cluster-specific command 253
2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] ZCL request 0x00fd: b'\x00'
2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] No handler for cluster command 253
2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=122 command_id=253>
2020-12-19 15:38:32 WARNING (MainThread) [zigpy.zcl] [0x7399:1:0x0006] Unknown cluster-specific command 253
2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] ZCL request 0x00fd: b'\x00'
2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] No handler for cluster command 253
```

mvroosmalen1970, 2021-07-01:
Whats needed to get the device integrated? all info seems readily uploaded or is something additional needed.

mvroosmalen1970, 2021-07-09:
> I have this remote control working in Home Assistant in ZHA, but when pressing the "on" button twice (without pushing "off" button first) no **zha_event** event gets fired and the following appears in the log:
> 
> ```
> 2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=122 command_id=253>
> 2020-12-19 15:38:32 WARNING (MainThread) [zigpy.zcl] [0x7399:1:0x0006] Unknown cluster-specific command 253
> 2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] ZCL request 0x00fd: b'\x00'
> 2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] No handler for cluster command 253
> 2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=122 command_id=253>
> 2020-12-19 15:38:32 WARNING (MainThread) [zigpy.zcl] [0x7399:1:0x0006] Unknown cluster-specific command 253
> 2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] ZCL request 0x00fd: b'\x00'
> 2020-12-19 15:38:32 DEBUG (MainThread) [zigpy.zcl] [0x7399:1:0x0006] No handler for cluster command 253
> ```

Can you share, how you solved this?

MartinKlingeling, 2021-08-08:
I got a LED panel that came with this remote. 

As expected, the remote only reports battery status and that is it. You cannot use the buttons in automations, they are not available for selection. 

The manual says the "on" button also functions as a "scene" switch between some hardcoded light temperatures. Similar to the left/right buttons on a 5-button-IKEA-remote.

So as gadgetchnnel wrote, pressing to "on" button a second time (without pressing "off" first) does not send the "on" command again, which is a bummer with regards to binding an automation to this button, should the remote be supported in the future. But three out of four is not too bad. Or maybe the above mentioned support in z2m has a workaround implemented?

This is what I get for the second "on" press. Looks like the command is fired four times? 

```
2021-08-08 23:24:39 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-08-08 23:24:39 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-08-08 23:24:39 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [27, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7c97>, 1, 260, 6, b'\x01E\xfd\x01', 0, 175, 255, 151, 94, 31, 0, -46]
2021-08-08 23:24:39 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=69 command_id=253>
2021-08-08 23:24:39 WARNING (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] Unknown cluster-specific command 253
2021-08-08 23:24:39 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] ZCL request 0x00fd: b'\x01'
2021-08-08 23:24:39 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] No handler for cluster command 253
2021-08-08 23:24:39 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7c97>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'0145fd01'
2021-08-08 23:24:40 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-08-08 23:24:40 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-08-08 23:24:40 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [27, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7c97>, 1, 260, 6, b'\x01E\xfd\x01', 0, 175, 255, 151, 94, 31, 0, -46]
2021-08-08 23:24:40 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=69 command_id=253>
2021-08-08 23:24:40 WARNING (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] Unknown cluster-specific command 253
2021-08-08 23:24:40 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] ZCL request 0x00fd: b'\x01'
2021-08-08 23:24:40 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] No handler for cluster command 253
2021-08-08 23:24:40 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7c97>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'0145fd01'
2021-08-08 23:24:41 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-08-08 23:24:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2021-08-08 23:24:41 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [27, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7c97>, 1, 260, 6, b'\x01E\xfd\x01', 0, 175, 255, 151, 94, 31, 0, -47]
2021-08-08 23:24:41 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=69 command_id=253>
2021-08-08 23:24:41 WARNING (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] Unknown cluster-specific command 253
2021-08-08 23:24:41 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] ZCL request 0x00fd: b'\x01'
2021-08-08 23:24:41 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] No handler for cluster command 253
2021-08-08 23:24:41 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7c97>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'0145fd01'
2021-08-08 23:24:42 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2021-08-08 23:24:42 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
[... cut ...]
2021-08-08 23:24:42 WARNING (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] Unknown cluster-specific command 253
2021-08-08 23:24:42 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] ZCL request 0x00fd: b'\x01'
2021-08-08 23:24:42 DEBUG (MainThread) [zigpy.zcl] [0x7c97:1:0x0006] No handler for cluster command 253
2021-08-08 23:24:42 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x7c97>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'0145fd01'
```

docgalaxyblock, 2021-08-09:
@MartinKlingeling I'm not 100% sure but I think the scene switching only works if you pair the panel directly to the remote without a bridge.
Here is a little workaround/other way to use the buttons at automations
<details><summary><b>requires HA 2021.7+</b></summary>

```
alias: Küche - Panel
description: ''
trigger:
  - platform: event
    event_type: zha_event
    event_data:
      device_ieee: xx:xx:xx:xx:xx:xx:xx:xx
      command: 'on'
    id: An
  - platform: event
    event_type: zha_event
    event_data:
      device_ieee: xx:xx:xx:xx:xx:xx:xx:xx
      command: step
      args:
        - 0
        - 51
        - 10
    id: Heller
  - platform: event
    event_type: zha_event
    event_data:
      device_ieee: xx:xx:xx:xx:xx:xx:xx:xx
      command: step
      args:
        - 1
        - 51
        - 10
    id: Dunkler
  - platform: event
    event_type: zha_event
    event_data:
      device_ieee: xx:xx:xx:xx:xx:xx:xx:xx
      command: 'off'
    id: Aus
condition: []
action:
  - choose:
      - conditions:
          - condition: trigger
            id: An
        sequence:
          - type: turn_on
            device_id: 6834f714e26fd1a8f9fcd60de8775afc
            entity_id: light.basement_panel
            domain: light
      - conditions:
          - condition: trigger
            id: Heller
        sequence:
          - device_id: 6834f714e26fd1a8f9fcd60de8775afc
            domain: light
            entity_id: light.basement_panel
            type: brightness_increase
      - conditions:
          - condition: trigger
            id: Dunkler
        sequence:
          - device_id: 6834f714e26fd1a8f9fcd60de8775afc
            domain: light
            entity_id: light.basement_panel
            type: brightness_decrease
      - conditions:
          - condition: trigger
            id: Aus
        sequence:
          - type: turn_off
            device_id: 6834f714e26fd1a8f9fcd60de8775afc
            entity_id: light.basement_panel
            domain: light
    default: []
mode: single

```

MartinKlingeling, 2021-08-10:
The scene switching will indeed only work when connected directly. But that is no big loss. The problem is that the "on" command is only if you pressed the off button in between. So you can't have an automation like "on button toggles a light, off button toggles a different light."

Thanks for the workaround, I will give it a try. I also found a blueprint for a remote that might be the same thing in a different shell. The event data looks like what I see in the log. https://community.home-assistant.io/t/immax-neo-07087-2-zigbee-remote-control/289200

docgalaxyblock, 2021-08-11:
year but that problem is caused by the cheaps remote firmware...

gavinsmith1971, 2021-08-21:
> @MartinKlingeling I'm not 100% sure but I think the scene switching only works if you pair the panel directly to the remote without a bridge.
> Here is a little workaround/other way to use the buttons at automations
> 
> requires HA 2021.7+

I'd love to try this, but I'm afraid I'm still new to HA yaml - which file do I put this in? (Doesn't work in configuration,yaml) 

I have the remote paired with ZHA and have replaced the ieee with my own - assume it doesn't need any more editing?

MartinKlingeling, 2021-08-22:
> I'd love to try this, but I'm afraid I'm still new to HA yaml - which file do I put this in? (Doesn't work in configuration,yaml)
 
Create a new, empty automation in Configuration > Automations, then switch to the YAML editor in the three-dots-menu. There you can put the automation definition from above.

gavinsmith1971, 2021-08-23:
> > I'd love to try this, but I'm afraid I'm still new to HA yaml - which file do I put this in? (Doesn't work in configuration,yaml)
> 
> Create a new, empty automation in Configuration > Automations, then switch to the YAML editor in the three-dots-menu. There you can put the automation definition from above.
> 
> Then you can even switch back to the visual editor to customize the actions.

Excellent, thank you! Have it working nicely now, with the limitations mentioned.

One oddity for anyone else trying this, when you switch back to the visual editor or open it again to edit, it adds 'brightness_pct: 0' for some reason, which threw me for a bit as it was actually turning 'on' but at zero brightness. Setting that value to any appropriate value works fine - but removing the line again in yaml lets it remember the last brightness instead, which is my preferred option.

Increasing Brightness when off actually turns it on too (at lowest brightness), and reducing brightness eventually turns it off too, so it's almost entirely functional other than the scenes/on button issue...

jquievreux, 2021-12-05:
thanks @MartinKlingeling and @gavinsmith1971  i made a blueprint from the source for those who want to save time.

https://gist.github.com/letroll/1ec683f977e5b0c8be290140f50955b4

elenril2, 2022-01-09:
Is there any chance to add this device support in ZHA?

In my home assistant I can't see any zha_events and I cannot use blueprint for this device. But strangely my zha_events are empty even from another sensors paired with ZHA Gateway - I'm using modified Silvercrest Gateway (https://paulbanks.org/projects/lidl-zigbee/ha/).

scooper1, 2022-01-09:
it would be nice to have this working

challs, 2022-01-11:
Some code I started a while back: https://github.com/challs/zha-device-handlers/tree/wip/ts1001

MattWestb, 2022-01-11:
Its one more device that is making tuya magic things and can changing from being one light controller and start sending tuya commands but little more ones https://github.com/zigpy/zha-device-handlers/issues/1268.
My experience with first round of TS004F is some users is having problems getting the remote sending commands to the coordinator and its not easy getting logs from ZHA.

The "scene commands" is implanted in tuya INIT but the above need little more commands for the color stepping but is manageable but implanting it on one OnOff cluster that shall also supporting the normal OnOff commands i dont knowing if its being easy or not.
Is the remote having on mode switch attribute for switching between light control and and scene commands ?
TS004F is having that so it can being possible using for switching decoding commands from the device.

One question @challs is you good on writing the py tests ?
We have making quirks for around 20 tuya TRVs and no one can doing the test so we cant getting the PR s merged. 
an you helping us explanting how to writing the test cone so we can getting the PRs merged ?
We having one issue open for discussion the tuya TRV implantation if you is interested helping us coming on track https://github.com/zigpy/zha-device-handlers/issues/1183.

Thanks in advance !!

challs, 2022-01-13:
> Is the remote having on mode switch attribute for switching between light control and and scene commands ?

I don't remember having seen this problem, but I wasn't aware that this might happen so I might have missed it. Thanks for the explanation about the scene commands - I didn't really understand what all that was about at the time.

I was only getting parsed IncomingMessageHandler frames for the '1' button. The '0' and dim down buttons were receiving messageSentHandler, which didn't seem to be doing anything useful. I need to retest against the latest version to check if this is still a problem.

> is you good on writing the py tests ?

I have written quite a few pytests before. I've commented in the issue you referenced.

MattWestb, 2022-01-13:
The tuya "sene commands" 0xFD is being pared here:
https://github.com/zigpy/zha-device-handlers/blob/6227c8b08403f536e13e9c48b647f66600f06145/zhaquirks/tuya/__init__.py#L776-L782
I have one test and adding 0xFC for rotating commands:
```
    rotate_type = {
        0x00: ROTATE_RIGHT,
        0x01: ROTATE_LEFT,
        0x02: ROTATE_STOP,
    }
```
that is being used for dimming and CT in command mode and rotating in scene / event mode.

I still need getting more logs from the user for getting all in place but the remote is switching mode with triple press the button. I think the LIDL is doing the same with pressing the on button one second time (without and other button pressed between).

Also the TS004F cant being bond to one light group in the noral way then binding is not working of out clusters. Instead its working adding group to group in cluster and its sending the light commands to the light group.

"Pure" tuya scene switches is the TS0044 that can being good looking on for getting it working scene part and the TS004F is default one light remote but can being switches to scene mode (the switch is working but is not making the 3 new endpoints so cant using them). The mode switch is here: https://github.com/zigpy/zha-device-handlers/blob/6227c8b08403f536e13e9c48b647f66600f06145/zhaquirks/tuya/__init__.py#L857.

mdeweerd, 2022-02-20:
Following from @MattWestb in other issue https://github.com/zigpy/zha-device-handlers/issues/1303#issuecomment-1046276369 , here are some sniffer logs for the remote.

Tried to make single, multiple and long presses.
[remote1.zip](https://github.com/zigpy/zha-device-handlers/files/8104547/remote1.zip)
[remote2.zip](https://github.com/zigpy/zha-device-handlers/files/8104548/remote2.zip)

Product code:
343820_2004
Manual can be fetched from https://www.lidl-service.com/en/index.html with the product code.

UK page:

https://customer-service.lidl.co.uk/SelfServiceUK/s/product2?productId=01t6N0000006z3OQAQ&language=en_GB#q=343820_2004

Label picture:
[image]

Reported in HA as : 
TS1001 by _TYZB01_bngwdjsr

MattWestb, 2022-02-20:
The first zipfile is OK and the second is the `LIDL_Remote.ncpcap` truncated to 48 byte.
With wireschark i can see its using "tuya standard scene commands" sent from the OnOff cluster but i cant see the data in WS but i think its the tuya standard.
Adding `TuyaSmartRemoteOnOffCluster` in one quirk shall decoding all commands and an can being used to do device automation with one, two and long press.
The data structure is described here https://github.com/zigpy/zha-device-handlers/pull/1376#issuecomment-1046285840.

MattWestb, 2022-02-20:
```
Frame 1: 49 bytes on wire (392 bits), 49 bytes captured (392 bits) on interface unknown, id 0
IEEE 802.15.4 Data, Dst: 0x2b5e, Src: 0xd988, Bad FCS
ZigBee Network Layer Data, Dst: 0x0000, Src: 0xd988
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 34
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
    Sequence Number: 75
    Command: Unknown (0xfd)
    Payload
```
Wireschark is decoding the command but is throwing the payload then its out of range in ZCL.

mdeweerd, 2022-02-20:
Sorry, I added raw packets to the output and that broke my packet filter script.

This is better.
[remote1.zip](https://github.com/zigpy/zha-device-handlers/files/8104790/remote1.zip)
[remote2.zip](https://github.com/zigpy/zha-device-handlers/files/8104791/remote2.zip)

mdeweerd, 2022-02-20:
Don't look at the 'short' logs - they are wrong now.

MattWestb, 2022-02-20:
Now both `LIDL_Remote.ncpcap` is truncated to 48 byte ;-((
The first `remote1.zip` was 4248 byte and was OK opening with wireshark.

mdeweerd, 2022-02-20:
Ok, the short log was required to do the filtering and it was the wrong one.
Checked the sizes:
[remote1.zip](https://github.com/zigpy/zha-device-handlers/files/8104843/remote1.zip)
[remote2.zip](https://github.com/zigpy/zha-device-handlers/files/8104844/remote2.zip)
c

MattWestb, 2022-02-20:
Both zip files looks OKand i can see all commands but WS is not showing the data payload.
One question is the remote sending some attributes the changing mode (I think double pressing on) that can being cached ?
Normal tuya Zigbee device is using (manufacture specific) switch config and mode attributes on the OnOff cluster that can being interesting but it can its dont have that implanted.

mdeweerd, 2022-02-20:
Wireshark does show the payload but does not decode it:
[image]

We can see that the command is 0xFD and the payload 0x02 here.

In fact I have FD 00 up to FD 03 so 4 payload values.

challs, 2022-02-20:
Hey guys, nice to see some activity here. Sorry I made more progress but have not updated my branch. I will prepare a new upload and post here shortly.

I have handlers for on/off and dim up/down. I can see the events for scenes but have not finished them yet.

MattWestb, 2022-02-20:
Thanks @mdeweerd !!
Then we need finding out what payload 0x03 is for "click" type and adding it in tuya INIT.

@challs I think you can using the decoding that is made in tuya INIT then its having the sending default response for the no ZCL commands (but i think it do the same with ZCL commands one more time) and can being reused for other devices in the future.
I was looking on your code and its looks good (but i dont have the coding knowledge) and i think you have finding all "light mode commands" but the TS004X quirks is using the "scene" commands so can looking on them for getting more input.

challs, 2022-02-20:
I've pushed the current state of my code to https://github.com/challs/zha-device-handlers/tree/wip/ts1001

I did quite a bit of work on improving the support for writing tests, and my tests use the new fixtures so I have included them on the branch.

I'm not yet settled on whether I can make use of the `TuyaSmartRemoteOnOffCluster` or not, so there is a custom class `TuyaRemote1001OnOffCluster` which I was working on but is unused right now.

The code already is good enough to write automations based on the `zha_event` events, but I was trying to also add support for [device automation triggers](https://github.com/zigpy/zha-device-handlers/blob/dev/README.md#how-device_automation_triggers-work) too.

challs, 2022-02-20:
This is what the current iteration of my code is doing with the on / scene switch:

```
# press off -> off event

[zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xF3xx, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=144, SecurityUse=<Bool.false: 0>, TimeStamp=4311070, TSN=0, Data=b'\x01\x48\x00', MacSrcAddr=0xF3xx, MsgResultRadius=11)
[zigpy.zcl] [0xf3xx:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=72 command_id=0>
[zigpy.zcl] [0xf3xx:1:0x0006] ZCL request 0x0000: []
[zigpy.zcl] [0xf3xx:1:0x0006] TS004X: send default response
[homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=bc:33:ac:ff:fe:fe:xx:xx, unique_id=bc:33:ac:ff:fe:fe:xx:xx:1:0x0006, device_id=d2a83b6f9ef3689cb2e5a51021dfxxxx, endpoint_id=1, cluster_id=6, command=off, args=[]>

# press on -> on event

[zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xF3xx, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=90, SecurityUse=<Bool.false: 0>, TimeStamp=4899481, TSN=0, Data=b'\x01\x49\x01', MacSrcAddr=0xF3xx, MsgResultRadius=11)
[zigpy.zcl] [0xf3xx:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=73 command_id=1>
[zigpy.zcl] [0xf3xx:1:0x0006] ZCL request 0x0001: []
[zigpy.zcl] [0xf3xx:1:0x0006] TS004X: send default response
[homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=bc:33:ac:ff:fe:fe:xx:xx, unique_id=bc:33:ac:ff:fe:fe:xx:xx:1:0x0006, device_id=d2a83b6f9ef3689cb2e5a51021dfxxxx, endpoint_id=1, cluster_id=6, command=on, args=[]>

# press on again -> scene 1

[zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xF3xx, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=99, SecurityUse=<Bool.false: 0>, TimeStamp=5362541, TSN=0, Data=b'\x01\x4A\xFD\x00', MacSrcAddr=0xF3xx, MsgResultRadius=29)
[zigpy.zcl] [0xf3xx:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=74 command_id=253>
[zigpy.zcl] [0xf3xx:1:0x0006] ZCL request 0x00fd: [0]
[zigpy.zcl] [0xf3xx:1:0x0006] TS004X: send default response
[homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=bc:33:ac:ff:fe:fe:xx:xx, unique_id=bc:33:ac:ff:fe:fe:xx:xx:1:0x0006, device_id=d2a83b6f9ef3689cb2e5a51021dfxxxx, endpoint_id=1, cluster_id=6, command=remote_button_short_press, args=[]>
[homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=bc:33:ac:ff:fe:fe:xx:xx, unique_id=bc:33:ac:ff:fe:fe:xx:xx:1:0x0006, device_id=d2a83b6f9ef3689cb2e5a51021dfxxxx, endpoint_id=1, cluster_id=6, command=press_type, args=[0]>

# press on again -> scene 2

[zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xF3xx, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=123, SecurityUse=<Bool.false: 0>, TimeStamp=5668728, TSN=0, Data=b'\x01\x4B\xFD\x01', MacSrcAddr=0xF3xx, MsgResultRadius=29)
[zigpy.zcl] [0xf3xx:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=75 command_id=253>
[zigpy.zcl] [0xf3xx:1:0x0006] ZCL request 0x00fd: [1]
[zigpy.zcl] [0xf3xx:1:0x0006] TS004X: send default response
[homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=bc:33:ac:ff:fe:fe:xx:xx, unique_id=bc:33:ac:ff:fe:fe:xx:xx:1:0x0006, device_id=d2a83b6f9ef3689cb2e5a51021dfxxxx, endpoint_id=1, cluster_id=6, command=remote_button_double_press, args=[]>
[homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=bc:33:ac:ff:fe:fe:xx:xx, unique_id=bc:33:ac:ff:fe:fe:xx:xx:1:0x0006, device_id=d2a83b6f9ef3689cb2e5a51021dfxxxx, endpoint_id=1, cluster_id=6, command=press_type, args=[1]>

# press on again -> scene 3

[zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0xF3xx, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=123, SecurityUse=<Bool.false: 0>, TimeStamp=6009511, TSN=0, Data=b'\x01\x4C\xFD\x02', MacSrcAddr=0xF3xx, MsgResultRadius=29)
[zigpy.zcl] [0xf3xx:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=76 command_id=253>
[zigpy.zcl] [0xf3xx:1:0x0006] ZCL request 0x00fd: [2]
[zigpy.zcl] [0xf3xx:1:0x0006] TS004X: send default response
[homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=bc:33:ac:ff:fe:fe:xx:xx, unique_id=bc:33:ac:ff:fe:fe:xx:xx:1:0x0006, device_id=d2a83b6f9ef3689cb2e5a51021dfxxxx, endpoint_id=1, cluster_id=6, command=remote_button_long_press, args=[]>
[homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=bc:33:ac:ff:fe:fe:xx:xx, unique_id=bc:33:ac:ff:fe:fe:xx:xx:1:0x0006, device_id=d2a83b6f9ef3689cb2e5a51021dfxxxx, endpoint_id=1, cluster_id=6, command=press_type, args=[2]>

# press on again -> scene 4

[... cut ...]
[homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=bc:33:ac:ff:fe:fe:xx:xx, unique_id=bc:33:ac:ff:fe:fe:xx:xx:1:0x0006, device_id=d2a83b6f9ef3689cb2e5a51021dfxxxx, endpoint_id=1, cluster_id=6, command=remote_button_short_press, args=[]>
[homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=bc:33:ac:ff:fe:fe:xx:xx, unique_id=bc:33:ac:ff:fe:fe:xx:xx:1:0x0006, device_id=d2a83b6f9ef3689cb2e5a51021dfxxxx, endpoint_id=1, cluster_id=6, command=press_type, args=[0]>
```

Clearly not right yet - I need to work on a cluster to map those scene ids.

challs, 2022-02-20:
@MattWestb 

> using the decoding that is made in tuya INIT then its having the sending default response for the no ZCL commands (but i think it do the same with ZCL commands one more time) and can being reused for other devices in the future.

Sorry, I don't understand what you are suggesting I change here. Maybe you can point me to an example of what you suggest I change?

mdeweerd, 2022-02-20:
https://github.com/challs/zha-device-handlers/blob/dev/zhaquirks/tuya/__init__.py#L791-L793

MattWestb, 2022-02-21:
All tuya scene switched is only doing zha_events then its no real scene commands only one tuya version of sending clicks.

@challs the data types is defined here https://github.com/challs/zha-device-handlers/blob/9a1401337a25a4adc33e0b37ace62d808b7b3098/zhaquirks/tuya/__init__.py#L775-L781
if its OK adding 0x03 or doing one new with if the switch need other versions of the commands its sending.

challs, 2022-02-21:
Thanks @MattWestb 

I was actually wondering whether I should be setting up a scene bus and arranging to map this to a scene recall?

MattWestb, 2022-02-21:
Its the cluster we is using for TS0041, TS0042, TS0043 and TS004 "scene switches" an shall being implanted in TS004F then we is getting reading and sending attribute thru quirk so we can getting the 3 extra EP working.

Is not real Zigbee (ZCL) scenes commands is sending but we is using them firing event i ZHA and using this in automation.

One small problem is the the cluster for your device need the normal ZCL light commands (on/off) on the 0x0006 and the extra "scene commands" working at the same time. The TS004F is needing the same later also the knob version plus one more command for rotating.
I have not looking so deep is your device is sending the same commands and data payload but its looks its doing it but perhaps all the same and only needing adding the function 0x03 that is new.

MattWestb, 2022-03-05:
31 useful DA is on the way for the TS004F https://github.com/zigpy/zha-device-handlers/issues/1322#issuecomment-1059771390 that can being used for the TS001.
Also need updating tuya INIT for working OK !!

mdeweerd, 2022-08-15:
Maybe the scene buttons could be implemented as in https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/inovelli/VZM31SN.py (device_automation_triggers).

Maybe with some inspiration from https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/adeo/color_controller.py as well

constapel, 2022-11-21:
Can somebody please xplain or help?
I have a LIVARNO LUX/LIDL Remote Control Dimmer Tuya TS1001. It is integrated to ZHA Home Assistent ( EZSP
by Silicon Labs )(LIDL patched).
When button 3 is fired all my ( 4 pieces) TS011F by _TZ3000_cphmq0q7 are switch of. When button 2 is fired all my ( 4 pieces) TS011F by _TZ3000_cphmq0q7 are switch on. It looks like the remote controle is sending a sort off overall ZHA_event on command to all my ZHA Zigbee devices. I first tought it had something to do with touchlink. But i never touchedlinked these devices and to my knowledge ZHA does not support Touch link. 
I think this has something to do with de implementation of TS1001 in ZHA...is the the case.
Now i can not use the TS1001 ( with Damian666 blueprint ) because everything all my TS011F are switch on /off..
Help needed...
Constapel

MattWestb, 2022-11-21:
The light remote is default sending its commands to Zigbee group 0x0000 and is its some device that is have the group 0x0000 added they is reacting on the commands sent to the group.
Delete the group 0x0000 from your devices that not need having it (some can have default after reset) or changing the group the remote is bonded to.

constapel, 2022-11-22:
OMG MattWestb you are most valuable player for this community. Your feedback is great en solved the problem!. I was not aware of ZHA LightLink Groups.  I moved the 4 switches out of group 0x0000 and the problem was solved. 
I can not move the LIVARNO LUX/LIDL Remote Control Dimmer Tuya TS1001 in any other group so unfortunately the other way to associate the rcontrol to a group does not work.  But with the Damiain666 blueprint i was able to control a helpergroup or other device with the control. Thank you very very much for you help and as always great share of knowledge. Many kudos!!!!
Constapel

MattWestb, 2022-11-22:
The TS1001 from LIDL have not getting any quirk (some user have "decoded" its function but is not finessed as You see in this issue) and most tuya device have very strange config / behavior like some remotes is added like lights = wrong.
Your device is looking in the first post looks having the On/Off cluster as out = its sending commands and shall possible binding to one group or devices.

Try from the device card > the three dots right from reconfigure > Manage Zigbee devices.
First unbindg its from the coordinator group (0x0000) and making all cluster you like unbinding and waking the remote up then sending the command.
Then you need doing binding to the coordinator from device binding and selectig your coordinator and the cluster you like to bod to the coordinator (If not bond to the coordinator group or direct too the coordinator you is not getting any events and automatons is not working) dont forget waking the remote up then sending somthing to it.
You can also binding it to one other Zigbee group is you like and dont need doing the binding to the coordinator.

I cant grantee its working then in the end its one tuya device that is making all kind of no standard crazy things.

challs, 2023-07-03:
This is still open. There was an attempted MR but this has problems still

MattWestb, 2023-07-03:
One device that is have little the same working mode is being added.
https://github.com/zigpy/zha-device-handlers/pull/2455#pullrequestreview-1509839238
Its using some of the same commands but is nit having so many buttons and not so strange "working modes".
Take one look if you like see how tuya have cooking there devices.

closed 2024-01-06 by github-actions[bot]

nothumbee, 2025-04-05:
Hey gyus, any chance of getting this done? Subsequent "on" clicks just do nothing and it is really annoying. I guess I have some newer design of the remote (TS1001 by _TYZB01_bngwdjsr), but it seems it works the same.