zigpy/zha-device-handlers #616: [Device Support Request] SilverCrest/LIDL Motion sensor SMSZ 1 A1 (HG06335) (TY0202 Tuya based)

docgalaxyblock, 2020-12-04:
https://zigbee.blakadder.com/Lidl_HG06335.html
Already supported at Z2M: https://www.zigbee2mqtt.io/devices/HG06335.html

**Describe the solution you'd like**
- [ ] Motion support 
- [ ] Battery support (first time it has worked)
- [ ] Tamper support
- [ ] Automations wizard support

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
[Device signature and pairing log at my dev VM](https://pastebin.com/F5X8xkji)
_Maybee something went wrong!?_

**Additional context**
looks like Heiman HS1MS
https://zigbee.blakadder.com/Heiman_HS1MS.html
https://www.zigbee2mqtt.io/devices/SMHM-I1.html

docgalaxyblock, 2020-12-04:
Signature
```
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ1800_fcdjzz3s",
  "model": "TY0202",
  "class": "zigpy.device.Device"
}

kkplein, 2021-01-31:
What can we do to help make this happen..? I have one too, and would like it to work :-)

kkplein, 2021-02-01:
For the record, this device works with the zigbee2mqtt integration.

docgalaxyblock, 2021-02-04:
@kkplein please read again the third line 😉 
> Already supported at Z2M: https://www.zigbee2mqtt.io/devices/HG06335.html

kkplein, 2021-02-04:
Isn't that what I said, above?
It didn't work with the ZHA HA zigbee integration, but does with Z2M.

docgalaxyblock, 2021-08-06:
Comment for the bot 😉

Joerg76, 2021-12-17:
please support LIDL Motion sensor SMSZ 1 B2 TY0202

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ1800_fcdjzz3s",
  "model": "TY0202",
  "class": "zigpy.device.Device"
}

and the debug log Nwk: 0x8709
[lidlbewegungsmelder_0x8709.txt](https://github.com/zigpy/zha-device-handlers/files/7736607/lidlbewegungsmelder_0x8709.txt)

MattWestb, 2022-03-05:
Battery and IAS Zone detecting and resetting is working out of the box with ZHA.

MattWestb, 2022-03-12:
Update: I was deleting it im my test network and trying adding it on my new NCP  based test network and its was paring nicely but was not resetting the IAS-Zone.
Trying manipulating the tamper but its working OK but no reset of the alarm.
Deleting it and waiting one minute and adding it and letting the tamper open and manipulating it only little then it joining and can see its working OK in the paring window log.
Also after some seconds after putting back the back plate so the tamper is OK i getting one ISA-Zone alarm reset.
Then its the device working OK (only the tamper is not being nicely rigging one own alarm) so its looks it can being little tricky then paring for getting the ISA-Zone status being OK updated.

docgalaxyblock, 2022-09-11:
Tested it today again at my testing vm (2022.09.1)
The tamper sensor still does not show up in the UI

flosen1304, 2023-03-26:
Can anyone help here? I have got the same device and when trying to pair it works. But it only recognize the first motion and doesn't reset...

MattWestb, 2023-03-27:
Taking the back plane away so you is sending one taper and putting it back its looks like the motion alarm is being retested OK then the time is running out.

@TheJulianJES I think this problem is the ISA-Zone status that you have working on.
If you need help testing code change give my one call and i can testing it (if not having **too** many devices and systems under test). 
I have both LIDL variants for testing plus the large door and window contact that also using ISA for testing.

closed 2023-09-30 by github-actions[bot]