zigpy/zha-device-handlers #618: [Device Support Request] LIVARNO LUX/LIDL Led Panel 60x60 (Tuya TS0502A)

docgalaxyblock, 2020-12-04:
[LIDL Service website](https://www.lidl-service.com/cps/rde/xchg/SID-3771F4F2-8A18D468/lsp/hs.xsl/product.html?id=5027306530&title=Smart+LED+Light+Panel&count=1)

**Describe the solution you'd like**
- [x] power control
- [x] brightness control
- [x] CWW control
- [ ] remove color control
[image]

Maybee because the board is also color capable
[image]

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
[Device signature and pairing log at my dev VM](https://pastebin.com/ifAkAXaF)

**Additional context**
https://zigbee.blakadder.com/Tuya_TS0502A.html
Touchlink resetable

docgalaxyblock, 2020-12-04:
Signature
```
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_9evm3otq",
  "model": "TS0502A",
  "class": "zigpy.device.Device"
}

MattWestb, 2020-12-05:
I like the PCB but someone have stolen the IR receiver and was not having time putting the drivers for RGB in place but the TC looks OK :-))

I was thinking buying one of then but they was not delivered to my store :-(

docgalaxyblock, 2020-12-05:
@MattWestb You are the guy who has made a teardown of the tradfri bulbs right?

MattWestb, 2020-12-05:
I have not making one large teardown of bulbs that's one netherland guy that have doing that. I have more "extracting" the Zigbee module for using it as one coordinator i ZHA and Tasmota.
What i have seen is the first gen bulbs potted with some heat transmitting mass and the newer one without only more metal inside and all components blues on the PCB. 
Normally its the coils that is not firm done that is making vibrations and then not being god glused its can making much sounds / vibrations.

Was reading that new Philips HUE Zigbee / BL bulbs was trying changing the PWM frequency to getting the noise away but its not easy if the components is not good done / fixed.

gijsje, 2020-12-11:
The round ceiling light is also shown als color lights but only has cold/warm white
`{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_rylaozuc",
  "model": "TS0502A",
  "class": "zigpy.device.Device"
}`

TheJulianJES, 2020-12-29:
This seems like the same issue as https://github.com/zigpy/zha-device-handlers/issues/643.
Created a PR here: https://github.com/zigpy/zha-device-handlers/pull/680 (Please test it if you can and report back)

PR #680 (2020-12-30): Add more LIDL CCT only devices (fix #618)
fixes: #618

Perhaps @gijsje and @docgalaxyblock can test this by downloading the "lidl" folder from here: https://github.com/TheJulianJES/zha-device-handlers/tree/lidl_cct/zhaquirks/lidl and moving it to their zha-quirks. (Instructions here for HassOS/Supervised: https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install)
commit c28f3cd1 (2020-12-30): Add more LIDL CCT only devices (fix #618) (#680)
closed 2020-12-30 by dmulcahey