zigpy/zha-device-handlers #621: Unsupported profile on FeiBit  FNB56-ZSW03LX2.0

adolfomendes, 2020-12-07:
**Describe the bug**
The switch FeiBit  FNB56-ZSW03LX2.0 doesn't support light level control (profile 0x0008).
Please remove profile 0x0008 from device description.

**Screenshots**
[image]
[image]

dmulcahey, 2020-12-11:
FYI I added the US version of this to my network and it caused non stop address conflicts. YMMV.

JMan7777, 2020-12-13:
Same on the FeiBit FNB56-SKT1JXN1.0 
The switch does not support light level control but the profile is available in the device. Need to be removed.

JMan7777, 2021-08-06:
The issue still exists on the FeiBit FNB56-SKT1JXN1.0 (tested with supervisor-2021.06.8).
The switch does not support light level control but the profile is available in the device. 

[image]